evcc-io/evcc #15354: In/Out not balanced in the UI

MehrCurry, 2024-08-12:
### Describe the bug

Today i connected my car and the displayed values in EVCC seems unplausible. Formerly the In/Out sum up to zero which is correct as I understand. Today the are unbalanced which looks like a bug to me.

[image]

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: https
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: xxx.xxx.xxx
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070
interval: 10s # control cycle interval

sponsortoken: xxxx

# log settings
log: debug
levels:
  core: debug
  lp-1: debug
  lp-2: debug

meters:
  - name: my_grid
    type: template
    template: fronius-gen24
    usage: grid
    host: 192.168.187.59 # IP-Adresse oder Hostname
    port: 502 # Port # Optional
  - name: my_pv
    type: template
    template: fronius-gen24
    usage: pv
    host: 192.168.187.59 # IP-Adresse oder Hostname
    port: 502 # Port # Optional
  - name: my_battery
    type: template
    template: fronius-gen24
    usage: battery
    host: 192.168.187.59 # IP-Adresse oder Hostname
    port: 502 # Port # Optional

chargers:
  - type: wallbe
    name: wallbox
    uri: 192.168.187.8:502 # TCP ModBus address
    legacy: true # enable for older Wallbes with Phoenix EV-CC-AC1-M3-CBC-RCM controller

loadpoints:
  - title: Wallbox # display name for UI
    charger: wallbox # charger reference
    # vehicles:
    #  - Kona1
    #  - Kona2
    mode: pv
    soc:
      # polling defines usage of the vehicle APIs
      # Modifying the default settings it NOT recommended. It MAY deplete your vehicle's battery
      # or lead to vehicle manufacturer banning you from API use. USE AT YOUR OWN RISK.
      poll:
        # poll mode defines under which condition the vehicle API is called:
        #   charging: update vehicle ONLY when charging (this is the recommended default)
        #   connected: update vehicle when connected (not only charging), interval defines how often
        #   always: always update vehicle regardless of connection state, interval defines how often
        mode: charging
        # poll interval defines how often the vehicle API may be polled if NOT charging
        interval: 60m
      estimate: true # set true to interpolate between api updates
    enable:
      threshold: 4100 # 3600W PV + 500W Netzbezug
      delay: 1m
    disable:
      threshold: 3600 # 3100W PV + 500W Netzbezug
      delay: 3m
    # resetOnDisconnect: true
    # maxcurrent: 16 # maximum charge current (default 16A)
    # mincurrent: 6
    # phases: 3
mqtt:
  broker: homeassistant.fritz.box:1883
  topic: evcc # root topic for publishing, set empty to disable
  user: mqtt
  password: xxxx
site:
  title: Zuhause # display name for UI
  residualPower: -500
  # prioritySoC: 75
  # bufferSoC: 90
  meters:
    grid: my_grid
    pv:
      - my_pv
    battery:
      - my_battery
influx:
  url: http://192.168.187.36:8086
  database: evcc
  user: evcc
  org: default
  token: xxxx
tariffs:
  currency: EUR # (default EUR)
  grid:
    type: fixed
    price: 0.2936 # [currency]/kWh

  feedin:
    type: fixed
    price: 0.1643 # [currency]/kWh 0.2443 - 0.08

plant: xxxx
```
```

### Log details

```text
[site  ] DEBUG 2024/08/12 12:49:32 ----
[lp-1  ] DEBUG 2024/08/12 12:49:32 charge power: 7590W
[site  ] DEBUG 2024/08/12 12:49:32 pv power: 4126W
[site  ] DEBUG 2024/08/12 12:49:32 battery soc: 58%
[site  ] DEBUG 2024/08/12 12:49:32 battery power: -1149W
[site  ] DEBUG 2024/08/12 12:49:32 grid meter: 15W
[site  ] DEBUG 2024/08/12 12:49:32 grid currents: [-4.4 -4.09 6.85]A
[site  ] DEBUG 2024/08/12 12:49:32 battery has priority at soc 58% (< 80%)
[site  ] DEBUG 2024/08/12 12:49:32 site power: -485W
[lp-1  ] DEBUG 2024/08/12 12:49:32 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[lp-1  ] DEBUG 2024/08/12 12:49:32 !! session: chargeRater.chargedEnergy=36.5 - chargedAtStartup=0.0
[lp-1  ] DEBUG 2024/08/12 12:49:32 charger status: C
[lp-1  ] DEBUG 2024/08/12 12:49:32 soc estimated: 31.03% (vehicle: 30.00%)
[lp-1  ] DEBUG 2024/08/12 12:49:32 vehicle soc: 31%
[lp-1  ] DEBUG 2024/08/12 12:49:32 vehicle soc limit: 100%
[lp-1  ] DEBUG 2024/08/12 12:49:32 vehicle range: 156km
[lp-1  ] DEBUG 2024/08/12 12:49:32 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[lp-1  ] DEBUG 2024/08/12 12:49:32 pv charge current: 11.7A = 11A + 0.703A (-485W @ 3p)
[site  ] DEBUG 2024/08/12 12:49:42 ----
[lp-1  ] DEBUG 2024/08/12 12:49:42 charge power: 7590W
[site  ] DEBUG 2024/08/12 12:49:42 pv power: 4126W
[site  ] DEBUG 2024/08/12 12:49:42 battery soc: 58%
[site  ] DEBUG 2024/08/12 12:49:42 battery power: -1187W
[site  ] DEBUG 2024/08/12 12:49:42 grid meter: -3W
[site  ] DEBUG 2024/08/12 12:49:42 grid currents: [-4.42 -4.1 6.83]A
[site  ] DEBUG 2024/08/12 12:49:42 battery has priority at soc 58% (< 80%)
[site  ] DEBUG 2024/08/12 12:49:42 site power: -503W
[lp-1  ] DEBUG 2024/08/12 12:49:42 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[lp-1  ] DEBUG 2024/08/12 12:49:42 !! session: chargeRater.chargedEnergy=36.5 - chargedAtStartup=0.0
[lp-1  ] DEBUG 2024/08/12 12:49:42 charger status: C
[lp-1  ] DEBUG 2024/08/12 12:49:42 soc estimated: 31.04% (vehicle: 30.00%)
[lp-1  ] DEBUG 2024/08/12 12:49:42 vehicle soc: 31%
[lp-1  ] DEBUG 2024/08/12 12:49:42 vehicle soc limit: 100%
[lp-1  ] DEBUG 2024/08/12 12:49:42 vehicle range: 156km
[lp-1  ] DEBUG 2024/08/12 12:49:42 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[lp-1  ] DEBUG 2024/08/12 12:49:42 pv charge current: 11.7A = 11A + 0.729A (-503W @ 3p)
[site  ] DEBUG 2024/08/12 12:49:52 ----
```

### What type of operating system are you running?

Linux

### Version

0.129.0

StefanSchoof, 2024-08-12:
Is it possible you have a charger without a meter and a car that can only 1p? Then evcc does not know how much the car is loading. You can set the phases of the car with https://docs.evcc.io/en/docs/reference/configuration/vehicles#phases

andig, 2024-08-12:
> Charger is a Wallbe Eco 2.0 without a meter.

It will be unbalanced when the car consumes less than the charger allows. Problem is the missing meter.

closed 2024-08-12 by andig

MehrCurry, 2024-08-13:
Strange: After a Restart of HomeAssistant every thing looks like before and is (more or less) in sync.

[image]