evcc-io/evcc #15356: Probleme mit der Steuerung vom Tesla <5A

Neffe1337, 2024-08-12:
### Describe the bug

Hi zusammen, 

es ist mir leider nicht mehr möglich meinen Tesla zu steuern.  
Damit meine ich unter 5A zu laden. 
Dies hat vor Ewigkeiten mal funktioniert und irgendwann ging es nicht mehr. 
Seit dem habe ich dutzend Sachen versucht um es wieder lauffähig zu bekommen. 
In der Config unten habe ich den Tesla und Sponsor Token via GUI hinzugefügt. Ich hatte den Tesla und den Token auch mal direkt in die YAML geschrieben aber auch dies half nicht. EVCC ist auf den Tesla berechtigt und den virtuellen Schlüssel habe ich auch zugewiesen. 
 Installiert habe ich auf einem Raspberry 5.
[image]
[image]
[image]

### Steps to reproduce

Ich habe die aktuelle stable Version 0.129.0 installiert und Tesla sowie Token via GUI hinzugefügt. 
Was mir aufgefallen ist als ich den Tesla während der Installation via evcc cinfigure hinzufügen wollte war eine Fehlermeldung das die Daten vom Tesla, Refresh Token und der andere, in einer anderen DB gespeichert werden würden und dann sich vielleicht nicht aktualisieren oder so. 
Ich habe evcc schon mehrfach neu installiert. Ich bin mir gerade unsicher, ob diese Meldung immer kam. 

Aus diesem Grund habe ich mir das hinzufügen während der Ersteinrichtung erspart und habe diesen Punkt übersprungen und den Wagen über die GUI hinzugefügt, in der Hoffnung, dass er dann richtig angelegt werden würde. 

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: XY

interval: 30s # control cycle interval

meters:
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: 1.1.1.1 
  port: 502  
  usage: grid  
  modbus: tcpip  
  timeout: 30s  
  name: grid1
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: 1.1.1.1  
  port: 502  
  usage: pv  
  modbus: tcpip  
  timeout: 30s  
  name: pv2
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: 1.1.1.1  
  port: 502  
  usage: battery  
  modbus: tcpip  
  timeout: 30s  
  name: battery3
- type: template
  template: ac-elwa-2 
  host: 1.1.1.1
  port: 502
  usage: aux
  name: Heizstab

chargers:
- type: template
  template: ellipro 
  ski: SKI  
  name: wallbox4

loadpoints:
- title: Garage
  charger: wallbox4
  mode: pv

site:
  title: solar
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3
    aux:
    - Heizstab
  residualPower: 100
eebus:
  certificate:
      private: |
          -----BEGIN EC PRIVATE KEY-----
          XY
          -----END EC PRIVATE KEY-----
      public: |
          -----BEGIN CERTIFICATE-----
          XY
          -----END CERTIFICATE-----
```

### Log details

```text
welche logs werden benötigt und wie kann ich die am besten aktivieren?
```

### What type of operating system are you running?

Linux

### Version

0.129.0

andig, 2024-08-12:
> Dies hat vor Ewigkeiten mal funktioniert und irgendwann ging es nicht mehr.

Das hat mit Elli noch nie funktioniert- schlicht weil es nicht geht. Der IEC Standard erlaubt nur 6A min.

closed 2024-08-12 by andig

Neffe1337, 2024-08-12:
Ich bin mir da ziemlich sicher, dass es funktioniert. 
1. weil es mal mit EVCC funktioniert hat
2. Weil es über clever-pv.de ebenfalls funktioniert

Er soll ja nicht die Wallbox runterregeln sondern den Tesla.

Neffe1337, 2024-08-12:
@andig kannst du den Thread bitte wieder öffnen?

Neffe1337, 2024-08-12:
Hi zusammen,

wieso glaubt ihr mir nicht wenn ich euch schreibe das es funktionieren kann?
Das untere Limit der Wallbox ist 6A, richtig. Ich kann aber in meiner Tesla App auf 5A stellen und dann lädt die Wallbox anstatt mit 4,2KW nur noch mit 3,5 KW. 

Wenn EVCC oder clever-PV die Steuerung richtig übernimmt, dann geht es sogar auf bis zu 1A runter. Klar ist es ineffizient mit 1A zu laden aber es geht.

Daher meine Bitte den Thread wieder eröffnen.

PS: Es gab auch andere Threads mit ähnlichem oder gleichem Verhalten:
https://github.com/evcc-io/evcc/issues/12427
https://github.com/evcc-io/evcc/issues/13007

Neffe1337, 2024-08-12:
Wenn der Tesla voll ist kommt der Fehler: 
[lp-1 ] ERROR 2024/08/12 15:45:37 vehicle soc limit: Get "https://fleet-api.prd.eu.vn.cloud.tesla.com/api/1/vehicles/XYZ/vehicle_data": net/http: request canceled (Client.Timeout exceeded while awaiting headers)

Vielleicht hat es damit etwas zu tun?

naltatis, 2024-08-12:
In zwei von den verlinkten Issues gehts um den Tesla Wall Connector. Da funktioniert die Regelung anders und die Kommandos werden an das Fahrzeug gesendet (weil technisch nicht anders möglich). Bei allen anderen Wallboxen steuern wir immer(!) nur die Wallbox. Du hast recht, wir könnten zusätzlich zu deiner Wallbox auch noch das Fahrzeug (nur bei Tesla) mitsteuern, aber das machen wir aus Komplexitäts- und Stabilitäsgründen bewusst nicht.

andig, 2024-08-12:
> Das untere Limit der Wallbox ist 6A, richtig. Ich kann aber in meiner Tesla App auf 5A stellen und dann lädt die Wallbox anstatt mit 4,2KW nur noch mit 3,5 KW.

Das kannst Du auch gerne weiter machen. Aber _evcc_ macht nur 6A an der Wallbox, den Rest für 5A musst Du also machen. Wenn der nicht geht liegts also bei Dir ;)