evcc-io/evcc #15379: UI: Waiting for Authentication?

premultiply, 2024-08-13:
mir ist aufgefallen das ich in EVCC leider nicht sehen kann ob die Wallbox die Authentifizierung erfolgreich durchgeführt hat. Stelle nur hin und wieder fest das der Ladevorgang überhaupt nicht startet obwohl der Überschuss erreicht wurde.

_Originally posted by @Timer2000 in https://github.com/evcc-io/evcc/discussions/13412#discussioncomment-10327618_

premultiply, 2024-08-13:
Das Thema haben wir vor allem auch bei OCPP.

Dort ist es momentan notdürftig und übergangsweise über eine Warnung im Log gelöst während auf eine lokale Authentifizierung gewartet wird.

Eine entsprechende Status- bzw. API-Erweiterung wäre daher sinnvoll.

andig, 2024-08-14:
Das bräuchte wohl ein weiteres API das bei `StatusB` abzufragen wäre. Allgemein vielleicht sowas wie

    StatusReason() (api.Reason, error)

Ausgeprägt wäre das dann nur bei OCPP und KEBA? Können wir das bei KEBA erkennen?

premultiply, 2024-08-14:
Muss ich mir im Detail ansehen.
Die besseren Kebas können ja auch OCPP.

An-Wi, 2024-08-14:
Was ist mit Wallboxen, die kein OCPP unterstützen, aber ein eigenes Modbus-Register wie bspw. `ChargePortState` für diese Info haben?

DADAPOWER Modbus Register V1.0 (Mögliche Werte für ChargePortState):
<pre>
0x00 Available
0x01 Unavailable
0x02 Connected
0x03 Authorization
0x04 SelfTest
0x05 WaitForCar
0x06 Charging
0x07 FinishedByStation
0x08 FinishedByCar
0x09 Reserved
0x0A Blocked
0x0B Faulted
0x0C Synchronization
</pre>

Das Thema ist auch bei mir sehr aktuell, da meine Wallbox für jederman zugänglich ist und daher per PIN oder NFC-Card freizuschalten ist.

andig, 2024-08-14:
Siehe Frage nach Keba. Ja!

linuxstony, 2024-08-24:
Wollte nur einmal einwerfen, dass hier bei meiner teilweise OCPP konformen Wallbox ebenfalls 

[garage-1] WARN 2024/08/24 11:00:43 waiting for local authentication

ausgeworfen wird und der Ladevorgang nicht startet. Freeload after restart ist aktiviert aber ich muss den Ladevorgang nun immer mit Chip freigeben (was bei 0.129.0 nicht der Fall war). 

`site ] DEBUG 2024/08/24 11:04:43 site power: 76W
[garage-1] WARN 2024/08/24 11:04:43 waiting for local authentication
[lp-1 ] DEBUG 2024/08/24 11:04:43 charger status: B
[ocpp ] TRACE 2024/08/24 11:04:43 send garage: [2,"3575687391","GetCompositeSchedule",{"connectorId":1,"duration":60}]
[ocpp ] TRACE 2024/08/24 11:04:43 recv garage: [3,"3575687391",{"status":"Accepted","connectorId":1,"scheduleStart":"2024-08-24T09:04:42.001Z","chargingSchedule":{"chargingRateUnit":"A","startSchedule":"2024-08-24T09:04:42.002Z","chargingSchedulePeriod":[{"startPeriod":0,"limit":6.0}]}}]
[lp-1 ] DEBUG 2024/08/24 11:04:43 pv charge current: min 6A > 0A (76W @ 1p, battery: false)
[mqtt ] TRACE 2024/08/24 11:05:01 recv DISnXQ5W/Q5mkHULJ/state: '{"packData":[{"maxTemp":293.1,"sn":"CO4HMDWMEK04770"}],"sn":"EE1HYMCFM230063"}'
[site ] DEBUG 2024/08/24 11:05:12 ----
[lp-1 ] DEBUG 2024/08/24 11:05:12 charge power: 0W
[fritzdect] TRACE 2024/08/24 11:05:12 GET https://192.168.178.1/webservices/homeautoswitch.lua?ain=116570558795&sid=04792657d98cf6ad&switchcmd=getswitchpower
[fritzdect] TRACE 2024/08/24 11:05:12 0
[lp-2 ] DEBUG 2024/08/24 11:05:12 charge power: 0W
[http ] TRACE 2024/08/24 11:05:12 GET http://192.168.178.28/api/inverter/id/0
[http ] TRACE 2024/08/24 11:05:12 {"id":0,"enabled":true,"name":"Balkonkraftwerk","serial":"114181869098","version":"10010","power_limit_read":125,"power_limit_ack":false,"max_pwr":600,"ts_last_success":1724490312,"generation":1,"status":2,"alarm_cnt":1,"rssi":-75,"ts_max_ac_pwr":1724404909,"ts_max_temp":1724411777,"ch":[[236.7,2.3,544.6,50.01,1,29.5,1256.032,290,570.3,95.494,0.5,612.2,46.8],[32.4,8.84,286,148,625.735,76.267,320.5],[32.4,8.78,284.3,142,630.297,75.813,320.5]],"ch_name":["AC","Panel1","Panel2"],"ch_max_pwr":[null,375,375]}
[http ] TRACE 2024/08/24 11:05:12 GET http://192.168.178.28/api/inverter/id/0
[http ] TRACE 2024/08/24 11:05:12 {"id":0,"enabled":true,"name":"Balkonkraftwerk","serial":"114181869098","version":"10010","power_limit_read":125,"power_limit_ack":false,"max_pwr":600,"ts_last_success":1724490312,"generation":1,"status":2,"alarm_cnt":1,"rssi":-75,"ts_max_ac_pwr":1724404909,"ts_max_temp":1724411777,"ch":[[236.7,2.3,544.6,50.01,1,29.5,1256.032,290,570.3,95.494,0.5,612.2,46.8],[32.4,8.84,286,148,625.735,76.267,320.5],[32.4,8.78,284.3,142,630.297,75.813,320.5]],"ch_name":["AC","Panel1","Panel2"],"ch_max_pwr":[null,375,375]}
[site ] DEBUG 2024/08/24 11:05:12 pv power: 545W
[http ] TRACE 2024/08/24 11:05:12 GET http://192.168.178.54/status
[http ] TRACE 2024/08/24 11:05:13 {"wifi_sta":{"connected":true,"ssid":"Schaefergarten","ip":"192.168.178.54","rssi":-44},"cloud":{"enabled":true,"connected":true},"mqtt":{"connected":false},"time":"11:05","unixtime":1724490312,"serial":44653,"has_update":false,"mac":"244CAB42CE42","cfg_changed_cnt":0,"actions_stats":{"skipped":0},"relays":[{"ison":true,"has_timer":false,"timer_started":0,"timer_duration":0,"timer_remaining":0,"overpower":false,"is_valid":true,"source":"input"}],"emeters":[{"power":28.45,"pf":0.22,"current":0.54,"voltage":233.09,"is_valid":true,"total":7803683.5,"total_returned":3560.7},{"power":336.89,"pf":0.80,"current":1.78,"voltage":233.13,"is_valid":true,"total":7396074.6,"total_returned":0.0},{"power":-464.62,"pf":-0.99,"current":1.97,"voltage":233.80,"is_valid":true,"total":851665.5,"total_returned":800585.9}],"total_power":-99.28,"emeter_n":{"current":0.00,"ixsum":2.41,"mismatch":false,"is_valid":false},"fs_mounted":true,"v_data":1,"ct_calst":0,"update":{"status":"idle","has_update":false,"new_version":"20230913-114244/v1.14.0-gcb84623","old_version":"20230913-114244/v1.14.0-gcb84623","beta_version":"20231107-165007/v1.14.1-rc1-g0617c15"},"ram_total":49920,"ram_free":31524,"fs_size":233681,"fs_free":154365,"uptime":2494968}
[site ] DEBUG 2024/08/24 11:05:13 grid meter: -99W
[http ] TRACE 2024/08/24 11:05:13 GET http://192.168.178.54/emeter/0
[http ] TRACE 2024/08/24 11:05:13 {"power":28.45,"pf":0.22,"current":0.54,"voltage":233.09,"is_valid":true,"total":7803683.5,"total_returned":3560.7}
[http ] TRACE 2024/08/24 11:05:13 GET http://192.168.178.54/emeter/1
[http ] TRACE 2024/08/24 11:05:13 {"power":336.89,"pf":0.80,"current":1.78,"voltage":233.13,"is_valid":true,"total":7396074.6,"total_returned":0.0}
[http ] TRACE 2024/08/24 11:05:13 GET http://192.168.178.54/emeter/2
[http ] TRACE 2024/08/24 11:05:13 {"power":-464.62,"pf":-0.99,"current":1.97,"voltage":233.80,"is_valid":true,"total":851665.5,"total_returned":800585.9}
[site ] DEBUG 2024/08/24 11:05:13 grid powers: [28 337 -465]W
[http ] TRACE 2024/08/24 11:05:13 GET http://192.168.178.54/emeter/0
[http ] TRACE 2024/08/24 11:05:13 {"power":28.45,"pf":0.22,"current":0.54,"voltage":233.09,"is_valid":true,"total":7803683.5,"total_returned":3560.7}
[http ] TRACE 2024/08/24 11:05:13 GET http://192.168.178.54/emeter/1
[http ] TRACE 2024/08/24 11:05:13 {"power":336.89,"pf":0.80,"current":1.78,"voltage":233.13,"is_valid":true,"total":7396074.6,"total_returned":0.0}
[http ] TRACE 2024/08/24 11:05:13 GET http://192.168.178.54/emeter/2
[... cut ...]
[ocpp ] TRACE 2024/08/24 11:05:43 send garage: [2,"2709756267","GetCompositeSchedule",{"connectorId":1,"duration":60}]
[ocpp ] TRACE 2024/08/24 11:05:43 recv garage: [3,"2709756267",{"status":"Accepted","connectorId":1,"scheduleStart":"2024-08-24T09:05:42.001Z","chargingSchedule":{"chargingRateUnit":"A","startSchedule":"2024-08-24T09:05:42.002Z","chargingSchedulePeriod":[{"startPeriod":0,"limit":6.0}]}}]
[lp-1 ] DEBUG 2024/08/24 11:05:43 pv charge current: min 6A > 0A (14W @ 1p, battery: false)`

andig, 2024-08-24:
Kauderwelsch. Magst Du das bitte formatieren?

linuxstony, 2024-08-24:
Habs als Datei angehängt wenn das okay ist. Und einen kleinen Teil als Code.

andig, 2024-08-24:
> ausgeworfen wird und der Ladevorgang nicht startet. Freeload after restart ist aktiviert aber ich muss den Ladevorgang nun immer mit Chip freigeben (was bei 0.129.0 nicht der Fall war).

Du hast zwei Möglichkeiten:
- entweder per Chip freischalten
- oder remotestart: true setzen

Kein Problem erkennbar?

linuxstony, 2024-08-24:
Danke - versuche letzteres. Hier besteht nicht die Gefahr des Fremdladens.

andig, 2024-08-24:
Ab morgen sollte die UI den Zustand nun anzeigen

andig, 2024-08-24:
> Was ist mit Wallboxen, die kein OCPP unterstützen, aber ein eigenes Modbus-Register wie bspw. ChargePortState für diese Info haben?

@An-Wi welches Register/Typ wäre das?

closed 2024-08-24 by andig

An-Wi, 2024-08-24:
> @An-Wi welches Register/Typ wäre das?

<pre>
Variable          |   | Data type | Bytes | ModBus 16bit registers | Modbus Register Start | Unit
ChargingPortState | R |  uint16   |   2   |    1                   |   1015                |
</pre>

Ich pack' die PDF einfach noch dazu: [DADAPOWER Modbus Registers.pdf](https://github.com/user-attachments/files/16737021/DADAPOWER.Modbus.Registers.pdf)

andig, 2024-08-24:
Danke- das wäre dann https://github.com/evcc-io/evcc/pull/15654

An-Wi, 2024-08-28:
⚠ Für DadaPower funktioniert es leider noch nicht!

2 Fragen habe ich zum neuen Codeteil in Zeile 120 der **charger/dadapower.go**:
`b, err := wb.conn.ReadHoldingRegisters(dadapowerRegChargingPortState, 2)`

1.) Warum wird hier Holding- und nicht InputRegister gelesen?
2.) Länge des Registers ist nur "1" nicht "2" (so wie auch bei dadapowerRegPlugState in Zeile 93)
`b, err := wb.conn.ReadInputRegisters(dadapowerRegPlugState+wb.regOffset, 1)`

Logisch für mich wäre demzufolge die neue Zeile 120 wie folgt:
`b, err := wb.conn.ReadInputRegisters(dadapowerRegChargingPortState+wb.regOffset, 1)`

andig, 2024-08-28:
> 1.) Warum wird hier Holding- und nicht InputRegister gelesen?

Weil Du nichts davon geschrieben hattest ;) Ist es Input?

andig, 2024-08-28:
So?

An-Wi, 2024-09-02:
> So?

Update 0.130.7 - funktioniert bestens 👍 - Danke ⚡
[image]