evcc-io/evcc #15383: evcc records wrong/implausible charging current

madkiss, 2024-08-14:
### Describe the bug

My EVCC instance will generally believe and record the chaging current to be 11 kW, no matter what the actual chaging current as reported by the charger is. I've seen a few times in the past weeks that EVCC displays a chaging rate of 11 kW despite the Audi app, for instance, showing 6kW only. This leads to wrong calculations of the price for indidual chages and to inappropriate amounts of recorded charge, as can be seen in the attached screenshot (car started charging at 20%, isn't even fully charget, yet the recorded amount of chaged energy exceeds the car's actual battery capacity).

[image]

When resetting my wallbox (Charge Amps Dawn) a few minutes ago to check its local configuration, I also noticed that the L2 fuse had popped (as reported correctly by the chager with a value of 0.000 amps), so there was no way it could have charged with 11 kW. Yet EVCC assumed chaging power to be 11040.

Looking at this again, something appears to be off here way more than I thought. I see the charging rate of 11040W straight and unchanged for hours in the logfile, even for lengthy period of times when the car wasn't charging at all. Is this possibly an interpretion problem between the charger and EVCC?

### Steps to reproduce

1. Connect car to chager, which is connected to EVCC.
2. Trigger charge. 
3. Check records of charged power for plausibility.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.lan # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  site: debug
  lp-1: debug
  cache: error
  db: error
  ocpp: trace

# unique installation id
plant: yalla

interval: 15s # control cycle interval

chargers:
- type: template
  template: ocpp
  timeout: 2m
  name: wallbox1
  meterInterval: 5s

vehicles:
- type: template
  template: audi
  title: e-tron S Sportback
  user: yalla
  password: yalla
  vin: yalla
  capacity: 95
  name: ev1
- type: template
  template: audi
  title: RS e-tron GT
  user: yalla
  password: yalla
  vin: yalla
  capacity: 93.4
  name: ev2

loadpoints:
- title: Garage
  charger: wallbox1
  mode: now

site:
  title: yalla
  meters:

tariffs:
  grid:
    type: tibber
    token: "yalla" # access token
  feedin:
    type: tibber
    token: "yalla" # access token
```

### Log details

```text
Aug 13 16:42:03 evcc evcc[7231]: [ocpp  ] TRACE 2024/08/13 16:42:03 recv C91B3C6D: [2,"dea3247c-e224-4d38-8e1c-be91163b875e","MeterValues",{"connectorId":1,"transactionId":7,"meterValue":[{"timestamp"
:"2024-08-13T14:42:03Z","sampledValue":[{"value":"275065","context":"Sample.Periodic","measurand":"Energy.Active.Import.Register","unit":"Wh"},{"value":"16.000","context":"Sample.Periodic","measurand"
:"Current.Offered","unit":"A"},{"value":"16.006","context":"Sample.Periodic","measurand":"Current.Import","phase":"L1","unit":"A"},{"value":"0.000","context":"Sample.Periodic","measurand":"Current.Imp
ort","phase":"L2","unit":"A"},{"value":"16.124","context":"Sample.Periodic","measurand":"Current.Import","phase":"L3","unit":"A"},{"value":"229.52","context":"Sample.Periodic","measurand":"Voltage","p
hase":"L1","unit":"V"},{"value":"0.78","context":"Sample.Periodic","measurand":"Voltage","phase":"L2","unit":"V"},{"value":"231.42","context":"Sample.Periodic","measurand":"Voltage","phase":"L3","unit
":"V"}]}]}]
Aug 13 16:42:03 evcc evcc[7231]: [ocpp  ] TRACE 2024/08/13 16:42:03 send C91B3C6D: [3,"dea3247c-e224-4d38-8e1c-be91163b875e",{}]
Aug 13 16:42:05 evcc evcc[7231]: [site  ] DEBUG 2024/08/13 16:42:05 ----
Aug 13 16:42:05 evcc evcc[7231]: [lp-1  ] DEBUG 2024/08/13 16:42:05 charge power: 11040W
Aug 13 16:42:05 evcc evcc[7231]: [site  ] DEBUG 2024/08/13 16:42:05 grid meter: 13057W
Aug 13 16:42:05 evcc evcc[7231]: [site  ] DEBUG 2024/08/13 16:42:05 grid currents: [0 0 0]A
Aug 13 16:42:05 evcc evcc[7231]: [site  ] DEBUG 2024/08/13 16:42:05 pv power: 0W
Aug 13 16:42:05 evcc evcc[7231]: [site  ] DEBUG 2024/08/13 16:42:05 site power: 13057W
Aug 13 16:42:05 evcc evcc[7231]: [lp-1  ] DEBUG 2024/08/13 16:42:05 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
Aug 13 16:42:05 evcc evcc[7231]: [lp-1  ] DEBUG 2024/08/13 16:42:05 !! session: chargeRater.chargedEnergy=93.1 - chargedAtStartup=0.0
Aug 13 16:42:05 evcc evcc[7231]: [lp-1  ] ERROR 2024/08/13 16:42:05 charger status: HighTemperature:
```

### What type of operating system are you running?

Linux

### Version

0.128.4

premultiply, 2024-08-14:
Please use the last nightly build and show the output of `evcc charger --diagnose`.

madkiss, 2024-08-14:
Thanks a bunch for the swift reply, here you go:

`root@evcc:~# evcc charger --diagnose
[main  ] INFO 2024/08/14 08:51:21 evcc 0.129.0 (9bf8dc7b)
[main  ] INFO 2024/08/14 08:51:21 using config file: /etc/evcc.yaml
[ocpp-1] DEBUG 2024/08/14 08:51:21 waiting for chargepoint: 5m0s
[ocpp  ] INFO 2024/08/14 08:51:52 charge point connected, registering: C91B3C6D
[ocpp  ] TRACE 2024/08/14 08:51:52 recv C91B3C6D: [2,"bda4c2fc-0bbe-4eec-9f90-ff5a750f6979","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"SuspendedEV"}]
[ocpp  ] TRACE 2024/08/14 08:51:52 send C91B3C6D: [3,"bda4c2fc-0bbe-4eec-9f90-ff5a750f6979",{}]
[ocpp  ] TRACE 2024/08/14 08:51:52 recv C91B3C6D: [2,"30e430b6-464e-47fb-b540-3a9b0d083657","StatusNotification",{"connectorId":0,"errorCode":"NoError","status":"Available"}]
[ocpp  ] TRACE 2024/08/14 08:51:52 send C91B3C6D: [3,"30e430b6-464e-47fb-b540-3a9b0d083657",{}]
[ocpp  ] TRACE 2024/08/14 08:51:53 send C91B3C6D: [2,"463511372","GetConfiguration",{}]
[ocpp  ] TRACE 2024/08/14 08:51:53 recv C91B3C6D: [3,"463511372",{"configurationKey":[{"key":"ACPhaseSwitchingSupported","readonly":true,"value":"true"},{"key":"AllowOfflineTxForUnknownId","readonly":false,"value":"false"},{"key":"AssignedCurrent","readonly":false,"value":"1.16"},{"key":"AssignedPhase","readonly":false,"value":"1.L123"},{"key":"AuthorizationCacheEnabled","readonly":false,"value":"false"},{"key":"AuthorizationKey","readonly":false,"value":""},{"key":"AuthorizeRemoteTxRequests","readonly":false,"value":"false"},{"key":"CableLock","readonly":false,"value":"false"},{"key":"ChargePointID","readonly":false,"value":""},{"key":"ChargeProfileMaxStackLevel","readonly":true,"value":"32"},{"key":"ChargingScheduleAllowedChargingRateUnit","readonly":true,"value":"Current,Power"},{"key":"ChargingScheduleMaxPeriods","readonly":true,"value":"100"},{"key":"ClockAlignedDataInterval","readonly":false,"value":"0"},{"key":"ConnectedWifiSSID","readonly":true,"value":"madkiss"},{"key":"ConnectionTimeOut","readonly":false,"value":"30"},{"key":"ConnectorPhaseRotation","readonly":false,"value":"0.RST"},{"key":"ConnectorPhaseRotationMaxLength","readonly":true,"value":"1"},{"key":"ConnectorSwitch3to1PhaseSupported","readonly":true,"value":"true"},{"key":"EarthingSystem","readonly":false,"value":"TN"},{"key":"FirmwareVersion","readonly":true,"value":"1.4.4"},{"key":"FreeCharging","readonly":false,"value":"true"},{"key":"GetConfigurationMaxKeys","readonly":true,"value":"128"},{"key":"HardwareVersion","readonly":true,"value":"Main board,3.5,1239580377,Processor Board,3.0,1239640577,MID module,2.1 2.0,020100012474C"},{"key":"HeartbeatInterval","readonly":false,"value":"60"},{"key":"InstallationChange","readonly":false,"value":""},{"key":"InstallationCurrent","readonly":false,"value":"0.16"},{"key":"InstallationOfflineCurrent","readonly":false,"value":"1.16"},{"key":"InstallationOfflinePhase","readonly":false,"value":"1.L123"},{"key":"InstallationPhases","readonly":false,"value":"3"},{"key":"LightIntensity","readonly":false,"value":"100"},{"key":"LocalAuthListEnabled","readonly":false,"value":"false"},{"key":"LocalAuthListMaxLength","readonly":true,"value":"200"},{"key":"LocalAuthorizeOffline","readonly":false,"value":"true"},{"key":"LocalPreAuthorize","readonly":false,"value":"false"},{"key":"LoggingStringDebug","readonly":false,"value":""},{"key":"LoggingStringError","readonly":false,"value":"*"},{"key":"LoggingStringInfo","readonly":false,"value":"*"},{"key":"LoggingStringWarning","readonly":false,"value":"*"},{"key":"MaxChargingProfilesInstalled","readonly":true,"value":"6400"},{"key":"MaxCurrent","readonly":true,"value":"0.32,1.32"},{"key":"MaxPhases","readonly":true,"value":"3"},{"key":"MeterValueSampleInterval","readonly":false,"value":"5"},{"key":"MeterValuesAlignedData","readonly":false,"value":""},{"key":"MeterValuesAlignedDataMaxLength","readonly":true,"value":"1"},{"key":"MeterValuesSampledData","readonly":false,"value":"Energy.Active.Import.Register,Current.Offered,Current.Import.L1,Current.Import.L2,Current.Import.L3,Voltage.L1,Voltage.L2,Voltage.L3"},{"key":"MeterValuesSampledDataMaxLength","readonly":true,"value":"10"},{"key":"NumberOfConnectors","readonly":true,"value":"1"},{"key":"ResetRetries","readonly":false,"value":"1"},{"key":"RetryBackOffRandomRange","readonly":false,"value":"60"},{"key":"RetryBackOffRepeatTimes","readonly":false,"value":"5"},{"key":"RetryBackOffWaitMinimum","readonly":false,"value":"30"},{"key":"RfidTagFreeCharging","readonly":false,"value":"00000000000000"},{"key":"SendLocalListMaxLength","readonly":true,"value":"200"},{"key":"Server","readonly":false,"value":"ws://10.42.0.70:8887"},{"key":"StopTransactionOnEVSideDisconnect","readonly":true,"value":"true"},{"key":"StopTransactionOnInvalidId","readonly":false,"value":"false"},{"key":"StopTxnAlignedData","readonly":false,"value":""},{"key":"StopTxnAlignedDataMaxLength","readonly":true,"value":"0"},{"key":"StopTxnSampledData","readonly":false,"value":""},{"key":"StopTxnSampledDataDataMaxLength","readonly":true,"value":"0"},{"key":"SupportedFeatureProfiles","readonly":true,"value":"Core,FirmwareManagement,SmartCharging,LocalAuthListManagement,RemoteTrigger"},{"key":"TransactionMessageAttempts","readonly":false,"value":"3"},{"key":"TransactionMessageRetryInterval","readonly":false,"value":"60"},{"key":"UnlockConnectorOnEVSideDisconnect","readonly":false,"value":"true"},{"key":"UserCurrentLimit","readonly":false,"value":"1.32"},{"key":"WebSocketPingInterva [... cut ...]
		StopTxnAlignedDataMaxLength (r/o): 0
		StopTxnSampledData (r/w):
		StopTxnSampledDataDataMaxLength (r/o): 0
		SupportedFeatureProfiles (r/o): Core,FirmwareManagement,SmartCharging,LocalAuthListManagement,RemoteTrigger
		TransactionMessageAttempts (r/w): 3
		TransactionMessageRetryInterval (r/w): 60
		UnlockConnectorOnEVSideDisconnect (r/w): true
		UserCurrentLimit (r/w): 1.32
		WebSocketPingInterval (r/w): 30
		WifiPassword (r/w):
		WifiSSID (r/w):`
		
Please let me know if you also need the debugging output. Thank you!

premultiply, 2024-08-14:
Should just work with this version now.

closed 2024-08-14 by premultiply