evcc-io/evcc #15389: Zeitstempel der Nissan API ignorierbar? / timestamp of nissan API ignorable?

An-Wi, 2024-08-14:
**Is your feature request related to a problem? Please describe.**
Die Nissan API hat jeden Tag mindestens 1x ein Problem mit dem "timestamp", den sie ausliefert. Meist ist es abends oder in der Nacht und war bisher eher selten ein Problem.
Aktuell ist der Zeitstempel aber etwa 10h alt, der über die API mit ausgeliefert wird (siehe Screenshot), obwohl die inhaltlichen Daten stimmen:
[image]
Die Angaben in der Nissan Connect App (rechts im Screenshot) stimmen vollständig (Standort, Zustand, Ladestand, Reichweite). 
Der reale Ladestand war von gestern Abend 20:00 Uhr bis heute früh 6:30 Uhr bei exakt 65% (also auch zum falschen "timestamp", der momentan via Nissan API zurückgegeben wird. Insofern liefert die Nissan API aktuell einfach nur keinen korrekten timestamp - warum auch immer!

Logischerweise ignoriert evcc die API-Werte, weil der Zeitstempel nicht passt bzw. zu alt ist. Wer geht schon davon aus, dass nur diese eine Datenspalte "veraltet" ist!

**Describe the solution you'd like**
Könnte man eine Möglichkeit schaffen, temporär (über einen manuellen Schalter in der Konfiguration beim Fahrzeug) genau diese "Unstimmigkeit beim Zeitstempel" zu ignorieren, damit evcc mit den "realen" Werten arbeiten kann?

**Describe alternatives you've considered**
Statt "temporär" einen permanent Schalter pro Fahrzeug zum Ignorieren des Zeitstempels (auf eigene Gefahr), wenn das einfach umsetzbar wäre.
Meist fängt sich die Nissan API innerhalb von 24h wieder - vielleicht sollten größere Zeitabweichungen vorsichtshalber dann doch so behandelt werden wie bisher und einen Fehler liefern.

**Additional context**
Schön wäre, wenn die Grund für den fehlenden Ladestand oder den Offline-Status als Info in der Konfigurationsansicht beim Fahrzeug mit angezeigt werden könnte.

andig, 2024-08-14:
> Könnte man eine Möglichkeit schaffen, temporär (über einen manuellen Schalter in der Konfiguration beim Fahrzeug) genau diese "Unstimmigkeit beim Zeitstempel" zu ignorieren, damit evcc mit den "realen" Werten arbeiten kann?

Wir könnten den Timestamp ignorieren oder nicht. Ein Anwender kann und soll nicht entscheiden ob das sinnvoll wäre.

An-Wi, 2024-08-14:
ok, verstehe ich voll.
Dann wäre ich für "ignorieren", weil der Timestamp mehr Problem macht als er Vorteile bringt. 
Gern aber auch in der Variante, dass Zeitstempel älter als 24h wieder zum Fehler führen, denn das spricht dann eher für ein ernstes Problem mit der API.
Bitte nur für `template: nissan-ariya` (ich kann nur für diese API konkrete Aussagen zum Zeitstempel-Verhalten machen)

An-Wi, 2024-08-18:
Zufällig ist der "Nissan Connect Services" seit Freitag für 2 Tage "geplant" offline (Wartungsarbeiten). Soll bis heute (Sonntag) 12:30 Uhr andauern. 

Eine aktuelle Abfrage der API funktioniert nach wie vor, bringt aber logischerweise wirklich alte Werte (Zeitstempel, SOC und Range sind in diesem Fall _alle_ veraltet):
[image]

Insofern wäre die Variante "Ignorie Zeitstempel, wenn dieser nicht älter als 24h ist, und verwende die gelieferten Werte" hierfür auch passend - vielleicht genügen vorerst "**12h**" als Abweichungszeitraum (das würde allgemeine API-Probleme dann wieder schneller sichtbar machen).

Generell auf den Zeitstempel verzichten würde ich nach der Situation heute allerdings **_nicht_** mehr für schlau halten!

Tidabblju, 2024-08-21:
Selbes Problem bei der CUPRA API. Ist mir gestern aufgefallen, dass die Zeiten um ca. 1-2 Minuten nicht stimmen.

closed 2024-09-02 by github-actions[bot]

An-Wi, 2024-09-13:
Das leidliche Thema der Nissan-API ist heute gerade wieder sehr aktuell!

`[nissan] TRACE 2024/09/13 12:20:06 {"id":"***","timestamp":"2024-09-13T07:18:08Z","batteryLevel":40,"batteryAutonomy":215,"batteryCapacity":87,"batteryAvailableEnergy":34,"plugStatus":0,"chargingStatus":0.0,"chargingRemainingTime":451}`

Zeitstempel ist 7h alt, aber alle Werte sind "frisch".
Wäre es denkbar, für das Nissan-Ariya-Template eine Sonderbehandlung beim Zeitstempel zu implementieren? (bis 12h Abweichung als "aktuell" verarbeiten)

Auch in den letzten Tagen trat das Problem der API hin und wieder auf, war aber meist innerhalb von 2-4h wieder "im Lot".

andig, 2024-09-15:
In der Config kann dafür `expiry` auf einen beliebigen Wert gesetzt werden. Geht allerdings nicht mit dem Template.

An-Wi, 2024-09-15:
Mh, zum einen finde ich `expiry` nicht in der Doku (weiß gar nicht, auf welche Einheit sich der Wert bezieht)…
Zum anderen brauche ich es gar nicht probieren, wenn es mit dem Template gar nicht funktioniert - oder könntest du das problemlos im Template integrieren? Denn es klingt sehr verlockend, wäre es doch die ideale Möglichkeit für mich, mit den Werten zu experimentieren…

andig, 2024-09-15:
Machs einfach ohne Template…

An-Wi, 2024-09-16:
Sorry für die doofe Nachfrage, aber stehe gerade total auf dem Schlauch...

Wenn ich:
<pre>
vehicles:
- name: my_car
  type: template
  template: nissan-ariya
</pre>
nicht verwenden soll, wie soll ich meinen Ariya dann definieren, damit er die Nissan API intern called?
So hier?
<pre>
vehicles:
- name: my_car
  type: nissan-ariya
</pre>

Und welche Zeiteinheit hat "expiry" (Sekunden, Minuten, Stunden)?

andig, 2024-09-16:
Die gibst Du mit an:

```yaml
type: nissan
version: v2 # ariya
expiry: 8h
```

An-Wi, 2024-09-16:
aaaaaah, ich liebe "undocumented features" 😁
Vielen herzlichen Dank! Ich habe es integriert und erstmal expiry nur auf "1h" gestellt. Sobald die Nissan API wieder Probleme macht und ich es mitbekomme, versuche ich eine größere Zeitspanne - genial!