evcc-io/evcc #15395: Home assistant addon does not start after gui config

atze09, 2024-08-14:
### Describe the bug

Maybe this issue is lnked to https://github.com/evcc-io/evcc/issues/15122
i just wanted to add a generic vehicle with identifier.
After save and restart the issue appeared

### Steps to reproduce

1. Added car with identifier with gui
2. Also clicked on eebus and couldn't revert, so i deleted the template konfig to empty field.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error
  mqtt: error
# unique installation id
plant: xx

interval: 30s # control cycle interval

sponsortoken: xx
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: true

meters:
- type: template
  template: victron-energy 
  usage: grid  
  host: 192.168.17.3  
  port: 502  
  name: grid1
- type: template
  template: victron-energy 
  usage: pv  
  host: 192.168.17.3  
  port: 502  
  name: pv2
- type: template
  template: victron-energy 
  usage: battery  
  host: 192.168.17.3  
  port: 502  
  name: battery3
  capacity: 29

chargers:
- type: template
  template: go-e 
  host: 192.168.17.83  
  name: wallbox5
- name: openwb
  type: template
  template: openwb-2.0
  # Modbus TCP
  modbus: tcpip
  id: 1
  host: 192.168.17.80 # Hostname
  port: 1502 # Port 
  
vehicles:
- type: template
  title: Corsa  
  icon: car  
  template: flobz
  url: http://192.168.17.7:5000
  vin: xx
  capacity: 46  
  phases: 1  
#  cache: 15m  
  mode: pv  
  minCurrent: 6  
  maxCurrent: 20
  name: ev4
  identifiers:
  - 2
  - 3
  - 0
  priority: 1
- type: template
  template: offline
  title: ZoePV
  icon: car
  capacity: 50
  phases: 3
  mode: pv
  minCurrent: 8
  maxCurrent: 16
  identifiers:
  - HenzePV
  name: ev5
  priority: 2
- type: template
  template: offline
  title: Zoe
  icon: car
  capacity: 50
  phases: 3
  mode: now
  minCurrent: 8
  maxCurrent: 16
  identifiers:
  - HenzeNetz
  name: ev6
  priority: 2
  
loadpoints:
- title: Garage
  charger: openwb
  mode: pv
  phases: 1
  mincurrent: 6
  maxcurrent: 20
  vehicle: ev4
- title: Outdoor
  charger: wallbox5
  mode: pv
  phases: 3
  mincurrent: 8
  maxcurrent: 16
  
mqtt:
  broker: 192.168.17.7:1883
  topic: evcc # root topic for publishing, set empty to disable publishing
  # clientid: foo
  user: xx
  password: xx
  
site:
  title: Home
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3
  residualpower: 200
```

### Log details

```text
Fehler beim Starten. Überprüfe deine Konfiguration und starte neu.
cannot create vehicle 'db:1': cannot create vehicle type 'template': template: eebus.tpl:7:3: executing "eebus.tpl" at <include "vehicle-identify" .>: error calling include: template: vehicle-identify.tpl:19:10: executing "vehicle-identify" at <.identifiers>: range can't iterate over '[1]'

[main ] FATAL 2024/08/14 15:19:20 cannot create vehicle 'db:1': cannot create vehicle type 'template': template: eebus.tpl:7:3: executing "eebus.tpl" at <include "vehicle-identify" .>: error calling include: template: vehicle-identify.tpl:19:10: executing "vehicle-identify" at <.identifiers>: range can't iterate over '[1]'
[main ] FATAL 2024/08/14 15:19:20 will attempt restart in: 15m0s
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.129.0

atze09, 2024-08-14:
Ich habe gerade einen Neustart mit einer leeren Datenbank Datei gemacht, dann startet evcc wieder.
Kann ich die bisherige Datenbank irgendwie reparieren? Gibt es dazu ein vorgeschlagenes addon in home assistant?

naltatis, 2024-08-14:
@atze09 ja, der Bug den du beschreibst sollte, wie du schon vermutest, mit https://github.com/evcc-io/evcc/pull/15224 behoben sein. Dieser Fix ist aber noch nicht im Stable-Release.

**Zur Datenbank:** Ja, das ist eine normale SQLite Datenbank. Der einfachste Weg die Datenbank mit den korrupten Daten wieder ans fliegen zu kriegen ist, wenn du die Tabellen `configs` und `config_details` leerst oder entfernst (werden von evcc neu angelegt). Darin sind nur die per UI konfigurierten Geräte (Fahrzeuge, Zähler, ...) gespeichert. Andere Einstellungen wie Ladelog und so sind davon nicht betroffen.

Eine konkrete Tool-Empfehlung fürs SQLite editieren (in Home Assistant oder auf deinem Rechner) kann ich dir nicht geben, aber da findest du bestimmt etwas.

closed 2024-08-14 by naltatis

atze09, 2024-08-16:
> @atze09 ja, der Bug den du beschreibst sollte, wie du schon vermutest, mit #15224 behoben sein. Dieser Fix ist aber noch nicht im Stable-Release.
>
> **Zur Datenbank:** Ja, das ist eine normale SQLite [...]

Hat funktioniert danke!

naltatis, 2024-08-16:
Das freut mich. Danke für deine Rückmeldung.