evcc-io/evcc #15401: OCPP ZJ Beny charger timeout during OCPP configuration

pdonovan, 2024-08-15:
### Describe the bug

When evcc starts it now configures the OCPP properties. When evcc sends the MeterValueSampleInterval key, my charger will not respond 9/10 times, causing evcc to fail preparing:

`cannot create charger 'zjbeny': cannot create charger type 'template': cannot create charger type 'ocpp': ocpp message (2597022782): GenericError - Request timed out`

Message 2597022782 was 
`send 240: [2,"2597022782","ChangeConfiguration",{"key":"MeterValueSampleInterval","value":"10"}]`

All other configuration keys do seem to work, although sometimes they don't reply instantly. The communication with the EVSE remains open as MeterValues continue to arrive.

This is in nightly build aefe76b2. The issue does not occur in release 0.129.0 because that version does not send configuration changes AFAIK.

The initial configuration response received from the EVSE shows that MeterValueSampleInterval is already set to 10, so the config send it's actually required in this case. I tried setting `meterinterval: 0s` in the evcc.yaml to see if I could stop the interval set being sent, but it still tried to set 10s.
 

### Steps to reproduce

1. Use nightly build aefe76b2
2. Add an ocpp charger
3. Launch evcc with a ZJ Beny CP

### Configuration details

```yaml
chargers:
 - name: zjbeny
   type: template
   template: ocpp
   stationid: 240
   timeout: 2m
#   remotestart: true

loadpoints:
  - title: Garage
    charger: zjbeny
    mode: "pv"
```

### Log details

```text
[main  ] INFO 2024/08/15 14:14:56 evcc 0.129.0 (aefe76b2)
[main  ] INFO 2024/08/15 14:14:56 using config file: /etc/evcc.yaml
[db    ] INFO 2024/08/15 14:14:56 using sqlite database: /home/pi/.evcc/evcc.db
[main  ] INFO 2024/08/15 14:14:59 listening at :7070
[240-1 ] DEBUG 2024/08/15 14:14:59 waiting for chargepoint: 5m0s
[ocpp  ] DEBUG 2024/08/15 14:15:01 charge point connected: 240
[ocpp  ] TRACE 2024/08/15 14:15:01 recv 240: [2,"dts72od2y1rbbwla4h2ryw8u","StatusNotification",{"connectorId":0,"errorCode":"NoError","status":"Available","timestamp":"2024-08-15T04:15:00Z"}]
[ocpp  ] TRACE 2024/08/15 14:15:01 send 240: [3,"dts72od2y1rbbwla4h2ryw8u",{}]
[ocpp  ] TRACE 2024/08/15 14:15:02 recv 240: [2,"n0k6yywmks7ba8vs5x88rj5r","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"SuspendedEVSE","timestamp":"2024-08-15T04:15:00Z"}]
[ocpp  ] TRACE 2024/08/15 14:15:02 send 240: [3,"n0k6yywmks7ba8vs5x88rj5r",{}]
[ocpp  ] TRACE 2024/08/15 14:15:02 send 240: [2,"751522245","GetConfiguration",{}]
[ocpp  ] TRACE 2024/08/15 14:15:02 recv 240: [2,"qd6yrxq15x3u3cc8athnrw1z","MeterValues",{"connectorId":1,"transactionId":1,"meterValue":[{"timestamp":"2024-08-15T04:15:01Z","sampledValue":[{"value":"0","measurand":"Power.Active.Import","format":"Raw","unit":"W"}]}]}]
[240-1 ] DEBUG 2024/08/15 14:15:02 hijacking transaction: 1
[ocpp  ] TRACE 2024/08/15 14:15:02 send 240: [3,"qd6yrxq15x3u3cc8athnrw1z",{}]
[ocpp  ] TRACE 2024/08/15 14:15:03 recv 240: [3,"751522245",{"configurationKey":[{"key":"AllowOfflineTxForUnknownId","readonly":false,"value":"true"},{"key":"AuthorizationCacheEnabled","readonly":false,"value":"true"},{"key":"AuthorizeRemoteTxRequests","readonly":false,"value":"false"},{"key":"ClockAlignedDataInterval","readonly":false,"value":"0"},{"key":"ConnectionTimeOut","readonly":false,"value":"120"},{"key":"ConnectorPhaseRotation","readonly":false,"value":"0.Unknown,1.Unknown"},{"key":"GetConfigurationMaxKeys","readonly":true,"value":"10"},{"key":"HeartbeatInterval","readonly":false,"value":"60"},{"key":"LocalAuthorizeOffline","readonly":false,"value":"true"},{"key":"LocalPreAuthorize","readonly":false,"value":"true"},{"key":"MeterValuesAlignedData","readonly":false,"value":"Voltage,Current.Import,Power.Active.Import,Energy.Active.Import.Register,Temperature"},{"key":"MeterValuesSampledData","readonly":false,"value":"Power.Active.Import"},{"key":"MeterValueSampleInterval","readonly":false,"value":"10"},{"key":"MinimumStatusDuration","readonly":false,"value":"1"},{"key":"NumberOfConnectors","readonly":true,"value":"1"},{"key":"ResetRetries","readonly":false,"value":"1"},{"key":"StopTransactionOnEVSideDisconnect","readonly":false,"value":"true"},{"key":"StopTransactionOnInvalidId","readonly":false,"value":"true"},{"key":"SupportedFeatureProfiles","readonly":true,"value":"Core,FirmwareManagement,LocalAuthListManagement,Reservation,SmartCharging,RemoteTrigger"},{"key":"TransactionMessageAttempts","readonly":false,"value":"3"},{"key":"TransactionMessageRetryInterval","readonly":false,"value":"5"},{"key":"UnlockConnectorOnEVSideDisconnect","readonly":false,"value":"true"},{"key":"WebSocketPingInterval","readonly":false,"value":"30"},{"key":"LocalAuthListEnabled","readonly":false,"value":"true"},{"key":"LocalAuthListMaxLength","readonly":true,"value":"200"},{"key":"SendLocalListMaxLength","readonly":true,"value":"10"},{"key":"ReserveConnectorZeroSupported","readonly":true,"value":"true"},{"key":"ChargeProfileMaxStackLevel","readonly":true,"value":"10"},{"key":"ChargingScheduleAllowedChargingRateUnit","readonly":true,"value":"Current,Power"},{"key":"ChargingScheduleMaxPeriods","readonly":true,"value":"10"},{"key":"ConnectorSwitch3to1PhaseSupported","readonly":true,"value":"false"},{"key":"MaxChargingProfilesInstalled","readonly":true,"value":"20"},{"key":"ChargePointURL","readonly":false,"value":"ws://192.168.0.132:8887/240"},{"key":"ChargePointVendor","readonly":false,"value":"ZJBENY"},{"key":"ChargePointModel","readonly":false,"value":"BCP-B2N-L"},{"key":"ChargePointAuthKey","readonly":false,"value":""},{"key":"ChargePointAuthEnable","readonly":false,"value":"false"}]}]
[ocpp  ] TRACE 2024/08/15 14:15:03 send 240: [2,"896499775","TriggerMessage",{"requestedMessage":"BootNotification","connectorId":1}]
[ocpp  ] TRACE 2024/08/15 14:15:03 recv 240: [3,"896499775",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/08/15 14:15:03 recv 240: [2,"q83lnl5kpuejo9iceg3272pp","BootNotification",{"chargeBoxSerialNumber":"230400240","chargePointModel":"BCP-B2N-L","chargePointVendor":"ZJBENY","firmwareVersion":"1.0.34"}]
[ocpp  ] TRACE 2024/08/15 14:15:03 send 240: [2,"3243444970","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Energy.Active.Import.Register"}]
[ocpp  ] TRACE 2024/08/15 14:15:03 send 240: [3,"q83lnl5kpuejo9iceg3272pp",{"currentTime":"2024-08-15T04:15:03Z","interval":60,"status":"Accepted"}]
[ocpp  ] TRACE 2024/08/15 14:15:03 recv 240: [3,"3243444970",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/08/15 14:15:03 send 240: [2,"3068109582","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Power.Active.Import"}]
[ocpp  ] TRACE 2024/08/15 14:15:04 recv 240: [3,"3068109582",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/08/15 14:15:04 send 240: [2,"171427550","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"SoC"}]
[ocpp  ] TRACE 2024/08/15 14:15:04 recv 240: [3,"171427550",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/08/15 14:15:04 send 240: [2,"909117265","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Current.Offered"}]
[ocpp  ] TRACE 2024/08/15 14:15:04 recv 240: [3,"909117265",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/08/15 14:15:04 send 240: [2,"2119552458","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Power.Offered"}]
[... 15 lines cut ...]
[ocpp  ] TRACE 2024/08/15 14:15:20 send 240: [3,"x01ouvesb9if9xo3xpdki5xb",{}]
[ocpp  ] TRACE 2024/08/15 14:15:30 recv 240: [2,"rteli9c2i8day6lt9ffhuigu","MeterValues",{"connectorId":1,"transactionId":1,"meterValue":[{"timestamp":"2024-08-15T04:15:29Z","sampledValue":[{"value":"423428","context":"Sample.Periodic","measurand":"Energy.Active.Import.Register","format":"Raw","unit":"Wh"}]}]}]
[ocpp  ] TRACE 2024/08/15 14:15:30 send 240: [3,"rteli9c2i8day6lt9ffhuigu",{}]
[ocpp  ] TRACE 2024/08/15 14:15:40 recv 240: [2,"pmgvox5a2gcmj9ejn1vkk7kx","MeterValues",{"connectorId":1,"transactionId":1,"meterValue":[{"timestamp":"2024-08-15T04:15:39Z","sampledValue":[{"value":"423428","context":"Sample.Periodic","measurand":"Energy.Active.Import.Register","format":"Raw","unit":"Wh"}]}]}]
[ocpp  ] TRACE 2024/08/15 14:15:40 send 240: [3,"pmgvox5a2gcmj9ejn1vkk7kx",{}]
[ocpp  ] TRACE 2024/08/15 14:15:50 recv 240: [2,"lm3qpuw73igdq2vfazkrnla8","MeterValues",{"connectorId":1,"transactionId":1,"meterValue":[{"timestamp":"2024-08-15T04:15:49Z","sampledValue":[{"value":"423428","context":"Sample.Periodic","measurand":"Energy.Active.Import.Register","format":"Raw","unit":"Wh"}]}]}]
[... cut ...]
[ocpp  ] TRACE 2024/08/15 14:16:20 send 240: [3,"kr789tgt0pao862hab861xg5",{}]
[ocpp  ] TRACE 2024/08/15 14:16:30 recv 240: [2,"p291bm42nrcmfqlstmqwhflc","MeterValues",{"connectorId":1,"transactionId":1,"meterValue":[{"timestamp":"2024-08-15T04:16:29Z","sampledValue":[{"value":"423428","context":"Sample.Periodic","measurand":"Energy.Active.Import.Register","format":"Raw","unit":"Wh"}]}]}]
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.129.0 (aefe76b2)

pdonovan, 2024-08-15:
I have also tried removing the `timeout: 2m` config line to see if that was a workaround, but it still happened (I haven't looked up what the default value is, or if that's even the correct timeout config variable.

premultiply, 2024-08-15:
@andig We may fix that by just ignoring the missing response and error here
https://github.com/evcc-io/evcc/blob/40b6d1b7f3963e4fea8ec9e291024730eaa9208f/charger/ocpp.go#L305

andig, 2024-08-15:
The "old" solution was not configuring it all by specifying the measurements. I can't see why we should ignore errors. If the charger errors (which is a bad defect) we should not parametrize it.

premultiply, 2024-08-15:
Of course, it is the fault of the wallbox that there is no response here. Clear violation of the spec.
But on the other hand, for us it is not a fatal error that justifies aborting the process as a last resort.

closed 2024-08-15 by andig