evcc-io/evcc #15406: Add Peblar / ProDrive WLAC Wallbox

ItsDominator, 2024-08-15:
Peblar is a brand of Prodrive based in the Netherlands. They have been on the market for several years and produce wallboxes for automotive brands. Now they have their own wallbox for the home, which is also available as a whitelable product. 

The wallbox is available as a non-calibration law version and as a calibration lawversion for the German market. 

I think this will be a competitor to Vestle, Keba and Alfen as the product has good quality and good features at a “cheap” price. 

I have a station with me and can gladly support.

Attachted you will find the documentation of Modbus and REST API. 
[SPD6004230484R03.pdf](https://github.com/user-attachments/files/16623282/SPD6004230484R03.pdf)

premultiply, 2024-08-15:
Looks like it supports OCPP.
Did you try that?

ItsDominator, 2024-08-15:
Yes they support OCPP 1.6 Smart Charging - but you can't enter every OCPP URL you want. The manufacturer has a list of backends and no manual field (as with Alfen). Only these presets can be used. Also, the manufacturer only allows wss connections. 

The wallbox is also connected to a backend for billing via my employer. Modbus control is therefore essential in this case.

andig, 2024-08-16:
Register map looks good- only 1p3p is ugly (but doable). Given we have very little capacity- do you by chance have contact to the vendor and would he support the integration? Please reach out to info@evcc.io.

andig, 2024-08-19:
It would be great if they wanted to become an evcc sponsor!

closed 2024-08-31 by github-actions[bot]

andig, 2024-09-01:
Seems interest in support this box is too rare to justify an implementation and vendor is not interested either. Might revisit at a later time.

andig, 2024-11-06:
Peblar support has been merged.

sneakyvv, 2024-11-07:
> Seems interest in support this box is too rare to justify an implementation and vendor is not interested either. Might revisit at a later time.

I was going to say I was looking for this as well, and ask for it to be implemented, but you guys already did, and only just yesterday... Great timing! :)

Two questions though:
1. Do you hava an idea how fast this might be in a release (I'm new to evcc and Home Assistant, so not sure about the release cycles)
2. How would one configure the charger? I see some [template in the PR](https://github.com/evcc-io/evcc/pull/16451/files#diff-9c8fc427eccddd64b64ffe82dc98fcceb0bd472420e462b8672bc15477d100eb) but first of all I don't know if that's what I should be looking for, and if it is, then I would expect at least an IP address for the configuration, right?

andig, 2024-11-08:
Check out the nightly release 👍🏻

ItsDominator, 2024-12-19:
Hi andig, 

that's fantastic. I am still in talks with the manufacturer regarding sponsorship. Further presumably in February.