evcc-io/evcc #15407: smart-hallo template

mecstroeken, 2024-08-15:
### Describe the bug

Errors when Reading the vehicle Status.
No SOC, Odometer etc. available

### Steps to reproduce

1. Start evcc
2. Switch manually to "my vehicle" 'Smart_BRABUS'
3.
...

### Configuration details

```yaml
vehicles:
	  - name: Smart_BRABUS
	    type: template
	    template: smart-hello
	    title: # Wird in der Benutzeroberfläche angezeigt (optional)
	    icon: car # Icon in der Benutzeroberfläche (optional)
	    user: xxx@gmx.de # Benutzerkonto (bspw. E-Mail Adresse, User Id, etc.)
	    password: xxx # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen)
	    vin: HESxxx # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind (optional)
	    capacity: 64 # Akkukapazität in kWh (optional)
	    phases: 3 # Die maximale Anzahl der Phasen welche genutzt werden können (optional)
	    cache: 15m # Zeitintervall nach dem Daten erneut vom Fahrzeug abgefragt werden (optional)
	    mode: MinPV # Möglich sind Off, Now, MinPV und PV, oder leer wenn keiner definiert werden soll (optional)
	    minCurrent: 6 # Definiert die minimale Stromstärke pro angeschlossener Phase mit welcher das Fahrzeug geladen werden soll (optional)
    maxCurrent: 32
```

### Log details

```text
/app # EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml vehicle
[main  ] INFO 2024/08/15 11:02:49 evcc 0.129.0
[main  ] INFO 2024/08/15 11:02:49 using config file: /config/evcc.yaml
[db    ] INFO 2024/08/15 11:02:49 using sqlite database: /data/evcc.db
[mqtt  ] INFO 2024/08/15 11:02:50 connecting evcc-1605013299 at tcp://192.168.178.100:1883
[mqtt  ] DEBUG 2024/08/15 11:02:50 tcp://192.168.178.100:1883 connected
.Soc:         json: cannot unmarshal number 0.0 into Go struct field .Data.VehicleStatus.AdditionalVehicleStatus.ElectricVehicleStatus.indPowerConsumption of type int: %!s(<nil>)
Capacity:     64.0kWh
Range:        json: cannot unmarshal number 0.0 into Go struct field .Data.VehicleStatus.AdditionalVehicleStatus.ElectricVehicleStatus.indPowerConsumption of type int: %!s(<nil>)
Odometer:     json: cannot unmarshal number 0.0 into Go struct field .Data.VehicleStatus.AdditionalVehicleStatus.ElectricVehicleStatus.indPowerConsumption of type int: %!s(<nil>)
Climater:     json: cannot unmarshal number 0.0 into Go struct field .Data.VehicleStatus.AdditionalVehicleStatus.ElectricVehicleStatus.indPowerConsumption of type int: %!s(<nil>)
Position:     json: cannot unmarshal number 0.0 into Go struct field .Data.VehicleStatus.AdditionalVehicleStatus.ElectricVehicleStatus.indPowerConsumption of type int: %!s(<nil>)
OnIdentified: Mode:minpv, MinCurrent:6, MaxCurrent:32
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.129

andig, 2024-08-15:
Fixed in master

closed 2024-08-15 by andig

mecstroeken, 2024-08-15:
> Fixed in master

Merci!