evcc-io/evcc #15411: Smart #1 - SOC does not work anymore

p-manent, 2024-08-15:
### Describe the bug

Hello, since yesterday (same evcc version 0.129) I cannot get the SOC during charging anymore.

Error:

Garage: vehicle soc: json: cannot unmarshal number 0.0 into Go struct field .Data.VehicleStatus.AdditionalVehicleStatus.ElectricVehicleStatus.indPowerConsumption of type int: %!s(<nil>)

### Steps to reproduce

1. just wait for the error
2. also a complete restart of HA (and evcc) does not solve the problem - still the same error

### Configuration details

```yaml
vehicles:
  - name: Smart
    type: template
    template: smart-hello
    title: Brabus
    user: ...
    password: ...
    vin: HES... 
    capacity: 64 # Akkukapazität in kWh (optional)
    cache: 5m

....

# loadpoint describes the charger, charge meter and connected vehicle
loadpoints:
   - title: Garage # display name for UI
     charger: charger # charger
     vehicle: Smart
```

### Log details

```text
[lp-1 ] ERROR 2024/08/15 13:41:16 vehicle soc: json: cannot unmarshal number 0.0 into Go struct field .Data.VehicleStatus.AdditionalVehicleStatus.ElectricVehicleStatus.indPowerConsumption of type int: %!s(<nil>)
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.129

andig, 2024-08-15:
Fixed in master

closed 2024-08-15 by andig

p-manent, 2024-08-15:
so this will be fixed in Version 0.130 ?