evcc-io/evcc #15421: Poll vehicle soc from mqtt error

atze09, 2024-08-16:
### Describe the bug

After plug in the car and start charging the soc is pulled from mqtt. EVCC gives me the error that the soc is invalid.
But in mqtt it is a plain number. here the mqtt broker information:
[image]

### Steps to reproduce

1. Plug in car
2. Start charging
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error
  mqtt: error
# unique installation id
plant: xx

interval: 30s # control cycle interval

sponsortoken: xx
telemetry: true

meters:
- type: template
  template: victron-energy 
  usage: grid  
  host: 192.168.17.3  
  port: 502  
  name: grid1
- type: template
  template: victron-energy 
  usage: pv  
  host: 192.168.17.3  
  port: 502  
  name: pv2
- type: template
  template: victron-energy 
  usage: battery  
  host: 192.168.17.3  
  port: 502  
  name: battery3
  capacity: 29

chargers:
- type: template
  template: go-e 
  host: 192.168.17.83  
  name: wallbox5
- name: openwb
  type: template
  template: openwb-2.0
  # Modbus TCP
  modbus: tcpip
  id: 1
  host: 192.168.17.80 # Hostname
  port: 1502 # Port 
  
vehicles:
- type: template
  title: Corsa  
  icon: car  
  template: flobz
  url: http://192.168.17.7:5000
  vin: xx
  capacity: 46  
  phases: 1  
#  cache: 15m  
  mode: pv  
  minCurrent: 6  
  maxCurrent: 20
  name: ev4
  identifiers:
  - 2
  - 3
  - 0
  priority: 1
- type: template
  template: offline
#- type: custom
  title: Corsa2
  icon: car
  capacity: 46
  phases: 1
  mode: pv
  minCurrent: 6
  maxCurrent: 20
  identifiers:
  - 2
  - 3
  - 0
  name: ev7
  priority: 1
- type: custom
  title: Corsa24
  icon: car
  capacity: 46
  soc: # battery soc (%)
    source: mqtt
    topic: wican/dc547550cd49/can/corsasoc2
  identifiers:
  - 2
  - 3
  - 0
  name: ev8
#  priority: 1
- type: template
  template: offline
  title: ZoePV
  icon: car
  capacity: 50
  phases: 3
  mode: pv
  minCurrent: 8
  maxCurrent: 16
  identifiers:
  - HenzePV
  name: ev5
  priority: 2
- type: template
  template: offline
  title: Zoe
  icon: car
  capacity: 50
  phases: 3
  mode: now
  minCurrent: 8
  maxCurrent: 16
  identifiers:
  - HenzeNetz
  name: ev6
  priority: 2
  
loadpoints:
- title: Garage
  charger: openwb
  mode: pv
  phases: 1
  mincurrent: 6
  maxcurrent: 20
  vehicle: ev7
- title: Outdoor
  charger: wallbox5
  mode: pv
  phases: 3
  mincurrent: 8
  maxcurrent: 16
  
mqtt:
  broker: 192.168.17.7:1883
  topic: evcc # root topic for publishing, set empty to disable publishing
  # clientid: foo
  user: hassio
  password: xx
  
site:
  title: Home
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3
  residualpower: 200
```

### Log details

```text
[lp-1 ] ERROR 2024/08/16 07:52:24 vehicle soc: wican/dc547550cd49/can/corsasoc2 invalid: '"76"'
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.129.0

StefanSchoof, 2024-08-16:
I am not mqtt expert, but for me this looks like this is a string with 76 and not the number 76.

atze09, 2024-08-16:
> I am not mqtt expert, but for me this looks like this is a string with 76 and not the number 76.

Thanks for the hint it was doubled "" in the mqtt publish automation, now it works

closed 2024-08-16 by atze09

tribun83, 2024-09-16:
> Thanks for the hint it was doubled "" in the mqtt publish automation, now it works

Sorry for capturing. I want to realize SoC pull via wican too (on MG ZS) and don't find much information if/how it works. Would you recommend the new wican Pro for $80 or the normal one for $40? I don't need LTE, just wlan and don't know the newly added protocols and their use... https://www.crowdsupply.com/meatpi-electronics/wican-pro#products