evcc-io/evcc #15435: Since upgrade 127.2 to 129 charging not possible anymore (Keba, Audi templates)

whentotrade, 2024-08-16:
### Describe the bug

I had the evcc and my keba running stable over a long period, until I upgraded from 127.2 to 129.0. 

Got a lot of error messages in the log, like "out of sync: expected enabled, got disabled", or "recv timeout message", I have no clue what to read out of that as my config worked for months now very stable without a single issue, did not change anything. I browsed the docs and comments here but this get above my level to understand all these possible discussions on "autostart" "nostop", OCCP communication, or whatever you might have changed and causes now new issues, I really spend 2 hours now trying almost all settings on my keba box (enabled OCPP completly, disabled RFID, etc.) but dont get it to work. And it worked absoltly perfect before. So there must have been a change on your end?

I moved back to version 127.2 and everything is working back again. But that cant be the solution.

I have no clue what you changed that make it broke. The documentation does not give me any clue and I am not a technician which understand the technical discussion in the issues coming up since the past 2-3 weeks on similar issues.

Can you please help to describe the changes the introduced between 127.2 to 129 that make it broke now? 
Using standard loadpoint and keba/audi templates.

I am even a sponsor.

Thanks &  Regards,
Lars

### Steps to reproduce

Connect car, start charging. Charging is not starting. Info in evcc is: Charging started. Waiting for vehicle...... and nothing is happening.

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

sponsortoken: *****

interval: 10s # control cycle interval

mqtt:
  broker: 192.168.0.40:1883
  topic: evcc # root topic for publishing, set empty to disable publishing
  # clientid: foo
  user: evcc
  password: ****

vehicles:
  - name: AudiEtron
    type: template
    template: audi
    title: ****
    user: ****
    password: ****
    vin: ****
    capacity: 95 # Akku-Kapa in kWh # Optional 
    phases: 3
    identifiers: 
      - "686f6d6561730d0a"

meters:
  - name: my_grid
    type: template
    template: sungrow-hybrid
    usage: grid
    modbus: tcpip
    id: 1
    host: 127.0.0.1  # 192.168.0.96    # LAN Connection WR1 ModBus 62  WiNET LAN WR1
    port: 9000       # 502

  - name: my_pv
    type: template
    template: sungrow-hybrid
    usage: pv
    modbus: tcpip
    id: 1
    host: 127.0.0.1 # 192.168.0.96    # LAN Connection WR1 ModBus 62  WiNET LAN WR1
    port: 9000      # 502

  - name: my_pv2
    type: template
    template: sungrow-hybrid
    usage: pv
    modbus: tcpip
    id: 2
    host: 127.0.0.1    # 192.168.0.166    # WiNET LAN Connection WR2 .97 WiNet / .166 Modbus
    port: 9001    # 502

  - name: my_bat
    type: template
    template: sungrow-hybrid
    usage: battery
    modbus: tcpip
    id: 1
    host: 127.0.0.1   # 192.168.0.96     # LAN Connection WR1 ModBus 62 WiNET LAN WR1
    port: 9000     # 502

chargers:
  - type: template
    template: keba 
    host: 192.168.0.69  
    serial: 23015762
    rfid: 686f6d6561730d0a  
    name: wallbox1

loadpoints:
- title: vt Parkplatz
  charger: wallbox1

site:
  - title: BredbeekCastle
    residualPower: 100
    maxGridSupplyWhileBatteryCharging: 100
    meters:
      battery:
        - my_bat
      grid: my_grid
      pv:
        - my_pv
        - my_pv2

tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.283 # [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.082 # [currency]/kWh
```

### Log details

```text
not available, rolled back to 127.2 to get it up and working again, problems seen are
out of sync: expected enabled, got disabled
recv timeout message
```

### What type of operating system are you running?

Linux

### Version

129.0

VolkerK62, 2024-08-16:
try to update release by release, to find out, which release causes the problem.
`sudo apt install evcc=x.xxx.x # Versionsnummer`

premultiply, 2024-08-16:
There is absolutly nothing about OCPP in your config.

Please try the latest nightly build (Master).

whentotrade, 2024-08-16:
@premultiply: not in the evcc config, but the keba box had a connection to the keba portal via OCPP, so tried to disable the opcc in the keba box (not evcc) and tested with ocpp enabled and disabled on the keba box - but this did not change anything and the problem was there regardless if I disabled or enabled the keba ocpp backend (as saied not in EVCC but I was thinking that the OCPP config in Keba could have caused problems with the latest updates, that was the reason I did almost a lot of trial and error to find a root-cause)

whentotrade, 2024-08-16:
@VolkerK62 : I could do that, but then it required another 1-2h of debug and trial and error. I was thinking there should be some kind of documentation about what has changed in the communication between charger, evcc and car fomr 127.2 to 129? For me it is clear something must have changed, as everything worked over a long period. And nothing touched here. That would help me instead of requiring to debug every release and read the logs. I am a little tiered of fiddling around with 129 now for hours and gave up with rolling back. Hard if breaking changes seem to be introduced without documentation or at least explanation what has changed. Need to find a free slot of time to walk through releases step by step to find the one where it failed. Will report back. At least a good proposal. Thx

whentotrade, 2024-08-16:
@premultiply : What is the sudo command to install the latest nighlty build? I understand the command to pick a version, but not how to install the latest nighly build. Thx

VolkerK62, 2024-08-16:
see the installation guide. https://docs.evcc.io/en/docs/installation/linux
You must add the unstable (nightly) repository.

VolkerK62, 2024-08-16:
just for info:
an other way is, to visit cloudsmith https://cloudsmith.io/~evcc/packages/
there you can pick up every version number and use it with the above mentioned install command
latest nightly = `0.129.0+1723775018`

andig, 2024-08-16:
/
Unfortunately your report is missing a detailed log file. Without log, we don‘t have the information for diagnosing the issue described. See https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen.

closed 2024-08-16 by andig

whentotrade, 2024-08-17:
This issue persists and I have done a complete test and you can see the full logs beldow. As stated already in the initial issue request:

**[lp-1 ] WARN 2024/08/17 07:22:05 charger out of sync: expected enabled, got disabled
[lp-1 ] DEBUG 2024/08/17 07:22:05 charger disable**

You can see in the full log that after connecting the car, switching to minpv, it is still stuck in the UI with the label "Ready. Waiting for vehicle..." and is stuck there forever.

I mentioned these issues already in the initial request. Here is the full log, it is not completed.

@TobiasHuber1980 , thanks for your suggestion! I did this, but the problem is still the same.

The Keba box itself still reports the state "idle" / "warten"

[site ] DEBUG 2024/08/17 07:20:04 ----
[lp-1 ] DEBUG 2024/08/17 07:20:05 charge power: 0W
[lp-1 ] DEBUG 2024/08/17 07:20:05 charge currents: [0 0 0]A
[site ] DEBUG 2024/08/17 07:20:05 pv power: 342W
[site ] DEBUG 2024/08/17 07:20:05 battery soc: 0%
[site ] DEBUG 2024/08/17 07:20:05 battery power: 0W
[site ] DEBUG 2024/08/17 07:20:05 grid meter: -0W
[site ] DEBUG 2024/08/17 07:20:05 site power: 100W
[lp-1 ] DEBUG 2024/08/17 07:20:05 !! session: chargeRater.chargedEnergy=1.0 - chargedAtStartup=0.0
[lp-1 ] DEBUG 2024/08/17 07:20:05 charge total import: 13312.418kWh
[lp-1 ] DEBUG 2024/08/17 07:20:05 charger status: A
[lp-1 ] INFO 2024/08/17 07:20:05 car disconnected
[lp-1 ] DEBUG 2024/08/17 07:20:05 !! active phases: 1p = min(0p measured 3p vehicle 1p physical 0p charger)
[lp-1 ] INFO 2024/08/17 07:20:05 vehicle updated: HH-VT-ETRON -> unknown
[lp-1 ] DEBUG 2024/08/17 07:20:05 !! active phases: 1p = min(0p measured 0p vehicle 1p physical 0p charger)
[lp-1 ] DEBUG 2024/08/17 07:20:05 set session soc limit: 0
[lp-1 ] DEBUG 2024/08/17 07:20:05 set session energy limit: 0
[site ] DEBUG 2024/08/17 07:20:14 ----
[lp-1 ] DEBUG 2024/08/17 07:20:14 charge power: 0W
[lp-1 ] DEBUG 2024/08/17 07:20:14 charge currents: [0 0 0]A
[site ] DEBUG 2024/08/17 07:20:14 pv power: 322W
[site ] DEBUG 2024/08/17 07:20:15 battery soc: 0%
[site ] DEBUG 2024/08/17 07:20:15 battery power: 0W
[site ] DEBUG 2024/08/17 07:20:15 grid meter: 5W
[site ] DEBUG 2024/08/17 07:20:15 site power: 105W
[lp-1 ] DEBUG 2024/08/17 07:20:15 !! session: chargeRater.chargedEnergy=1.0 - chargedAtStartup=0.0
[lp-1 ] DEBUG 2024/08/17 07:20:15 charge total import: 13312.418kWh
[lp-1 ] DEBUG 2024/08/17 07:20:15 charger status: A
[site ] DEBUG 2024/08/17 07:20:24 ----
[lp-1 ] DEBUG 2024/08/17 07:20:24 charge power: 0W
[lp-1 ] DEBUG 2024/08/17 07:20:24 charge currents: [0 0 0]A
[site ] DEBUG 2024/08/17 07:20:24 pv power: 310W
[site ] DEBUG 2024/08/17 07:20:25 battery soc: 0%
[site ] DEBUG 2024/08/17 07:20:25 battery power: 0W
[site ] DEBUG 2024/08/17 07:20:25 grid meter: 13W
[site ] DEBUG 2024/08/17 07:20:25 site power: 113W
[lp-1 ] DEBUG 2024/08/17 07:20:25 !! session: chargeRater.chargedEnergy=1.0 - chargedAtStartup=0.0
[lp-1 ] DEBUG 2024/08/17 07:20:25 charge total import: 13312.418kWh
[lp-1 ] DEBUG 2024/08/17 07:20:25 charger status: A
[site ] DEBUG 2024/08/17 07:20:34 ----
[lp-1 ] DEBUG 2024/08/17 07:20:34 charge power: 0W
[lp-1 ] DEBUG 2024/08/17 07:20:34 charge currents: [0 0 0]A
[site ] DEBUG 2024/08/17 07:20:34 pv power: 310W
[site ] DEBUG 2024/08/17 07:20:35 battery soc: 0%
[site ] DEBUG 2024/08/17 07:20:35 battery power: 0W
[site ] DEBUG 2024/08/17 07:20:35 grid meter: 14W
[site ] DEBUG 2024/08/17 07:20:35 site power: 114W
[lp-1 ] DEBUG 2024/08/17 07:20:35 !! session: chargeRater.chargedEnergy=1.0 - chargedAtStartup=0.0
[lp-1 ] DEBUG 2024/08/17 07:20:35 charge total import: 13312.418kWh
[lp-1 ] DEBUG 2024/08/17 07:20:35 charger status: A
[site ] DEBUG 2024/08/17 07:20:44 ----
[lp-1 ] DEBUG 2024/08/17 07:20:44 charge power: 0W
[lp-1 ] DEBUG 2024/08/17 07:20:44 charge currents: [0 0 0]A
[site ] DEBUG 2024/08/17 07:20:44 pv power: 298W
[site ] DEBUG 2024/08/17 07:20:45 battery soc: 0%
[site ] DEBUG 2024/08/17 07:20:45 battery power: 0W
[site ] DEBUG 2024/08/17 07:20:45 grid meter: 18W
[... 194 lines cut ...]
[site ] DEBUG 2024/08/17 07:22:55 battery soc: 0%
[site ] DEBUG 2024/08/17 07:22:55 battery power: 0W
[site ] DEBUG 2024/08/17 07:22:55 grid meter: 1883W
[site ] DEBUG 2024/08/17 07:22:55 site power: 1983W
[lp-1 ] DEBUG 2024/08/17 07:22:55 charge total import: 13312.418kWh
[lp-1 ] DEBUG 2024/08/17 07:22:55 charger status: B
[lp-1 ] DEBUG 2024/08/17 07:22:55 !! active phases: 1p = min(0p measured 3p vehicle 1p physical 0p charger)
[lp-1 ] DEBUG 2024/08/17 07:22:55 pv charge current: min 6A > 0A (1983W @ 1p, battery: false)
[site ] DEBUG 2024/08/17 07:23:04 ----
[lp-1 ] DEBUG 2024/08/17 07:23:05 charge power: 0W
[lp-1 ] DEBUG 2024/08/17 07:23:05 charge currents: [0 0 0]A
[site ] DEBUG 2024/08/17 07:23:05 pv power: 175W
[site ] DEBUG 2024/08/17 07:23:05 battery soc: 0%
[site ] DEBUG 2024/08/17 07:23:05 battery power: 0W
[site ] DEBUG 2024/08/17 07:23:05 grid meter: 1162W
[site ] DEBUG 2024/08/17 07:23:05 site power: 1262W
[lp-1 ] DEBUG 2024/08/17 07:23:05 charge total import: 13312.418kWh
[lp-1 ] DEBUG 2024/08/17 07:23:05 charger status: B
[lp-1 ] WARN 2024/08/17 07:23:05 charger out of sync: expected enabled, got disabled
[lp-1 ] DEBUG 2024/08/17 07:23:05 charger disable
[lp-1 ] DEBUG 2024/08/17 07:23:05 !! active phases: 1p = min(0p measured 3p vehicle 1p physical 0p charger)
[lp-1 ] DEBUG 2024/08/17 07:23:05 pv charge current: min 6A > 0A (1262W @ 1p, battery: false)
[lp-1 ] DEBUG 2024/08/17 07:23:05 charger enable
[lp-1 ] DEBUG 2024/08/17 07:23:05 wake-up timer: start
[site ] DEBUG 2024/08/17 07:23:14 ----

whentotrade, 2024-08-17:
@andig :Not sure why you closed it so swiftly now forcing me to reopen as new issue: #15441 #15441

andig, 2024-08-17:
> Not sure why you closed it so swiftly

Geschlossen da unvollständig. Siehe Issuetemplate.