evcc-io/evcc #15437: Heidelberg Wallbox Timeout issues

KalkiDE, 2024-08-16:
### Describe the bug

After update of EVCC from v0.122.1 to v0.129.0 (required due to update of SEAT/CUPRA API),
I have regular timeout issues with my Heidelberg Wallbox, connected via TCP/IP gateway.
Readout of power meter works w/o issues with same TCP/IP gateway.

### Steps to reproduce

1. Multiple restarts of EVCC
2. Multiple restarts of TCP/IP gateway
3. Multiple restarts of Heiderberg Wallbox

### Configuration details

```yaml
- name: meter_wallbe
    type: template
    template: eastron
    usage: charge
    # RS485 via TCP/IP (Modbus RTU)
    modbus: rs485tcpip
    id: 1
    host: 10.1.12.20 # Hostname
    port: 502 # Port

  - name: meter_heidelberg
    type: template
    template: eastron
    usage: charge
    # RS485 via TCP/IP (Modbus RTU)
    modbus: rs485tcpip
    id: 2
    host: 10.1.12.20 # Hostname
    port: 502 # Port 

chargers:
  - name: charger_heidelberg
    type: template
    template: heidelberg 
    # RS485 via TCP/IP (Modbus RTU)
    modbus: rs485tcpip
    id: 3
    host: 10.1.12.20 # Hostname
    port: 502 # Port 

vehicles:
  - name: vehicle_leon
    title: SEAT Leon Sportstourer
    type: seat
    user: xxx
    password: xxx
    capacity: 10.4
    onIdentify:
      mode: pv

loadpoints:
  - title: Heidelberg
    charger: charger_heidelberg
    mode: "off"
    meter: meter_heidelberg
    vehicle: vehicle_leon
    enable:
      delay: 2m
      threshold: -1600 # es wird eingeschaltet, wenn 2 Minuten lang 200 W Einspeisung vorhanden sind
    disable:
      delay: 5m
      threshold: 300 # es wird ausgeschaltet, wenn 5 Minuten lang mehr als 300 W aus dem Netz bezogen werden
```

### Log details
[... 410 lines cut ...]
[lp-1 ] DEBUG 2024/08/16 19:11:07 charge voltages: [231 233 232]V
[lp-1 ] DEBUG 2024/08/16 19:11:07 detected connected phases: 3p
[lp-1 ] DEBUG 2024/08/16 19:11:07 charge total import: 1151.609kWh
[lp-1 ] DEBUG 2024/08/16 19:11:07 charger status: B
[lp-1 ] DEBUG 2024/08/16 19:11:07 limitSoc reached: 92.0% > 80%
[site ] DEBUG 2024/08/16 19:11:37 ----
[lp-1 ] DEBUG 2024/08/16 19:11:37 charge power: 0W
[lp-1 ] DEBUG 2024/08/16 19:11:37 charge currents: [0 0 0]A
[site ] DEBUG 2024/08/16 19:11:37 pv power: 106W
[site ] DEBUG 2024/08/16 19:11:37 grid meter: 317W
[site ] DEBUG 2024/08/16 19:11:37 site power: 417W
[lp-1 ] DEBUG 2024/08/16 19:11:37 charge voltages: [232 233 232]V
[lp-1 ] DEBUG 2024/08/16 19:11:37 detected connected phases: 3p
[lp-1 ] DEBUG 2024/08/16 19:11:37 charge total import: 1151.609kWh
[lp-1 ] DEBUG 2024/08/16 19:11:37 charger status: B
[lp-1 ] DEBUG 2024/08/16 19:11:37 limitSoc reached: 92.0% > 80%
```

### What type of operating system are you running?

Linux

### Version

0.129.0

premultiply, 2024-08-16:
Please try the latest nightly build.

KalkiDE, 2024-08-17:
Thanks - and I have to correct my statement of the used version (I'm yet not so familiar with github).
I notified that I falsly downloaded the latest commit (therefore more or less what you mentioned with nightly build) and not the tagged release 0.129.0.

I just "updated" now to the released version and communication is running stable.
It looks like the latest change (Modbus: add UDP and allow concurrent access (https://github.com/evcc-io/evcc/pull/13676)), is causing issues.
Please let me know if I can support here with testing before next release.

andig, 2024-08-17:
@KalkiDE kannst du selbst compilieren? Oder: deine beiden Geräte über socat/ssh exportieren so dass ich das lokal verifizieren kann? Dann bitte Mail an info@evcc.io.

KalkiDE, 2024-08-18:
> @KalkiDE kannst du selbst compilieren? Oder: deine beiden Geräte über socat/ssh exportieren so dass ich das lokal verifizieren kann? Dann bitte Mail an [info@evcc.io](mailto:info@evcc.io).

Kompilieren ist kein Problem - läuft bei mir in einem Docker-Container.

andig, 2024-08-18:
Probier mal bitte den aktuellen Master- da sind 100ms Delay drin die es im alten Code- eigentlich unnötigerweise- auch gab. Nightly baut auch gerade.

KalkiDE, 2024-08-18:
So, erstes Feedback (Auto ist aktuell zu voll, um die Ladung selber auszuprobieren):
Es scheint mit dem Delay zu funktionieren ... mir ist aber generell schon häufiger aufgefallen, dass die Heidelberg WB auch etwas zickig ist, was die Kommunikation angeht. 
Die Daten vom Powermeter (gleicher Modbus, Abfrage erfolgt aber, soweit ich das im Trace sehe, vor der Kommunikation mit der Wallbox) kommen bislang ohne Probleme an.

Danke erstmal!

andig, 2024-08-18:
Closed in https://github.com/evcc-io/evcc/commit/9ad7399b1d738605964cbb8d483ed83c356f6f4b

closed 2024-08-18 by andig

andig, 2024-09-25:
@KalkiDE könntest Du mal probieren ob es auch ohne das Delay geht wenn Du stattdessen noch zusätzlich https://github.com/grid-x/modbus/pull/93 mit ein baust?