evcc-io/evcc #15438: [EEBus] Porsche Mobile Charger Plus Probleme beim Verbindungsaufbau

Neverminda, 2024-08-16:
### Describe the bug

Als erstes ein riesiges Lob an alle die EVCC möglich gemacht haben.

Da meine Pulsar in Verbindung mit EVCC beim Taycan nicht ganz so smooth läuft wie mein ID dachte ich mit ich versuche mal mein Glück mit dem PMCP. Nach dem ich es nun eine (hoffentlich) vernünftige config hinbekommen haben, und der EEBus Support noch Experimentell ist wie es nicht verpassen und ein paar Erfahrungen mit euch teilen.

Es war ziemlich schwierig den PMCP dazu zu überreden eine Verbindung mit EVCC herzustellen. Geklappt hat es letztendlich mit ver Version 0.128.1. Andere Versionen (0.129.0, 0.129.0 (3798d4ea), 0.127.1) wurden vom PMCP gar nicht "akzeptiert"

Leider kann auch die Version 0.128.1 keinen Status des PMCP erkennen, zumindest in der UI ist nicht zu sehen.

In den Logs sind mir 2 Dinge aufgefallen:
`[eebus ] DEBUG 2024/08/16 16:41:38 connection to 6c401f7671f64d682ed5a2e048a3efb9de7669db failed: dial tcp: lookup MobileChargerPlus-0048524.local on 192.168.9.1:53: no such host
`
MobileChargerPlus-0048524 läuft unter der IP **192.168.9.11** auf **192.168.9.1** läuft der auch der DNS (scheint nur einer von mehreren Versuchen zu sein)

interessanter ist glaube ich:
`SHIP handshake error: invalid character '\x00' after top-level value`

Ihr wisst sicher mehr mit den Logs anzufangen.

Config:
[evcc.yaml.txt](https://github.com/user-attachments/files/16639795/evcc.yaml.txt)

Logs: 
[evcc-20240816-173503-debug.log](https://github.com/user-attachments/files/16639580/evcc-20240816-173503-debug.log)
[evcc-20240816-170025-debug.log](https://github.com/user-attachments/files/16639583/evcc-20240816-170025-debug.log)
[evcc-20240816-164714-debug.log](https://github.com/user-attachments/files/16639586/evcc-20240816-164714-debug.log)
[evcc-20240816-154719-debug.log](https://github.com/user-attachments/files/16639588/evcc-20240816-154719-debug.log)

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local  #192.168.9.8 # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

#database configuration for persisting charge sessions and settings
database:
  type: sqlite
  dsn: /var/lib/evcc/evcc.db

# unique installation id
plant: 76ae1e651abbdb6bd4deecdb7caf6feb3f0292cef776fc0c4c6af9d4541fa524

interval: 30s # control cycle interval

tariffs:
  currency: EUR
  grid:
    type: fixed
    price: 0.2882
    zones:
      - days: Mo-So
        hours: 22-0
        price: 0.2024
      - days: Mo-So
        hours: 0-6
[... 200 lines cut ...]
[eebus ] DEBUG 2024/08/16 16:41:38 connection to 6c401f7671f64d682ed5a2e048a3efb9de7669db failed: dial tcp: lookup MobileChargerPlus-0048524.local on 192.168.9.1:53: no such host
[eebus ] DEBUG 2024/08/16 16:41:38 trying to connect to 6c401f7671f64d682ed5a2e048a3efb9de7669db at 192.168.9.11
[eebus ] DEBUG 2024/08/16 16:41:38 initiating connection to 6c401f7671f64d682ed5a2e048a3efb9de7669db at 192.168.9.11:82/ship/
[eebus ] DEBUG 2024/08/16 16:41:39 6c401f7671f64d682ed5a2e048a3efb9de7669db SHIP handshake error: invalid character '\x00' after top-level value
[eebus ] DEBUG 2024/08/16 16:41:39 ski 6c401f7671f64d682ed5a2e048a3efb9de7669db disconnected
[eebus ] DEBUG 2024/08/16 16:41:39 mdns: announce
[eebus ] DEBUG 2024/08/16 16:41:39 mdns: using avahi
[eebus ] DEBUG 2024/08/16 16:41:39 delaying connection to 6c401f7671f64d682ed5a2e048a3efb9de7669db by 18.253s to minimize double connection probability
[eebus ] DEBUG 2024/08/16 16:41:41 ski: 6c401f7671f64d682ed5a2e048a3efb9de7669db name: Porsche-EVSE-0048524 brand: Porsche model: BasisPlus-EVSE typ: EVSE identifier: Porsche-EVSE-0048524 register: false host: MobileChargerPlus-0048524.local port: 82 addresses: [192.168.9.11]
[eebus ] DEBUG 2024/08/16 16:41:57 trying to connect to 6c401f7671f64d682ed5a2e048a3efb9de7669db at MobileChargerPlus-0048524.local
[eebus ] DEBUG 2024/08/16 16:41:57 initiating connection to 6c401f7671f64d682ed5a2e048a3efb9de7669db at MobileChargerPlus-0048524.local:82/ship/
[eebus ] DEBUG 2024/08/16 16:41:58 connection to 6c401f7671f64d682ed5a2e048a3efb9de7669db failed: dial tcp: lookup MobileChargerPlus-0048524.local on 192.168.9.1:53: no such host
[eebus ] DEBUG 2024/08/16 16:41:58 trying to connect to 6c401f7671f64d682ed5a2e048a3efb9de7669db at 192.168.9.11
[eebus ] DEBUG 2024/08/16 16:41:58 initiating connection to 6c401f7671f64d682ed5a2e048a3efb9de7669db at 192.168.9.11:82/ship/
[eebus ] DEBUG 2024/08/16 16:41:59 6c401f7671f64d682ed5a2e048a3efb9de7669db SHIP handshake error: invalid character '\x00' after top-level value
[eebus ] DEBUG 2024/08/16 16:41:59 ski 6c401f7671f64d682ed5a2e048a3efb9de7669db disconnected
```

### What type of operating system are you running?

Linux

### Version

multiple

DerAndereAndi, 2024-08-16:
Danke für den Report. Wäre es möglich ein Log mit EEBUS auf 'TRACE'  level zu bekommen?

Neverminda, 2024-08-16:
Klar,
hier zwei von der aktuellen nightly.
Einmal mit restart ohne restart der PMCP und der 2. mit restart.

[evcc-20240816-233038-trace.log](https://github.com/user-attachments/files/16641966/evcc-20240816-233038-trace.log)
[evcc-20240816-233546-trace.log](https://github.com/user-attachments/files/16641968/evcc-20240816-233546-trace.log)

DerAndereAndi, 2024-08-17:
Danke für die neuen Logs. Die PMCP schickt ungültige Daten in 2 Nachrichten welche nicht verarbeitet werden können. Das sieht für mich nach einem Bug in der Software des Gerätes aus. Ich könnte versuchen rauszufiltern, aber sehe das nicht als oberste Prio zur Zeit an.

andig, 2024-08-17:
Machen wir hier zu- kein evcc Thema 👍🏻

closed 2024-08-17 by andig