evcc-io/evcc #15441: KEBA/Audi templates not working after mirgration from 127.2 to 129 - charger got disabled due to "out of sync" logs message

whentotrade, 2024-08-17:
### Describe the bug

Keba box not starting and kept stuck into mode **"Ready. Waiting for vehicle..."** The instance was working in version 127.2 without problems stable for a very long time. Not touched. Now after upgrading to 129 it has broken the instance.

The key lines I read seem to be:
[lp-1 ] WARN 2024/08/17 07:22:05 charger out of sync: expected enabled, got disabled
[lp-1 ] DEBUG 2024/08/17 07:22:05 charger disable

Changed the Keba configuration from UDP to modbus according to new docs. Did not help.

Browsed docs and issues here for hours. Got out of ideas and time to debug test. My only solution is to roll back to 127.2

Any help and suggestion is welcome.

### Steps to reproduce

1. Connect car
2. Switch mode to minpv (or now)
3. EVCC shows "Ready. Waiting for verhicle..."
4. Nothing happens.
5. Keba UI shows box is idle/waiting

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

sponsortoken: ***

interval: 10s # control cycle interval

mqtt:
  broker: 192.168.0.40:1883
  topic: evcc # root topic for publishing, set empty to disable publishing
  user: evcc
  password: ***

vehicles:
  - name: AudiEtron
    type: template
    template: audi
    title: HH-VT-ETRON
    user: ***
    password: ***
    vin: ***
    capacity: 95 # Akku-Kapa in kWh # Optional 
    phases: 3
    identifiers: 
       - 686f6d65
 #     - "686f6d6561730d0a"

meters:
  - name: my_grid
    type: template
    template: sungrow-hybrid
    usage: grid
    modbus: tcpip
    id: 1
    host: 127.0.0.1  # 192.168.0.96    # LAN Connection WR1 ModBus 62  WiNET LAN WR1
    port: 9000       # 502

  - name: my_pv
    type: template
[... 334 lines cut ...]
[lp-1 ] DEBUG 2024/08/17 07:23:05 charge power: 0W
[lp-1 ] DEBUG 2024/08/17 07:23:05 charge currents: [0 0 0]A
[site ] DEBUG 2024/08/17 07:23:05 pv power: 175W
[site ] DEBUG 2024/08/17 07:23:05 battery soc: 0%
[site ] DEBUG 2024/08/17 07:23:05 battery power: 0W
[site ] DEBUG 2024/08/17 07:23:05 grid meter: 1162W
[site ] DEBUG 2024/08/17 07:23:05 site power: 1262W
[lp-1 ] DEBUG 2024/08/17 07:23:05 charge total import: 13312.418kWh
[lp-1 ] DEBUG 2024/08/17 07:23:05 charger status: B
[lp-1 ] WARN 2024/08/17 07:23:05 charger out of sync: expected enabled, got disabled
[lp-1 ] DEBUG 2024/08/17 07:23:05 charger disable
[lp-1 ] DEBUG 2024/08/17 07:23:05 !! active phases: 1p = min(0p measured 3p vehicle 1p physical 0p charger)
[lp-1 ] DEBUG 2024/08/17 07:23:05 pv charge current: min 6A > 0A (1262W @ 1p, battery: false)
[lp-1 ] DEBUG 2024/08/17 07:23:05 charger enable
[lp-1 ] DEBUG 2024/08/17 07:23:05 wake-up timer: start
[site ] DEBUG 2024/08/17 07:23:14 ----
```

### What type of operating system are you running?

Linux

### Version

129.0

whentotrade, 2024-08-17:
Reference: #15435

whentotrade, 2024-08-17:
At least some trial and error follow ups:

If I disable RFID authorization in the KEBA portal, it seems to work. So maybe it get stuck in authorization of the charging start? But at least this must be due to a change between 127.2 and 129 and the RFID worked without any issues before. Maybe the root caues is somewhere in how RFID authorization is handled? As I need RFID, disabling authorization in the box is not a final solution. But might help nailing down the root cause?

TobiasHuber1980, 2024-08-17:
welche keba hast du? auch mal bitte die produktnummer. sind die rfid-karten auch in der weboberfläche der WB sichtbar?

whentotrade, 2024-08-17:
Keba P30 x series, firmware 1.18
RFID karten sind lokal in der box hinterlegt, hatte vorher die einstellung Authorization auf "OnlyLocal" hatte wie gesagt vorher prima funktioniert. Habe jetzt die komplette RFID authorization of "no" gesetzt.

whentotrade, 2024-08-17:
Das waren meine RFID Einstellung in WebUI der Keba:
[image]

premultiply, 2024-08-17:
Bitte mal zwei Dinge probieren:

1. aktuelles Nightly Build verwenden und funktional vergleichen
2. damit die Box per OCPP anbinden und funktional vergleichen

whentotrade, 2024-08-17:
@premultiply , danke, aber sudo apt install evcc=x.xxx.x mit der nighly funktioniert nicht

`sudo apt install evcc=0.129.0+1723861311`

gibt:

```
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Version '0.129.0+1723861311' for 'evcc' was not found
```
Mach ich da was falsch?

premultiply, 2024-08-17:
Ja, siehe Doku.
Erstmal das Nightly Repository installieren.

TobiasHuber1980, 2024-08-17:
umstieg auf nightly:
`curl -1sLf 'https://dl.evcc.io/public/evcc/unstable/setup.deb.sh' | sudo -E bash`

dann installieren:
```
sudo systemctl stop evcc
sudo apt update && sudo apt upgrade -y && sudo apt autoremove -y
sudo systemctl start evcc
```

whentotrade, 2024-08-17:
Danke für Eure Hinweise und Hilfe erstmal!

Update auf Nightly erfolgt, gleiches verhalten. Wenn Ladekabel neu eingesteckt wird, startet der Ladevorgang nicht. EVCC bleibt im Modus `"Ready. Waiting for vehicle..."` stecken und ich sehe im Log wieder die Warnung `"charger out of sync: expected enabled, got disabled"`

Wenn ich im Keba UI die RFID Authorization ausschalte, tritt das Problem nicht auf. Die Karten sind in der Keba UI Wallbox Whitelist hinterlegt, das funktionierte mit der 127.2 und RFID Autorisierung bislang einwandfrei. Genau mit diesen Einstellungen.

Irgendwo muss im Flow und der RFID Autorisierung was schief gehen?

Ich habe mir mal da direkte KEBA diagnose log von der Box geladen und da sieht man, das es einen authorisierungsversuch ohne RFID gab. Danach kommen keine Einträge mehr im Diagnose Log - und in EVCC bleibt es auf "Ready. Waiting for vehicle.."
Ich hab das komplette KEBA Log angehängt.
[wallbox_23015762.log](https://github.com/user-attachments/files/16644123/wallbox_23015762.log)

KEBA Wallbox Log:
```
2024-08-17 08:03:11,866 DEBUG - LogEvt: Event time: [2024-08-17 08:03:11,853] [23015762] [CPLD] (0:14,01) EvPlugged,Ready4Charge;HW_Lock
2024-08-17 08:03:11,920 DEBUG - LogEvt: Event time: [2024-08-17 08:03:11,875] [23015762] Pilot mapped : 1
2024-08-17 08:03:12,039 DEBUG - LogEvt: Event time: [2024-08-17 08:03:11,876] [23015762] Check 4 HostAuthReq without RFID 0x7 0x7c 0x8 0
2024-08-17 08:03:12,057 DEBUG - LogEvt: Event time: [2024-08-17 08:03:11,955] [23015762] [CPLD] (0:04,01) EvPlugged;HW_Lock
```

EVCC log:
```
[lp-1 ] DEBUG 2024/08/17 10:03:54 pv charge current: 0A = 0A + -0.691A (159W @ 1p)
[lp-1 ] DEBUG 2024/08/17 10:03:54 pv enable timer reset
[lp-1 ] DEBUG 2024/08/17 10:03:54 pv timer inactive
**[lp-1 ] DEBUG 2024/08/17 10:04:14 set charge mode: minpv**
[site ] DEBUG 2024/08/17 10:04:14 ----
[lp-1 ] DEBUG 2024/08/17 10:04:14 charge power: 0W
[lp-1 ] DEBUG 2024/08/17 10:04:14 charge currents: [0 0 0]A
[site ] DEBUG 2024/08/17 10:04:15 pv power: 1282W
[site ] DEBUG 2024/08/17 10:04:15 battery soc: 1%
[site ] DEBUG 2024/08/17 10:04:15 battery power: -698W
[site ] DEBUG 2024/08/17 10:04:15 grid meter: 44W
[site ] DEBUG 2024/08/17 10:04:15 battery has priority at soc 1% (< 20%)
[site ] DEBUG 2024/08/17 10:04:15 site power: 144W
[lp-1 ] DEBUG 2024/08/17 10:04:15 charge total import: 13314.324kWh
[lp-1 ] DEBUG 2024/08/17 10:04:15 charger status: B
[lp-1 ] DEBUG 2024/08/17 10:04:15 pv charge current: min 6A > 0A (144W @ 1p, battery: false)
[lp-1 ] DEBUG 2024/08/17 10:04:15 max charge current: 6A
[lp-1 ] DEBUG 2024/08/17 10:04:15 charger enable
[lp-1 ] DEBUG 2024/08/17 10:04:15 wake-up timer: start
[site ] DEBUG 2024/08/17 10:04:51 ----
[lp-1 ] DEBUG 2024/08/17 10:04:51 charge power: 0W
[lp-1 ] DEBUG 2024/08/17 10:04:51 charge currents: [0 0 0]A
[site ] DEBUG 2024/08/17 10:04:51 pv power: 1215W
[site ] DEBUG 2024/08/17 10:04:51 battery soc: 1%
[site ] DEBUG 2024/08/17 10:04:51 battery power: -677W
[site ] DEBUG 2024/08/17 10:04:51 grid meter: 41W
[site ] DEBUG 2024/08/17 10:04:51 battery has priority at soc 1% (< 20%)
[site ] DEBUG 2024/08/17 10:04:51 site power: 141W
[lp-1 ] DEBUG 2024/08/17 10:04:51 charge total import: 13314.324kWh
[lp-1 ] DEBUG 2024/08/17 10:04:51 charger status: B
[lp-1 ] DEBUG 2024/08/17 10:04:51 pv charge current: min 6A > 0A (141W @ 1p, battery: false)
[site ] DEBUG 2024/08/17 10:05:51 ----
[lp-1 ] DEBUG 2024/08/17 10:05:51 charge power: 0W
[lp-1 ] DEBUG 2024/08/17 10:05:51 charge currents: [0 0 0]A
[site ] DEBUG 2024/08/17 10:05:52 pv power: 1121W
[site ] DEBUG 2024/08/17 10:05:52 battery soc: 1%
[site ] DEBUG 2024/08/17 10:05:52 battery power: -525W
[site ] DEBUG 2024/08/17 10:05:52 grid meter: 25W
[site ] DEBUG 2024/08/17 10:05:52 battery has priority at soc 1% (< 20%)
[site ] DEBUG 2024/08/17 10:05:52 site power: 125W
[lp-1 ] DEBUG 2024/08/17 10:05:52 charge total import: 13314.324kWh
[lp-1 ] DEBUG 2024/08/17 10:05:52 charger status: B
**[lp-1 ] WARN 2024/08/17 10:05:52 charger out of sync: expected enabled, got disabled**
[lp-1 ] DEBUG 2024/08/17 10:05:52 charger disable
[lp-1 ] DEBUG 2024/08/17 10:05:52 pv charge current: min 6A > 0A (125W @ 1p, battery: false)
[lp-1 ] DEBUG 2024/08/17 10:05:52 charger enable
[lp-1 ] DEBUG 2024/08/17 10:05:52 wake-up timer: start
```

TobiasHuber1980, 2024-08-17:
1) Keba mal rebootet/komplett stromlos gemacht?
2) schau mal mit `evcc charger` ob die rfid-karte erkannt wurde und ob die korrekt ist.
3) `evcc charger --log trace --diagnose` mal posten

whentotrade, 2024-08-17:
@TobiasHuber1980 : Danke für deine schnelle Hilfe!

evcc charger --log trace --diagnose
```
[main  ] INFO 2024/08/17 10:42:19 evcc 0.129.0 (f73072d8)
[main  ] INFO 2024/08/17 10:42:19 using config file: /etc/evcc.yaml
[db    ] INFO 2024/08/17 10:42:19 using sqlite database: /home/datapool/.evcc/evcc.db
[db    ] TRACE 2024/08/17 10:42:19 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2024/08/17 10:42:19 CREATE TABLE `settings` (`key` text,`value` text,PRIMARY KEY (`key`)) 0 <nil>
[db    ] TRACE 2024/08/17 10:42:19 SELECT * FROM `settings` 0 <nil>
[mqtt  ] INFO 2024/08/17 10:42:20 connecting evcc-2048084546 at tcp://192.168.0.40:1883
[mqtt  ] DEBUG 2024/08/17 10:42:21 tcp://192.168.0.40:1883 connected
[db    ] TRACE 2024/08/17 10:42:21 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2024/08/17 10:42:21 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2024/08/17 10:42:21 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2024/08/17 10:42:21 CREATE TABLE `configs` (`id` integer PRIMARY KEY AUTOINCREMENT,`class` integer,`type` text,`value` tex                                   t) 0 <nil>
[db    ] TRACE 2024/08/17 10:42:21 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2024/08/17 10:42:21 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2024/08/17 10:42:21 SELECT * FROM `configs` WHERE `configs`.`class` = 1 0 <nil>
[keba  ] TRACE 2024/08/17 10:42:21 modbus: send 00 01 00 00 00 06 ff 03 03 f8 00 02
[keba  ] TRACE 2024/08/17 10:42:21 modbus: recv 00 01 00 00 00 07 ff 03 04 00 04 ca a5
[keba  ] TRACE 2024/08/17 10:42:21 modbus: send 00 02 00 00 00 06 ff 03 06 0e 00 02
[keba  ] TRACE 2024/08/17 10:42:21 modbus: recv 00 02 00 00 00 07 ff 03 04 00 00 00 00
[keba  ] TRACE 2024/08/17 10:42:21 modbus: send 00 03 00 00 00 06 ff 03 06 42 00 02
[keba  ] TRACE 2024/08/17 10:42:21 modbus: recv 00 03 00 00 00 07 ff 03 04 00 00 00 00
[keba  ] TRACE 2024/08/17 10:42:21 modbus: send 00 04 00 00 00 06 ff 03 03 fc 00 02
[keba  ] TRACE 2024/08/17 10:42:21 modbus: recv 00 04 00 00 00 07 ff 03 04 00 00 00 00
[keba  ] TRACE 2024/08/17 10:42:21 modbus: send 00 05 00 00 00 06 ff 03 04 0c 00 02
[keba  ] TRACE 2024/08/17 10:42:21 modbus: recv 00 05 00 00 00 07 ff 03 04 07 ef 9a c8
[keba  ] TRACE 2024/08/17 10:42:21 modbus: send 00 06 00 00 00 06 ff 03 03 f0 00 02
[keba  ] TRACE 2024/08/17 10:42:21 modbus: recv 00 06 00 00 00 07 ff 03 04 00 00 00 00
[keba  ] TRACE 2024/08/17 10:42:21 modbus: send 00 07 00 00 00 06 ff 03 03 f2 00 02
[keba  ] TRACE 2024/08/17 10:42:21 modbus: recv 00 07 00 00 00 07 ff 03 04 00 00 00 00
[keba  ] TRACE 2024/08/17 10:42:21 modbus: send 00 08 00 00 00 06 ff 03 03 f4 00 02
[keba  ] TRACE 2024/08/17 10:42:21 modbus: recv 00 08 00 00 00 07 ff 03 04 00 00 00 00
[keba  ] TRACE 2024/08/17 10:42:21 modbus: send 00 09 00 00 00 06 ff 03 03 ec 00 02
[keba  ] TRACE 2024/08/17 10:42:21 modbus: recv 00 09 00 00 00 07 ff 03 04 00 00 00 07
[keba  ] TRACE 2024/08/17 10:42:21 modbus: send 00 0a 00 00 00 06 ff 03 03 e8 00 02
[keba  ] TRACE 2024/08/17 10:42:21 modbus: recv 00 0a 00 00 00 07 ff 03 04 00 00 00 01
[keba  ] TRACE 2024/08/17 10:42:21 modbus: send 00 0b 00 00 00 06 ff 03 03 e8 00 02
[keba  ] TRACE 2024/08/17 10:42:21 modbus: recv 00 0b 00 00 00 07 ff 03 04 00 00 00 01
[keba  ] TRACE 2024/08/17 10:42:21 modbus: send 00 0c 00 00 00 06 ff 03 05 dc 00 02
[keba  ] TRACE 2024/08/17 10:42:21 modbus: recv 00 0c 00 00 00 07 ff 03 04 00 00 00 00
Power:          0W
Energy:         13314.3kWh
Current L1..L3: 0A 0A 0A
Charge status:  B
Enabled:        false
Identifier:     <none>
Features:       [WelcomeCharge]
Diagnostic dump:
[keba  ] TRACE 2024/08/17 10:42:21 modbus: send 00 0d 00 00 00 06 ff 03 03 f6 00 02
[keba  ] TRACE 2024/08/17 10:42:21 modbus: recv 00 0d 00 00 00 07 ff 03 04 01 5f 31 52
        Serial: 23015762
[keba  ] TRACE 2024/08/17 10:42:21 modbus: send 00 0e 00 00 00 06 ff 03 03 fa 00 02
[keba  ] TRACE 2024/08/17 10:42:21 modbus: recv 00 0e 00 00 00 07 ff 03 04 03 0a 35 00
        Firmware:       3.10.53
[keba  ] TRACE 2024/08/17 10:42:21 modbus: send 00 0f 00 00 00 06 ff 03 03 f8 00 02
[keba  ] TRACE 2024/08/17 10:42:21 modbus: recv 00 0f 00 00 00 07 ff 03 04 00 04 ca a5
        Product:        314021
[keba  ] TRACE 2024/08/17 10:42:21 modbus: send 00 10 00 00 00 06 ff 03 06 0e 00 02
[keba  ] TRACE 2024/08/17 10:42:21 modbus: recv 00 10 00 00 00 07 ff 03 04 00 00 00 00
        Phases source:  0
[keba  ] TRACE 2024/08/17 10:42:21 modbus: send 00 11 00 00 00 06 ff 03 06 10 00 02
[keba  ] TRACE 2024/08/17 10:42:21 modbus: recv 00 11 00 00 00 07 ff 03 04 00 00 00 00
        Phases state:   0
[keba  ] TRACE 2024/08/17 10:42:21 modbus: send 00 12 00 00 00 06 ff 03 06 42 00 02
[keba  ] TRACE 2024/08/17 10:42:21 modbus: recv 00 12 00 00 00 07 ff 03 04 00 00 00 00
        Failsafe timeout:       0s
```

evcc charger
```
[main  ] INFO 2024/08/17 10:44:30 evcc 0.129.0 (f73072d8)
[main  ] INFO 2024/08/17 10:44:30 using config file: /etc/evcc.yaml
[db    ] INFO 2024/08/17 10:44:30 using sqlite database: /home/datapool/.evcc/evcc.db
[mqtt  ] INFO 2024/08/17 10:44:30 connecting evcc-822174201 at tcp://192.168.0.40:1883
Power:          0W
Energy:         13314.3kWh
Current L1..L3: 0A 0A 0A
Charge status:  B
Enabled:        false
Identifier:     <none>
Features:       [WelcomeCharge]
```

Ja, Keba reboot hatte ich schon mehrfach gemacht. Mich wundert halt das mit V. 127.2 das alles problemlos klappt.

TobiasHuber1980, 2024-08-17:
`Identifier:     <none>`
da ist der fehler. so wie mich das anschaut, gibt die keba den rfid nicht/nicht korrekt an evcc weiter.

ich habe eine ähnliche konfiguration wie du (keba x-serie + c-serie im verbund) und keine probleme mit 0.129.0.

kontrolliere mal die einrückungen bei vehicle/indentifiers, es könnte sein, dass ein leerzeichen zu viel vor dem "-" bei identifiers bei dir ist.

```
 vehicles:
  - name: AudiEtron
    type: template
    template: audi
    title: HH-VT-ETRON
    user: ***
    password: ***
    vin: ***
    capacity: 95 # Akku-Kapa in kWh # Optional 
    phases: 3
    identifiers: 
      - 686f6d65
 #   - "686f6d6561730d0a"
```

whentotrade, 2024-08-17:
ich habe mehrere einrückungen ausprobiert, immer das gleiche...
aber kurze rückfrage: der identifiers is hier doch beim fahrzeug hinterlegt, nicht beim charger in der config?

whentotrade, 2024-08-17:
Beim vehicle sehe ich den identifier:

sudo evcc -l debug vehicle
```
[main  ] INFO 2024/08/17 11:09:45 evcc 0.129.0 (f73072d8)
[main  ] INFO 2024/08/17 11:09:45 using config file: /etc/evcc.yaml
[db    ] INFO 2024/08/17 11:09:45 using sqlite database: /root/.evcc/evcc.db
[mqtt  ] INFO 2024/08/17 11:09:45 connecting evcc-1020574094 at tcp://192.168.0.40:1883
[mqtt  ] DEBUG 2024/08/17 11:09:45 tcp://192.168.0.40:1883 connected
Soc:            51%
Capacity:       95.0kWh
Charge status:  B
Range:          185km
Odometer:       41656km
Finish time:    2024-08-17 10:42:00 +0200 CEST
Climate active: false
Limit Soc:      100%
Identifiers:    [686f6d65]
```

whentotrade, 2024-08-17:
@TobiasHuber1980 : danke weiterhin für die unterstützung.

Ja, in der Tat der identifier wird nicht durchgereicht. Wenn ich manuell mit der RFID Karte freigebe, sehe ich den identifier hinterher im debug mit evcc charger. Aber das hinterlegen in der config funktioniert nicht mehr mit 129.0

whentotrade, 2024-08-17:
Ich hab ein Downgrade auf 127.2 gemacht und auf KEBA UDP umgestellt, dann wieder charger evcc ausgelesen, und voila, der identifier ist da und alles klappt wieder:

```
datapool@evcc:~ $ evcc charger
[main  ] INFO 2024/08/17 11:53:49 evcc 0.127.2
[main  ] INFO 2024/08/17 11:53:49 using config file: /etc/evcc.yaml
[db    ] INFO 2024/08/17 11:53:49 using sqlite database: /home/datapool/.evcc/evcc.db
[mqtt  ] INFO 2024/08/17 11:53:49 connecting evcc-598236826 at tcp://192.168.0.40:1883
Power:          1324W
Energy:         13314.6kWh
Current L1..L3: 5.91A 0A 0A
Charge status:  C
Enabled:        true
Identifier:     686f6d6561730d0a
datapool@evcc:~ $
```

Da muss doch irgendwas in der 129, Umstellung auf Keba ModBus und RFID Kommunikation verändert worden sein. Es klappt alles mit 127.2. 129.0 nicht mehr - fehler liegt in der nicht-weitergabe des RFID Identifiers unter 129.0. Mehr kann ich wirklich nicht mehr testen auf meiner Seite und muss dann auf 127.2 bleiben solange der Fehler besteht?

@TobiasHuber1980: 1000 Dank für es saubere Tracken hin zur Ursache, meine Mittel sind jetzt am Ende. Was kann ich weiter tun als downgraden?

andig, 2024-08-17:
> Changed the Keba configuration from UDP to modbus according to new docs. Did not help.

Gabs das Problem vor Änderung oder nach?

> damit die Box per OCPP anbinden und funktional vergleichen

Bitte nicht die Dinge vermischen. Hier geht es erstmal um die Modbus Implementierung.

whentotrade, 2024-08-17:
@andig: Auf Deine Fragen: Ich hatte 127.2 laufen mit Keba UDP und RFID Autorisierung, alles stabile über Monate. Ich habe nix geändert. Update auf 129.0 vorgenommen. Ladevorgang startet nicht mehr. Danach aufgrund der Hinweise dann statt UDP auf Keba Modbus umgestellt in der Konfig. Gleiches Problem (Details siehe oben). fehler liegt im nicht-weitergeben des identifiers. Test Protokolle und Ergebnisse sind alle oben, dank Tobias Hilfe.

Jetzt wieder auf 127.2 zurück. Da läuft es und ich bekomme den Identifier mit beiden Versionen, sowohl modbus und UDP wieder sauber:
```
evcc charger
[main  ] INFO 2024/08/17 12:05:34 evcc 0.127.2
[main  ] INFO 2024/08/17 12:05:34 using config file: /etc/evcc.yaml
[db    ] INFO 2024/08/17 12:05:35 using sqlite database: /home/datapool/.evcc/evcc.db
[mqtt  ] INFO 2024/08/17 12:05:35 connecting evcc-1098870803 at tcp://192.168.0.40:1883
Power:          1330W
Energy:         13314.9kWh
Current L1..L3: 5.91A 0A 0A
Charge status:  C
Enabled:        true
Identifier:     686f6d65
```

Config:
```
chargers:
  - name: wallbox1
    type: template
    template: keba-modbus
    modbus: tcpip
    id: 255
    host: 192.168.0.69
    port: 502
    
vehicles:
  - name: AudiEtron
    type: template
    template: audi
    title: **
    user: ***
    password: ***
    vin: **
    capacity: 95 # Akku-Kapa in kWh # Optional 
    phases: 3
    identifiers:
      - 686f6d65
```

Ich habe an meinem System nix verändert. Diese Config funktioniert mit ModBus unter 127.2 aber nicht unter 129.0. Protokolle unter 129 siehe oben.

andig, 2024-08-17:
Dann brauchts mal bitte altes und neues trace log, modbus und udp. Insbesondere bei udp hat sich nichts geändert, daher schwer zu glauben.

whentotrade, 2024-08-17:
nee leute, ich habe jetzt über 4 stunden ge-debugged, auto an- & abgeschlossen, protokolle runtergeladen, sitze hier seit 4 stunden am testen und habe alle infos hier immer zurückgegeben.

irgendjemand hat da einen bug eingebaut - ich habe nix geändert und doch hier gezeigt was unter 127.2 läuft und was unter 129.0 nicht. wenn ihr mir nicht glaubt kann ich das nicht ändern. ich sitze hier aber nicht stunden um dann zu hören das sei nicht zu glauben und ich nochmal wieder alles testen und tracen soll - da ist ein bug und breaking change. Steht alles oben.

127.2 läuft und ich bin zufrieden. irgendjemand hat auf dem weg zur 129 was zerschossen. der fehler liegt im nicht-weitergeben des identifiers - wie hier eindeutig aufgezeigt. klappt bei 127.2, nicht bei 129. Das sind doch schon jede Menge details.

closed 2024-08-17 by andig

TobiasHuber1980, 2024-08-17:
ich habe bei mir die indentifiers nur bei vehicle, nicht beim charger.
logik: rfid-karte an die wallbox, wallbox gibt ladung frei (daher auch manchmal [Warn: charger out of sync: expected disabled, got enabled](https://github.com/evcc-io/evcc/issues/15288).

Hier mal auszugsweise meine Konfig, mit der funktioniert die 0.129 ohne Probleme.

```
chargers:
- type: template
  template: keba-modbus 
  host: 192.168.1.**
  name: wallbox1
  modbus: tcpip
  id: 255
  port: 502
  welcomecharge: true
```
```
loadpoints:
- title: expert links
  charger: wallbox1
  mode: pv
```
```
vehicles:
  - name: car1
    type: template
    template: vw
    title: VW ID.5 (Tobias Huber)
    user: ***
    password: ***
    vin: ***
    capacity: 77
    timeout: 10s
    identifiers:
      - 04115e12  
```
bau doch das einfach mal mit deinen daten nach.

andig, 2024-08-17:
Das ganze Gebastel ist völlig sinn- und nutzlos.

andig, 2024-08-17:
> Ich habe an meinem System nix verändert. Diese Config funktioniert mit ModBus unter 127.2 aber nicht unter 129.0. Protokolle unter 129 siehe oben.

Dann braucht es ja einfach mal ein

    evcc charger --log trace

mit alter und neuer Version.

> nee leute

Oder auch nicht 

>  fehler liegt im nicht-weitergeben des identifiers. 

Was soll da weiter gegeben werden? evcc liest den aus, nicht mehr und nicht weniger.

whentotrade, 2024-08-18:
Nochmals herzlichen Dank an @TobiasHuber1980, dass er das Problem so klar herausgearbeitet hat. Dies hat mir nun dabei geholfen, die Ursache für den fehlenden "Identifier" zu finden.

Für alle, die möglicherweise in eine ähnliche Situation geraten, könnte die kurze Zusammenfassung hilfreich sein.

In meinem Kontext warte ich auf das MQTT-Event, bis ein Fahrzeug gemäß `evcc/loadpoints/1/connected` angeschlossen wird. Sobald dies erkannt wird, starte ich den Ladevorgang über die REST-API mit `POST /api/loadpoints/1/mode/minpv`. Dies hatte wie erwähnt bisher immer funktioniert.

Nun geht die Logik folgendermaßen: Es wurde zwar der Anschluss des Fahrzeugs erkannt, aber ich habe dann den REST-API-Befehl zum Starten des Ladevorgangs wohl zu früh ausgeführt, da das Fahrzeug zu diesem Zeitpunkt vermutlich noch nicht vollständig in EVCC erkannt war.

Da der Identifier im Fahrzeug in der vehicle Konfiguration hinterlegt ist, hatte der charger natürlich keinen Identifier mitbekommen. Daraufhin konnte die KEBA-Ladestation aufgrund des fehlenden RFID-Identifikators den Ladevorgang nicht starten. So haben wir es oben in den Protokollen ja auch gesehen.

Auf der EVCC-Benutzeroberfläche stand dann `Ready. Waiting for vehicle..`, und im KEBA-Logbuch war zu sehen, dass der Ladevorgang abgelehnt wurde. Einige Zeit später erschien dann im Logbuch die Meldung` charger out of sync: expected enabled, got disabled`

Die Fehler und Protokolle machen also wieder Sinn. 

Jedoch ist die Benutzeroberfläche in dieser Race Condition hängen geblieben, sodass der Ladevorgang auch manuell im EVCC UI nicht mehr gestartet werden konnte.

Meine Lösung ist jetzt, dass ich vor dem Ausführen des REST-API-Befehls zum Starten des Ladevorgangs zusätzlich mit` POST /api/loadpoints//vehicle/name` vorher sicherstelle, dass das richtige Fahrzeug eingestellt ist. Dadurch wird aufgrund des definierten Fahrzeugs dann auch der Identifier aus der vehicle Konfiguration abgerufen und dem charger mitgegeben. Anschließend autorisiert die KEBA-Ladestation den Ladevorgang.

Das läuft so jetzt auch mit Version 129 reibungslos. Warum diese Race Condition nicht bereits früher aufgefallen ist, lässt sich schwer sagen. Egal, Ursache erkannt und gebannt.

Das mag jetzt vielleicht alles nicht in der richtigen Entwickler Sprache beschrieben sein. Aber zumindest wollte ich hier die Lösung auch für andere die darüber vielleicht mal stolpern, hinterlegen.

andig, 2024-08-18:
> Da der Identifier im Fahrzeug in der vehicle Konfiguration hinterlegt ist, hatte der charger natürlich keinen Identifier mitbekommen. 

Das ist- mit Verlaub- völliger Blödsinn. Die Erkennung des Fahrzeugs durch evcc hat mit der Wallbox überhaupt gar nichts zu tun.

> Daraufhin konnte die KEBA-Ladestation aufgrund des fehlenden RFID-Identifikators den Ladevorgang nicht starten. So haben wir es oben in den Protokollen ja auch gesehen.

Woher soll dieser RFID-Identifikator denn bitte kommen?

> Jedoch ist die Benutzeroberfläche in dieser Race Condition hängen geblieben, sodass der Ladevorgang auch manuell im EVCC UI nicht mehr gestartet werden konnte.

Sicher nicht.

> Das mag jetzt vielleicht alles nicht in der richtigen Entwickler Sprache beschrieben sein. Aber zumindest wollte ich hier die Lösung auch für andere die darüber vielleicht mal stolpern, hinterlegen.

Die vermuteten Zusammenhänge existieren im Code einfach nicht. Was auch immer es war- es hat eine andere Ursache...

whentotrade, 2024-08-18:
@andig  Mit Verlaub und hohem Respekt vor dem was ihr hier macht. Aber Deine sehr kurz gehaltenen Kommentare oder Fragen im Entwickler Slang haben mich nur frustriert und nicht weitergebracht. Ich habe jetzt eine Lösung gefunden die funktioniert und ich habe beschrieben was ich geändert habe damit es funktioniert. Warum werde ich dafür nun wieder angemeckert? Was sollte denn jetzt dieser Kommentar - versteh ich nicht mehr - ich bin raus.

andig, 2024-08-18:
> Ich habe jetzt eine Lösung gefunden die funktioniert und ich habe beschrieben was ich geändert habe damit es funktioniert.

Das ist super. Ändert aber überhaupt nix daran, dass die Zusammenhänge so nicht existieren. In https://github.com/evcc-io/evcc/issues/15441#issuecomment-2294890912 hatte ich geschrieben was zur Diagnose notwendig ist.

> Warum werde ich dafür nun wieder angemeckert? Was sollte denn jetzt dieser Kommentar - versteh ich nicht mehr - ich bin raus.

Wieso meckern? Ich möchte Dich davon abhalten, einer Scheinlösung nachzulaufen die genauso schnell wieder auseinander fallen kann.

juerlink, 2024-08-19:
Hallo,
ich lese hier schon einige Zeit mit, da ich mit meiner Keba P30X ein ähnliches Problem habe (irgendwas mit Authorisierung).
Könnte es nicht auch sein, dass die Keba Firmware einen Bug hat? Ich habe etwa im gleichen Zeitraum die Keba FW auf V1.18.0 upgedatet und EVCC auf 0.129.0. Danach hat die Keba sporadisch angefangen das Laden zu verweigern (gefühlt einmal in 2 Wochen bei nahezu täglichem Laden). Downgrade von EVCC hatte erst gut ausgesehen, aber der Fehler kam dann doch. Man musste leider immer sehr lange warten, da sporadisch.
@whentotrade: Hast Du vielleicht im gleichen Zeitraum auch auf die neue Keba-FW geupdatet?

Anfangs war es so:
* Keba mit ausgeschalteter Authorisierung, auch keine RFID-Karten angelernt
* EVCC mit Keba-UDP-Template
* sporadisch Fehler: missing credentials for rfid authorization
* Fehler nur zu beseitigen durch Neustart der Wallbox

Umstieg auf ModbusTCP
* Hat ein paar mal funktioniert
* Dann Fehler: charger out of sync: expected enabled, got disabled
* Fehler wieder nur zu beseitigen durch Neustart der Wallbox

Zurück auf UDP-Modus:
Im Code hatte ich gesehen, dass EVCC eine RFID schicken will, wenn die Wallbox auf "report 2" ein Authreq != 0 liefert.

// no auth required
if kr.AuthReq == 0 {
	return nil
}

// auth required but missing tag
if c.rfid.Tag == "" {
	return errors.New("missing credentials for RFID authorization")
}

daher habe ich testweise eine Dummy-RFID in die evcc.yaml eingetragen, in der Annahme, dass die Wallbox ja eigentlich gar keine RFID braucht, es ist ja ausgeschaltet. Das lief dann auch ca. 2 Wochen ohne Probleme, danach kam wieder ein Ladehänger. Die Fehlermeldung "missing credentials... blieb aber aus, da ich ja eine Dummy-RFID konfiguriert hatte. Ich konnte die Wallbox auch nicht über die Keba-App starten und musste die Wallbox wieder mal neu starten.

Ich habe bei Keba jetzt ein Ticket geöffnet, mit dem Hinweis, dass fälschlicherweise sporadisch ein Authreq != 0 geschickt wird, obwohl die Authorisierung ausgeschaltet ist. Aktuell warte ich auf den Antwort und erneutes Auftreten des Fehlers. Wenn er wieder kommt lese ich den "report 2" mal mit einem anderen Tool aus. Sollte da dauerhaft Authreq != 0 kommen, kann ich dies an Keba melden und es wäre ein Beweis, dass die Keba da Müll schickt.

Sorry für die lange Ausführung, aber vielleicht hilft es bei der Fehlersuche. Wie gesagt vermute ich den Fehler eher bei Keba, außer evcc hat einen Fehler beim Parsen des Rückgabe-json-Strings von "report 2". Das müsste ich dann aber sehen, wenn der Fehler wieder auftritt.

Grüße, juerlink

andig, 2024-08-19:
Danke für die Erläuterung- das klingt etwas nach KEBA FW Issue.
Als zusätzlicher Kontext dazu: das RFID schicken gibts nur in der UDP Variante. Auch dort wir die _konfigurierte_ RFID geschickt, nicht irgendwas aus der Fahrzeugerkennung. Mit https://github.com/evcc-io/evcc/pull/15399 und v.a. mit "WelcomeCharge" sollte auch das Senden dann nicht mehr notwendig sein.

whentotrade, 2024-08-19:
> @whentotrade: Hast Du vielleicht im gleichen Zeitraum auch auf die neue Keba-FW geupdatet?

Ja, habe ich. 

Aber die zeitlichen Zusammenhänge kann ich nicht mehr genau nachvollziehen. Das einzige, was bei mit jetzt reproduzierbar ist aus Anwednersicht ist folgendes:

Alt (funktioniert unter evcc 129.0 und Keba 118.0 nicht mehr):

1. warten auf mqtt `evcc/loadpoints/1/connected`
2. starten des ladevorgangs mit `POST /api/loadpoints/<id>/mode/<mode>`, je nach interner logik PV oder MinPV

Führt dazu, das im UI die Meldung 

- `"Ready. Waiting for vehicle..."` und im Log dann später
- `"charger out of sync: expected enabled, got disabled"` auftaucht.

Im charger debug log ist dann 
- `Identifier: <none>`
zu sehen, das der identifier leer geblieben ist und in der KEBA Box die RFID Authorisierung gescheitert ist. Das kann ich auch im KEBA log sehen.

Irgendwas geht da also bei der Authorisierung/RFID weitergabe aus der Config schief.

Die Lösung, die das Problem für mich erstmal beseitigt ist wie folgt. Zwischen die MQTT Message das das Fahrzeug angeschlossen ist und das senden des POST ladevorgangs PV oder MinPV hänge ich einfach noch manuell die Info zum Fahrtzeug per REST API:

1. warten auf mqtt `evcc/loadpoints/1/connected`
**2. NEU: `POST /api/loadpoints/vehicle/Myname`**
2. starten des ladevorgangs mit `POST /api/loadpoints/1/mode/<mode>`, je nach interner logik PV oder MinPV

Jetzt ist im charger debug log der identifier auch wieder zu finden und die Keba authorisiert den Vorgang mit der RFID Karte. 
`Identifier:     686f6d65`

Siehe auch: https://github.com/evcc-io/evcc/issues/15441#issuecomment-2294815715

Dieses Verhalten und Problem ist das gleiche bei UDP und modbus. Es ist reproduzierbar wenn ich zwischen diesen beiden Varianten wechsele. Zumindest funktioniert es mit dem zusätzlichen setzen des vehicles so wieder bei mir.

Irgendwo hängt da was in der Authorisierung und Weitergabe der RFID Identifier aus der config. Ob das jetzt nachhaltig oder dem erwarteten Verhalten gem. Entwickler-Code entspricht, dem letzten KEBA oder EVCC Update entspringt, entzieht sich meiner Kenntnis. So konnte ich das zumindest bei mir lösen.

Aber falls das alte Problem sporadisch doch wieder auftaucht, dann geb ich gerne bescheid. Dann schlummert da tatsächlich noch irgendwo was anderes.

juerlink, 2024-08-19:
Ich bin etwas verwirrt:
Mein Verständnis (bitte steinigt mich nicht, wenn das falsch ist): evcc schickt nur im udp Modus eine in der evcc.yaml unter charger konfigurierte RFID an die Wallbox (wirkt, als würde man die RFID Karte an die Wallbox halten). Im Modbus Modus wird das nicht gemacht.
Der ID-Eintrag im vehicle dient nur zur Erkennung, welches Fahrzeug dran steckt, wenn das Fahrzeug fest der RFID zugeordnet ist durch den Anwender.

Mein Verständnis ist daher, dass man im Modbus Modus immer die Karte manuell dran halten muss. In Deiner Beschreibung klingt es aber so, als sollte die Authorisierung sowohl im udp, als auch im Modbus Modus durch evcc erfolgen. Wenn ich hier falsch liege, bitte korrigieren. Ich würde es gerne selbst besser verstehen.

andig, 2024-08-19:
Ich habs ja oben schonmal korrigiert. Es ist genauso wie du sagst. 

Und damit sollten wir hier schliessen…