evcc-io/evcc #15442: SMA Tripower Smart Energy 10.0  grid charging without function

unim0trix, 2024-08-17:
### Describe the bug

The modbus parameters for the active power specification is not used.
SMA STP-SE (10) / BYD HVS 7.7

[sma-hybrid.yaml](https://github.com/evcc-io/evcc/blob/master/templates/definition/meter/sma-hybrid.yaml)

129: value: -2147483648 # Wirkleistungsvorgabe

The battery can be controlled manually as expected.
For example:
modbus write -s 3 -i 10.128.0.4 40152 0 802
modbus write -s 3 -i 10.128.0.4 40150 -1 -2500
modbus write -s 3 -i 10.128.0.4 40150 0 2500
modbus write -s 3 -i 10.128.0.4 40152 0 803

### Steps to reproduce

1. Set the price limit correspondingly low

### Configuration details

```yaml
network:
  schema: http
  host: evcc.internal
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: #####

interval: 10s 

sponsortoken: #####

telemetry: true

meters:
- type: sma
  name: grid1
  serial: #####
- type: template
  template: sma-hybrid
  id: 3
  host: 10.128.0.4
  port: 502
  usage: pv
  modbus: tcpip
  name: pv2
  watchdog: 30s
- type: template
  template: sma-hybrid
  id: 3
  host: 10.128.0.4
  port: 502
  usage: battery
  modbus: tcpip
  name: battery3
  watchdog: 30s
  capacity: 7.7
  #- name: grid2
    #type: template
    #template: tibber-pulse
    #usage: grid
    #token: "#####"
- name: wp_meter
[... 200 lines cut ...]
[modbus] TRACE 2024/08/17 07:59:29 modbus: recv 60 19 00 00 00 0b 03 03 08 00 00 00 00 01 0a 1e 6b
[site ] DEBUG 2024/08/17 07:59:29 pv power: 69W
[modbus] TRACE 2024/08/17 07:59:29 modbus: send 60 28 00 00 00 06 03 04 7a a3 00 02
[modbus] TRACE 2024/08/17 07:59:29 modbus: recv 60 28 00 00 00 07 03 04 04 00 00 01 3b
[modbus] TRACE 2024/08/17 07:59:29 modbus: send 60 19 00 00 00 06 03 04 7a a1 00 02
[modbus] TRACE 2024/08/17 07:59:29 modbus: recv 60 19 00 00 00 07 03 04 04 00 00 00 00
[modbus] TRACE 2024/08/17 07:59:29 modbus: send 60 19 00 00 00 06 03 03 7a a9 00 04
[modbus] TRACE 2024/08/17 07:59:29 modbus: recv 60 19 00 00 00 0b 03 03 08 00 00 00 00 00 26 f4 98
[modbus] TRACE 2024/08/17 07:59:29 modbus: send 60 19 00 00 00 06 03 03 78 7d 00 02
[modbus] TRACE 2024/08/17 07:59:29 modbus: recv 60 19 00 00 00 07 03 03 04 00 00 00 17
[site ] DEBUG 2024/08/17 07:59:29 battery soc: 23%
[site ] DEBUG 2024/08/17 07:59:29 battery power: 315W
[site ] DEBUG 2024/08/17 07:59:29 grid meter: 22W
[site ] DEBUG 2024/08/17 07:59:29 grid powers: [105 -60 -23]W
[site ] DEBUG 2024/08/17 07:59:29 grid currents: [0.835 -0.404 -0.538]A
[site ] DEBUG 2024/08/17 07:59:29 site power: 437W
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.129.0 (9bf8dc7b)

andig, 2024-08-17:
Same as https://github.com/evcc-io/evcc/issues/15294?

premultiply, 2024-08-17:
SMA hat vor Wochen eine fehlerhafte Firmware an diverse Anlagenkomponenten verteilt.
Bitte an SMA wenden.

unim0trix, 2024-08-17:
SHM Update ist hier aber nicht relevant.
Die FW vom STP-SE ist von 2023 und hat sich seitdem auch nicht verändert.

premultiply, 2024-08-17:
Doch, genau das ist relevant, da es ständig die Einstellungen überschreibt.

unim0trix, 2024-08-17:
Der SHM ist resettet und das sunny portal bzw. ennexos gelöscht.
Verhält er sich so nicht nur als EnergyMeter?

unim0trix, 2024-08-17:
Der Wechselrichter wird durch die Aktivierung des "Netzladen" korrekt durch EVCC per modbus in den "manuellen" Modus für Wirk- und Blindleistungsregelung über Kommunikation gesetzt.

Die dann zusätzlich notwendige Wirkleistungsvorgabe wird dann aber offenbar nicht korrekt übergeben, bzw. nicht korrekt in das entsprechende Register geschrieben. In der sma-hybrid.yaml steht für den Fall  "-2147483648"
Ohne diese Vorgabe, z.B. -10000 (W am Netzanschlusspunkt) passiert halt nichts. Die Batterie lädt nicht.

Wenn man diesen Wert jedoch manuell per Modbus ins Register schreibt, funktioniert das Laden aus dem Netz wie es eigentlich funktionieren sollte.

wittmannchris, 2024-08-17:
> Doch, genau das ist relevant, da es ständig die Einstellungen überschreibt.

Der Release der neuen FW für ALLE ist in der ersten Sep Woche. 

Nichts desto trotz passen die Register für den SBS 2.5 nicht mehr. Die gehen nicht. Bitte ändern. 

Danke

premultiply, 2024-08-17:
So lange SMA das Problem mit dem SHM nicht behebt passiert von meiner Seite nichts mehr.

wittmannchris, 2024-08-17:
> Hello fieldtester,
 
Thank you for your testing and your feedback. We have examined your feedback carefully even if you did not always receive an immediate response.  
We are planning to end the fieldtest next week and starting the gradual rollout at the beginning of September.
The final release note will be at the SMA homepage when the roll starts.
We hope it was helpful for you to be involved in the early development and testing phase and to be able to ensure the use of Modbus again.
 

Na das ist ja bald 👍

andig, 2024-08-17:
> Nichts desto trotz passen die Register für den SBS 2.5 nicht mehr.

Das wiederrum hat ja mit diesem Issue gar nix zu tun. Also warten wir hier einfach mal, dass die Zeit vergeht…

unim0trix, 2024-08-17:
> So lange SMA das Problem mit dem SHM nicht behebt passiert von meiner Seite nichts mehr.

Hallo premultiply, nichts für ungut!
Das ist nachvollziehbar und ich kann gut damit leben!

Die Liste ist bei SMA ist leider im Moment lang und etwas frustrierend... 
Genau dem Grund nutze ich ja EVCC und habe die Regelung selbst in die Hand genommen.
Wie gesagt. Die Batteriesteuerung rundet die Sache ab.

Habt Ihr mal überlegt mit Andras Schmitz zusammenzuarbeiten?
Ich weiß, der Fokus liegt beim Laden von EVs, aber ein Fork für ein Energymanagement wäre doch sinnvoller als dieses gebastel mit HA, OH etc...

Nochmals DANKE!

andig, 2024-08-17:
> Die dann zusätzlich notwendige Wirkleistungsvorgabe wird dann aber offenbar nicht korrekt übergeben, bzw. nicht korrekt in das entsprechende Register geschrieben. In der sma-hybrid.yaml steht für den Fall "-2147483648"
Ohne diese Vorgabe, z.B. -10000 (W am Netzanschlusspunkt) passiert halt nichts. Die Batterie lädt nicht.

Es wird ja eine Vorgabe geschrieben- nämlich der minimal mögliche Wert. Was passiert wenn Du den ins Register schreibst? Falls das auch nicht geht- welcher Maximalwert ginge denn?

unim0trix, 2024-08-18:
> welcher Maximalwert ginge denn?

modbus-cli kann leider nur Sint16 und kein Sint32.
Mit Modbus Poll habe ich es jetzt mal nachgestellt. Wenn ich es richtig sehe ist die Implementation von SMA einfach Mist.  SInt32 verhält sich mit -2147483648 abnorm.
Mit -2147483647 lädt er dann mit maximaler Leistung.

> Soweit ich das verstanden habe geht es nicht um den Netzanschlusspunkt, ist hier aber unerheblich.
Ja, da hast du Recht. Es bezieht sich wohl auf die Wechselrichterleistung.
Bei einer PV Leistung von z.B. 6 kW kann ich mit mit dem Parameter auf 10 kW (also max. WR Leistung) durch 4 kW der Batterie ergänzen.

premultiply, 2024-08-18:
> verhält sich mit -2147483648 abnorm.
Mit -2147483647 lädt er dann mit maximaler Leistung.

Danke.

unim0trix, 2024-08-19:
TOP! Danke dir!

[image]

closed 2024-08-19 by premultiply