evcc-io/evcc #15445: Power values from Solar Edge hybrid inverter are wrong (far too high / negative).

grobiballon, 2024-08-17:
### Describe the bug

I set-up EVCC first time. In beginning all seemd to be correct. Data from Hybrid Inverter was plausible. Today (the battery is completely discharged) the power values are showing wrong values. Seems to be that the MODBUS TCP values are interpreted incorrect. I will review if values will be correct if the battery has a higher level.

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: 21143e16283763d1eab236cd5d040895fc2ea55a745ff5a651e152d890f182f8

interval: 30s # control cycle interval

sponsortoken: eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJldmNjLmlvIiwic3ViIjoidHJpYWwiLCJleHAiOjE3MjQ0MDM2MDAsImlhdCI6MTcyMTgxMTYwMCwic3BlIjp0cnVlLCJzcmMiOiJtYSJ9.VYed-tDIlZtHLGHmYpEhKVY1-9kgZXVebYYAXU90dcs

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: solaredge-hybrid 
  id: 1  
  host: 192.168.1.134  
  port: 1502  
  usage: grid  
  modbus: tcpip  
  watchdog: 60s  
  name: grid1
- type: template
  template: solaredge-hybrid 
  id: 1  
  host: 192.168.1.134  
  port: 1502  
  usage: pv  
  modbus: tcpip  
  watchdog: 60s  
  name: pv2
- type: template
  template: solaredge-hybrid 
  id: 1  
  host: 192.168.1.134  
  port: 1502  
  usage: battery  
  modbus: tcpip  
  watchdog: 60s  
  name: battery3

chargers:
- type: template
  template: vestel 
  id: 255  
  host: 192.168.1.123  
  port: 502  
  modbus: tcpip  
  name: wallbox4
- type: template
  template: vestel 
  id: 255  
  host: 192.168.1.121  
  port: 502  
  modbus: tcpip  
  name: wallbox5

loadpoints:
- title: Einfahrt
  charger: wallbox4
  mode: pv
- title: Mieter
  charger: wallbox5
  mode: pv

site:
  title: Schuetzenstr.
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3
```

### Log details

```text
[site ] WARN 2024/08/17 10:28:52 pv 1 power: **-340282346638528859811704183484516925440W** is negative - check configuration if sign is correct
[site ] DEBUG 2024/08/17 10:28:52 pv power: 0W
[site ] DEBUG 2024/08/17 10:28:52 battery soc: 10%
[site ] DEBUG 2024/08/17 10:28:52 battery power: **340282346638528859811704183484516925440W**
[site ] DEBUG 2024/08/17 10:28:52 grid meter: 542W
[site ] DEBUG 2024/08/17 10:28:52 grid powers: [-162 501 203]W
[site ] DEBUG 2024/08/17 10:28:53 grid currents: [-1.7 2.8 1.2]A
[site ] DEBUG 2024/08/17 10:28:53 site power: **340282346638528859811704183484516925440W**
[lp-2 ] DEBUG 2024/08/17 10:28:53 charge voltages: [236 234 237]V
[lp-2 ] DEBUG 2024/08/17 10:28:53 detected connected phases: 3p
[lp-2 ] DEBUG 2024/08/17 10:28:53 charge total import: 37.400kWh
[lp-2 ] DEBUG 2024/08/17 10:28:53 charger status: A

C:\Users\grobi\OneDrive\Desktop\evcc>evcc meter
[main  ] INFO 2024/08/17 10:43:33 evcc 0.129.0
[main  ] INFO 2024/08/17 10:43:33 using config file: C:\Users\grobi\OneDrive\Desktop\evcc\evcc.yaml
[db    ] INFO 2024/08/17 10:43:33 using sqlite database: C:\Users\grobi\.evcc\evcc.db
grid1
-----
Power:          168W
Energy:         18089544.0kWh
Current L1..L3: 2.6A 2.8A 1.2A
Voltage L1..L3: 235V 234V 236V
Power L1..L3:   -509W 500W 176W

pv2
---
Power: -340282346638528859811704183484516925440W

battery3
--------
Power: 340282346638528859811704183484516925440W
Soc:   10%
```

### What type of operating system are you running?

Windows

### Version

0.129.0

andig, 2024-08-17:
Please show

    evcc meter —log trace

grobiballon, 2024-08-17:
`C:\Users\grobi\OneDrive\Desktop\evcc>evcc meter --log trace
[main  ] INFO 2024/08/17 17:22:16 evcc 0.129.0
[main  ] INFO 2024/08/17 17:22:16 using config file: C:\Users\grobi\OneDrive\Desktop\evcc\evcc.yaml
[db    ] INFO 2024/08/17 17:22:16 using sqlite database: C:\Users\grobi\.evcc\evcc.db
[db    ] TRACE 2024/08/17 17:22:16 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2024/08/17 17:22:16 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/08/17 17:22:16 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/08/17 17:22:16 SELECT * FROM `settings` 8 <nil>
[db    ] TRACE 2024/08/17 17:22:17 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2024/08/17 17:22:17 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2024/08/17 17:22:17 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2024/08/17 17:22:17 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/08/17 17:22:17 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/08/17 17:22:17 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2024/08/17 17:22:17 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "config_details" AND sql IS NOT NULL order by type = "table" desc 2 <nil>
[db    ] TRACE 2024/08/17 17:22:17 SELECT * FROM `config_details` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/08/17 17:22:17 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_configs_details"" %" OR sql LIKE "%CONSTRAINT fk_configs_details %" OR sql LIKE "%CONSTRAINT `fk_configs_details`%" OR sql LIKE "%CONSTRAINT [fk_configs_details]%" OR sql LIKE "%CONSTRAINT         fk_configs_details      %") -1 <nil>
[db    ] TRACE 2024/08/17 17:22:17 SELECT count(*) FROM sqlite_master WHERE type = "index" AND tbl_name = "config_details" AND name = "idx_unique" -1 <nil>
[db    ] TRACE 2024/08/17 17:22:17 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_devices_details"" %" OR sql LIKE "%CONSTRAINT fk_devices_details %" OR sql LIKE "%CONSTRAINT `fk_devices_details`%" OR sql LIKE "%CONSTRAINT [fk_devices_details]%" OR sql LIKE "%CONSTRAINT         fk_devices_details      %") -1 <nil>
[db    ] TRACE 2024/08/17 17:22:17 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%""device_id"" %" OR sql LIKE "%device_id %" OR sql LIKE "%`device_id`%" OR sql LIKE "%[device_id]%" OR sql LIKE "%   device_id       %") -1 <nil>
[sunspec] TRACE 2024/08/17 17:22:17 modbus: send 00 01 00 00 00 06 01 03 9c 40 00 02
[sunspec] TRACE 2024/08/17 17:22:17 modbus: recv 00 01 00 00 00 07 01 03 04 53 75 6e 53
[sunspec] TRACE 2024/08/17 17:22:17 modbus: send 00 02 00 00 00 06 01 03 9c 42 00 02
[sunspec] TRACE 2024/08/17 17:22:17 modbus: recv 00 02 00 00 00 07 01 03 04 00 01 00 41
[sunspec] TRACE 2024/08/17 17:22:17 modbus: send 00 03 00 00 00 06 01 03 9c 85 00 02
[sunspec] TRACE 2024/08/17 17:22:17 modbus: recv 00 03 00 00 00 07 01 03 04 00 65 00 32
[sunspec] TRACE 2024/08/17 17:22:17 modbus: send 00 04 00 00 00 06 01 03 9c b9 00 02
[sunspec] TRACE 2024/08/17 17:22:17 modbus: recv 00 04 00 00 00 07 01 03 04 00 01 00 41
[sunspec] TRACE 2024/08/17 17:22:17 modbus: send 00 05 00 00 00 06 01 03 9c fc 00 02
[sunspec] TRACE 2024/08/17 17:22:17 modbus: recv 00 05 00 00 00 07 01 03 04 00 cb 00 69
[sunspec] TRACE 2024/08/17 17:22:17 modbus: send 00 06 00 00 00 06 01 03 9d 67 00 02
[sunspec] TRACE 2024/08/17 17:22:17 modbus: recv 00 06 00 00 00 07 01 03 04 02 bd 00 99
[sunspec] TRACE 2024/08/17 17:22:17 modbus: send 00 07 00 00 00 06 01 03 9e 02 00 02
[sunspec] TRACE 2024/08/17 17:22:17 modbus: recv 00 07 00 00 00 07 01 03 04 02 be 00 32
[sunspec] TRACE 2024/08/17 17:22:17 modbus: send 00 08 00 00 00 06 01 03 9e 36 00 02
[sunspec] TRACE 2024/08/17 17:22:17 modbus: recv 00 08 00 00 00 07 01 03 04 02 bf 00 11
[sunspec] TRACE 2024/08/17 17:22:17 modbus: send 00 09 00 00 00 06 01 03 9e 49 00 02
[sunspec] TRACE 2024/08/17 17:22:17 modbus: recv 00 09 00 00 00 07 01 03 04 02 c0 00 41
[sunspec] TRACE 2024/08/17 17:22:17 modbus: send 00 0a 00 00 00 06 01 03 9e 8c 00 02
[sunspec] TRACE 2024/08/17 17:22:17 modbus: recv 00 0a 00 00 00 07 01 03 04 02 c1 00 39
[sunspec] TRACE 2024/08/17 17:22:17 modbus: send 00 0b 00 00 00 06 01 03 9e c7 00 02
[sunspec] TRACE 2024/08/17 17:22:17 modbus: recv 00 0b 00 00 00 07 01 03 04 02 c2 00 2f
[sunspec] TRACE 2024/08/17 17:22:17 modbus: send 00 0c 00 00 00 06 01 03 9e f8 00 02
[sunspec] TRACE 2024/08/17 17:22:17 modbus: recv 00 0c 00 00 00 07 01 03 04 02 c3 00 69
[sunspec] TRACE 2024/08/17 17:22:17 modbus: send 00 0d 00 00 00 06 01 03 9f 63 00 02
[sunspec] TRACE 2024/08/17 17:22:17 modbus: recv 00 0d 00 00 00 07 01 03 04 02 c4 00 69
[sunspec] TRACE 2024/08/17 17:22:17 modbus: send 00 0e 00 00 00 06 01 03 9f ce 00 02
[sunspec] TRACE 2024/08/17 17:22:17 modbus: recv 00 0e 00 00 00 07 01 03 04 02 c5 00 87
[sunspec] TRACE 2024/08/17 17:22:17 modbus: send 00 0f 00 00 00 06 01 03 a0 57 00 02
[... cut ...]
[modbus] TRACE 2024/08/17 17:22:26 modbus: send 01 09 00 00 00 06 01 03 f5 74 00 02
[modbus] TRACE 2024/08/17 17:22:26 modbus: recv 01 09 00 00 00 07 01 03 04 80 00 c3 ca
Power: 1006W

battery3
--------
[modbus] TRACE 2024/08/17 17:22:26 modbus: send 01 0a 00 00 00 06 01 03 e1 74 00 02
[modbus] TRACE 2024/08/17 17:22:26 modbus: recv 01 0a 00 00 00 07 01 03 04 80 00 c3 ca
[modbus] TRACE 2024/08/17 17:22:26 modbus: send 01 0b 00 00 00 06 01 03 e1 84 00 02
[modbus] TRACE 2024/08/17 17:22:27 modbus: recv 01 0b 00 00 00 07 01 03 04 33 33 42 b8
Power: 405W
Soc:   92%`

grobiballon, 2024-08-17:
And now the log where the issue appears again (low battery)
[meter with error.txt](https://github.com/user-attachments/files/16646625/meter.with.error.txt)

premultiply, 2024-08-18:
SE has still the bug in their firmware where they suddenly swap the words of the value on NaN. So this can no longer interpreted as NaN value but as ordinary value.
Please contact SE to have it fixed in their firmware!

grobiballon, 2024-08-18:
Hi, many thanks for your investigation. Is there anyone out there who could tell me which MODBUS value (Modbus Adress) is causing the issue? Otherwise it will be hard to explain to themfor me.

premultiply, 2024-08-18:
https://github.com/evcc-io/evcc/blob/07f54e878e5e2aa43b5f4eb17897eeb1e7abf4ad/templates/definition/meter/solaredge-hybrid.yaml#L122

andig, 2024-08-18:
@premultiply was spricht denn dagegen, dass wir die falschen NaNs- auch wenn es theoretisch auch richtige Werte sein könnten- einfach auch raus filtern?

grobiballon, 2024-08-18:
Ich hab mal ein SE Ticket aufgemacht. Mal sehen was da kommt.

andig, 2024-08-18:
@premultiply lass uns doch bitte nochmal schauen, was hier wie falsch ist. Ich sehe im Log:

    ff ff ff 7f

Das wäre (direkt) `NaN`, Aufgrund `float32s` swapped Encoding aber `ff 7f ff ff` und damit 

    -3.4028235e38

Da ich wenig Hoffnung habe, dass SE nach sovielen Jahren irgendwas vernünftiges tut würde ich das Problem lieber selbst lösen.

premultiply, 2024-08-18:
Alles bekannt.
Was ist jetzt anders als bei den vorherigen Issues zum gleichen Thema?

andig, 2024-08-18:
Bei mir die Erkenntnis, dass das lösbar ist. Spricht denn was dagegen? Ich habe spontan keinen Link gefunden wo wir das ausgeschlossen hatten. Aber vielleicht übersehe ich was?

closed 2024-08-18 by andig

grobiballon, 2024-09-17:
Sola rEdge hat mir gemeldet, dass (zumindest meine) Firmware geändert wurde und ich teste jetzt mit der EVCC 0.129.0 ob der Fehler noch auftritt.  Muss nun warten bis der Akku leer ist.

grobiballon, 2024-11-19:
Hallo, nach endlosem Schriftverkehr über Monate im Rahmen eines Tickets (und außerordentlicher Penetranz meinerseits) ist - so wie ich das sehe - das Problem seitens SolarEdge behoben. Getestet habe ich das mit der EVCC Version 0.129.0

@premultiply: Kannst du das einmal gegenprüfen? Dann könnte man das Thema hier ein für allemal abhaken.

[meter problem solved.txt](https://github.com/user-attachments/files/17822247/meter.problem.solved.txt)

Gruß
Andreas