evcc-io/evcc #15448: Pool WP Steuerung tuts nach Update auf 0.129.0 nicht mehr

ThiloGa, 2024-08-17:
### Describe the bug

Seit dem Update auf 0.129.0 tuts bei mir die Pool WP Steuerung nicht mehr.

Gehe ich auf 0.128.4 zurück gehts.

### Steps to reproduce

PV Überschuss vorhanden
60 sek Startup-Timer läuft herunter
WP wird per Tasmota Switch eingeschaltet, WP fährt hoch, läuft aber noch nicht, daher wenig Stromverbrauch.
nach 30 Sekunden wird wieder ausgeschaltet.
goto TOP

### Configuration details

```yaml
meters:
- name: wp-shellyplugs
  type: custom

  power:
    source: http
    uri: http://wp-pool/cm?cmnd=status%2010
    jq: if .StatusSNS.ENERGY.Power > 0 then .StatusSNS.ENERGY.Power + 0 else 0 end
  soc:
    source: http
    uri: https://ha:8123/api/states/sensor.temp_pool_temp
    method: GET # default HTTP method
    headers:
      - authorization: Bearer xxx
      - content-type: application/json
    jq: .state 

  currents:
    - source: http
      uri: http://wp-pool/cm?cmnd=status%2010
      method: GET # default HTTP method
      headers:
        - authorization: Bearer xxxx
        - content-type: application/json
      jq: .StatusSNS.ENERGY.Current 
    - source: const
      value: 0
    - source: const
      value: 0      

chargers:
  - name: wp_pool
  type: template
  template: tasmota
  host: wp-pool
  channel: 1
  integrateddevice: true
  icon: waterheater
  heating: true

vehicles:

  - name: wp_pool2
    type: custom
    title: Wärmepumpe Pool
    icon: waterheater
    phases: 1
    onIdentify:
      mode: pv
      mincurrent: 7
      maxcurrent: 8
    soc:
      source: http
      uri: https://ha:8123/api/states/sensor.temp_pool_temp
      method: GET # default HTTP method
      headers:
        - authorization: Bearer xxx
        - content-type: application/json
      jq: .state 

loadpoints:
  - title: Pool
  charger: wp_pool
  vehicle: wp_pool2
  meter: wp-shellyplugs
  mode: pv 
  enable:
    threshold: 0
    delay: 1m
  disable:
    threshold: 0
    delay: 15m    
  soc:
    estimate: false
    poll:
      mode: always
      interval: 5m
```

### Log details

```text
[lp-2  ] DEBUG 2024/08/17 11:15:27 set session soc limit: 27
[lp-2  ] DEBUG 2024/08/17 11:15:27 charge power: 0W
[lp-2  ] DEBUG 2024/08/17 11:15:27 charge currents: [0 0 0]A
[lp-2  ] DEBUG 2024/08/17 11:15:27 charger status: B
[lp-2  ] DEBUG 2024/08/17 11:15:27 vehicle soc: 23%
[lp-2  ] DEBUG 2024/08/17 11:15:27 !! active phases: 1p = min(0p measured 1p vehicle 1p physical 1p charger)
[lp-2  ] DEBUG 2024/08/17 11:15:27 pv charge current: 0A = 0A + -2.15A (494W @ 1p)
[lp-2  ] DEBUG 2024/08/17 11:15:40 charge power: 0W
[lp-2  ] DEBUG 2024/08/17 11:15:40 charge currents: [0 0 0]A
[lp-2  ] DEBUG 2024/08/17 11:15:55 charge power: 0W
[lp-2  ] DEBUG 2024/08/17 11:15:55 charge currents: [0 0 0]A
[lp-2  ] DEBUG 2024/08/17 11:15:55 charger status: B
[lp-2  ] DEBUG 2024/08/17 11:15:55 vehicle soc: 23%
[lp-2  ] DEBUG 2024/08/17 11:15:56 !! active phases: 1p = min(0p measured 1p vehicle 1p physical 1p charger)
[lp-2  ] DEBUG 2024/08/17 11:15:56 pv charge current: 0A = 0A + -2.24A (516W @ 1p)
[lp-2  ] DEBUG 2024/08/17 11:16:10 charge power: 0W
[lp-2  ] DEBUG 2024/08/17 11:16:10 charge currents: [0 0 0]A
[lp-2  ] DEBUG 2024/08/17 11:16:25 charge power: 0W
[lp-2  ] DEBUG 2024/08/17 11:16:25 charge currents: [0 0 0]A
[lp-2  ] DEBUG 2024/08/17 11:16:25 charger status: B
[lp-2  ] DEBUG 2024/08/17 11:16:25 vehicle soc: 23%
[lp-2  ] DEBUG 2024/08/17 11:16:26 !! active phases: 1p = min(0p measured 1p vehicle 1p physical 1p charger)
[lp-2  ] DEBUG 2024/08/17 11:16:26 pv charge current: 0A = 0A + -2.35A (540W @ 1p)
[lp-2  ] DEBUG 2024/08/17 11:16:40 charge power: 0W
[lp-2  ] DEBUG 2024/08/17 11:16:40 charge currents: [0 0 0]A
[lp-2  ] DEBUG 2024/08/17 11:16:55 charge power: 0W
[lp-2  ] DEBUG 2024/08/17 11:16:55 charge currents: [0 0 0]A
[lp-2  ] DEBUG 2024/08/17 11:16:55 charger status: B
[lp-2  ] DEBUG 2024/08/17 11:16:55 vehicle soc: 23%
[lp-2  ] DEBUG 2024/08/17 11:16:55 !! active phases: 1p = min(0p measured 1p vehicle 1p physical 1p charger)
[lp-2  ] DEBUG 2024/08/17 11:16:55 pv charge current: 0A = 0A + -2.26A (520W @ 1p)
[lp-2  ] DEBUG 2024/08/17 11:17:10 charge power: 0W
[lp-2  ] DEBUG 2024/08/17 11:17:10 charge currents: [0 0 0]A
[lp-2  ] DEBUG 2024/08/17 11:17:25 charge power: 0W
[lp-2  ] DEBUG 2024/08/17 11:17:25 charge currents: [0 0 0]A
[lp-2  ] DEBUG 2024/08/17 11:17:25 charger status: B
[lp-2  ] DEBUG 2024/08/17 11:17:25 vehicle soc: 23%
[lp-2  ] DEBUG 2024/08/17 11:17:26 !! active phases: 1p = min(0p measured 1p vehicle 1p physical 1p charger)
[lp-2  ] DEBUG 2024/08/17 11:17:26 pv charge current: 10.9A = 0A + 10.9A (-2499W @ 1p)
[lp-2  ] DEBUG 2024/08/17 11:17:26 site power -2499W <= 0W enable threshold
[lp-2  ] DEBUG 2024/08/17 11:17:26 pv enable timer start: 1m0s
[lp-2  ] DEBUG 2024/08/17 11:17:26 pv enable in 1m0s
[lp-2  ] DEBUG 2024/08/17 11:17:40 charge power: 0W
[lp-2  ] DEBUG 2024/08/17 11:17:40 charge currents: [0 0 0]A
[lp-2  ] DEBUG 2024/08/17 11:17:55 charge power: 0W
[lp-2  ] DEBUG 2024/08/17 11:17:55 charge currents: [0 0 0]A
[lp-2  ] DEBUG 2024/08/17 11:17:55 charger status: B
[lp-2  ] DEBUG 2024/08/17 11:17:55 vehicle soc: 23%
[lp-2  ] DEBUG 2024/08/17 11:17:56 !! active phases: 1p = min(0p measured 1p vehicle 1p physical 1p charger)
[lp-2  ] DEBUG 2024/08/17 11:17:56 pv charge current: 11.3A = 0A + 11.3A (-2597W @ 1p)
[lp-2  ] DEBUG 2024/08/17 11:17:56 site power -2597W <= 0W enable threshold
[lp-2  ] DEBUG 2024/08/17 11:17:56 pv enable in 30s
[lp-2  ] DEBUG 2024/08/17 11:17:56 pv enable timer remaining: 30s
[lp-2  ] DEBUG 2024/08/17 11:18:10 charge power: 0W
[lp-2  ] DEBUG 2024/08/17 11:18:10 charge currents: [0 0 0]A
[lp-2  ] DEBUG 2024/08/17 11:18:25 charge power: 0W
[lp-2  ] DEBUG 2024/08/17 11:18:25 charge currents: [0 0 0]A
[lp-2  ] DEBUG 2024/08/17 11:18:25 charger status: B
[lp-2  ] DEBUG 2024/08/17 11:18:25 vehicle soc: 23%
[lp-2  ] DEBUG 2024/08/17 11:18:26 !! active phases: 1p = min(0p measured 1p vehicle 1p physical 1p charger)
[lp-2  ] DEBUG 2024/08/17 11:18:26 pv charge current: 12.1A = 0A + 12.1A (-2789W @ 1p)
[lp-2  ] DEBUG 2024/08/17 11:18:26 site power -2789W <= 0W enable threshold
[lp-2  ] DEBUG 2024/08/17 11:18:26 pv enable in 0s
[lp-2  ] DEBUG 2024/08/17 11:18:26 pv enable timer elapsed
[lp-2  ] DEBUG 2024/08/17 11:18:26 max charge current: 7A
[lp-2  ] DEBUG 2024/08/17 11:18:26 charger enable
[lp-2  ] DEBUG 2024/08/17 11:18:26 wake-up timer: start
[lp-2  ] DEBUG 2024/08/17 11:18:40 charge power: 3W
[lp-2  ] DEBUG 2024/08/17 11:18:40 charge currents: [0.034 0 0]A
[lp-2  ] DEBUG 2024/08/17 11:18:55 charge power: 3W
[lp-2  ] DEBUG 2024/08/17 11:18:55 charge currents: [0.034 0 0]A
[lp-2  ] DEBUG 2024/08/17 11:18:55 charger status: B
[lp-2  ] DEBUG 2024/08/17 11:18:55 vehicle soc: 23%
[lp-2  ] DEBUG 2024/08/17 11:18:56 !! active phases: 1p = min(0p measured 1p vehicle 1p physical 1p charger)
[lp-2  ] DEBUG 2024/08/17 11:18:56 pv charge current: 11.3A = 0A + 11.3A (-2593W @ 1p)
[lp-2  ] DEBUG 2024/08/17 11:18:56 pv timer reset
[lp-2  ] DEBUG 2024/08/17 11:18:56 pv timer inactive
[lp-2  ] DEBUG 2024/08/17 11:18:56 charger disable
[lp-2  ] DEBUG 2024/08/17 11:18:56 wake-up timer: stop
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.129.0

andig, 2024-08-17:
Es braucht mal bitte (und wie immer!!!) ein vollständiges Log. Alle Ladepunkte, Site. Nicht bits&pieces. V.a. das ist komisch:

    [lp-2  ] DEBUG 2024/08/17 11:18:56 charger disable

ThiloGa, 2024-08-18:
> Es braucht mal bitte (und wie immer!!!) ein vollständiges Log. Alle Ladepunkte, Site. Nicht bits&pieces. V.a. das ist komisch:

Sry, habe es oben ergänzt.

andig, 2024-08-18:
@VolkerK62 hast Du evtl. eine Idee, warum lp2 hier wieder abschaltet? Aus dem Logfile sehe ich da keinen Grund?

VolkerK62, 2024-08-18:
Kann es an fehlendem `standbypower` liegen?

ThiloGa, 2024-08-18:
> Kann es an fehlendem `standbypower` liegen?
> 
> Edit: Habs mit einem Shelly probiert. Daran liegt es nicht. Bleibt zwar auf Status B, aber bleibt eingeschaltet.

Hier ist ein tasmotaisierter Shelly Plug.
Es wird immer exakt nach 30 sekunden abgeschaltet.

ThiloGa, 2024-08-18:
> Ich habe mal etwas mit meinem Shelly rumgespielt. Keine Ahnung warum, aber es liegt an den vehicle currents, wenn diese nicht zu den Laodpoint-Einstellungen passen
> 
> ```
> onIdentify:
>       mode: pv
>       mincurrent: 7
>       maxcurrent: 8
> ```
> 
> Sobald ich am Loadpoint den Max.Ladestrom auf 6A einstelle, zeigt sich genau das beschriebene Verhalten.
> 
> @ThiloGa welche Ladestrom-Einstellungen hast du in den Loadpoint-Settings?

[image]

ThiloGa, 2024-08-18:
> Ich habe mal etwas mit meinem Shelly rumgespielt. Keine Ahnung warum, aber es liegt an den vehicle currents, wenn diese nicht zu den Laodpoint-Einstellungen passen
> 
> ```
> onIdentify:
>       mode: pv
>       mincurrent: 7
>       maxcurrent: 8
> ```

Hab die Einstellungen angeglichen -> Läuft

andig, 2024-08-18:
Was hat sich eingeschlichen? Welche Einstellung funktioniert und welche nicht?

ThiloGa, 2024-08-18:
> Was hat sich eingeschlichen? 
Die oben beschriebene Problematik :-)
Mit 0.128.4 lief es hier

> Welche Einstellung funktioniert und welche nicht?
Hab in der GUI beim Loadpoint die gleichen min/max Stromangaben wie in den vehicle-stromangeben in der yaml eingestellt. Seit dem rennt hier die WP.

andig, 2024-08-18:
Ok, ich geb auf...

closed 2024-08-18 by andig

VolkerK62, 2024-08-18:
Bei meinem Test hatte ich am Vehicle
```
mincurrent: 7
Maxcurrent: 8
```
eingestellt.
Am Loadpoint
```
Min.Ladestrom: 1A
Max.Ladestrom: 8A
```
Das hat mit 0.130.0 einwandfrei funktioniert.
In dem Moment, wo ich am Loadpoint
```
Max.Ladestrom: 6A
```
eingestellt habe wurde abgeschaltet. 30s später wieder ein, 30s später wieder aus ...usw.

VolkerK62, 2024-08-18:
@ThiloGa  da du `integrateddevice` nutzt, kannst du dir die `onIdentify` Einstellungen beim Vehicle eigentlich sparen.

andig, 2024-08-18:
Danke Volker, das ist nachvollziehbar. Die eine Einstellung sagt "muss aus", die andere will einschalten.

Generell machen die min/max Angaben bei den Shellys überhaupt keinen Sinn. Hier sollte man den _einen_ festen Wert einstellen mit dem das Gerät dahinter erwartungsgemäß läuft.

VolkerK62, 2024-08-18:
>Die eine Einstellung sagt "muss aus", die andere will einschalten. Die Konfiguration widerspricht sich.

Was mich aber wundert, dann hätte es mit den Einstellungen von @ThiloGa  funktionieren müssen. 

>Generell machen die min/max Angaben bei den Shellys überhaupt keinen Sinn

evtl bei den switchsockets nur noch "Min Ladestrom" (für die Einschaltschwelle) anbieten?