evcc-io/evcc #15450: charging  with lowest price for x hours (instead of defining a price limit)

walburgf, 2024-08-17:
**Is your feature request related to a problem? Please describe.**
At the moment it is just possible to define a price limit.
[image]

**Describe the solution you'd like**
It would be great, if you also would be able to define: charge during x hours with the cheapest price in the next period
It also would be good if you could implement this a permanent setting (for a certain period). THis could be useful especially in winter times, when there is not so much PV power available.

**Describe alternatives you've considered**

**Additional context**
Add any other context or screenshots about the feature request here.
[image]

andig, 2024-08-17:
Instead of specifying hours you can already specify the target soc or energy which is equivalent.

walburgf, 2024-08-17:
This is only the case, when you have constant prices for the energy.
If you have prices changing on an houlry base (like with Tibber), just usind the target SOC would load immediatly until the SOC is reached, but not when the prices are cheapest. This would be the goal.

mdkeil, 2024-08-17:
combine target soc with a time and you have the implemented charge-plan feature, with charge in cheapest time frames.. were is your problem with this?

StefanSchoof, 2024-08-17:
Maybe a dump question: Have you found the planner feature? (https://docs.evcc.io/en/docs/features/plans)

andig, 2024-08-17:
Closing as not being an issue. Seems this is a general misunderstanding of what evcc already provides or the interfaces it has. Can ask questions in discussions.

closed 2024-08-17 by andig