evcc-io/evcc #15451: VW ID Buzz stopps charging when canging from 3p to 1p - OpenWB 2.x

angeldelivery86, 2024-08-17:
### Describe the bug

Charging with PV mode (Wallbox is OpenWB with software 2.x):
1p to 3p works perfect. When PV surplus is not enough evcc reduces to 1p. After that the Buzz stopps charging and get an Error (charging light is red - and cable cannot be released).
It can be reproduced.
ID software is 5.0

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
N/A
```

### Log details

```text
Aug 16 17:59:12 evcc evcc[25402]: [site  ] DEBUG 2024/08/16 17:59:12 ----
Aug 16 17:59:12 evcc evcc[25402]: [lp-1  ] DEBUG 2024/08/16 17:59:12 charge power: 0W
Aug 16 17:59:12 evcc evcc[25402]: [lp-1  ] DEBUG 2024/08/16 17:59:12 charge currents: [0 0 0]A
Aug 16 17:59:12 evcc evcc[25402]: [lp-2  ] DEBUG 2024/08/16 17:59:12 charge power: 4189W
Aug 16 17:59:12 evcc evcc[25402]: [lp-2  ] DEBUG 2024/08/16 17:59:12 charge currents: [6.02 5.99 6.17]A
Aug 16 17:59:12 evcc evcc[25402]: [site  ] DEBUG 2024/08/16 17:59:12 pv power: 3842W
Aug 16 17:59:12 evcc evcc[25402]: [site  ] DEBUG 2024/08/16 17:59:12 battery soc: 99%
Aug 16 17:59:12 evcc evcc[25402]: [site  ] DEBUG 2024/08/16 17:59:12 battery power: 2045W
Aug 16 17:59:12 evcc evcc[25402]: [site  ] DEBUG 2024/08/16 17:59:12 grid meter: -76W
Aug 16 17:59:12 evcc evcc[25402]: [site  ] DEBUG 2024/08/16 17:59:12 site power: 1969W
Aug 16 17:59:12 evcc evcc[25402]: [lp-2  ] DEBUG 2024/08/16 17:59:12 detected active phases: 3p
Aug 16 17:59:12 evcc evcc[25402]: [lp-2  ] DEBUG 2024/08/16 17:59:12 !! session: chargeRater.chargedEnergy=17.1 - chargedAtStartup=0.0
Aug 16 17:59:13 evcc evcc[25402]: [lp-2  ] DEBUG 2024/08/16 17:59:13 charge total import: 4910.125kWh
Aug 16 17:59:13 evcc evcc[25402]: [lp-2  ] DEBUG 2024/08/16 17:59:13 charger status: C
Aug 16 17:59:13 evcc evcc[25402]: [lp-2  ] DEBUG 2024/08/16 17:59:13 soc estimated: 82.43% (vehicle: 82.00%)
Aug 16 17:59:13 evcc evcc[25402]: [lp-2  ] DEBUG 2024/08/16 17:59:13 vehicle soc: 82%
Aug 16 17:59:13 evcc evcc[25402]: [lp-2  ] DEBUG 2024/08/16 17:59:13 vehicle soc limit: 100%
Aug 16 17:59:13 evcc evcc[25402]: [lp-2  ] DEBUG 2024/08/16 17:59:13 vehicle range: 319km
Aug 16 17:59:13 evcc evcc[25402]: [lp-2  ] DEBUG 2024/08/16 17:59:13 !! active phases: 3p = min(3p measured 3p vehicle 3p physical 0p charger)
Aug 16 17:59:13 evcc evcc[25402]: [lp-2  ] DEBUG 2024/08/16 17:59:13 available power 2220W < 4140W min 3p threshold
Aug 16 17:59:13 evcc evcc[25402]: [lp-2  ] DEBUG 2024/08/16 17:59:13 phase scale1p in 4s
Aug 16 17:59:13 evcc evcc[25402]: [lp-2  ] DEBUG 2024/08/16 17:59:13 !! active phases: 3p = min(3p measured 3p vehicle 3p physical 0p charger)
Aug 16 17:59:13 evcc evcc[25402]: [lp-2  ] DEBUG 2024/08/16 17:59:13 pv charge current: min 6A > 3.15A (1969W @ 3p, battery: true)
Aug 16 17:59:20 evcc evcc[25402]: [site  ] DEBUG 2024/08/16 17:59:20 ----
Aug 16 17:59:20 evcc evcc[25402]: [lp-1  ] DEBUG 2024/08/16 17:59:20 charge power: 0W
Aug 16 17:59:20 evcc evcc[25402]: [lp-1  ] DEBUG 2024/08/16 17:59:20 charge currents: [0 0 0]A
Aug 16 17:59:20 evcc evcc[25402]: [lp-2  ] DEBUG 2024/08/16 17:59:20 charge power: 4189W
Aug 16 17:59:20 evcc evcc[25402]: [lp-2  ] DEBUG 2024/08/16 17:59:20 charge currents: [6.02 5.99 6.17]A
Aug 16 17:59:20 evcc evcc[25402]: [site  ] DEBUG 2024/08/16 17:59:20 pv power: 4026W
Aug 16 17:59:20 evcc evcc[25402]: [site  ] DEBUG 2024/08/16 17:59:20 battery soc: 99%
Aug 16 17:59:20 evcc evcc[25402]: [site  ] DEBUG 2024/08/16 17:59:20 battery power: 1511W
Aug 16 17:59:20 evcc evcc[25402]: [site  ] DEBUG 2024/08/16 17:59:20 grid meter: -24W
Aug 16 17:59:20 evcc evcc[25402]: [site  ] DEBUG 2024/08/16 17:59:20 site power: 1487W
Aug 16 17:59:20 evcc evcc[25402]: [lp-1  ] DEBUG 2024/08/16 17:59:20 !! session: chargeRater.chargedEnergy=5.3 - chargedAtStartup=0.0
Aug 16 17:59:20 evcc evcc[25402]: [lp-1  ] DEBUG 2024/08/16 17:59:20 charge total import: 5.295kWh
Aug 16 17:59:20 evcc evcc[25402]: [lp-1  ] DEBUG 2024/08/16 17:59:20 charger status: C
Aug 16 17:59:21 evcc evcc[25402]: [lp-1  ] DEBUG 2024/08/16 17:59:21 soc estimated: 100.00% (vehicle: 100.00%)
Aug 16 17:59:21 evcc evcc[25402]: [lp-1  ] DEBUG 2024/08/16 17:59:21 vehicle soc: 100%
Aug 16 17:59:21 evcc evcc[25402]: [lp-1  ] DEBUG 2024/08/16 17:59:21 vehicle range: 400km
Aug 16 17:59:21 evcc evcc[25402]: [lp-1  ] DEBUG 2024/08/16 17:59:21 !! active phases: 1p = min(0p measured 3p vehicle 1p physical 0p charger)
Aug 16 17:59:21 evcc evcc[25402]: [lp-1  ] DEBUG 2024/08/16 17:59:21 !! active phases: 1p = min(0p measured 3p vehicle 1p physical 0p charger)
Aug 16 17:59:21 evcc evcc[25402]: [lp-1  ] DEBUG 2024/08/16 17:59:21 pv charge current: min 6A > 0A (1487W @ 1p, battery: true)
Aug 16 17:59:28 evcc evcc[25402]: [site  ] DEBUG 2024/08/16 17:59:28 ----
Aug 16 17:59:28 evcc evcc[25402]: [lp-1  ] DEBUG 2024/08/16 17:59:28 charge power: 0W
Aug 16 17:59:28 evcc evcc[25402]: [lp-1  ] DEBUG 2024/08/16 17:59:28 charge currents: [0 0 0]A
Aug 16 17:59:28 evcc evcc[25402]: [lp-2  ] DEBUG 2024/08/16 17:59:28 charge power: 4189W
Aug 16 17:59:28 evcc evcc[25402]: [lp-2  ] DEBUG 2024/08/16 17:59:28 charge currents: [6.02 5.99 6.17]A
[... 161 lines cut ...]
Aug 16 18:00:32 evcc evcc[25402]: [lp-1  ] DEBUG 2024/08/16 18:00:32 charge currents: [0 0 0]A
Aug 16 18:00:32 evcc evcc[25402]: [lp-2  ] DEBUG 2024/08/16 18:00:32 charge power: 0W
Aug 16 18:00:32 evcc evcc[25402]: [lp-2  ] DEBUG 2024/08/16 18:00:32 charge currents: [0 0 0]A
Aug 16 18:00:32 evcc evcc[25402]: [site  ] DEBUG 2024/08/16 18:00:32 pv power: 4564W
Aug 16 18:00:32 evcc evcc[25402]: [site  ] DEBUG 2024/08/16 18:00:32 battery soc: 99%
Aug 16 18:00:32 evcc evcc[25402]: [site  ] DEBUG 2024/08/16 18:00:32 battery power: -3227W
Aug 16 18:00:32 evcc evcc[25402]: [site  ] DEBUG 2024/08/16 18:00:32 grid meter: -77W
Aug 16 18:00:32 evcc evcc[25402]: [site  ] DEBUG 2024/08/16 18:00:32 site power: -3304W
Aug 16 18:00:32 evcc evcc[25402]: [lp-2  ] DEBUG 2024/08/16 18:00:32 !! session: chargeRater.chargedEnergy=17.1 - chargedAtStartup=0.0
Aug 16 18:00:32 evcc evcc[25402]: [lp-2  ] DEBUG 2024/08/16 18:00:32 charge total import: 4910.149kWh
Aug 16 18:00:33 evcc evcc[25402]: [lp-2  ] DEBUG 2024/08/16 18:00:33 charger status: B
Aug 16 18:00:33 evcc evcc[25402]: [lp-2  ] DEBUG 2024/08/16 18:00:33 !! active phases: 1p = min(0p measured 3p vehicle 1p physical 0p charger)
Aug 16 18:00:33 evcc evcc[25402]: [lp-2  ] DEBUG 2024/08/16 18:00:33 !! active phases: 1p = min(0p measured 3p vehicle 1p physical 0p charger)
Aug 16 18:00:33 evcc evcc[25402]: [lp-2  ] DEBUG 2024/08/16 18:00:33 pv charge current: 14.4A = 0A + 14.4A (-3304W @ 1p)
Aug 16 18:00:33 evcc evcc[25402]: [lp-2  ] DEBUG 2024/08/16 18:00:33 max charge current: 14.4A
Aug 16 18:00:40 evcc evcc[25402]: [site  ] DEBUG 2024/08/16 18:00:40 ----
```

### What type of operating system are you running?

Linux

### Version

0.128.4

andig, 2024-08-17:
Missing config

closed 2024-08-17 by andig

angeldelivery86, 2024-08-17:
Konfiguration (/etc/evcc.yaml):

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: *****

interval: 8s # control cycle interval

meters:
- type: template
  template: victron-energy 
  usage: grid  
  host: 192.168.1.233  
  port: 502  
  minsoc: 15
  maxsoc: 100  
  name: grid1
- type: template
  template: victron-energy 
  usage: pv  
  host: 192.168.1.233  
  port: 502  
  minsoc: 15 
  maxsoc: 100  
  name: pv2
- type: template
  template: victron-energy 
  usage: battery  
  host: 192.168.1.233  
  port: 502  
  minsoc: 15  
  maxsoc: 100  
  name: battery3

chargers:
- type: template
  template: openwb-2.0
  id: 1
  host: 192.168.1.88
  port: 1502
  modbus: tcpip
  connector: 1
  name: wallbox2
- type: template
  template: openwb-2.0
  id: 1
  host: 192.168.1.89
  port: 1502
  modbus: tcpip
  connector: 1
  name: wallbox1

vehicles:
- type: template
  template: polestar 
  title: Polestar 2  
  icon: car  
  user: *****
  password: *****
  vin: *****
  capacity: 78  
  phases: 3  
  cache: 30m  
  mode: pv  
  minCurrent: 6  
  maxCurrent: 16  
  priority: 1  
  name: polestar
- type: template
  template: vw 
  title: ID Buzz
  icon: car  
  user: *****
  password: *****
  vin: *****
  capacity: 79 
  phases: 3  
  cache: 30m  
  mode: pv  
  minCurrent: 6  
  maxCurrent: 16  
  priority: 1  
  name: buzz

loadpoints:
- title: Wallbox 1
  charger: wallbox1
  vehicle: polestar
  mode: pv
  soc:
    poll:
      mode: charging
      interval: 60m
    estimate: true
- title: Wallbox 2
  charger: wallbox2
  vehicle: buzz
  mode: pv
  soc:
    poll:
      mode: charging
      interval: 60m
    estimate: true

site:
  title: Villa Kunterbunt
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3

tariffs:
  currency: EUR
  grid:
    type: tibber
    token: *****
    homeid:
  feedin:
    type: fixed
    price: 0.075

Version: `0.129.0 (f73072d8)`

angeldelivery86, 2024-08-17:
@andig bitte nochmal aufmachen. Habe die Yaml mit eingefügt

andig, 2024-08-17:
Die WB sagt abwechselnd löd/ löd nicht. Am Besten mal mit einem anderen Auto ausprobieren. Ich tippe auf Fehler bei VW.

angeldelivery86, 2024-08-17:
Ja mit dem Polestar funktioniert es bestens. Ich wollte ausschließen das es an meiner Seite liegt.

angeldelivery86, 2024-08-17:
Kann es mit der Phasenumschlaltung zu tun haben?

andig, 2024-08-17:
Ich fürchte nicht. Die OpenWB läuft sonst sehr unauffällig. Ich würde aber auch vorschlagen, den OpenWB Support mal zu fragen. Vllt. gibts ja ein Massenphänomän dass sich dann doch noch heilen lässt.

angeldelivery86, 2024-08-17:
Alles klar. Danke dir.

andig, 2024-08-17:
> Kann es mit der Phasenumschlaltung zu tun haben?

Warum nicht wenn das Auto komische Dinge tut.