evcc-io/evcc #15467: V 0.130 im HA Docker schmiert nach Update ab.

HaareistNase, 2024-08-18:
### Describe the bug

Nach dem Update von 1.28 auf 1.3 schmiert evcc beim starten ab.

Sieht so aus, als würde ein programmierender Fehler auftreten.

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

influx: 
  url: http://192.168.178.xxx:8086
  database: homeassistant
  user: xxxxx
  password: xxxxx
  
mqtt:
  broker: 192.168.178.xxxxx:1883
  topic: evcc
  user: xxxxx
  password: xxxxx
  clientid: evcc
  
log: info
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error

# unique installation id
plant: a34e9891f3f851942995bc6b38eafffd9a2aa49c1787b7ec87d24ac08c8be018

interval: 10s # control cycle interval

sponsortoken: xxxxx
# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: true

meters:

- type: template
  template: growatt-hybrid
  id: 1
  usage: grid
  modbus: tcpip
  host: 192.168.178.xxxxx
  port: 5021
  name: grid1

- type: template
  template: growatt-hybrid
  id: 1
  usage: pv
  modbus: tcpip
  host: 192.168.178.xxxxx
  port: 5021
  name: pv1

- type: template
  template: growatt-hybrid
  id: 1
  usage: pv
  modbus: tcpip
  host: 192.168.178.xxxx
  port: 5021
  name: pv2

- type: template
  template: growatt-hybrid
  id: 1
  usage: battery
  modbus: tcpip
  host: 192.168.178.xxxx
  port: 5021
  capacity: 19.2
  name: battery3

chargers:
- type: template
  template: innogy-ebox
  id: 1
  host: 192.168.178.xxxx
  port: 502
  modbus: tcpip
  name: wallbox5

vehicles:

- type: custom
  identifiers:
  - Upita
  name: ev2
  title: E-Up
  icon: car
  phases: 3
  capacity: 32
  soc:
    source: mqtt
    topic: export/e-up
    
- type: custom
  identifiers:
  - Penita
  name: ev1
  title: Kia
  icon: car
  phases: 3
  capacity: 64
  soc:
    source: mqtt
    topic: export/niro

loadpoints:
- title: Front
  charger: wallbox5
  mode: off
  soc:
   poll:
     mode: allways
     interval: 2m
   estimate: false

# tariffs are the fixed or variable tariffs
tariffs:
  currency: EUR # three letter ISO-4217 currency code (default EUR)
  grid:
    type: tibber
    token: "xxxxxx" # access token
  feedin:
    type: fixed
    price: 0.068

site:
  title: My home
  meters:
    grid: grid1
    pv:
    - pv1
    - pv2

    battery:
    - battery3
  residualPower: 100 # additional household usage margin
  maxGridSupplyWhileBatteryCharging: 1000 # ignore battery charging if AC consumption is above this value
```

### Log details

```text
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2024/08/18 15:01:58 evcc 0.130.0
[main  ] INFO 2024/08/18 15:01:58 using config file: /config/evcc.yaml
[mqtt  ] INFO 2024/08/18 15:01:58 connecting evcc at tcp://192.168.178.21:1883
[main  ] INFO 2024/08/18 15:01:58 listening at :7070
[lp-1  ] WARN 2024/08/18 15:01:58 invalid poll mode: allways
[site  ] WARN 2024/08/18 15:02:00 vehicle 'Kia' does not support automatic detection
[site  ] WARN 2024/08/18 15:02:00 vehicle 'E-Up' does not support automatic detection
[site  ] INFO 2024/08/18 15:02:00 site config:
[site  ] INFO 2024/08/18 15:02:00   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2024/08/18 15:02:00     grid:      power ✓ energy ✓ currents ✗
[site  ] INFO 2024/08/18 15:02:00     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2024/08/18 15:02:00     pv 2:      power ✓ energy ✓ currents ✗
[site  ] INFO 2024/08/18 15:02:00     battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✓
[site  ] INFO 2024/08/18 15:02:00   vehicles:
[site  ] INFO 2024/08/18 15:02:00     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[site  ] INFO 2024/08/18 15:02:00     vehicle 2: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2024/08/18 15:02:00 loadpoint 1:
[lp-1  ] INFO 2024/08/18 15:02:00   mode:        off
[lp-1  ] INFO 2024/08/18 15:02:00   charger:     power ✗ energy ✗ currents ✗ phases ✗ wakeup ✗
[lp-1  ] INFO 2024/08/18 15:02:00   meters:      charge ✗
[site  ] DEBUG 2024/08/18 15:02:00 set buffer soc: 90
[site  ] DEBUG 2024/08/18 15:02:00 set buffer start soc: 100
[site  ] DEBUG 2024/08/18 15:02:00 set priority soc: 60
[lp-1  ] DEBUG 2024/08/18 15:02:00 set smart cost limit: 0.0
[lp-1  ] DEBUG 2024/08/18 15:02:00 phase timer inactive
[lp-1  ] DEBUG 2024/08/18 15:02:00 pv timer inactive
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x20 pc=0x25c7588]

goroutine 1 [running]:
github.com/evcc-io/evcc/core.(*Loadpoint).Prepare(0xc001089408, 0x0?, 0x5bddde0?, 0xc00162d848?)
	github.com/evcc-io/evcc/core/loadpoint.go:651 +0xf28
github.com/evcc-io/evcc/core.(*Site).Prepare(0xc001636000, 0xc00002cb60, 0xc00056a150)
	github.com/evcc-io/evcc/core/site.go:968 +0x17b
github.com/evcc-io/evcc/cmd.runRoot(0x5b82360, {0x2e47152?, 0x7?, 0x2e3a116?})
	github.com/evcc-io/evcc/cmd/root.go:283 +0x1d92
github.com/spf13/cobra.(*Command).execute(0x5b82360, {0xc0000501f0, 0x2, 0x2})
	github.com/spf13/cobra@v1.8.1/command.go:989 +0xa91
github.com/spf13/cobra.(*Command).ExecuteC(0x5b82360)
	github.com/spf13/cobra@v1.8.1/command.go:1117 +0x3ff
github.com/spf13/cobra.(*Command).Execute(...)
	github.com/spf13/cobra@v1.8.1/command.go:1041
github.com/evcc-io/evcc/cmd.Execute()
	github.com/evcc-io/evcc/cmd/root.go:107 +0x1a
main.main()
	github.com/evcc-io/evcc/main.go:44 +0x2d
```

### What type of operating system are you running?

Linux

andig, 2024-08-18:
Das wäre 

	if enabled, err := lp.charger.Enabled(); err == nil {

Da müsste die WB nil sein. Keine Ahnung, wie das passieren soll?

andig, 2024-08-18:
Am einfachsten wäre, wenn Du ein VPN aufmachen könntest (Fritzbox?!) und info@evcc.io dazu einladen. Dann könnte ich Deine Config in Deinem Netz im Debugger testen. Sonst hab ich grad keine gute Idee...

SolarPower2024, 2024-08-18:
Als ein allgemeiner Fehler scheint es nicht zu sein, ich habe soeben von 129.0 auf 130.0 upgedatet in HA, ohne Probleme.
Vielleicht ist bei der Installation etwas schief gegangen? Versuche mal das Add-on nochmals neu zu installieren.

andig, 2024-08-18:
Das hat damit nix zu tun. panic ist panic :/

HaareistNase, 2024-08-18:
> Am einfachsten wäre, wenn Du ein VPN aufmachen könntest (Fritzbox?!) und [info@evcc.io](mailto:info@evcc.io) dazu einladen. Dann könnte ich Deine Config in Deinem Netz im Debugger testen. Sonst hab ich grad keine gute Idee...

Würde ich ja sofort machen (Fritzbox ist da), aber ich bin Planlos, wie ich Dir Zugangsdaten und den Weg in meinem Netz übermitteln kann :-)

HaareistNase, 2024-08-18:
Update.
Ich habe wirklich nichts geändert, aber inzwischen ist ein Startversuch um 15:50 erfolgreich gewesen.
HA versucht ja im Hintergrund ein nicht laufendes Addon zu starten. Das scheint hier erfolgreich gewesen zu sein.

HaareistNase, 2024-08-18:
Was mich auch gleich zu einem Schreibfehler führt .. allways .. ich werde es nicht lernen ... ein l

HaareistNase, 2024-08-18:
Übrigens Wiederholbar.
Ich hab jetzt von always auf connected gestellt.
EVCC gestoppt.
EVCC gestartet.

Absturz

````
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2024/08/18 16:15:17 evcc 0.130.0
[main  ] INFO 2024/08/18 16:15:17 using config file: /config/evcc.yaml
[mqtt  ] INFO 2024/08/18 16:15:17 connecting evcc at tcp://192.168.178.21:1883
[main  ] INFO 2024/08/18 16:15:17 listening at :7070
[lp-1  ] WARN 2024/08/18 16:15:18 poll mode '{connected 2m0s}' may deplete your battery or lead to API misuse. USE AT YOUR OWN RISK.
[site  ] WARN 2024/08/18 16:15:19 vehicle 'Kia' does not support automatic detection
[site  ] WARN 2024/08/18 16:15:19 vehicle 'E-Up' does not support automatic detection
[site  ] INFO 2024/08/18 16:15:19 site config:
[site  ] INFO 2024/08/18 16:15:19   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2024/08/18 16:15:19     grid:      power ✓ energy ✓ currents ✗
[site  ] INFO 2024/08/18 16:15:19     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2024/08/18 16:15:19     pv 2:      power ✓ energy ✓ currents ✗
[site  ] INFO 2024/08/18 16:15:19     battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✓
[site  ] INFO 2024/08/18 16:15:19   vehicles:
[site  ] INFO 2024/08/18 16:15:19     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[site  ] INFO 2024/08/18 16:15:19     vehicle 2: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2024/08/18 16:15:19 loadpoint 1:
[lp-1  ] INFO 2024/08/18 16:15:19   mode:        off
[lp-1  ] INFO 2024/08/18 16:15:19   charger:     power ✗ energy ✗ currents ✗ phases ✗ wakeup ✗
[lp-1  ] INFO 2024/08/18 16:15:19   meters:      charge ✗
[site  ] DEBUG 2024/08/18 16:15:19 set buffer soc: 90
[site  ] DEBUG 2024/08/18 16:15:19 set buffer start soc: 100
[site  ] DEBUG 2024/08/18 16:15:19 set priority soc: 60
[lp-1  ] DEBUG 2024/08/18 16:15:19 set smart cost limit: 0.0
[lp-1  ] DEBUG 2024/08/18 16:15:19 phase timer inactive
[lp-1  ] DEBUG 2024/08/18 16:15:19 pv timer inactive
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x20 pc=0x25c7588]

goroutine 1 [running]:
github.com/evcc-io/evcc/core.(*Loadpoint).Prepare(0xc001162408, 0x0?, 0x5bddde0?, 0xc001595f38?)
	github.com/evcc-io/evcc/core/loadpoint.go:651 +0xf28
github.com/evcc-io/evcc/core.(*Site).Prepare(0xc0015a4000, 0xc0002fe4d0, 0xc000549030)
	github.com/evcc-io/evcc/core/site.go:968 +0x17b
github.com/evcc-io/evcc/cmd.runRoot(0x5b82360, {0x2e47152?, 0x7?, 0x2e3a116?})
	github.com/evcc-io/evcc/cmd/root.go:283 +0x1d92
github.com/spf13/cobra.(*Command).execute(0x5b82360, {0xc0000501f0, 0x2, 0x2})
	github.com/spf13/cobra@v1.8.1/command.go:989 +0xa91
github.com/spf13/cobra.(*Command).ExecuteC(0x5b82360)
	github.com/spf13/cobra@v1.8.1/command.go:1117 +0x3ff
github.com/spf13/cobra.(*Command).Execute(...)
	github.com/spf13/cobra@v1.8.1/command.go:1041
github.com/evcc-io/evcc/cmd.Execute()
	github.com/evcc-io/evcc/cmd/root.go:107 +0x1a
main.main()
	github.com/evcc-io/evcc/main.go:44 +0x2d
````

andig, 2024-08-18:
Ändert ja nix. Die Maßnahme ist klar. Email steht oben.

HaareistNase, 2024-08-18:
> Ändert ja nix. Die Maßnahme ist klar. Email steht oben.

Mail ist unterwegs :-)

andig, 2024-08-18:
Ähnlicher Vorfall- hie fehlt ein Zähler: https://github.com/evcc-io/evcc/issues/14911#issuecomment-2295283226

Timer2000, 2024-08-18:
Hier im HA Docker das selbe Problem, spiele gerade das Backup ein. Noch ist Sonne und ich muß erstmal laden.

Was mir sehr gut gefällt ist wie schnell hier auf solche Meldungen reagiert wird.

Liebes EVCC Team, vielen Dank für euren Einsatz.

Gruß
Timer

grunaur, 2024-08-18:
Hallo,
hier nochmal zur Info mein Fehlerbild:

C:\evcc>evcc
[main  ] INFO 2024/08/18 21:28:48 evcc 0.130.0
[main  ] INFO 2024/08/18 21:28:48 using config file: C:\evcc\evcc.yaml
[db    ] INFO 2024/08/18 21:28:48 using sqlite database: C:\Users\rgrunau\.evcc\evcc.db
[main  ] INFO 2024/08/18 21:28:48 listening at :7070
[lp-2  ] DEBUG 2024/08/18 21:28:53 charge total import: 351.242kWh
[lp-3  ] DEBUG 2024/08/18 21:28:53 charge total import: 324.893kWh
[lp-4  ] DEBUG 2024/08/18 21:28:53 charge total import: 343.192kWh
[site  ] INFO 2024/08/18 21:28:53 site config:
[site  ] INFO 2024/08/18 21:28:53   meters:      grid ✗ pv ✓ battery ✓
[site  ] INFO 2024/08/18 21:28:53     pv 1:      power ✗ energy ✗ currents ✗
[site  ] INFO 2024/08/18 21:28:53     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✓
[site  ] INFO 2024/08/18 21:28:53   vehicles:
[site  ] INFO 2024/08/18 21:28:53     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[site  ] INFO 2024/08/18 21:28:53     vehicle 2: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[site  ] INFO 2024/08/18 21:28:53     vehicle 3: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[lp-1  ] INFO 2024/08/18 21:28:53 loadpoint 1:
[lp-1  ] INFO 2024/08/18 21:28:53   mode:        off
[lp-1  ] INFO 2024/08/18 21:28:53   charger:     power ✓ energy ✗ currents ✗ phases ✗ wakeup ✗
[lp-1  ] INFO 2024/08/18 21:28:53   meters:      charge ✓
[lp-1  ] INFO 2024/08/18 21:28:53     charge:    power ✓ energy ✗ currents ✗
[lp-2  ] INFO 2024/08/18 21:28:53 loadpoint 2:
[lp-2  ] INFO 2024/08/18 21:28:53   mode:        pv
[lp-2  ] INFO 2024/08/18 21:28:53   charger:     power ✓ energy ✓ currents ✗ phases ✗ wakeup ✗
[lp-2  ] INFO 2024/08/18 21:28:53   meters:      charge ✓
[lp-2  ] INFO 2024/08/18 21:28:53     charge:    power ✓ energy ✓ currents ✗
[lp-3  ] INFO 2024/08/18 21:28:53 loadpoint 3:
[lp-3  ] INFO 2024/08/18 21:28:53   mode:        pv
[lp-3  ] INFO 2024/08/18 21:28:53   charger:     power ✓ energy ✓ currents ✗ phases ✗ wakeup ✗
[lp-3  ] INFO 2024/08/18 21:28:53   meters:      charge ✓
[lp-3  ] INFO 2024/08/18 21:28:53     charge:    power ✓ energy ✓ currents ✗
[lp-4  ] INFO 2024/08/18 21:28:53 loadpoint 4:
[lp-4  ] INFO 2024/08/18 21:28:53   mode:        pv
[lp-4  ] INFO 2024/08/18 21:28:53   charger:     power ✓ energy ✓ currents ✗ phases ✗ wakeup ✗
[lp-4  ] INFO 2024/08/18 21:28:53   meters:      charge ✓
[lp-4  ] INFO 2024/08/18 21:28:53     charge:    power ✓ energy ✓ currents ✗
[site  ] DEBUG 2024/08/18 21:28:53 set buffer soc: 100
[site  ] DEBUG 2024/08/18 21:28:53 set buffer start soc: 100
[site  ] DEBUG 2024/08/18 21:28:53 set priority soc: 100
[site  ] DEBUG 2024/08/18 21:28:53 set battery discharge control: true
[lp-1  ] DEBUG 2024/08/18 21:28:53 set smart cost limit: 0.2
[lp-1  ] DEBUG 2024/08/18 21:28:53 phase timer inactive
[lp-1  ] DEBUG 2024/08/18 21:28:53 pv timer inactive
[lp-2  ] DEBUG 2024/08/18 21:28:53 set smart cost limit: 0.0
[lp-2  ] DEBUG 2024/08/18 21:28:53 phase timer inactive
[lp-2  ] DEBUG 2024/08/18 21:28:53 pv timer inactive
[lp-3  ] DEBUG 2024/08/18 21:28:53 set smart cost limit: 0.0
[lp-3  ] DEBUG 2024/08/18 21:28:53 phase timer inactive
[lp-3  ] DEBUG 2024/08/18 21:28:53 pv timer inactive
[lp-4  ] DEBUG 2024/08/18 21:28:53 set smart cost limit: 0.0
[lp-4  ] DEBUG 2024/08/18 21:28:53 phase timer inactive
[lp-4  ] DEBUG 2024/08/18 21:28:53 pv timer inactive
[site  ] WARN 2024/08/18 21:28:53 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site  ] DEBUG 2024/08/18 21:28:53 ----
[lp-1  ] DEBUG 2024/08/18 21:28:53 charge power: 0W
[lp-2  ] DEBUG 2024/08/18 21:28:53 charge power: 0W
[lp-3  ] DEBUG 2024/08/18 21:28:53 charge power: 0W
[lp-4  ] DEBUG 2024/08/18 21:28:53 charge power: 0W
panic: runtime error: invalid memory address or nil pointer dereference
[signal 0xc0000005 code=0x0 addr=0x0 pc=0x2570418]

goroutine 358 [running]:
github.com/evcc-io/evcc/core.(*Site).updatePvMeters(0xc001c55800)
        github.com/evcc-io/evcc/core/site.go:452 +0xf8
github.com/evcc-io/evcc/core.(*Site).updateMeters(0xc001c55800)
        github.com/evcc-io/evcc/core/site.go:661 +0x18
github.com/evcc-io/evcc/core.(*Site).sitePower(0xc001c55800, 0x0, 0x0)
        github.com/evcc-io/evcc/core/site.go:674 +0x67
github.com/evcc-io/evcc/core.(*Site).update(0xc001c55800, {0x36f1470, 0xc000507408})
        github.com/evcc-io/evcc/core/site.go:870 +0x64b
github.com/evcc-io/evcc/core.(*Site).Run(0xc001c55800, 0xc00116e380, 0x37e11d600)
        github.com/evcc-io/evcc/core/site.go:994 +0x217
github.com/evcc-io/evcc/cmd.runRoot.func4()
        github.com/evcc-io/evcc/cmd/root.go:288 +0x25
created by github.com/evcc-io/evcc/cmd.runRoot in goroutine 1
        github.com/evcc-io/evcc/cmd/root.go:287 +0x1e13

C:\evcc>

Ich könnte auf dem Rechner TeamViewer starten, zum testen wenn gewünscht.
Auch von meiner Seite vielen Dank an das evcc Team für euren Einsatz!!

VG
Roland

andig, 2024-08-18:
In dem Fall brauchts Remotezugang via VPN, s.o.

ross-w, 2024-08-19:
Looks like this is well under control here, but for what it's worth I have the same issue. It only appeared for me in the nightly (Docker) version released around 0200 UTC on Aug 19. The previous nightly did not exhibit the issue. When I then tried release version `0.130.0` the same issue occurred. Hope that helps narrow it down.

andig, 2024-08-19:
That's interesting, however I couldn't find anything obvious. Could you share the commit hashes of working and non-working nightly?

RettungsTim, 2024-08-19:
Ich habe leider das selbe Problem mit EVCC in Homeassistant.

`
[main  ] INFO 2024/08/19 17:10:02 evcc 0.130.0
[main  ] INFO 2024/08/19 17:10:02 using config file: /config/evcc.yaml
[db    ] INFO 2024/08/19 17:10:02 using sqlite database: /data/evcc.db
[main  ] INFO 2024/08/19 17:10:05 listening at :7070
[lp-1  ] DEBUG 2024/08/19 17:10:17 charge total import: 1234.737kWh
[site  ] INFO 2024/08/19 17:10:17 site config:
[site  ] INFO 2024/08/19 17:10:17   meters:      grid ✗ pv ✓ battery ✓
[site  ] INFO 2024/08/19 17:10:17     pv 1:      power ✗ energy ✗ currents ✗
[site  ] INFO 2024/08/19 17:10:17     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✓
[site  ] INFO 2024/08/19 17:10:17   vehicles:
[site  ] INFO 2024/08/19 17:10:17     vehicle 1: range ✓ finish ✗ status ✓ climate ✗ wakeup ✓
[lp-1  ] INFO 2024/08/19 17:10:17 loadpoint 1:
[lp-1  ] INFO 2024/08/19 17:10:17   mode:        pv
[lp-1  ] INFO 2024/08/19 17:10:17   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2024/08/19 17:10:17   meters:      charge ✓
[lp-1  ] INFO 2024/08/19 17:10:17     charge:    power ✓ energy ✓ currents ✓
[site  ] DEBUG 2024/08/19 17:10:17 set buffer soc: 95
[site  ] DEBUG 2024/08/19 17:10:17 set buffer start soc: 95
[site  ] DEBUG 2024/08/19 17:10:17 set priority soc: 75
[site  ] DEBUG 2024/08/19 17:10:17 set battery discharge control: true
[lp-1  ] DEBUG 2024/08/19 17:10:17 set smart cost limit: 0.3
[lp-1  ] DEBUG 2024/08/19 17:10:17 phase timer inactive
[lp-1  ] DEBUG 2024/08/19 17:10:17 pv timer inactive
[lp-1  ] DEBUG 2024/08/19 17:10:17 max charge current: 6A
[site  ] WARN 2024/08/19 17:10:17 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site  ] DEBUG 2024/08/19 17:10:17 ----
[lp-1  ] DEBUG 2024/08/19 17:10:17 charge power: 10319W
[lp-1  ] DEBUG 2024/08/19 17:10:17 charge currents: [16 15.2 14.4]A
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0x25e1f78]

goroutine 218 [running]:
github.com/evcc-io/evcc/core.(*Site).updatePvMeters(0xc000da4a00)
	github.com/evcc-io/evcc/core/site.go:452 +0xf8
github.com/evcc-io/evcc/core.(*Site).updateMeters(0xc000da4a00)
	github.com/evcc-io/evcc/core/site.go:661 +0x18
github.com/evcc-io/evcc/core.(*Site).sitePower(0xc000da4a00, 0x40c427bd4024b33e, 0x0)
	github.com/evcc-io/evcc/core/site.go:674 +0x67
github.com/evcc-io/evcc/core.(*Site).update(0xc000da4a00, {0x3760dd0, 0xc0010f8c08})
	github.com/evcc-io/evcc/core/site.go:870 +0x64b
github.com/evcc-io/evcc/core.(*Site).Run(0xc000da4a00, 0xc0012cefc0, 0x2540be400)
	github.com/evcc-io/evcc/core/site.go:994 +0x217
github.com/evcc-io/evcc/cmd.runRoot.func4()
	github.com/evcc-io/evcc/cmd/root.go:288 +0x25
created by github.com/evcc-io/evcc/cmd.runRoot in goroutine 1
	github.com/evcc-io/evcc/cmd/root.go:287 +0x1e13
`

Arturfrain, 2024-08-19:
Auch bei mir in Homeassistant selbes Problem.

[main  ] INFO 2024/08/19 19:27:05 evcc 0.130.0
[main  ] INFO 2024/08/19 19:27:05 using config file: /config/evcc.yaml
[mqtt  ] INFO 2024/08/19 19:27:05 connecting foo at tcp://192.168.178.79:1883
[main  ] INFO 2024/08/19 19:27:05 listening at :7070
[tesla ] INFO 2024/08/19 19:27:10 !! tesla proxy for Emma: http://192.168.178.106:8080
[site  ] WARN 2024/08/19 19:27:10 battery configured but residualPower is missing or <= 0 (add residualPower: 100 to site), see https://docs.evcc.io/en/docs/reference/configuration/site#residualpower
[site  ] INFO 2024/08/19 19:27:11 site config:
[site  ] INFO 2024/08/19 19:27:11   meters:      grid ✗ pv ✓ battery ✓
[site  ] INFO 2024/08/19 19:27:11     pv 1:      power ✗ energy ✗ currents ✗
[site  ] INFO 2024/08/19 19:27:11     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✓
[site  ] INFO 2024/08/19 19:27:11   vehicles:
[site  ] INFO 2024/08/19 19:27:11     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[site  ] INFO 2024/08/19 19:27:11     vehicle 2: range ✓ finish ✓ status ✓ climate ✗ wakeup ✓
[lp-1  ] INFO 2024/08/19 19:27:11 loadpoint 1:
[lp-1  ] INFO 2024/08/19 19:27:11   mode:        pv
[lp-1  ] INFO 2024/08/19 19:27:11   charger:     power ✓ energy ✗ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2024/08/19 19:27:11   meters:      charge ✓
[lp-1  ] INFO 2024/08/19 19:27:11     charge:    power ✓ energy ✗ currents ✓
[site  ] DEBUG 2024/08/19 19:27:11 set buffer soc: 100
[site  ] DEBUG 2024/08/19 19:27:11 set buffer start soc: 100
[site  ] DEBUG 2024/08/19 19:27:11 set priority soc: 80
[site  ] DEBUG 2024/08/19 19:27:11 set battery discharge control: false
[lp-1  ] DEBUG 2024/08/19 19:27:11 set smart cost limit: 0.0
[lp-1  ] DEBUG 2024/08/19 19:27:11 phase timer inactive
[lp-1  ] DEBUG 2024/08/19 19:27:11 pv timer inactive
[lp-1  ] INFO 2024/08/19 19:27:11 vehicle updated: unknown -> Emma
[site  ] DEBUG 2024/08/19 19:27:11 ----
[lp-1  ] DEBUG 2024/08/19 19:27:11 charge power: 0W
[lp-1  ] DEBUG 2024/08/19 19:27:11 charge currents: [0 0.3 0.6]A
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0x25e1f78]

WordsOfMe, 2024-08-19:
Der Fehler scheint nicht zwangsweise aufzutreten, hier mit aktuellem HA Core 2024.8.2 und HA OS 12.4 startet evcc nach dem Update auf 0.130.0 problemlos.

andig, 2024-08-19:
Siehe https://github.com/evcc-io/evcc/issues/15467#issuecomment-2295369264. Es braucht bitte Remotezugang mittels VPN um das im Debugger testen zu können. Bisher ist es nicht gelungen den Fehler nachzustellen. 

Ebenfalls hilfreich wäre es, per `git bisect` die Commits ein paar Tage vor 0.130.0 zu testen.

WordsOfMe, 2024-08-19:
VPN-Zugang könnte ich einrichten, aber "leider" kann ich den Fehler auch nicht reproduzieren... hilft das dann irgendwie?

Arturfrain, 2024-08-19:
> Der Fehler scheint nicht zwangsweise aufzutreten, hier mit aktuellem HA Core 2024.8.2 und HA OS 12.4 startet evcc nach dem Update auf 0.130.0 problemlos.

Mit Core
2024.8.2
Supervisor
2024.08.0
Operating System
13.0
Frontend
20240809.0
Funktioniert evcc 0.130.0 bei mir nicht. Logs sind die gleichen wie bei den anderen oben.

andig, 2024-08-19:
Weitere "ich auch" Einträge helfen nicht. VPN Zugang und Config dazu schon...

andig, 2024-08-19:
Wir müssen weiter eingrenzen. Auf welchen Betriebssystemen (Linux) und CPU Architekturen wurde das Problem bisher beobachtet? 

Unter MacOS/Arm64 kann ich es bisher nicht reproduzieren. Gibts jemanden der das Problem auf Mac+Arm sieht?

WordsOfMe, 2024-08-19:
Bei @Arturfrain (https://github.com/evcc-io/evcc/issues/15467#issuecomment-2297379767) tritt der Fehler mit HA OS 13.0 auf, bei mir mit HA OS 12.4 (in Proxmox VM) klappt alles. Ich kann auf HA OS 13.0 momentan auch gar nicht updaten, wurde wegen eines Kernel-Fehlers inzwischen zurückgezogen - siehe hier: https://community.home-assistant.io/t/update-to-haos-13-0-completely-broke-all-network-integrations/759709/16

grunaur, 2024-08-19:
Bei mir folgendes Betriebssystem:

Prozessor	Intel(R) Core(TM) i3-8109U CPU @ 3.00GHz   3.00 GHz
Installierter RAM	8,00 GB (7,87 GB verwendbar)
Systemtyp	64-Bit-Betriebssystem, x64-basierter Prozessor
Edition	Windows 10 Pro
Version	21H2
Installiert am	‎24.‎01.‎2021
Betriebssystembuild	19044.3086

auf einem INTEL NUC 7

andig, 2024-08-19:
> Ich kann auf HA OS 13.0 momentan auch gar nicht updaten, wurde wegen eines Kernel-Fehlers inzwischen zurückgezogen

Das wäre leider zu einfach- denn Windows ist ja auch betroffen.

KG887, 2024-08-20:
Hallo, ich habe das gleiche Problem.
0.129 läuft stabil, nach update auf 0.130 startet evcc nicht mehr. Im Log taucht ebenfalls der im Verlauf beschriebene Fehler auf.

HA: 13.0
Umgebung: Proxmox VE
Intel Alder Lake N100 16GB RAM

physiko, 2024-08-20:
> Unter MacOS `arm64` kann ich es bisher nicht reproduzieren. Gibts jemanden der das Problem auf Mac+Arm sieht?

Bei mir tritt das Problem auch auf einem MacMini M1 auf.

HaareistNase, 2024-08-20:
Ich Reihe mich ein mit einem Intel Nuc, aber Linux (HA OS) direkt und nicht virtualisiert.

130.1 startet beim 1. Versuch.

andig, 2024-08-20:
> 130.1 startet beim 1. Versuch.

Was meinst du damit?

HaareistNase, 2024-08-20:
Gerade kam ein Update. 
Hab ich installiert.
Das ist ohne Absturz sofort durchgestartet.

HaareistNase, 2024-08-20:
Aber schon der 2. Versuch crashed wieder.

Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2024/08/20 09:36:24 evcc 0.130.1
[main  ] INFO 2024/08/20 09:36:24 using config file: /config/evcc.yaml
[mqtt  ] INFO 2024/08/20 09:36:24 connecting evcc at tcp://192.168.178.21:1883
[main  ] INFO 2024/08/20 09:36:24 listening at :7070
[lp-1  ] WARN 2024/08/20 09:36:25 poll mode '{connected 2m0s}' may deplete your battery or lead to API misuse. USE AT YOUR OWN RISK.
[site  ] WARN 2024/08/20 09:36:27 vehicle 'Kia' does not support automatic detection
[site  ] WARN 2024/08/20 09:36:27 vehicle 'E-Up' does not support automatic detection
[site  ] INFO 2024/08/20 09:36:27 site config:
[site  ] INFO 2024/08/20 09:36:27   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2024/08/20 09:36:27     grid:      power ✓ energy ✓ currents ✗
[site  ] INFO 2024/08/20 09:36:27     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2024/08/20 09:36:27     pv 2:      power ✓ energy ✓ currents ✗
[site  ] INFO 2024/08/20 09:36:27     battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✓
[site  ] INFO 2024/08/20 09:36:27   vehicles:
[site  ] INFO 2024/08/20 09:36:27     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[site  ] INFO 2024/08/20 09:36:27     vehicle 2: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2024/08/20 09:36:27 loadpoint 1:
[lp-1  ] INFO 2024/08/20 09:36:27   mode:        off
[lp-1  ] INFO 2024/08/20 09:36:27   charger:     power ✗ energy ✗ currents ✗ phases ✗ wakeup ✗
[lp-1  ] INFO 2024/08/20 09:36:27   meters:      charge ✗
[site  ] DEBUG 2024/08/20 09:36:27 set buffer soc: 90
[site  ] DEBUG 2024/08/20 09:36:27 set buffer start soc: 95
[site  ] DEBUG 2024/08/20 09:36:27 set priority soc: 60
[lp-1  ] DEBUG 2024/08/20 09:36:27 set smart cost limit: 0.0
[lp-1  ] DEBUG 2024/08/20 09:36:27 phase timer inactive
[lp-1  ] DEBUG 2024/08/20 09:36:27 pv timer inactive
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x20 pc=0x25c74e8]

goroutine 1 [running]:
github.com/evcc-io/evcc/core.(*Loadpoint).Prepare(0xc0011f3408, 0x0?, 0x5bddde0?, 0xc0017245d0?)
	github.com/evcc-io/evcc/core/loadpoint.go:651 +0xf28
github.com/evcc-io/evcc/core.(*Site).Prepare(0xc001499a00, 0xc000554230, 0xc000595030)
	github.com/evcc-io/evcc/core/site.go:968 +0x17b
github.com/evcc-io/evcc/cmd.runRoot(0x5b82360, {0x2e47132?, 0x7?, 0x2e3a0f6?})
	github.com/evcc-io/evcc/cmd/root.go:283 +0x1d92
github.com/spf13/cobra.(*Command).execute(0x5b82360, {0xc00011a1c0, 0x2, 0x2})
	github.com/spf13/cobra@v1.8.1/command.go:989 +0xa91
github.com/spf13/cobra.(*Command).ExecuteC(0x5b82360)
	github.com/spf13/cobra@v1.8.1/command.go:1117 +0x3ff
github.com/spf13/cobra.(*Command).Execute(...)
	github.com/spf13/cobra@v1.8.1/command.go:1041
github.com/evcc-io/evcc/cmd.Execute()
	github.com/evcc-io/evcc/cmd/root.go:107 +0x1a
main.main()
	github.com/evcc-io/evcc/main.go:44 +0x2d

HaareistNase, 2024-08-20:
10-15x gestartet … irgendwann gehts dann wieder.

Arturfrain, 2024-08-20:
> Gerade kam ein Update. Hab ich installiert. Das ist ohne Absturz sofort durchgestartet.

Bei mir startet auch 130.1 nicht.

grunaur, 2024-08-20:
Hallo andig, VPN ist nun eingerichtet.
Siehe Mail 
VG
Roland

andig, 2024-08-20:
@grunaur bitte noch die Config dazu schicken. Danke.

HaareistNase, 2024-08-20:
Hallo Andreas,
vielen Dank für Deine Mühe. Da Du schon VPN-Zugriff hast, habe ich evcc in meinem HA-Docker gestoppt und meinen alten Test-Pi für Dich zugänglich gemacht. Darauf ist das aktuelle evcc und meine config für Dich erreichbar im Home-Verzeichnis, damit Du live testen kannst. Auch auf dem alten Pi schmiert evcc beim start mit der Panic-Fehlermeldung ab. Es scheint also keinen Unterschied zu machen worauf evcc läuft für diesen Fehler.

Alex

HaareistNase, 2024-08-20:
Beobachtung:
Ich klemme das Auto vom Kabel ab und höre wie die Wallbox ihr Kabel-Ende entriegelt. Ich versuche evcc zu starten: Nichts. Ich klemme das Auto wieder an, höre wie die Wallbox ihr Ende verriegelt, und in dem Moment lässt sich evcc starten. 
Maybe Zufall. 
Ich probiere das später noch mal.

ross-w, 2024-08-20:
@andig working nightly was 9ad7399b, started failing from nightly d74fbd38

andig, 2024-08-20:
Does anyone have the issue on docker and could try the different commits in between?

dakky, 2024-08-20:
Yes have this issue on docker.  happend between 0.129.0 and 0.130.0. How can I switch the commit in the docker container without all the build cycle?

andig, 2024-08-20:
Fehler gefunden. Lösung kommt!

andig, 2024-08-20:
Fixed in master, Release läuft. Der Fehler war https://github.com/evcc-io/evcc/pull/15434. Die Idee war gut, fehlerfreie Anwendung aber fast unmöglich. Neuer Versuch zu einem späteren Zeitpunkt...

Danke an alle die sich am Debugging beteiligt haben!

closed 2024-08-20 by andig