evcc-io/evcc #15474: 0.130.0 cannot create charger 'Wall-E': cannot create charger type 'template': cannot create charger type 'ocpp': the mandatory SmartCharging profile is not supported

linuxstony, 2024-08-18:
### Describe the bug

Siehe oben. mit 0.129.0 hat evcc die ocpp Verbindung nach einigen Versuchen aufgebaut und die Box auch normal steuern können. 

Die Wallbox ist eine alte Weidmüller-Box mit Wallbe-Backend. OCPP model ist auf "Smart Charge Control" eingestellt. 

Die box habe ich über die GUI konfiguriert - daher habe ich nicht wirklich alle Punkte unten in der YAML drin, die ihr braucht :(

Die Box ist so wie ich verstanden habe nicht zu 100% OCPP konform, hat aber bisher mit evcc gut funktioniert. 

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
loadpoints:
- title: Wallbox
  charger: Wall-E
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  enable:
    threshold: -300
    delay: 3m
  disable:
    threshold: 2100
    delay: 3m
```

### Log details

```text
[db ] TRACE 2024/08/18 21:50:23 SELECT * FROM `configs` WHERE `configs`.`class` = 2 0 <nil>
[db ] TRACE 2024/08/18 21:50:23 SELECT * FROM `configs` WHERE `configs`.`class` = 1 0 <nil>
[garage-1] DEBUG 2024/08/18 21:50:23 waiting for chargepoint: 5m0s
[mqtt ] TRACE 2024/08/18 21:51:37 recv DISnXQ5W/Q5mkHULJ/state: '{"electricLevel":21,"sn":"EE1HYMCFM230063"}'
[ocpp ] DEBUG 2024/08/18 21:52:32 charge point connected: garage
[ocpp ] TRACE 2024/08/18 21:52:32 recv garage: [2,"45149","BootNotification",{"chargePointVendor":"wallbe","chargePointModel":"Smart Charge Control","chargePointSerialNumber":"00000000596","firmwareVersion":"BF-01.04.13"}]
[ocpp ] TRACE 2024/08/18 21:52:32 send garage: [3,"45149",{"currentTime":"2024-08-18T19:52:32Z","interval":60,"status":"Accepted"}]
[ocpp ] TRACE 2024/08/18 21:52:32 recv garage: [2,"45128","MeterValues",{"connectorId":1,"transactionId":1,"meterValue":[{"timestamp":"2024-08-18T19:31:47.000Z","sampledValue":[{"measurand":"Voltage","phase":"L1","unit":"V","value":"234"},{"measurand":"Voltage","phase":"L2","unit":"V","value":"235"},{"measurand":"Voltage","phase":"L3","unit":"V","value":"236"},{"measurand":"Current.Import","phase":"L1","unit":"A","value":"7"},{"measurand":"Current.Import","phase":"L2","unit":"A","value":"0"},{"measurand":"Current.Import","phase":"L3","unit":"A","value":"0"},{"measurand":"Energy.Active.Import.Register","unit":"Wh","value":"740"},{"measurand":"Power.Active.Import","unit":"W","value":"1436"}]}]}]
[ocpp ] TRACE 2024/08/18 21:52:32 send garage: [3,"45128",{}]
[ocpp ] TRACE 2024/08/18 21:52:33 recv garage: [2,"45129","MeterValues",{"connectorId":1,"transactionId":1,"meterValue":[{"timestamp":"2024-08-18T19:32:47.000Z","sampledValue":[{"measurand":"Voltage","phase":"L1","unit":"V","value":"234"},{"measurand":"Voltage","phase":"L2","unit":"V","value":"236"},{"measurand":"Voltage","phase":"L3","unit":"V","value":"236"},{"measurand":"Current.Import","phase":"L1","unit":"A","value":"7"},{"measurand":"Current.Import","phase":"L2","unit":"A","value":"0"},{"measurand":"Current.Import","phase":"L3","unit":"A","value":"0"},{"measurand":"Energy.Active.Import.Register","unit":"Wh","value":"764"},{"measurand":"Power.Active.Import","unit":"W","value":"1436"}]}]}]
[ocpp ] TRACE 2024/08/18 21:52:33 send garage: [3,"45129",{}]
[ocpp ] TRACE 2024/08/18 21:52:33 recv garage: [2,"45130","MeterValues",{"connectorId":1,"transactionId":1,"meterValue":[{"timestamp":"2024-08-18T19:33:47.000Z","sampledValue":[{"measurand":"Voltage","phase":"L1","unit":"V","value":"234"},{"measurand":"Voltage","phase":"L2","unit":"V","value":"236"},{"measurand":"Voltage","phase":"L3","unit":"V","value":"236"},{"measurand":"Current.Import","phase":"L1","unit":"A","value":"7"},{"measurand":"Current.Import","phase":"L2","unit":"A","value":"0"},{"measurand":"Current.Import","phase":"L3","unit":"A","value":"0"},{"measurand":"Energy.Active.Import.Register","unit":"Wh","value":"788"},{"measurand":"Power.Active.Import","unit":"W","value":"1435"}]}]}]
[ocpp ] TRACE 2024/08/18 21:52:33 send garage: [3,"45130",{}]
[ocpp ] TRACE 2024/08/18 21:52:33 send garage: [2,"1771892434","GetConfiguration",{}]
[ocpp ] TRACE 2024/08/18 21:52:33 recv garage: [2,"45131","MeterValues",{"connectorId":1,"transactionId":1,"meterValue":[{"timestamp":"2024-08-18T19:34:47.000Z","sampledValue":[{"measurand":"Voltage","phase":"L1","unit":"V","value":"234"},{"measurand":"Voltage","phase":"L2","unit":"V","value":"236"},{"measurand":"Voltage","phase":"L3","unit":"V","value":"236"},{"measurand":"Current.Import","phase":"L1","unit":"A","value":"7"},{"measurand":"Current.Import","phase":"L2","unit":"A","value":"0"},{"measurand":"Current.Import","phase":"L3","unit":"A","value":"0"},{"measurand":"Energy.Active.Import.Register","unit":"Wh","value":"812"},{"measurand":"Power.Active.Import","unit":"W","value":"1436"}]}]}]
[ocpp ] TRACE 2024/08/18 21:52:33 send garage: [3,"45131",{}]
[ocpp ] TRACE 2024/08/18 21:52:33 recv garage: [3,"1771892434",{"configurationKey":[{"key":"AuthorizeRemoteTxRequests","readonly":true,"value":"true"},{"key":"ClockAlignedDataInterval","readonly":true,"value":"0"},{"key":"ConnectionTimeOut","readonly":false,"value":"60"},{"key":"GetConfigurationMaxKeys","readonly":true,"value":"0"},{"key":"HeartbeatInterval","readonly":false,"value":"60"},{"key":"LocalAuthorizeOffline","readonly":true,"value":"false"},{"key":"LocalPreAuthorize","readonly":true,"value":"false"},{"key":"MeterValuesAlignedData","readonly":true,"value":"false"},{"key":"MeterValuesSampledData","readonly":true,"value":"true"},{"key":"StopTransactionOnEVSideDisconnect","readonly":true,"value":"true"},{"key":"StopTransactionOnInvalidId","readonly":true,"value":"true"},{"key":"StopTxnAlignedData","readonly":true,"value":"false"},{"key":"StopTxnSampledData","readonly":true,"value":"false"},{"key":"MeterValueSampleInterval","readonly":false,"value":"60"},{"key":"NumberOfConnectors","readonly":true,"value":"1"},{"key":"ConnectorPhaseRotation","readonly":true,"value":"false"},{"key":"SupportedFeatureProfiles","readonly":true,"value":"true"},{"key":"ResetRetries","readonly":false,"value":"3"},{"key":"TransactionMessageAttempts","readonly":false,"value":"3"},{"key":"TransactionMessageRetryInterval","readonly":false,"value":"60"},{"key":"UnlockConnectorOnEVSideDisconnect","readonly":true,"value":"true"},{"key":"ChargeProfileMaxStackLevel","readonly":true,"value":"1"},{"key":"ChargingScheduleAllowedChargingRateUnit","readonly":true,"value":"true"},{"key":"ChargingScheduleMaxPeriods","readonly":true,"value":"1"},{"key":"MaxChargingProfilesInstalled","readonly":true,"value":"1"},{"key":"PricePerKwh","readonly":false,"value":"0"},{"key":"AllowOfflineTxForFreemode","readonly":true,"value":"false"},{"key":"WebSocketPingInterval","readonly":false,"value":"60"},{"key":"UID_HighRegisterFirst","readonly":false,"value":"false"},{"key":"ConnectorMaximumCurrent","readonly":false,"value":"255"},{"key":"ConnectorMinimumCurrent","readonly":false,"value":"1"},{"key":"GlobalMaximumCurrent","readonly":false,"value":"16"},{"key":"MeterValueSendTotalEnergy","readonly":false,"value":"false"}],"unknownKey":[]}]
[main ] FATAL 2024/08/18 21:52:33 cannot create charger 'Wall-E': cannot create charger type 'template': cannot create charger type 'ocpp': the mandatory SmartCharging profile is not supported
[main ] FATAL 2024/08/18 21:52:33 will attempt restart in: 15m0s
[main ] TRACE 2024/08/18 21:52:33 GET https://api.github.com/repos/evcc-io/evcc/releases/latest
[ocpp ] DEBUG 2024/08/18 21:52:33 charge point disconnected: garage
```

### What type of operating system are you running?

Linux

### Version

0.130.0

JackJackBauer, 2024-08-18:
Gleiches Verhalten hier bei meiner Wallbox PulsarPlus. (Ebenfalls per OCPP eingebunden). 

Zusätzlich wird der ID4 nicht mehr erkannt. 
Musste kurzfristig ein Downgrade auf 0.129.0 durchführen, wonach alles wieder einwandfrei funktioniert. 
Die Logfiles und konkrete Fehler muss ich noch schuldig bleiben, da wir unterwegs sind…

premultiply, 2024-08-18:
Zurecht, denn

> `{"key":"SupportedFeatureProfiles","readonly":true,"value":"true"},`

`true` ist einfach keine standardkonforme Aufzählung der unterstützten Feature Profiles.

Fehlerbereinigte Firmware beim Hersteller anfordern. Alte Wallbe-Boxen haben doch bestimmt einen Phoenix Controller drin.
Notfalls einfach die aktuelle Phoenix-Originalfirmware einspielen.

premultiply, 2024-08-18:
Weidmüller, Wall-E, Wallbe - was denn nun?

premultiply, 2024-08-18:
Wenn es wie per OCPP BootNotification ausgegeben eine Wallbe-Box ist, dann ist da vermutlich dieser Phoenix-Controller drin:
https://www.phoenixcontact.com/de-de/produkte/ac-ladesteuerung-ev-cc-ac1-m3-cbc-rcm-eth-3g-1018702#downloads-link-target
Hier gibt es die Firmware dazu.

linuxstony, 2024-08-19:
Moin @premultiply, 

Über dem Backend prangert in riesigen Buchstaben Wallbe - Hersteller ist Weidmüller.

ich hatte vor Jahren versucht über Weidmüller an eine neue Firmware zu kommen, da ich schon bei der Einbindung in Homeassistant gemerkt habe, dass die Box nicht 100% OCPP konform ist. 
(den Zugang zum Backend hab ich nur über einen Forenbeitrag wo das geleakt wurde bekommen).

Lapidare Antwort: "Eine neue Firmware gibt es nicht, wenden Sie sich an ihren Elektroinstallateur." /zitatende

Du hast mir mit dem Link schon mehr geholfen, als der Weidmüller-Support (nebenbei - nie wieder Weidmüller)

Ich werde die Box nachher mal aufschrauben und schauen, welcher Controller verbaut ist, damit ich mir die nicht zerschieße und dann wage ich das Update.

DANKE euch allen !

jgantenberg, 2024-08-19:
Ich bekomme leider mit evcc 0.130.0 eine ähnliche Fehermeldung: cannot create charger type 'ocpp': the mandatory smartcharging profile is not supported
`[ocpp  ] TRACE 2024/08/19 08:19:06 recv xxxxxxxxxx: [3, "3859682090", {"configurationKey": [{"key": "AuthorizeRemoteTxRequests", "readonly": false, "value": "0"}, {"key": "ClockAlignedDataInterval", "readonly": false, "value": "900"}, {"key": "ConnectionTimeOut", "readonly": false, "value": "120"}, {"key": "ConnectorPhaseRotation", "readonly": false, "value": "RST"}, {"key": "ChargingScheduleAllowedChargingRateUnit", "readonly": true, "value": "Current, Power"}, {"key": "ChargingScheduleMaxPeriods", "readonly": true, "value": "5"}, {"key": "GetConfigurationMaxKeys", "readonly": true, "value": "30"}, {"key": "HeartbeatInterval", "readonly": false, "value": "60"}, {"key": "LocalAuthorizeOffline", "readonly": false, "value": "1"}, {"key": "LocalPreAuthorize", "readonly": false, "value": "1"}, {"key": "MeterValuesAlignedData", "readonly": false, "value": "Current.Import,Energy.Active.Import.Interval,Power.Active.Import,Temperature,Voltage"}, {"key": "MeterValuesSampledData", "readonly": false, "value": "Power.Offered,Current.Offered,Current.Import,Energy.Active.Import.Register,Energy.Active.Import.Interval,Power.Active.Import,Temperature,Temperature,Temperature,Voltage"}, {"key": "MeterValueSampleInterval", "readonly": false, "value": "30"}, {"key": "NumberOfConnectors", "readonly": true, "value": "1"}, {"key": "ResetRetries", "readonly": false, "value": "0"}, {"key": "StopTransactionOnInvalidId", "readonly": false, "value": "1"}, {"key": "StopTxnAlignedData", "readonly": false, "value": "Current.Import,Energy.Active.Import.Interval,Power.Active.Import,Temperature,Voltage"}, {"key": "StopTxnSampledData", "readonly": false, "value": "Current.Import,Energy.Active.Import.Register,Power.Active.Import,Temperature,Voltage"}, {"key": "SupportedFeatureProfiles", "readonly": true, "value": ""}, {"key": "TransactionMessageAttempts", "readonly": false, "value": "3"}, {"key": "TransactionMessageRetryInterval", "readonly": false, "value": "60"}, {"key": "UnlockConnectorOnEVSideDisconnect", "readonly": false, "value": "1"}, {"key": "LocalAuthListEnabled", "readonly": false, "value": "0"}, {"key": "LocalAuthListMaxLength", "readonly": true, "value": "60"}, {"key": "SendLocalListMaxLength", "readonly": true, "value": "8"}, {"key": "ChargeProfileMaxStackLevel", "readonly": true, "value": "8"}, {"key": "MaxChargingProfilesInstalled", "readonly": true, "value": "16"}, {"key": "VendorMaxWorkCurrent", "readonly": false, "value": "16.00"}, {"key": "VendorDirectWorkMode", "readonly": false, "value": "1"}, {"key": "GunInsertUpdateStatus", "readonly": false, "value": "1"}]}]
`
Mit Version 0.129.0 funktioniert die Kommunikation dem Anschein nach korrekt und stabil, das Laden funktioniert einwandfrei.
Wallbox: Entratek Power Dot Pro (alte Version)
Da die Wallbox inzwischen ca. 3 Jahre alt ist, erwarte ich keinen Support vom Hersteller. Ich werde dennoch eine Anfrage stellen.

andig, 2024-08-19:
Es ist wirklich unglaublich, wieviel Sch***** sich die Hersteller da zusammenfrickeln. Eure Boxen verhalten sich beide nicht Standard-konform.

@premultiply mein Vorschlag wäre, den Check in eine Warnung umzubauen. Im Zweifel für den Angeklagten.

closed 2024-08-19 by andig