evcc-io/evcc #15476: Meters wrong by a factor of 100 (solax)

WordsOfMe, 2024-08-18:
### Describe the bug

The grid and PV energy meters shown on the configuration page (when experimental features are turned on) both show 100 times the correct value.  E.g., here for total solar energy the correct value would be 20110 kWh (20 MWh), but I get a value of 2011040 kWh (2000 MWh). I have a SolaX inverter which is queried via modbus.

My guess is this is due to line 38 and and 62 of evcc/tree/master/templates/definition/meter/solax.yaml where
`    scale: 10`
should be
`    scale: 0.1`
because from the manufacturer's description the queried register 0x0094 reports units of 0.1 kWh:
[image]

### Steps to reproduce

1. Use solax meter template
2. Turn experimental features on
3. Look for energy values (kWh) on configuration page
4. Compare with correct value e.g. displayed in HomeAssistant or in SolaX cloud

### Configuration details

```yaml
meters:
  - name: grid
    type: template
    template: solax
    usage: grid
    modbus: tcpip
    id: 1
    host: 127.0.0.1 # Hostname
    port: 502 # Port
  - name: pv
    type: template
    template: solax
    usage: pv
    modbus: tcpip
    id: 1
    host: 127.0.0.1 # Hostname
    port: 502 # Port
```

### Log details

```text
(skipped)
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.129.0

andig, 2024-08-19:
Danke- magst Du gleich einen PR machen?

WordsOfMe, 2024-08-19:
...wäre mein erster, werd's versuchen :)

closed 2024-08-19 by andig