evcc-io/evcc #15480: Add support for Acona Stromat

hd42, 2024-08-19:
I have a "Stromat" wallbox from Acona (see [Acona website](https://www.alcona.info/stromat.html)). They offer a custom hardware solution (["Solarkopplung"](https://www.alcona.info/stromat.html#Solarkopplung)) to connect it to a PV system, but I'd rather use an open source solution like evcc. The wallbox' web interface is very basic and the Solarkupplung suggests that it should also offer an API with all the required features to manage the charging process.
If anyone would be interested in writing the integration for this kind of wallbox, I'd be glad to assist in any way I can.

andig, 2024-08-19:
Is there any documentation? Does the box support OCPP?

hd42, 2024-08-19:
I haven't found any documentation on the official website, OCPP is not mentioned in the marketing PDFs. I'll write the manufacturer and ask about it.

andig, 2024-08-20:
Closing until docs available.

closed 2024-08-20 by andig