evcc-io/evcc #15484: Password not working after 0.130 update

passuff, 2024-08-19:
### Describe the bug

I never had a password configured or at least it was not necessary to use it when changing the config. After updating to 0.130 I was forced to enter it. I then tried to reset it but it didn't force me to set a new pw after rebooting. I then set a new pw in the prompt and rebooted. Again, the new pw is not working when entering the config. It seems like, at least for me, the password functionality to enter the config, is broken.

### Steps to reproduce

1. update to v0.130
2. enter config
3. enter pw
4. reset pw
...

### Configuration details

```yaml
not necessary
```

### Log details

```text
can't open logs as pw does not work
```

### What type of operating system are you running?

Linux

### Version

0.130

naltatis, 2024-08-19:
Directly via apt on Linux or in Docker? This sounds like running it in docker but without proper mounting of the `/root/.evcc` folder.

passuff, 2024-08-19:
No docker - direct Linux install...

naltatis, 2024-08-19:
Are your charging sessions from before the update still there?

passuff, 2024-08-19:
Yes. Everything is there. No changes discovered.

naltatis, 2024-08-19:
Strange, can you (or anyone else) reproduce this? Don't see a reason why you should be prompted again for a new password other than having lost the original database.

passuff, 2024-08-19:
> @passuff <https://github.com/passuff> Did you try another browser?

I tried also from another device. Same behaviour. I still can't access log
or config...

naltatis, 2024-08-19:
This command should return a hash of you're current admin password and the config and db location that are used

```
evcc settings get adminPassword
```

The value should change if you run `evcc password set` and set another password.

Can you verify that the db location is the same as in your evcc service?
You can see it in the first log lines
https://docs.evcc.io/docs/installation/linux#testen

passuff, 2024-08-19:
> This command should return a hash of you're current admin password and the config and db location that are used
>
> ```
> evcc settings get adminPassword
> ```
>
> The value should change if you run `evcc [...]

`passuff@influxdb2024:~$ evcc settings get adminPassword
[main  ] INFO 2024/08/19 18:23:44 evcc 0.130.0
[main  ] INFO 2024/08/19 18:23:44 using config file: /etc/evcc.yaml
[db    ] INFO 2024/08/19 18:23:44 using sqlite database: /home/passuff/.evcc/evcc.db
adminPassword:
`
I dont receive any hash....

I can't access logs as it asks for pw...

naltatis, 2024-08-19:
> I can't access logs as it asks for pw...

but you can access logs via cli (see link above)

passuff, 2024-08-21:
> This command should return a hash of you're current admin password and the config and db location that are used
>
> ```
> evcc settings get adminPassword
> ```
>
> The value should change if you run `evcc [...]

sudo systemctl status evcc
● evcc.service - evcc
     Loaded: loaded (/lib/systemd/system/evcc.service; enabled; preset: enabled)
     Active: active (running) since Wed 2024-08-21 16:28:29 CEST; 1min 24s ago
   Main PID: 3487 (evcc)
      Tasks: 7 (limit: 2307)
     Memory: 41.8M
        CPU: 2.275s
     CGroup: /system.slice/evcc.service
             └─3487 /usr/bin/evcc

----------------------------

passuff@influxdb2024:~$ evcc settings get adminPassword
[main  ] INFO 2024/08/21 16:30:35 evcc 0.130.2
[main  ] INFO 2024/08/21 16:30:35 using config file: /etc/evcc.yaml
[db    ] INFO 2024/08/21 16:30:35 using sqlite database: /home/passuff/.evcc/evcc.db

Sorry for the delay. I hope this answers your question.

passuff, 2024-08-21:
> Reference

Working now - thanks a lot. 
Why did I suddenly witn 0.130 need a password at all (was not necessary before)?
Why couldn't I reset it the common way?

naltatis, 2024-08-22:
Nice!
We introduced the admin password feature a couple of releases ago. It's a prerequisite for moving to UI based configuration.

closed 2024-08-22 by naltatis