evcc-io/evcc #15487: v0.130.0 keine Verbindung zu Cupra Born

digitalonkel, 2024-08-19:
### Describe the bug

Fehlermeldung:
cannot create vehicle type 'template': cannot create vehicle type 'cupra': missing id_token

### Steps to reproduce

1. create Cupra Born
2.
3.
...

### Configuration details

```yaml
Über UI configuriert
```

### Log details

```text
cannot create vehicle type 'template': cannot create vehicle type 'cupra': missing id_token
```

### What type of operating system are you running?

Linux

### Version

0.130.0

naltatis, 2024-08-19:
> Über UI configuriert

Konfiguriere bitte noch einmal. Config UI ist noch experimental und mit diesem Release hat sich die Datenstruktur teilweise geändert.

closed 2024-08-19 by naltatis

digitalonkel, 2024-08-19:
Fluppt wieder.