evcc-io/evcc #15490: Wallbox.com/Pulsar: charger logic error: current mismatch (got 15.5A measured, expected 6A)

mrhund, 2024-08-19:
### Describe the bug

Bereits heiß diskutiert in #14040, mit 0.130.0 sind nun endlich im Log brauchbare Dinge zu sehen. EVCC erkennt nun nämlich den mißmatch zwischen geforderter Charging Power und tatsächlicher Charging Power: "current mismatch (got 15.5A measured, expected 6A)".

### Steps to reproduce

1. Use Wallbox/Pulsar
2. Start Charging with PV or PV+MIN
3. Wait a few minutes, charging power will increase to 11KW even if requestes charing power is less
...

### Configuration details

```yaml
loadpoints:
- title: Garage
  charger: wallbox6
  mode: pv
  enable:
    threshold: -3699
    delay: 2m
  disable:
    threshold: 500 # maximum import power (W)
    delay: 10m

chargers:
- type: template
  template: pulsarplus 
  timeout: 2m  
  name: wallbox6
  remotestart: true //notwendig seit 0.130.0, da ansonsten "waiting for local authentication kommt"
```

### Log details

```text
[ocpp-1] DEBUG 2024/08/19 11:14:15 waiting for chargepoint: 5m0s
[ocpp  ] INFO 2024/08/19 11:14:43 charge point connected, registering: g001
[ocpp  ] TRACE 2024/08/19 11:14:43 recv g001: [2,"1019295848","StatusNotification",{"info": "","vendorId": "com.wallbox","vendorErrorCode": "","connectorId": 0,"errorCode": "NoError","status": "Available","timestamp": "2024-08-19T09:14:43Z"}]
[ocpp  ] TRACE 2024/08/19 11:14:43 send g001: [3,"1019295848",{}]
[ocpp  ] TRACE 2024/08/19 11:14:43 recv g001: [2,"469654653","StatusNotification",{"info": "","vendorId": "com.wallbox","vendorErrorCode": "","connectorId": 1,"errorCode": "NoError","status": "SuspendedEVSE","timestamp": "2024-08-19T09:14:43Z"}]
[ocpp  ] TRACE 2024/08/19 11:14:43 send g001: [3,"469654653",{}]
[ocpp  ] TRACE 2024/08/19 11:14:43 recv g001: [2,"1347599670","MeterValues",{"meterValue": [{"timestamp": "2024-08-19T09:14:43Z","sampledValue": [{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Energy.Active.Import.Register","unit": "Wh","value": "532218.4"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Power.Active.Import","unit": "W","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Current.Offered","unit": "A","value": "0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Power.Offered","unit": "W","value": "0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L1","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L2","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L3","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L1-N","measurand": "Voltage","unit": "V","value": "233.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L2-N","measurand": "Voltage","unit": "V","value": "244.0"} ]} ],"connectorId": 1,"transactionId": 2}]
[ocpp-1] DEBUG 2024/08/19 11:14:43 hijacking transaction: 2
[ocpp  ] TRACE 2024/08/19 11:14:43 send g001: [3,"1347599670",{}]
[ocpp  ] TRACE 2024/08/19 11:14:43 recv g001: [2,"2027165033","MeterValues",{"meterValue": [{"timestamp": "2024-08-19T09:14:12Z","sampledValue": [{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Energy.Active.Import.Register","unit": "Wh","value": "532218.4"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Power.Active.Import","unit": "W","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Current.Offered","unit": "A","value": "0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Power.Offered","unit": "W","value": "0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L1","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L2","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L3","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L1-N","measurand": "Voltage","unit": "V","value": "233.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L2-N","measurand": "Voltage","unit": "V","value": "245.0"} ]} ],"connectorId": 1,"transactionId": 2}]
[ocpp  ] TRACE 2024/08/19 11:14:43 send g001: [3,"2027165033",{}]
[ocpp  ] TRACE 2024/08/19 11:14:43 recv g001: [2,"1297993920","MeterValues",{"meterValue": [{"timestamp": "2024-08-19T09:14:22Z","sampledValue": [{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Energy.Active.Import.Register","unit": "Wh","value": "532218.4"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Power.Active.Import","unit": "W","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Current.Offered","unit": "A","value": "0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Power.Offered","unit": "W","value": "0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L1","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L2","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L3","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L1-N","measurand": "Voltage","unit": "V","value": "233.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L2-N","measurand": "Voltage","unit": "V","value": "245.0"} ]} ],"connectorId": 1,"transactionId": 2}]
[ocpp  ] TRACE 2024/08/19 11:14:43 send g001: [3,"1297993920",{}]
[ocpp  ] TRACE 2024/08/19 11:14:43 recv g001: [2,"779233100","MeterValues",{"meterValue": [{"timestamp": "2024-08-19T09:14:32Z","sampledValue": [{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Energy.Active.Import.Register","unit": "Wh","value": "532218.4"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Power.Active.Import","unit": "W","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Current.Offered","unit": "A","value": "0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Power.Offered","unit": "W","value": "0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L1","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L2","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L3","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L1-N","measurand": "Voltage","unit": "V","value": "233.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L2-N","measurand": "Voltage","unit": "V","value": "244.0"} ]} ],"connectorId": 1,"transactionId": 2}]
[ocpp  ] TRACE 2024/08/19 11:14:43 send g001: [3,"779233100",{}]
[ocpp  ] TRACE 2024/08/19 11:14:44 send g001: [2,"4253539538","GetConfiguration",{}]
[ocpp  ] TRACE 2024/08/19 11:14:44 recv g001: [3,"4253539538",{"unknownKey": [],"configurationKey": [{"readonly": false,"key": "AllowOfflineTxForUnknownId","value": "true"},{"readonly": false,"key": "AuthorizationCacheEnabled","value": "true"},{"readonly": false,"key": "AuthorizeRemoteTxRequests","value": "false"},{"readonly": false,"key": "BlinkRepeat","value": "10"},{"readonly": false,"key": "ClockAlignedDataInterval","value": "0"},{"readonly": false,"key": "ConnectionTimeOut","value": "120"},{"readonly": false,"key": "ConnectorPhaseRotation","value": "1.Unknown"},{"readonly": true,"key": "ConnectorPhaseRotationMaxLength","value": "1"},{"readonly": true,"key": "GetConfigurationMaxKeys","value": "50"},{"readonly": false,"key": "HeartbeatInterval","value": "60"},{"readonly": false,"key": "LightIntensity","value": "100"},{"re [... cut ...]
[lp-1  ] DEBUG 2024/08/19 11:18:59 charge total import: 532.671kWh
[lp-1  ] DEBUG 2024/08/19 11:18:59 charger status: C
[lp-1  ] DEBUG 2024/08/19 11:18:59 soc estimated: 38.55% (vehicle: 38.00%)
[lp-1  ] DEBUG 2024/08/19 11:18:59 vehicle soc: 39%
[lp-1  ] DEBUG 2024/08/19 11:18:59 vehicle soc limit: 85%
[lp-1  ] DEBUG 2024/08/19 11:18:59 vehicle range: 157km
[lp-1  ] WARN 2024/08/19 11:18:59 charger logic error: current mismatch (got 15.4A measured, expected 6A)
```

### What type of operating system are you running?

Docker container

### Version

0.130.0

premultiply, 2024-08-19:
Die Box missachtet einfach die von ihr sogar bestätigten Anweisungen.
Ich kann keinen Fehler seitens evcc erkennen.

Davon abgesehen: `remoteStart` (Box lokal richtig konfigurieren!) und `timeout` sind überflüssig.

mrhund, 2024-08-19:
Thx, dann ist das dokumentiert und ich werde wieder einen Case beim Hersteller aufmachen.

mrhund, 2024-08-19:
Geschlossen, da kein Fehler im EVCC. Ich schreibe ggf. Updates, falls sich seiten Hersteller was ergibt.

premultiply, 2024-08-21:
Ja, scheint so zu sein.
Kein Bedienkonzept ist absurd genug, dass es nicht doch irgendjemand implementiert hat. :)

Trotzdem solltest du damit nochmal beim Hersteller vorstellig werden.
Sollen sie halt den fehlenden Schalter mal irgendwo in der Konfiguration nachrüsten.
Andere Hersteller schaffen das ja auch...

mrhund, 2024-08-21:
Ich bin eh schon in Kontakt wegen den ignorierten OCPP Befehlen, aber auch der Support ist sch****. Scheint sich durchzuziehen :D

flepp4ever, 2024-09-10:
Hallo @mrhund !
Ich habe mich auch beim Wallbox Support gemeldet und bin mal gespannt was als Antwort kommt.
Was ich nicht verstehe ist, warum bis v0.129 diese Probleme bei der Pulsar Plus nicht auftraten?

flepp4ever, 2024-09-10:
Tatsächlich hat mir der Support gerade geantwortet und das angeraten, was ich eine Zeile drüber angemerkt habe. Nichtsdestotrotz haben sie einen Update der FW versprochen.

[image]

mrhund, 2024-09-15:
> Hallo @mrhund ! Ich habe mich auch beim Wallbox Support gemeldet und bin mal gespannt was als Antwort kommt. Was ich nicht verstehe ist, warum bis v0.129 diese Probleme bei der Pulsar Plus nicht auftraten?

Bei mir war das vorher schon so. Da gibt es gravierende Unterschiede zwischen der 5.X Firmare und der 6.X Firmware der Box. Die Probleme mit der 5.X Firmware sind wohl nun behoben durch Anpassungen von EVCC, die Probleme mit 6.X jedoch bestehen noch.

mrhund, 2024-09-15:
Ich hab jetzt durch das schlechte Wetter der letzten Tage herausgefunden, woher das Verhalten imho ziemlich sicher kommt. Nämlich wenn mehrmals hintereinander SetChargingProfile mit der selben Amperzahl geschickt wird. Wenn man im Log ganz oben schaut, dann sieht man, dass immer wieder 6A gesetzt werden, obwohl diese schon gesetzt sind. Die Box sagt zwar "Accepted" springt aber auf 16A.

`[ocpp  ] TRACE 2024/08/19 11:16:15 send g001: [2,"1909664084","SetChargingProfile",{"connectorId":1,"csChargingProfiles":{"chargingProfileId":10,"stackLevel":3,"chargingProfilePurpose":"TxDefaultProfile","chargingProfileKind":"Absolute","chargingSchedule":{"startSchedule":"2024-08-19T09:16:15Z","chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"**limit":6**}]}}}]
[ocpp  ] TRACE 2024/08/19 11:16:16 recv g001: [3,"1909664084",{"status": "Accepted"}]
[lp-1  ] DEBUG 2024/08/19 11:16:16 max charge current: 6A
[ocpp  ] TRACE 2024/08/19 11:16:16 send g001: [2,"636018276","SetChargingProfile",{"connectorId":1,"csChargingProfiles":{"chargingProfileId":10,"stackLevel":3,"chargingProfilePurpose":"TxDefaultProfile","chargingProfileKind":"Absolute","chargingSchedule":{"startSchedule":"2024-08-19T09:16:16Z","chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"**limit":6**}]}}}]
[ocpp  ] TRACE 2024/08/19 11:16:16 recv g001: [2,"1365555378","StatusNotification",{"info": "","vendorId": "com.wallbox","vendorErrorCode": "","connectorId": 1,"errorCode": "NoError","status": "SuspendedEV","timestamp": "2024-08-19T09:16:16Z"}]
[ocpp  ] TRACE 2024/08/19 11:16:16 send g001: [3,"1365555378",{}]
[ocpp  ] TRACE 2024/08/19 11:16:17 recv g001: [3,"636018276",{"status": "Accepted"}]`

Jetzt, wo das Wetter schlecht ist und nichts in der PV kommt und man mit MIN+PV lädt, werden die 6A nur einmal zu Beginn gesetzt und dann nicht mehr und jetzt bleibt die Box auf 6A und spring nicht mehr auf 16A. 

@premultiply meinte in einem früheren Kommentar mal, dass man mehrmals den gleichen Befehl schicken darf laut Protokoll, ok verstehe ich. Die Frage ist, warum sie geschickt werden und vielleicht könnte man das Problem lösen, wenn man den selben Befehl nur einmal schickt...?

andig, 2024-09-15:
Unleserlich :(

andig, 2024-09-15:
> Wenn man im Log ganz oben schaut, dann sieht man, dass immer wieder 6A gesetzt werden, obwohl diese schon gesetzt sind. Die Box sagt zwar "Accepted" springt aber auf 16A.

Das sollte ja dann der Support beheben können wenn der Fehler so klar nachvollziehbar ist!

mrhund, 2024-09-15:
> > Wenn man im Log ganz oben schaut, dann sieht man, dass immer wieder 6A gesetzt werden, obwohl diese schon gesetzt sind. Die Box sagt zwar "Accepted" springt aber auf 16A.
> 
> Das sollte ja dann der Support beheben können wenn der Fehler so klar nachvollziehbar ist!

Glaubt man, unmöglich der Support. Zum 100sten Mall soll ich ein Video der App machen, wo man sieht dass sich die Amper ändern. Ich schick ihnen Logfiles, sie wollen wieder ein Video. Ich erklr ihnen das selbe wie euch, sie wollen ein Video.

mrhund, 2024-09-15:
> Unleserlich :(

Sorry, sollte nun besser sein.

andig, 2024-09-15:
Der Schnipsel ist nutzlos. Minischipsel sind fast immer nutzlos mangels Kontext. Interessant wäre ja was die Box dann in `Current.Offered` meldet.

mrhund, 2024-09-16:
> Der Schnipsel ist nutzlos. Minischipsel sind fast immer nutzlos mangels Kontext. Interessant wäre ja was die Box dann in `Current.Offered` meldet.

Hab den ersten Post nochmal runter kopiert für die bessere Übersicht - da sieht man, dass Current.Offered beim Starten des Ladens direkt von 0A auf 16A springt.

andig, 2024-09-16:
Wenn die Meldung auftritt

    charger logic error: current mismatch 

und die WB auf `Current.Offered` den physisch richtigen Wert zeigt- was sie ja hier tut- dann wird evcc den wieder korrigieren. Das erzeugt zwar eine Fehlermeldung, aber die ist ja richtig und wird auch behoben.

In dem Sinne sollte das Problem gelöst sein. Bzw. mir ist nicht klar, was das verbleibende Problem ist?

premultiply, 2024-09-16:
Bräuchten wir dazu nicht `CurrentGetter`?
Hier wird ja nur festgestellt, dass die Messwerte erheblich vom Soll abweichen.

andig, 2024-09-16:
@premultiply genau- aber den hast Du ja schon:

```go
	if c.cp.HasMeasurement(types.MeasurandCurrentImport) {
		currentsG = c.conn.Currents
	}
```

andig, 2024-09-16:
@mrhund zur Erläuterung: Deine WB ändert selbstständig den Vorgabestrom (schlecht), gibt darüber aber auch Auskunft (gut). Das könnten wir zum Nachsteuern benutzen. Der eigentliche Bug ist aber in der FW und sollte dort behoben werden.

@premultiply ich erinnere mich: eine andere WB hatte Vorgabe=0 akzeptiert, aber weiterhin Vorgabe=6A angezeigt. Das geht natürlich nicht und müssten wir dann special-casen?

andig, 2024-09-16:
@premultiply PR erstellt. Das zweite Problem würde ich hier ignorieren- das können wir dann separat über Überschreiben der `metervalues` lösen.

premultiply, 2024-09-16:
Ich meine das war die Vestel-Box (oder die Evbox Elvi?) wo wir dann im Template manuell die meterValues ohne Current.Offered gesetzt haben.

Ist also dort entschärft.

Kann natürlich immer wieder irgendwo vorkommen...

andig, 2024-09-16:
Dann mal rein damit. @mrhund bitte morgen im Nightly probieren.

closed 2024-09-16 by andig

mrhund, 2024-09-17:
Ich weiß, langer Post - aber bitte kurz Zeit nehmen :)

Ich habe mit dem nightly getestet, springt noch immer auf 16A. Aber ich habe die Bestätigung, dass der Fehler immer dann auftritt und Befehle ignoriert werden, wenn man mehrmals hintereinander die gleiche Amperzahl mit ```SetChargingProfile``` sendet, obwohl sie schon gesetzt ist.

Laden wird gestartet:
```text
[ocpp  ] TRACE 2024/09/17 12:19:41 sent JSON message to g001: [2,"4113053374","SetChargingProfile",{"connectorId":1,"csChargingProfiles":{"chargingProfileId":10,"stackLevel":3,"chargingProfilePurpose":"TxDefaultProfile","chargingProfileKind":"Absolute","chargingSchedule":{"startSchedule":"2024-09-17T10:19:41Z","chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":6}]}}}]
```
und gleich danach:
```text
[ocpp  ] TRACE 2024/09/17 12:19:42 sent JSON message to g001: [2,"604768236","SetChargingProfile",{"connectorId":1,"csChargingProfiles":{"chargingProfileId":10,"stackLevel":3,"chargingProfilePurpose":"TxDefaultProfile","chargingProfileKind":"Absolute","chargingSchedule":{"startSchedule":"2024-09-17T10:19:42Z","chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":6}]}}}]
```
Ergebnis:
```text
[lp-1  ] DEBUG 2024/09/17 12:20:08 charge power: 11500W
[lp-1  ] DEBUG 2024/09/17 12:20:08 charge currents: [15.6 15.7 15.7]A
```
Danach der Versuch zu korrigieren:
```text
[lp-1  ] WARN 2024/09/17 12:20:45 charger logic error: current mismatch (got 15.6A measured, expected 6A)
[lp-1  ] DEBUG 2024/09/17 12:20:45 pv charge current: min 6A > 3.26A (8512W @ 3p, battery: false)
[ocpp  ] TRACE 2024/09/17 12:20:45 enqueued CALL [2111151039, SetChargingProfile] for g001
[ocpp  ] TRACE 2024/09/17 12:20:45 dispatched request 2111151039 for g001
[ocpp  ] TRACE 2024/09/17 12:20:45 sent JSON message to g001: [2,"2111151039","SetChargingProfile",{"connectorId":1,"csChargingProfiles":{"chargingProfileId":10,"stackLevel":3,"chargingProfilePurpose":"TxDefaultProfile","chargingProfileKind":"Absolute","chargingSchedule":{"startSchedule":"2024-09-17T10:20:45Z","chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":6}]}}}]
```
Ergebnis, Box ignoriert Befehl:
```text
[lp-1  ] DEBUG 2024/09/17 12:21:08 charge power: 11441W
[lp-1  ] DEBUG 2024/09/17 12:21:08 charge currents: [15.6 15.6 15.5]A
```
Danach eine Änderung der Amper:
```text
[ocpp  ] TRACE 2024/09/17 12:21:46 sent JSON message to g001: [2,"309212367","SetChargingProfile",{"connectorId":1,"csChargingProfiles":{"chargingProfileId":10,"stackLevel":3,"chargingProfilePurpose":"TxDefaultProfile","chargingProfileKind":"Absolute","chargingSchedule":{"startSchedule":"2024-09-17T10:21:46Z","chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":6.2}]}}}]
```
...und siehe da:
```text
[lp-1  ] DEBUG 2024/09/17 12:22:08 charge power: 4294W
[lp-1  ] DEBUG 2024/09/17 12:22:08 charge currents: [5.8 5.9 5.8]A
```
Gesamtes Log: 
[evcc-20240917-122639-trace.log](https://github.com/user-attachments/files/17026989/evcc-20240917-122639-trace.log)

Im Log ab 12:22:45 sieht man Ganze nochmal. 2x 6A hintereinander -> springt auf 16A. Danch wird auf 7.4A gestellt und es funktioniert wieder.

Ich schicke das genauso auch an den Hersteller, aber vielleicht habt ihr ja eine schnelle Idee oder könnt abprüfen, dass eben nicht mehrmals der selbe Befehl gesendet wird.

Kenterfie, 2024-09-30:
Gibt es da schon etwas neues. Das Problem ist sehr nervig und habe das Gefühl das es an der Art wie evcc mit der wallbox kommuniziert liegt. Habe es jetzt eine Zeitlang mit einem anderen ocpp tool getestet und dort das Verhalten nicht festgestellt, aber da das im Grunde nur ein paar Python Scripte sind wäre es schön wenn es mit evcc auch zuverlässig laufen würde. Hatte auch schon den Hersteller Support kontaktiert und dort wurde im Grunde auch nur auf die ocpp spec verwiesen.

andig, 2024-09-30:
> Gibt es da schon etwas neues.

Meine Lieblings fragt. Wenn oben nix neues steht wohl nicht.

> Das Problem ist sehr nervig

Welches? Dass die WB auf 2-maliges Setzen des Stromes völlig erratisch dadurch reagiert einfach einen beliebigen anderen Strom einzustellen?

> und habe das Gefühl das es an der Art wie evcc mit der wallbox kommuniziert liegt

Das liegt einfach daran, dass die Firmware einen Fehler macht. Was sagt der Hersteller dazu? Hast Du schon Rückmeldung?

> Hatte auch schon den Hersteller Support kontaktiert und dort wurde im Grunde auch nur auf die ocpp spec verwiesen.

Prima. Warum ändert die WB dann den Strom entgegen der Vorgabe? Was sagt der Hersteller dazu?

mrhund, 2024-10-02:
@Kenterfie Also ich hab aufgegeben. Der Support versteht das Problem trotz Logfiles und gefühlt 100facher Erklärung nicht, sagen sie kennen das Problem nicht und wollen, dass ich OCPP deaktiviere. An diesem Punkt hab ich Kommunikation eingestellt.

Welche "anderen ocpp tool" verwendest du zum Testen? Da würd ich auch mal eine Testsession starten...

andig, 2024-10-02:
Wovon reden wir denn hier jetzt? Soweit ich das sehe funktioniert das Laden. Evtl. bringt sogar https://github.com/evcc-io/evcc/pull/16430 nochmal eine Besserung.

Also was ist hier das "sehr nervige" Problem das absolut noch gelöst werden muss?

Kenterfie, 2025-05-15:
Update

Das Problem wurde inzwischen behoben. Wer noch nicht auf die neuste Firmware (6.7.24) gewechselt ist, sollte dies nun tun und testen ob das Problem noch besteht.

andig, 2025-05-15:
Danke für das Update- sehr hilfreich 👍🏻