evcc-io/evcc #15492: Wallbox.com/Pulsar: charge meter: not available

mrhund, 2024-08-19:
### Describe the bug

Seit 0.130.0 wird die oa. Meldung alle paar Sekunden ausgegeben, egal ob gelanden wird oder nicht. Die aktuellen Werte des Ladevorgangs usw. werden aber korrekt verarbeitet und auch in der GUI angezeigt.

### Steps to reproduce

1. Use Wallbox.com/Pulsar
2. Start EVCC

### Configuration details

```yaml
chargers:
- type: template
  template: pulsarplus 
  timeout: 2m  
  name: wallbox6
  remotestart: true

loadpoints:
- title: Garage
  charger: wallbox6
  mode: pv
  enable:
    threshold: -3699
    delay: 2m
  disable:
    threshold: 500 # maximum import power (W)
    delay: 10m
```

### Log details

```text
[ocpp-1] DEBUG 2024/08/19 11:14:15 waiting for chargepoint: 5m0s
[ocpp  ] INFO 2024/08/19 11:14:43 charge point connected, registering: g001
[ocpp  ] TRACE 2024/08/19 11:14:43 recv g001: [2,"1019295848","StatusNotification",{"info": "","vendorId": "com.wallbox","vendorErrorCode": "","connectorId": 0,"errorCode": "NoError","status": "Available","timestamp": "2024-08-19T09:14:43Z"}]
[ocpp  ] TRACE 2024/08/19 11:14:43 send g001: [3,"1019295848",{}]
[ocpp  ] TRACE 2024/08/19 11:14:43 recv g001: [2,"469654653","StatusNotification",{"info": "","vendorId": "com.wallbox","vendorErrorCode": "","connectorId": 1,"errorCode": "NoError","status": "SuspendedEVSE","timestamp": "2024-08-19T09:14:43Z"}]
[ocpp  ] TRACE 2024/08/19 11:14:43 send g001: [3,"469654653",{}]
[ocpp  ] TRACE 2024/08/19 11:14:43 recv g001: [2,"1347599670","MeterValues",{"meterValue": [{"timestamp": "2024-08-19T09:14:43Z","sampledValue": [{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Energy.Active.Import.Register","unit": "Wh","value": "532218.4"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Power.Active.Import","unit": "W","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Current.Offered","unit": "A","value": "0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Power.Offered","unit": "W","value": "0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L1","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L2","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L3","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L1-N","measurand": "Voltage","unit": "V","value": "233.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L2-N","measurand": "Voltage","unit": "V","value": "244.0"} ]} ],"connectorId": 1,"transactionId": 2}]
[ocpp-1] DEBUG 2024/08/19 11:14:43 hijacking transaction: 2
[ocpp  ] TRACE 2024/08/19 11:14:43 send g001: [3,"1347599670",{}]
[ocpp  ] TRACE 2024/08/19 11:14:43 recv g001: [2,"2027165033","MeterValues",{"meterValue": [{"timestamp": "2024-08-19T09:14:12Z","sampledValue": [{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Energy.Active.Import.Register","unit": "Wh","value": "532218.4"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Power.Active.Import","unit": "W","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Current.Offered","unit": "A","value": "0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Power.Offered","unit": "W","value": "0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L1","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L2","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L3","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L1-N","measurand": "Voltage","unit": "V","value": "233.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L2-N","measurand": "Voltage","unit": "V","value": "245.0"} ]} ],"connectorId": 1,"transactionId": 2}]
[ocpp  ] TRACE 2024/08/19 11:14:43 send g001: [3,"2027165033",{}]
[ocpp  ] TRACE 2024/08/19 11:14:43 recv g001: [2,"1297993920","MeterValues",{"meterValue": [{"timestamp": "2024-08-19T09:14:22Z","sampledValue": [{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Energy.Active.Import.Register","unit": "Wh","value": "532218.4"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Power.Active.Import","unit": "W","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Current.Offered","unit": "A","value": "0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Power.Offered","unit": "W","value": "0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L1","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L2","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L3","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L1-N","measurand": "Voltage","unit": "V","value": "233.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L2-N","measurand": "Voltage","unit": "V","value": "245.0"} ]} ],"connectorId": 1,"transactionId": 2}]
[ocpp  ] TRACE 2024/08/19 11:14:43 send g001: [3,"1297993920",{}]
[ocpp  ] TRACE 2024/08/19 11:14:43 recv g001: [2,"779233100","MeterValues",{"meterValue": [{"timestamp": "2024-08-19T09:14:32Z","sampledValue": [{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Energy.Active.Import.Register","unit": "Wh","value": "532218.4"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Power.Active.Import","unit": "W","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Current.Offered","unit": "A","value": "0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Power.Offered","unit": "W","value": "0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L1","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L2","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L3","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L1-N","measurand": "Voltage","unit": "V","value": "233.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L2-N","measurand": "Voltage","unit": "V","value": "244.0"} ]} ],"connectorId": 1,"transactionId": 2}]
[ocpp  ] TRACE 2024/08/19 11:14:43 send g001: [3,"779233100",{}]
[ocpp  ] TRACE 2024/08/19 11:14:44 send g001: [2,"4253539538","GetConfiguration",{}]
[ocpp  ] TRACE 2024/08/19 11:14:44 recv g001: [3,"4253539538",{"unknownKey": [],"configurationKey": [{"readonly": false,"key": "AllowOfflineTxForUnknownId","value": "true"},{"readonly": false,"key": "AuthorizationCacheEnabled","value": "true"},{"readonly": false,"key": "AuthorizeRemoteTxRequests","value": "false"},{"readonly": false,"key": "BlinkRepeat","value": "10"},{"readonly": false,"key": "ClockAlignedDataInterval","value": "0"},{"readonly": false,"key": "ConnectionTimeOut","value": "120"},{"readonly": false,"key": "ConnectorPhaseRotation","value": "1.Unknown"},{"readonly": true,"key": "ConnectorPhaseRotationMaxLength","value": "1"},{"readonly": true,"key": "GetConfigurationMaxKeys","value": "50"},{"readonly": false,"key": "HeartbeatInterval","value": "60"},{"readonly": false,"key": "LightIntensity","value": "100"},{"readonly": false,"key": "LocalAuthorizeOffline","value": "true"},{"readonly": false,"key": "LocalPreAuthorize","value": "true"},{"readonly": false,"key": "MaxEnergyOnInvalidId","value": "5000"},{"readonly": false,"key": "MeterValuesAlignedData","value": "Energ [... cut ...] and": "Voltage","unit": "V","value": "232.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L2-N","measurand": "Voltage","unit": "V","value": "243.0"} ]} ],"connectorId": 1,"transactionId": 2}]
[ocpp  ] TRACE 2024/08/19 11:17:44 send g001: [3,"186886598",{}]
[lp-1  ] DEBUG 2024/08/19 11:17:47 charge power: 11097W
[lp-1  ] DEBUG 2024/08/19 11:17:47 charge currents: [15.4 15.4 15.5]A
```

### What type of operating system are you running?

Linux

### Version

0.130.0

pmknowles, 2024-08-19:
I get the same error followed by the charge current being rejected when a charge request is sent so won't charge. Works fine in 0.129.0

andig, 2024-08-19:
@premultiply ich kann mir nicht erklären, warum die Energie hier "not available" sein sollte. Da die Fehlermeldung von verschiedenen Stellen im Code kommen kann habe ich diese angepasst: d970fc94cbb683747eb4a350321cdd276eaaf62c (in 130.1 nicht enthalten).

Hast Du eine Idee, was hier los ist?

pdonovan, 2024-08-20:
I also see "charge meter: not available" on my ZJ Beny charger in 0.130.1. It was not doing this in nightly 3798d4ea.

mrhund, 2024-08-20:
> @premultiply ich kann mir nicht erklären, warum die Energie hier "not available" sein sollte. Da die Fehlermeldung von verschiedenen Stellen im Code kommen kann habe ich diese angepasst: [d970fc9](https://github.com/evcc-io/evcc/commit/d970fc94cbb683747eb4a350321cdd276eaaf62c) (in 130.1 nicht enthalten).
> 
> Hast Du eine Idee, was hier los ist?

@andig Eben mit dem nightly getestet. Fehlermeldung ist nun "charge voltages: not available"

premultiply, 2024-08-20:
Bitte `evcc charger --log trace --diagnose`

mrhund, 2024-08-20:
@premultiply [Charger trace.csv](https://github.com/user-attachments/files/16673744/Charger.trace.csv)

premultiply, 2024-08-20:
```json
[
  2,
  "917891895",
  "MeterValues",
  {
    "meterValue": [
      {
        "timestamp": "2024-08-19T09:17:23Z",
        "sampledValue": [
          {
            "format": "Raw",
            "location": "Outlet",
            "context": "Sample.Periodic",
            "measurand": "Energy.Active.Import.Register",
            "unit": "Wh",
            "value": "532393.7"
          },
          {
            "format": "Raw",
            "location": "Outlet",
            "context": "Sample.Periodic",
            "measurand": "Power.Active.Import",
            "unit": "W",
            "value": "11035.3"
          },
          {
            "format": "Raw",
            "location": "Outlet",
            "context": "Sample.Periodic",
            "measurand": "Current.Offered",
            "unit": "A",
            "value": "16"
          },
          {
            "format": "Raw",
            "location": "Outlet",
            "context": "Sample.Periodic",
            "measurand": "Power.Offered",
            "unit": "W",
            "value": "11085"
          },
          {
            "format": "Raw",
            "location": "Outlet",
            "context": "Sample.Periodic",
            "phase": "L1",
            "measurand": "Current.Import",
            "unit": "A",
            "value": "15.4"
          },
          {
            "format": "Raw",
            "location": "Outlet",
            "context": "Sample.Periodic",
            "phase": "L2",
            "measurand": "Current.Import",
            "unit": "A",
            "value": "15.4"
          },
          {
            "format": "Raw",
            "location": "Outlet",
            "context": "Sample.Periodic",
            "phase": "L3",
            "measurand": "Current.Import",
            "unit": "A",
            "value": "15.5"
          },
          {
            "format": "Raw",
            "location": "Outlet",
            "context": "Sample.Periodic",
            "phase": "L1-N",
            "measurand": "Voltage",
            "unit": "V",
            "value": "231.0"
          },
          {
            "format": "Raw",
            "location": "Outlet",
            "context": "Sample.Periodic",
            "phase": "L2-N",
            "measurand": "Voltage",
            "unit": "V",
            "value": "241.0"
          }
        ]
      }
    ],
    "connectorId": 1,
    "transactionId": 2
  }
]
```

@andig Hier fehlt Voltage L3.
Hat wohl firmwareintern nicht mehr in den Ausgabepuffer gepasst?
Man fragt besser nicht, was die sich da zurecht programmiert haben.
Hab mal die Reihenfolge der Measurands angepasst um das Problem damit hoffentlich zu umschiffen.

May be fixed by https://github.com/evcc-io/evcc/commit/29efcceea9e5fd0b558d053a699aa1ddbf7e43d9

closed 2024-08-20 by premultiply

andig, 2024-08-20:
@premultiply den Fix verstehe ich nicht. Warum prüfen wir die Measurands nicht auf die Verfügbarkeit aller drei Phasen wenn wir für einen Messwert Phasen erwarten bevor wir das dann in den Decorator stecken? Da scheint der alte Code weniger fehleranfällig gewesen zu sein.

mfuchs1984, 2024-08-20:
Jusr speculating based on some numbers:

The charger reports `{"readonly": true,"key": "MeterValuesSampledDataMaxLength","value": "10"}`. This means, the 
>Maximum number of items in a MeterValuesSampledData Configuration Key

is 10

Evcc is sending `[2,"1919348389","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Energy.Active.Import.Register,Power.Active.Import,SoC,Current.Offered,Power.Offered,Current.Import,Voltage"}]`

This looks like 7 keys. But the current import can, according to the specs, also be sent as current.import.l1, l2, l3, same for voltage. In total these would then be 11, one more than the charger allows. The last one is voltage.l3, which is missing. Looks like they might internally split current and voltage.

premultiply, 2024-08-20:
Weil die in den Measurands nicht auftauchen. Man KANN die zwar theoretisch auch einzeln explitit abonnieren, aber das brauch ja dann noch mehr der meist stark beschränkten Listenlänge.
Die Aufsplittung von Voltage zu Voltage.L1,Voltage.L2,Voltage.L3 macht erst der Charger intern.
Und hier ist es wohl besonders doof implementiert.

mfuchs1984, 2024-08-20:
Exakt das war mein Gedanke

premultiply, 2024-08-20:
@andig Es ist mir aber in diesem Zusammenhang jetzt doch noch ein Problem aufgefallen:

andig, 2024-08-20:
Magst du ein Issue machen? Hier hätte das ja auch nix genutzt weil L3 komplett fehlt?

premultiply, 2024-08-20:
Das ist richtig, aber manchmal stolpert man beim stöbern in den Diagnose-Outputs eben über neue Gemeinheiten...

mrhund, 2024-08-22:
> May be fixed by [29efcce](https://github.com/evcc-io/evcc/commit/29efcceea9e5fd0b558d053a699aa1ddbf7e43d9)

Fehler mit 0.130.2 nicht mehr da. Danke!

pmknowles, 2024-08-22:
It stops the charge meter not available message put it doesn't fix the charging profile rejected which I think might be caused by Absolute rather than Relative (and/or SetChargingProfile v RemoteStartTransaction)
v0.130.2
[lp-1 ] DEBUG 2024/08/22 09:23:36 charge power: 0W
[lp-1 ] DEBUG 2024/08/22 09:23:36 charge currents: [0 0 0]A
[site ] DEBUG 2024/08/22 09:23:36 pv power: 315W
[site ] DEBUG 2024/08/22 09:23:36 battery soc: 21%
[site ] DEBUG 2024/08/22 09:23:36 battery power: -221W
[site ] DEBUG 2024/08/22 09:23:36 grid meter: 0W
[site ] DEBUG 2024/08/22 09:23:36 site power: -219W
[lp-1 ] DEBUG 2024/08/22 09:23:36 charge voltages: [227 0 0]V
[lp-1 ] DEBUG 2024/08/22 09:23:36 detected connected phases: 1p
[lp-1 ] DEBUG 2024/08/22 09:23:36 charge total import: 1475.943kWh
[733419-1] WARN 2024/08/22 09:23:36 waiting for local authentication This is new
[lp-1 ] DEBUG 2024/08/22 09:23:36 charger status: B
[lp-1 ] DEBUG 2024/08/22 09:23:36 pv charge current: 0.952A = 0A + 0.952A (-219W @ 1p)
[ocpp ] TRACE 2024/08/22 09:23:36 send 733419: [2,"1160549648","SetChargingProfile",{"connectorId":1,"csChargingProfiles":{"chargingProfileId":1,"stackLevel":1,"chargingProfilePurpose":"TxDefaultProfile","chargingProfileKind":"Absolute","chargingSchedule":{"startSchedule":"2024-08-22T08:23:36Z","chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":10}]}}}]
[ocpp ] TRACE 2024/08/22 09:23:36 recv 733419: [3,"1160549648",{"status": "Rejected"}]
[lp-1 ] ERROR 2024/08/22 09:23:36 max charge current 10A: set charging profile: Rejected
[ocpp ] TRACE 2024/08/22 09:23:44 send 733419: [2,"3407976403","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp ] TRACE 2024/08/22 09:23:44 recv 733419: [3,"3407976403",{"status": "Accepted"}]
[ocpp ] TRACE 2024/08/22 09:23:44 recv 733419: [2,"119509984","MeterValues",{"meterValue": [{"timestamp": "2024-08-22T08:23:44Z","sampledValue": [{"format": "Raw","location": "Outlet","context": "Trigger","measurand": "Power.Active.Import","unit": "W","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Trigger","measurand": "Energy.Active.Import.Register","unit": "Wh","value": "1475943.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L1","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L2","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L3","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L1-N","measurand": "Voltage","unit": "V","value": "227.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L2-N","measurand": "Voltage","unit": "V","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L3-N","measurand": "Voltage","unit": "V","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Trigger","measurand": "Current.Offered","unit": "A","value": "32"},{"format": "Raw","location": "Outlet","context": "Trigger","measurand": "Power.Offered","unit": "W","value": "22170"} ]} ],"connectorId": 1}]

v0.129.0
[ocpp ] TRACE 2024/08/22 09:32:30 send 733419: [2,"1757804546","RemoteStartTransaction",{"connectorId":1,"idTag":"evcc","chargingProfile":{"chargingProfileId":1,"stackLevel":0,"chargingProfilePurpose":"TxProfile","chargingProfileKind":"Relative","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":16}]}}}]
[ocpp ] TRACE 2024/08/22 09:32:30 recv 733419: [3,"1757804546",{"status": "Accepted"}]

premultiply, 2024-08-22:
> doesn't fix the charging profile rejected

OT. This is https://github.com/evcc-io/evcc/issues/15581

premultiply, 2024-08-22:
> [733419-1] WARN 2024/08/22 09:23:36 waiting for local authentication

See https://docs.evcc.io/en/blog#rfid-and-authorization and https://docs.evcc.io/en/docs/devices/chargers#ocpp-16j-compatible-charger-with-smart-charging-profile

pmknowles, 2024-08-22:
That fixes the warning (although the Wallbox Pulsar Max seems to have the setting - but it's disabled when OCPP is selected)
[cid:38d728e0-1a02-4583-ae61-ba99ef8f321b]
It doesn't resolve the Charging Profile Rejected issue

mfuchs1984, 2024-08-22:
This guy has a more recent charger firmware an no rejected issue

https://github.com/evcc-io/evcc/issues/15581#issuecomment-2304288134

pmknowles, 2024-08-22:
Latest firmware for Pulsar Max is 6.1.13. There may  be one due...

[cid:703bb0af-35dd-4d1b-ab53-7034d4dc99c4]
[cid:7cfc1fa0-c72a-4cd0-86fc-2ecde32d0596]

#15581 (comment)<https://github.com/evcc-io/evcc/issues/15581#issuecomment-2304288134>

premultiply, 2024-08-22:
@pmknowles Your images are not visible.
Please comment in GitHub directly and do not reply to the mails. This does not work correctly. Thanks.

pmknowles, 2024-08-22:
FW version
[image]
Wallbox FW versions
[image]
Wallbox security settings which are disabled by using OCPP
[image]

cdyna43, 2024-08-22:
Die Angabe auf der HP von Wallbox ist nicht richtig. Es gibt offenbar verschiedene Versionen der Pulsar Plus und für einige (ältere?) ist die momentan aktuelle FW die 5.17.87. Wurde mir von Seiten Wallbox ausdrücklich auf Nachfrage bestätigt. Ich befürchte, da kommt auch nichts mehr...

premultiply, 2024-08-22:
Prima. Danke für die Recherche.