evcc-io/evcc #15493: 0.130.0 - OCPP working - ABB TerraAC (FW 1.8.0)

sreichholf, 2024-08-19:
### Describe the bug

Charging with my TerraAC doesn't work anymore due to "SetChargingProfile" always being rejected by the charger.
I can't see the transactionId anywhere but in MeterValues, maybe that's an initial hint.

I am aware of the changes with this release.
I have tried it with "Free Vending" enabled, but also "the old way" by setting remotestart: true and free vending to false, no difference.

Attached log is with the wallbox in Free Vending mode.
Should any additional information be required, I'll do my best to delivery it asap.

### Steps to reproduce

1. Set TerraAC to free vending
2. start charging
3. evcc cannot set charge due to SetRemoteChargingProfile being rejected by the Wallbox

### Configuration details

```yaml
chargers:
  - name: TerraAC
    type: ocpp
#    autostart: true
    connecttimeout: 30m
    timeout: 20m
    remotestart: false
#    getconfiguration: true
#    bootnotification: true
```

### Log details

```text
[ocpp-1] DEBUG 2024/08/19 10:51:28 waiting for chargepoint: 30m0s
[ocpp  ] INFO 2024/08/19 10:53:49 charge point connected, registering: TACW11**********
[ocpp  ] TRACE 2024/08/19 10:53:50 recv TACW11**********: [2, "1984704", "StatusNotification", {"connectorId": 0, "errorCode": "NoError", "info": "null", "status": "Available", "vendorErrorCode": "0x0000"}]
[ocpp  ] TRACE 2024/08/19 10:53:50 send TACW11**********: [3,"1984704",{}]
[ocpp  ] TRACE 2024/08/19 10:53:50 send TACW11**********: [2,"1553186888","GetConfiguration",{}]
[ocpp  ] TRACE 2024/08/19 10:53:51 recv TACW11**********: [2, "7542564", "StatusNotification", {"connectorId": 1, "errorCode": "NoError", "info": "null", "status": "Charging", "vendorErrorCode": "0x0000"}]
[ocpp  ] TRACE 2024/08/19 10:53:51 send TACW11**********: [3,"7542564",{}]
[ocpp  ] TRACE 2024/08/19 10:53:53 recv TACW11**********: [3, "1553186888", {"configurationKey": [{"key": "GetConfigurationMaxKeys", "readonly": true, "value": "20"}, {"key": "HeartbeatInterval", "readonly": false, "value": "60"}, {"key": "MeterValueSampleInterval", "readonly": false, "value": "10"}, {"key": "AuthorizationKey", "readonly": false}, {"key": "LocalAuthorizeOffline", "readonly": false, "value": "true"}, {"key": "LocalPreAuthorize", "readonly": false, "value": "false"}, {"key": "LocalAuthListEnabled", "readonly": false, "value": "true"}, {"key": "LocalAuthListMaxLength", "readonly": true, "value": "16"}, {"key": "SendLocalListMaxLength", "readonly": true, "value": "8"}, {"key": "SupportedFeatureProfiles", "readonly": true, "value": "Core,FirmwareManagement,LocalAuthListManagement,SmartCharging,RemoteTrigger"}, {"key": "ChargeProfileMaxStackLevel", "readonly": true, "value": "3"}, {"key": "ChargingScheduleAllowedChargingRateUnit", "readonly": true, "value": "Current,Power"}, {"key": "ChargingScheduleMaxPeriods", "readonly": true, "value": "25"}, {"key": "MaxChargingProfilesInstalled", "readonly": true, "value": "16"}, {"key": "NumberOfConnectors", "readonly": true, "value": "1"}, {"key": "AllowOfflineTxForUnknownId", "readonly": false, "value": "false"}, {"key": "AuthorizeRemoteTxRequests", "readonly": false, "value": "false"}, {"key": "ConnectionTimeOut", "readonly": false, "value": "120"}, {"key": "WebSocketPingInterval", "readonly": false, "value": "30"}, {"key": "StopTransactionOnInvalidId", "readonly": false, "value": "true"}, {"key": "MaxEnergyOnInvalidId", "readonly": false, "value": "1000"}, {"key": "MeterValuesAlignedData", "readonly": false, "value": "Energy.Active.Import.Register"}, {"key": "MeterValuesAlignedDataMaxLength", "readonly": true, "value": "4"}, {"key": "ClockAlignedDataInterval", "readonly": false, "value": "0"}, {"key": "MeterValuesSampledData", "readonly": false, "value": "Current.Import,Current.Offered,Energy.Active.Import.Register,Power.Active.Import,Power.Offered,SoC,Voltage"}, {"key": "MeterValuesSampledDataMaxLength", "readonly": true, "value": "22"}, {"key": "ResetRetries", "readonly": false, "value": "0"}, {"key": "TransactionMessageAttempts", "readonly": false, "value": "10"}, {"key": "TransactionMessageRetryInterval", "readonly": false, "value": "10"}, {"key": "ChangeNFCPower", "readonly": false, "value": "10,1"}]}]
[ocpp  ] TRACE 2024/08/19 10:53:53 send TACW11**********: [2,"201601688","TriggerMessage",{"requestedMessage":"BootNotification"}]
[ocpp  ] TRACE 2024/08/19 10:53:55 recv TACW11**********: [3, "201601688", {"status": "Accepted"}]
[ocpp  ] TRACE 2024/08/19 10:53:56 recv TACW11**********: [2, "1135050", "BootNotification", {"chargePointModel": "CDT_TACW11::NET_WIFI", "chargePointVendor": "ABB", "chargeBoxSerialNumber": "TACW11**********", "chargePointSerialNumber": "TACW11**********", "firmwareVersion": "TAC3Z9119006710273::V1.8.0", "meterType": "V1"}]
[ocpp  ] TRACE 2024/08/19 10:53:56 send TACW11**********: [3,"1135050",{"currentTime":"2024-08-19T08:53:56Z","interval":60,"status":"Accepted"}]
[ocpp  ] TRACE 2024/08/19 10:53:56 send TACW11**********: [2,"333035995","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Energy.Active.Import.Register"}]
[ocpp  ] TRACE 2024/08/19 10:53:57 recv TACW11**********: [2, "2843554", "SecurityEventNotification", {"type": "StartupOfTheDevice", "timestamp": "2024-03-08T13:45:45.000Z"}]
[ocpp  ] ERROR 2024/08/19 10:53:57 ocpp message (2843554): NotSupported - Unsupported feature SecurityEventNotification ([2, "2843554", "SecurityEventNotification", {"type": "StartupOfTheDevice", "timestamp": "2024-03-08T13:45:45.000Z"}])
[ocpp  ] TRACE 2024/08/19 10:53:58 recv TACW11**********: [2, "0993052", "SecurityEventNotification", {"type": "SettingSystemTime", "techInfo": "ocppBoot", "timestamp": "2024-08-19T08:53:56.000Z"}]
[ocpp  ] ERROR 2024/08/19 10:53:58 ocpp message (0993052): NotSupported - Unsupported feature SecurityEventNotification ([2, "0993052", "SecurityEventNotification", {"type": "SettingSystemTime", "techInfo": "ocppBoot", "timestamp": "2024-08-19T08:53:56.000Z"}])
[ocpp  ] TRACE 2024/08/19 10:53:58 recv TACW11**********: [3, "333035995", {"status": "Accepted"}]
[ocpp  ] TRACE 2024/08/19 10:53:58 send TACW11**********: [2,"1364174338","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Power.Active.Import"}]
[ocpp  ] TRACE 2024/08/19 10:53:59 recv TACW11**********: [3, "1364174338", {"status": "Accepted"}]
[ocpp  ] TRACE 2024/08/19 10:53:59 send TACW11**********: [2,"4115962588","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"SoC"}]
[ocpp  ] TRACE 2024/08/19 10:54:00 recv TACW11**********: [3, "4115962588", {"status": "Accepted"}]
[ocpp  ] TRACE 2024/08/19 10:54:00 send TACW11**********: [2,"2345987780","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Current.Offered"}]
[ocpp  ] TRACE 2024/08/19 10:54:01 recv TACW11**********: [3, "2345987780", {"status": "Accepted"}]
[ocpp  ] TRACE 2024/08/19 10:54:01 send TACW11**********: [2,"1510698206","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Power.Offered"}]
[ocpp  ] TRACE 2024/08/19 10:54:02 recv TACW11**********: [3, "1510698206", {"status": "Accepted"}]
[ocpp  ] TRACE 2024/08/19 10:54:02 send TACW11**********: [2,"3897740803","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Current.Import"}]
[ocpp  ] TRACE 2024/08/19 10:54:02 recv TACW11**********: [3, "3897740803", {"status": "Accepted"}]
[ocpp  ] TRACE 2024/08/19 10:54:02 send TACW11**********: [2,"4033040641","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Voltage"}]
[ocpp  ] TRACE 2024/08/19 10:54:03 recv TACW11**********: [3, "4033040641", {"status": "Accepted"}]
[ocpp  ] TRACE 2024/08/19 10:54:03 send TACW11**********: [2,"3169484987","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Energy.Active.Import.Register,Power.Active.Import,SoC,Current.Offered,Power.Offered,Current.Import,Voltage"}]
[ocpp  ] TRACE 2024/08/19 10:54:04 recv TACW11**********: [3, "3169484987", {"status": "Accepted"}]
[ocpp  ] TRACE 2024/08/19 10:54:04 send TACW11**********: [2,"4128797202","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp  ] TRACE 2024/08/19 10:54:05 recv TACW11**********: [3, "4128797202", {"status": "Accepted"}]
[ocpp  ] TRACE 2024/08/19 10:54:07 recv TACW11**********: [2, "2115240", "MeterValues", {"connectorId": 1, "meterValue": [{"timestamp": "2024-08-19T08:54:04.000Z", "sampledValue": [{"value": "224.80", "context": "Trigger", "format": "Raw", "measurand": "Voltage", "phase": "L1-N", "unit": "V"}, {"value": "12.59", "context": "Trigger", "format": "Raw", "measurand": "Curr [... cut ...] A"}, {"value": "0.00", "context": "Transaction.Begin", "format": "Raw", "measurand": "Current.Import", "unit": "A"}, {"value": "0", "context": "Transaction.Begin", "format": "Raw", "measurand": "Power.Active.Import", "unit": "W"}]}]}]
[ocpp  ] TRACE 2024/08/19 10:54:17 send TACW11**********: [3,"8425676",{}]
[ocpp  ] TRACE 2024/08/19 10:54:17 recv TACW11**********: [3, "3634723683", {"status": "Rejected"}]
.....
```

### What type of operating system are you running?

Linux

### Version

0.130.0

hyperbart, 2024-08-19:
I am also getting the red warning message "charger disable: set charging profile: Rejected". Warning wasn't present in 0.129 but is in 0.130.

This is with a Wallbox Pulsar Plus.

premultiply, 2024-08-19:
```json
[2,"2150461646","SetChargingProfile",{"connectorId":1,"csChargingProfiles":{"chargingProfileId":16,"stackLevel":3,"chargingProfilePurpose":"TxDefaultProfile","chargingProfileKind":"Absolute","chargingSchedule":{"startSchedule":"2024-08-19T08:54:14Z","chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":6}]}}}]

[3, "2150461646", {"status": "Rejected"}]
```

Bitte mal beim Hersteller nachfragen was die Box an dieser klaren und standardkonformen Anweisung zu bemängeln hat.

premultiply, 2024-08-19:
Vielleicht auch mal auf die aktuelle Firmware aktualisieren.

HelloAccount, 2024-08-19:
Das Problem besteht seit deiner Umstrukturierung, da du die TransactionID wieder aus dem Code entfernt hast. Die Thematik wurde letzes Jahr von pjjjv mit ABB abgeklärt. Nachzulesen unter:
https://github.com/evcc-io/evcc/discussions/8138#discussioncomment-7062007

sreichholf, 2024-08-19:
Nach einem Blick in die Spec ist es doch aber so spezifiziert, dass man TxDefaultProfile nur für neue Transaktionen nutzt. Ich verstehe das als "Default Werte beim Start einer neuen Transaktion.

Für bereits laufende Transaktionen ist TxProfile gedacht und das benötigt dann auch die zugehörige transactionId.

Chapter 7.10

premultiply, 2024-08-19:
Wir senden keine transaktionsgebundenen TxProfiles mehr, sondern nur noch TxDefaultProfiles. Dort gibt es keine Transaktionsnummer.

Irgendwas fehlt wohl in der OCPP-Implementierung von ABB...
Aber was?

Vielleicht können sie kein Kind "Absolute"? 🤷🏻‍♂️
Das wird nur ABB beantworten können oder irgendwer muss es ausprobieren.

premultiply, 2024-08-19:
@sreichholf Das verstehst du falsch.
Die Einlassungen dazu in der Spec sind deutlich umfangreicher und über mehrere Abschnittschnitte zu Charging Profiles generell,  Smart Charging usw. verstreut.
Bitte genauer lesen!

Kurz: TxDefaultProfile greift immer wenn nicht von TxProfile überlagert.
TxProfile nutzen wir nicht mehr sondern nur noch TxDefaultProfile.
Das ist absolut Standardkonform und auch sonst die übliche Variante.

HelloAccount, 2024-08-19:
Wie bereits geschrieben, ABB sagt dazu https://github.com/evcc-io/evcc/discussions/8138#discussioncomment-7062007

premultiply, 2024-08-19:
Wir benutzen kein TxProfile!!!

HelloAccount, 2024-08-19:
Schon klar, das habe ich verstanden, aber nur weil die TransactionID nicht für ein TxDefaultProfile benötig wird, heißt es doch nicht direkt, dass sie nicht als Standard 0-Wert mitgesendet werden darf.
Das TxDefaultProfile wird ja schließlich in einen SetChargingProfile Request gesendetet und der beinhaltet grundsätzlich die TransactionID.
Ich sehe da kein Problem bezüglich des Standards, wenn ein 0-Wert mitgesendet wird.

Die von ABB scheinen die ID immer zu benötigen, ja, das nicht optional und widerspricht dem Standard, aber nochmal, ein 0-Wert spricht grundsätzlich nicht gegen den Standard.

premultiply, 2024-08-19:
[image]

HelloAccount, 2024-08-19:
invalid und optional heißt aber nicht, dass man es weglassen muß, es kann auch als 0-Wert mitgesendet werden:

{
       "connectorId": 3,
       "csChargingProfiles": {
               "chargingProfileId": 26771,
               "chargingProfileKind": "Absolute",
               "chargingProfilePurpose": "TxDefaultProfile",
               "chargingSchedule": {
                       "chargingRateUnit": "kW",
                       "chargingSchedulePeriod": [
                               {
                                       "limit": 2,
                                       "startPeriod": 0
                               }
                       ],
                       "duration": 86400
               },
               "stackLevel": 1,
               **"transactionId": null,**
               "validFrom": "2021-10-25T21:57:33.222092",
               "validTo": "2021-10-26T21:57:33.222092"
       }
}

Ich will mich nicht mit dir streiten, gibt es eine Chance, dass unser Problem von euch gelöst werden kann?

premultiply, 2024-08-19:
Mmmmmh. Spec sagt: Integer
Wie passt das zusammen?

Ich versuche ja eine gute Lösung zu finden. Diese soll aber keinesfalls absehbar neue Probleme mit anderen Boxen provozieren, da sie ggf. gegen die Spec verstößt oder diese zu kreativ interpretiert.

Fakt ist aber auch, dass ABB hier ein Problem hat, was es vor allem von deren Seite zu lösen gilt.

HelloAccount, 2024-08-19:
Ich verstehe das Dilemma, ihr macht habt ein tolles Projekt und macht einen tollen Job. Es wird leider nie die ideale Lösung für alle geben und einer wird immer hintenüberfallen.    ... wenn es jetzt die TerraAC-Box Besitzer mit OCPP-Anbindung sind, dann soll es eben so sein :-)

Ich werde selber mal den Default-Wert im ChargingProfile setzen und sehen, ob die Box das akzteptiert. 

Vielen Dank und gute Nacht.

sreichholf, 2024-08-20:
@premultiply erstmals sorry, wie gesagt hatte ich nur einen kurzen Blick in die Spec geworfen.
Ich kenne solche Probleme mit Spezifikation vs. "real world" und weiß dass das unfassbar nervig ist...

Persönlich würde ich mich freuen wenn man dieses Problem durch erneutes mitsenden der transactionId (auch wenn der standard sagt die darf an der stelle nicht benötigt werden) wieder lösen könnte.
Das hat jetzt knapp 1 Jahr lang sehr gut funktioniert und scheinbar auch nicht wirklich negative Auswirkungen auf andere wallboxen.
Das ist aber nicht meine Entscheidung sondern lediglich eine Bitte :).

ABB benötigt für jedes Update ~6 Monate und das aktuelle kann ich nichtmal einspielen weil die Wallbox einfach mit "Fehler beim Aktualisieren" aussteigt. Hier auf eine Lösung von deren Seite zu hoffen halte ich für, milde formuliert, eine liebevoll gemeinte Freizeitbeschäftigung mit wenig Aussicht auf Erfolg.

premultiply, 2024-08-20:
Vor allem ist das alles bisher reine Spekulation ob das überhaupt das Problem lösen würde.

smiley1803, 2024-08-20:
Leider habe ich bei mir das gleiche Problem. Vielleicht ist es eine naive Frage von mir als Laie, aber könnte man nicht für die ABB Terra AC ein eigenes OCPP Template (z.b. ocpp_abb) erstellen, dass dann die Transaction ID drinnen hat? Bis Version 0.129.0 hat ja alles einwandfrei funktioniert. Das würde dann (hoffentlich) für uns das Problem wieder lösen und gleichzeitig aber bei anderen Wallboxen keine Probleme verursachen.

pmknowles, 2024-08-20:
V0.129.0 trace
`[ocpp  ] TRACE 2024/08/20 09:39:06 send 733419: [2,"2723582184","RemoteStartTransaction",{"connectorId":1,"idTag":"evcc","chargingProfile":{"chargingProfileId":1,"stackLevel":0,"chargingProfilePurpose":"TxProfile","chargingProfileKind":"**Relative**","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":16}]}}}]
[ocpp  ] TRACE 2024/08/20 09:39:07 recv 733419: [3,"2723582184",{"status": "Accepted"}]`
V0.130.0 trace
`[ocpp  ] TRACE 2024/08/20 09:29:02 send 733419: [2,"1617144868","SetChargingProfile",{"connectorId":1,"csChargingProfiles":{"chargingProfileId":1,"stackLevel":1,"chargingProfilePurpose":"TxDefaultProfile","chargingProfileKind":"**Absolute**","chargingSchedule":{"startSchedule":"2024-08-20T08:29:02Z","chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":16}]}}}]
[ocpp  ] TRACE 2024/08/20 09:29:02 recv 733419: [3,"1617144868",{"status": "Rejected"}]
[lp-1  ] ERROR 2024/08/20 09:29:02 max charge current 16A: set charging profile: Rejected`
Differences seems to be
RemoteStartTransaction and SetChargingPeofile
Relative and Absolute
TxProfile and TxDefaultProfile
stackLevel 0 and stackLevel 1
V0.130.1 is rejected too

pmknowles, 2024-08-20:
@premultiply It's not just ABB - the Wallbox Pulsar Max has the same issue

premultiply, 2024-08-20:
This issue is about ABB.

Please open another issue and provide all required informations.

pmknowles, 2024-08-20:
It IS the same issue. I get the same error message and it's only started since 0.130.0 - I've been running evcc for over a year. If I change the template to OCPP I get the same error in 0.130.0 but not in 0.129.0. It's something to do with the change in the OCPP message. When I used to debug software for a living it was always useful to know if different systems had the same issue. My trace shows exactly the same message being sent as the ABB trace with exactly the same result. It may be that the issue is Absolute and Relative (or something else) but the fact that two different manufacturers are having the same issue is important - maybe they are both non-compliant - but you can't ignore it.

dscho, 2024-08-20:
> It's not just ABB - the Wallbox Pulsar Max has the same issue

I suspect that is true.

> Please open another issue and provide all required informations.

I have done that: https://github.com/evcc-io/evcc/issues/15535. @pmknowles maybe you want to add your logs there, too, just for completeness' sake?

premultiply, 2024-08-20:
> Ich werde selber mal den Default-Wert im ChargingProfile setzen und sehen, ob die Box das akzteptiert.

@HelloAccount Haben deine Tests schon ein Ergebnis gebracht?

matti1993, 2024-08-21:
Oder gibt es die Möglichkeit wieder auf V29 zurück zugehen? 
So hat es ja für die ABB immer gut funktioniert

matti1993, 2024-08-21:
also ich hab jetzt mal die neueste Version für die Wollbox geladen es funktioniert aber trotzdem nicht.

premultiply, 2024-08-21:
Was sagt denn ABB?

sreichholf, 2024-08-24:
Hi,
meine TerraAC (ohne Display/MID) zeigt keine derartig fatalen Probleme. Mittlerweile ist sie ebenfalls auf FW 1.8.21.
Ja, die Steuerung funktioniert aktuell nicht mehr - seit 0.130 aber die Box stürzt nicht ab.
Ich weiß aber, dass ich solche Probleme (iirc nach dem Update auf 1.8.0) schon hatte. Bei Mir hat damals ein kompletter Reset der Wallbox geholfen.

mik5z, 2024-08-24:
[image]

Hier meine ABB Konfiguration: free mode Ist an

andig, 2024-08-24:
Long story short. Das funzt nicht:

```
[ocpp  ] TRACE 2024/08/19 10:54:14 send TACW11**********: [2,"2150461646","SetChargingProfile",{"connectorId":1,"csChargingProfiles":{"chargingProfileId":16,"stackLevel":3,"chargingProfilePurpose":"TxDefaultProfile","chargingProfileKind":"Absolute","chargingSchedule":{"startSchedule":"2024-08-19T08:54:14Z","chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":6}]}}}]
[ocpp  ] TRACE 2024/08/19 10:54:14 recv TACW11**********: [3, "2150461646", {"status": "Rejected"}]
```

Frage an ABB: warum sollte das nicht gehen? Bitte Support kontaktieren.

Ansonsten siehe https://github.com/evcc-io/evcc/issues/15581#issuecomment-2308356630 und https://github.com/evcc-io/evcc/issues/15581#issuecomment-2306376275. Jemand (tm) muss herausfinden, welchen Teil an `SetChargingProfile` die Box nicht mag. Da wir die Box nicht haben gibts für uns nichts zu tun.

Sorry :(

Frintrop, 2024-08-25:
Ich habe leider auch das selber Problem nun mit meinen beiden Wallboxen Pulsar Plus / Commander 2, FW hat sich nicht geändert. Mit oder ohne "remotestart: true". Es betrifft anscheinend OCCP Implementierungen mehrerer Hersteller.

#15535 wurde ja geschlossen, deswegen poste ich hier.

Was kann man tun ?

andig, 2024-08-25:
@Frintrop das steht genau einen Post über Deinem. Bitte lesen. Danke.

Frintrop, 2024-08-25:
Support von Wallbox Chargers angefragt mit Trace Log von SetChargingProfile

sreichholf, 2024-08-26:
Ich hab das jetzt soweit an ABB gemeldet.

smiley1803, 2024-08-26:
Ich habe eine Anfrage an ABB gesendet:
Leider gibt es aktuell ein Problem damit, denn die Terra AC hat offenbar mit den Befehl "SetChargingProfile" ihre Schwierigkeiten. Hier ein Auszug aus dem Log von Evcc:

 

[ocpp ] TRACE 2024/08/24 22:19:20 send TACW2241621T0751: [2,"1510332861","SetChargingProfile",{"connectorId":1,"csChargingProfiles":{"chargingProfileId":16,"stackLevel":3,"chargingProfilePurpose":"TxDefaultProfile","chargingProfileKind":"Absolute","chargingSchedule":{"startSchedule":"2024-08-24T20:19:20Z","chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":0}]}}}]

[ocpp ] TRACE 2024/08/24 22:19:20 recv TACW2241621T0751: [3, "1510332861", {"status": "Rejected"}]

[lp-1 ] ERROR 2024/08/24 22:19:20 charger disable: set charging profile: Rejected

 

Können Sie mir dazu irgendwelche Infos geben, woran das liegen kann? Erwartet die Terra AC irgendwelche zusätzlichen oder abweichenden Angaben?

Die Firmware ist aktuell (1.8.21)

Die Antwort von ABB kam gerade in 2 Etappen:
Mitarbeiter 1:
0 Limit vielleicht?
Hat die Frage an einen Kollegen weiter gegeben.

2.Mitarbeiter:
Ja das limit sollte natürlich gesetzt sein - wenn die WB nicht plausible werte bekommt macht sie einen reject: - auch Ap oder Watt sollen gesetz werden..

[image]
[image]

Das wäre noch das komplette Schema:  -> Vorlage

{

    "$schema": "http://json-schema.org/draft-04/schema#",

    "id": "urn:OCPP:1.6:2019:12:SetChargingProfileRequest",

    "title": "SetChargingProfileRequest",

    "type": "object",

    "properties": {

        "connectorId": {

            "type": "integer"

        },

        "csChargingProfiles": {

            "type": "object",

            "properties": {

                "chargingProfileId": {

                    "type": "integer"

                },

                "transactionId": {

                    "type": "integer"

                },

                "stackLevel": {

                    "type": "integer"

                },

                "chargingProfilePurpose": {

                    "type": "string",

                    "additionalProperties": false,

                    "enum": [

                        "ChargePointMaxProfile",

                        "TxDefaultProfile",

                        "TxProfile"

                    ]

                },

                "chargingProfileKind": {

                    "type": "string",

                    "additionalProperties": false,

                    "enum": [

                        "Absolute",

                        "Recurring",

                        "Relative"

                    ]

                },

                "recurrencyKind": {

                    "type": "string",

                    "additionalProperties": false,

                    "enum": [

                        "Daily",

                        "Weekly"

                    ]

                },

                "validFrom": {

                    "type": "string",

                    "format": "date-time"

                },

                "validTo": {

                    "type": "string",

                    "format": "date-time"

                },

                "chargingSchedule": {

                    "type": "object",

                    "properties": {

                        "duration": {

                            "type": "integer"

                        },

                        "startSchedule": {

                            "type": "string",

                            "format": "date-time"

                        },

                        "chargingRateUnit": {

                            "type": "string",

                            "additionalProperties": false,

                            "enum": [

                                "A",

                                "W"

                            ]

                        },

                        "chargingSchedulePeriod": {

                            "type": "array",

                            "items": {

                                "type": "object",

                                "properties": {

                                    "startPeriod": {

                                        "type": "integer"

                                    },

                                "limit": {

                                    "type": "number",

                                    "multipleOf" : 0.1

                                },

                                "numberPhases": {

                                        "type": "integer"

                                    }

                                },

                                "additionalProperties": false,

                                "required": [

                                    "startPeriod",

                                    "limit"

                                ]

                            }

                        },

                        "minChargingRate": {

                            "type": "number",

                            "multipleOf" : 0.1

                        }

                    },

                    "additionalProperties": false,

                    "required": [

                        "chargingRateUnit",

                        "chargingSchedulePeriod"

                    ]

                }

            },

            "additionalProperties": false,

            "required": [

                "chargingProfileId",

                "stackLevel",

                "chargingProfilePurpose",

                "chargingProfileKind",

                "chargingSchedule"

            ]

        }

    },

    "additionalProperties": false,

    "required": [

        "connectorId",

        "csChargingProfiles"

    ]

}

Bringt das jemanden weiter?

premultiply, 2024-08-26:
Nein, das hilft leider genau nichts. Wir sind so schlau wie vorher.
Frag mal wo genau sie jetzt Abweichungen sehen, denn ich kann keine erkennen?
Warum lehnt die Box den validen Request ab?

dscho, 2024-08-26:
> Nein, das hilft leider genau nichts. Wir sind so schlau wie vorher. Frag mal wo genau sie jetzt Abweichungen sehen, denn ich kann keine erkennen? Warum lehnt die Box den validen Request ab?

Na ja, beide Mitarbeiter scheinen 0 nicht für einen validen Wert für Limit zu halten, scheint es.

mfuchs1984, 2024-08-26:
`[ocpp  ] TRACE 2024/08/19 10:54:14 send TACW11**********: [2,"2150461646","SetChargingProfile",{"connectorId":1,"csChargingProfiles":{"chargingProfileId":16,"stackLevel":3,"chargingProfilePurpose":"TxDefaultProfile","chargingProfileKind":"Absolute","chargingSchedule":{"startSchedule":"2024-08-19T08:54:14Z","chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":6}]}}}]
[ocpp  ] TRACE 2024/08/19 10:54:14 recv TACW11**********: [3, "2150461646", {"status": "Rejected"}]`

Aus dem ursprünglichen Beitrag, hier wird Limit 6 rejected

dscho, 2024-08-26:
> `[ocpp ] TRACE 2024/08/19 10:54:14 send TACW11**********: [...]

Stimmt. @smiley1803 kannst Du vielleicht die Mitarbeiter auf dieses Ticket verweisen und dazu schreiben, dass nicht nur `"limit":0` sondern auch `"limit":6` abgelehnt wurde?

smiley1803, 2024-08-26:
Ja sicher, werde nochmals an ABB schreiben und darauf hinweisen, dass auch ein anderes Limit zum gleichen Problem führt.
Mir ist nur aufgefallen, der 2.Mitarbeiter schreibt nicht nur vom Limit sondern auch von
- auch Ap oder Watt sollen gesetz werden..

mfuchs1984, 2024-08-26:
Wenn ich raten müsste, meint er die Einheit des limits, die ist aber drin

`"chargingRateUnit":"A"`

dscho, 2024-08-26:
Ich gehe auch davon aus, dass "A" für Ampere steht, weil es ja auch so spezifiziert ist in [OCPP 1.6](https://openchargealliance.org/my-oca/ocpp/):

[image]

Jetzt habe ich aber doch noch einige interessanten Sachen bemerkt:

- wenn man ein _Standard_ Profil setzen möchte, dann macht eigentlich ein absoluter Startzeitpunkt herzlich wenig Sinn, denn das Profil kann ja wirklich nur einmal starten, ist also keinesfalls eine gute Standardeinstellungen für zukünftige Ladevorgänge.
- der "startSchedule" Eintrag könnte doch eigentlich wie früher weggelassen werden, oder etwa nicht?
- der `stackLevel` war eigentlich immer 0. Nun ist er stets größer als 0, was wenig sinnvoll erscheint.
- da sind jetzt plötzlich wirklich sehr, sehr viele Dinge ganz, ganz anders als früher. Ist also eigentlich wenig überraschend, dass da irgendwas davon nicht überall funktioniert, und genauso wenig überraschend, dass es jetzt ziemlich schwierig ist, zwischen all den Neuerungen diejenige zu identifizieren, die in einem gegebenen Einzelfall dafür verantwortlich ist, dass es nicht funktioniert. Das wäre vielleicht ja eventuell vielleicht doch einfacher, wenn man solche Änderungen inkrementell vornähme.

dscho, 2024-08-26:
> * der "startSchedule" Eintrag könnte doch eigentlich wie früher weggelassen werden, oder etwa nicht?

Und wo wir schon bei dem Eintrag sind, wenn man mal von einer zentral-europäischen Zeitzone ausgeht, dann entspricht der in [diesem Kommentar](https://github.com/evcc-io/evcc/issues/15493#issuecomment-2310161378) erwähnte Zeitpunkt ("2024/08/19 10:54:14" der UTC Zeit "2024-08-19T08:54:14Z", also exakt der Zeit, die per `startSchedule` angegeben wurde. Es wäre daher also durchaus denkbar, dass die Firmware davon ausgeht, dass das jetzt schon zu spät wäre, weil dieser Zeitpunkt nicht mehr in der Zukunft liegt, und deshalb ablehnt.

Vielleicht könnte man ja mal andenken, solche Sachen wesentlich einfacher von Nicht-Programmierern testbar zu machen, z.B. in Form eines neuen `test` Kommandos oder so etwas ähnlichem, was Benutzer mit minimalstem Aufwand und völlig ohne Kompilierung ausführen könnten. Andernfalls wäre das Risiko groß, dass weiterhin Entwickler und Benutzer windschief aneinander vorbei kommunizieren.

andig, 2024-08-26:
> da sind jetzt plötzlich wirklich sehr, sehr viele Dinge ganz, ganz anders als früher

Das ist bekannt. Und deshalb wurde schon x-fach geschrieben, dass die einzeln probiert werden müssen.

premultiply, 2024-08-26:
> - wenn man ein Standard Profil setzen möchte, dann macht eigentlich ein absoluter Startzeitpunkt herzlich wenig Sinn, denn das Profil kann ja dann wirklich nur ein einziges Mal starten, ist also keinesfalls eine gute Standardeinstellung für zukünftige Ladevorgänge.

Es gibt einige Boxen wie z. B. Easee, die schon in der Doku angeben dass sie kein `ProfileKind` "Relative" unterstützen (obwohl eine derartige Rosinenpickerei in der Spezifikation schlichtweg nicht vorgesehen ist). Daher zur Verbesserung der Kompatiblität die Verwendung von "Absolute".

Das Profil gilt ab dem angegebenen Zeitpunkt. Da `DefaultTxProfile` gilt es global und unabhängig von Transaktionen.

Die Uhr des CP wird mit dem CS synchronisiert.

Wie bei allem haben aber auch dies mit Sicherheit einige Firmwareentwickler nicht verstanden oder einfach auch nur nicht, nicht vollständig oder eben fehlerhaft implementiert.

> - der "startSchedule" Eintrag könnte doch eigentlich wie früher weggelassen werden, oder etwa nicht?

Bei ProfileKind "Absolute" verpflichtende Angabe. Kann daher nicht entfallen.

> - der stackLevel war eigentlich immer 0. Nun ist er stets größer als 0, was wenig sinnvoll erscheint.

Der `stackLevel` ist genau so groß wie er laut `ChargeProfileMaxStackLevel` eben maximal sein darf. Der stackLevel und dabei der Wert des erlaubten Maximum definiert die höchste Prioriät des Profils und stellt somit die elementare Funktion der Steuerung durch evcc sicher auch wenn bereits andere Profile vorhanden sein sollten. Sollte diese Angabe unzulässigerweise gänzlich fehlen wird 0 als Fallback verwendet.

Bei so manchen Firmwaren mit angeblicher OCPP SmartCharging-Unterstützung fragt man sich schon ob dies jemals von irgendwem irgendwie getestet wurde.

Das klassische "Transaktion starten, volle Power laden und am Anfang und Ende den Zählerstand übermitteln", wie halt an öffentlichen Ladepunkten oder bei Firmenwagenkunden landläufig üblich, bekommen sie ja meist noch irgendwie hin, aber bei allem was über das Core-FeatureProfile hinaus geht wird es qualitativ schnell dünn.

Und noch etwas persönliches @dscho am Rande: Einige deiner Kommentare hier in den Issues und Discussions empfinde ich freundlich gesagt recht _unangemessen_ und leider wenig sachdienlich.

dscho, 2024-08-26:
> deshalb wurde schon x-fach geschrieben, dass die einzeln probiert werden müssen. Ich hatte Dir schon im Slack geschrieben, was konkret zu tun ist.

Nur um sicherzugehen, dass ich nicht falsch verstanden werde: das einzeln bei der Fehlerprobe zu machen ist weniger gut, als es bereits bei der Einführung hübsch nacheinander einzuführen.

Ich hatte Dir im Slack schon geschrieben, dass selbst ein Revert aller vier angegebenen Commits den Fehler nicht behebt. Das hatte ich dann auch noch [öffentlich dokumentiert](https://github.com/evcc-io/evcc/issues/15581#issuecomment-2309899118).

> > * der "startSchedule" Eintrag könnte doch eigentlich wie früher weggelassen werden, oder etwa nicht?
> 
> Bei ProfileKind "Absolute" verpflichtende Angabe. Kann daher nicht entfallen.

Da hätte ich die Spezifikation anders interpretiert:

[image]

> Bei so manchen Firmwaren mit angeblicher OCPP SmartCharging-Unterstützung fragt man sich schon ob dies jemals von irgendwem irgendwie getestet wurde.

Also dafür, dass Du Dich über meine Kritik der Änderungen (die ja dokumentierterweise funktionierende Systeme in nicht mehr funktionierende verwandelt haben) beschwerst, teilst Du ja gegen andere ganz schön aus. Ich find's auch nicht wirklich nett, wie Du Cross-Links zu ähnlichen Tickets und andere Beiträge, die offensichtlich in guter Absicht verfasst wurden, als Off-Topic klassifizierst, obwohl sich ja jetzt herauskristallisiert, dass diese ganzen Probleme sehr wohl doch viel miteinander zu tun haben. Dieses Verhalten hat unter anderem dazu geführt, dass ich schon gar nicht mehr mit Dir zusammenarbeiten und testen wollte und eigentlich hier gar nicht mehr hereinschauen. @andig hat mich dann noch mal überredet, es doch noch lokal zu kompilieren, ein paar Reverts hier und da zu testen, dann noch mal andere Änderungen zu testen, mit kaputter Datenbank zu ringen, und eben doch noch Mal zu versuchen, hier zu helfen. Hoffentlich wird das alles in Zukunft besser.

andig, 2024-08-27:
Und was hast du jetzt getestet? Das wäre ja hilfreich!

premultiply, 2024-08-27:
Hilft leider auch nicht weiter, denn das war bereits eine der verfügbaren Referenzen zur Wahl der Parameter bei der Neuimplementierung.

smiley1803, 2024-08-27:
Leider scheinen auch die ABB Mitarbeiter etwas ratlos zu sein. Es wurde nun aber auch als Vermutung die fehlende transactionId genannt. Bei TxProfile wird die unbedingt von der Terra AC benötigt, können sich vorstellen dass das auch für das TxDefaultProfile gilt - auch wenn es da laut Spezifikation nicht zwingend gesendet werden muss...

Hat das inzwischen jemand getestet ob das was ändert?

smiley1803, 2024-08-28:
Das mag schon so sein, aber nachdem inzwischen mehrfach die Vermutung kam, dass die Wallbox eine transactionId "Null" haben will, wäre es wirklich interessant zu wissen ob damit das Problem gelöst werden kann oder nicht. Egal ob das die Spezifikationen vorgeben oder nicht.
Es geht um eine Lösung, damit die Steuerung wieder klappt und ich bin mir sicher, ABB wird hier nicht für EVCC die Firmware ändern, zumal ja diverse andere Ocpp Steuerungen mit der ABB Terra AC problemlos klar kommen. z.b. clever-pv läuft reibungslos
Ich kann somit nur darum bitten, dass jemand der weiß wie das geht, Mal testet ob eine transactionId etwas ändert

andig, 2024-08-28:
> Und was hast du jetzt getestet? Das wäre ja hilfreich!

Wir drehen uns hier weiter fröhlich im Kreis.

>aber nachdem inzwischen mehrfach die Vermutung kam, dass die Wallbox eine transactionId "Null" haben will, wäre es wirklich interessant zu wissen ob damit das Problem gelöst werden kann oder nicht. 

Hat das denn einfach mal jemand ausprobiert? Mangels Box können wir das nicht tun.

smiley1803, 2024-08-28:
Ich würde es ja gerne testen, hab aber keine Ahnung wie. Bin leider absoluter Laie und froh, dass ich die Basics mit Home Assistant geschafft hab. Wenn wer eine Anleitung für mich hat schau ich es mir gerne an...

sreichholf, 2024-08-28:
Ich werde (hoffentlich heute) als erstes mal die Option testen StartSchedule auf ein paar sekunden in der Zukunft zu setzen.
Momentan erscheint mir das noch als "naheliegendste" Option.
Ich geb bescheid was rauskam sobald ich mehr weiß.

sreichholf, 2024-08-28:
Mit stackLevel 0 wird von der request von der TerraAC akzeptiert

```
[ocpp ] TRACE 2024/08/28 13:14:54 send TACW1143421G1770: [2,"3345117854","SetChargingProfile", "connectorId":1,"csChargingProfiles":{"chargingProfileId":16 ,"stackLevel":0,"chargingProfilePurpose":"TxDefaultProfile","chargingProfileKind":"Absolute","chargingSchedule":{"startSchedule":"2024-08-28T11:14:54Z","chargingRateUnit"
/:"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":6}]}}}] 
[ocpp ] TRACE 2024/08/28 13:14:54 recv TACW1143421G1770: [3, "3345117854", {"status": "Accepted"}]
``` 

Die Regelung funktioniert augenscheinlich ohne Probleme - ich habs nur getestet indem ich ein paar mal zwischen aus, pv und schnell gewechselt habe.

andig, 2024-08-28:
@premultiply macht es nicht Sinn, für Default Profile davon ausgehen, dass es nur eins geben kann und damit Stacklevel == 0 korrekt wäre?

mfuchs1984, 2024-08-28:
@andig zumindest laut OCCP spec kann es mehrere TxDefaultProfiles geben
 
**3.13.2. Stacking charging profiles**
>It is allowed to stack charging profiles of the same charging profile purpose in order to describe complex
calendars. For example, one can define a charging profile of purpose TxDefaultProfile with a duration and
recurrence of one week that allows full power or current charging on weekdays from 23:00h to 06:00h and from
00:00h to 24:00h in weekends and reduced power or current charging at other times. On top of that, one can
define other TxDefaultProfiles that define exception to this rule, for example for holidays.

sreichholf, 2024-08-28:
Ich denke der Kerngedanke war vmtl. dass man sich nicht darum kümmern muss was es sonst noch gibt, wenn man ein TxDefaultProfile auf max. stack level setzt. Das ist dann letztlich immer standardmäßig gültig.
Die Spec ist aus meiner Sicht tats. recht eindeutig und so würde ich persönlich das als Bug auf Seiten der TerraAC einordnen (das kann aber am Ende @premultiply erheblich besser beurteilen).

Das Problem das ich potentiell sehe wenn man das jetzt einfach ändert:
Die scheinbar meisten Wallboxen haben damit offenbar kein Problem und damit bereits ein TxDefaultProfile mit maximalen stackLevel gesetzt (afaik sind die permanent gespeichert).
Ich vermute diese Profile müsste man nun vorab löschen damit die 0 überhaupt wieder beachtet wird?

mfuchs1984, 2024-08-28:
@sreichholf liegt richtig, die Profile sind wohl tatsächlich persistent, ich hatte probehalber meine Bender Wallbox (Spelsberg Smart Pro) per ocpp und eebus angebunden, hat beides super funktioniert, aber nach der OCPP Geschichte hatte die Wallbox das letzte Limit gespeichert und die Wallbox konnte nicht mehr über 8A laden, weil das ocpp Limit nach wie vor bei 8A lag (die Wallbox kann über diverse Schnittstellen gleichzeitig gesteuert werden, das kleinste Limit gewinnt). Erst nachdem ich die Box erneut mit evcc verbunden und auf "schnell" gestellt hatte, gingen über Modbus und eebus wieder mehr als 8A. Wenn ich mich richtig erinnere, habe ich sie zwischendurch ohne Erfolg neu gestartet.

andig, 2024-08-28:
Hätte, könnte, müsste... die Implementierungen sind in weiten Teilen nicht spec-konform und/oder die spec nicht klar genug. Hier muss doch schlicht entschieden werden:

- so lassen (Pech für ABB)
- ein anderes Default nehmen (0 funktionierte in der Vergangenheit)
- konfigurierbar machen

i.S. keep it simple plädiere ich für b).

mfuchs1984, 2024-08-28:
@andig ich kann gerne testen, ob es mit 0 funktioniert, wenn vorher schon eine höheres gesetzt wurde.

mfuchs1984, 2024-08-28:
> @andig ich kann gerne testen, ob es mit 0 funktioniert, wenn vorher schon eine höheres gesetzt wurde.

Der Bender Controller akzeptiert TxDefaultProfile stackLevel 0 auch nachdem vorher TxDefaultProfile mit StackLevel 255 gesendet wurde.

[22 more replies]

closed 2024-08-29 by andig