evcc-io/evcc #15496: Version 0.130.0 - Startup Error - Cannot create meter type - Tesla Powerwall

manuelgruber, 2024-08-19:
### Describe the bug

Since version 0.130.0 I get these error messages:

**Error during startup. Check your configuration and restart.**
**cannot create meter 'grid1': cannot create meter type 'template': cannot create meter type 'powerwall': 404 Not Found**

(Get the errors for pv1 and battery1 as well)

### Steps to reproduce

1. Setup evcc with 0.129.0 with the attached evcc.yaml
2. evcc is working
3. Update to 0.130.0
4. evcc shows "startup error"

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

interval: 10s # control cycle interval
log: trace

site:
  title: Home
  meters:
    grid: grid1
    pv:
    - pv1
    battery:
    - battery1
  residualPower: 300

meters:
- type: template
  template: tesla-powerwall
  usage: grid
  host: 10.0.0.2
  password: Abc123
  name: grid1
  refreshToken: Abc123
- type: template
  template: tesla-powerwall
  usage: pv
  host: 10.0.0.2
  password: Abc123
  name: pv1
  refreshToken: Abc123
- type: template
  template: tesla-powerwall
  usage: battery
  host: 10.0.0.2
  password: Abc123
  name: battery1
  refreshToken: Abc123

...
```

### Log details

```text
[main ] INFO 2024/08/19 15:22:34 evcc 0.130.0
[main ] INFO 2024/08/19 15:22:34 using config file: /etc/evcc.yaml
[db ] INFO 2024/08/19 15:22:34 using sqlite database: /root/.evcc/evcc.db
[main ] INFO 2024/08/19 15:22:34 listening at :7070
[main ] FATAL 2024/08/19 15:22:36 cannot create meter 'grid1': cannot create meter type 'template': cannot create meter type 'powerwall': 404 Not Found
[main ] FATAL 2024/08/19 15:22:36 will attempt restart in: 15m0s
```

### What type of operating system are you running?

Docker container

### Version

0.130.0

manuelgruber, 2024-08-19:
FYI
In the config I'm using 'tesla-powerwall', but the error refers to 'powerwall'.

manuelgruber, 2024-08-19:
EVcc is starting again after I removed "refreshToken: Abc123"...

andig, 2024-08-19:
Mach mal bitte ein trace log für `powerwall`

/cc @GrimmiMeloni

bill340, 2024-08-19:
Same issue here. if I remove "refreshToken: ..." line, the new "Netzladen" Tab is not available anymore as it was in the latest 0.129.0 nightlys.

...
meters:
- type: template
  template: tesla-powerwall 
  usage: grid  
  host: 192.168.2.xxx
  password: xxx  
#  refreshToken: eyJh...
  minSoc: 1  
  maxSoc: 100  
  name: grid1
- type: template
  template: tesla-powerwall 
  usage: pv  
  host: 192.168.2.xxx  
  password: xxx  
#  refreshToken: eyJhb...
  minSoc: 1  
  maxSoc: 100  
  name: pv2
- type: template
  template: tesla-powerwall 
  usage: battery  
  host: 192.168.2.xxx  
  password: xxx  
#  refreshToken: eyJh...  
  minSoc: 1  
  maxSoc: 100  
  name: battery3

andig, 2024-08-19:
https://github.com/evcc-io/evcc/issues/15496#issuecomment-2296890985

GrimmiMeloni, 2024-08-20:
Cannot reproduce this, neither with 0.130.0 nor with 0.130.1

For reference here's my meter section from the configuration. Refresh token is only required on the battery meter.

```yaml
meters:
- type: template
  template: tesla-powerwall
  usage: grid
  host: 192.168.3.38
  password: XXXXX
  name: grid1
- type: template
  template: tesla-powerwall
  usage: pv
  host: 192.168.3.38
  password: XXXXX
  name: pv2
- type: template
  template: tesla-powerwall
  usage: battery
  host: 192.168.3.38
  password: XXXXX
  name: battery3
  refreshToken: eyJ…
  minSoc: 20
  maxSoc: 95
```

GrimmiMeloni, 2024-08-20:
> > [#15496 (comment)](https://github.com/evcc-io/evcc/issues/15496#issuecomment-2296890985)
> 
> Sorry, not familiar how to do this...

Adjust the logging section of `evcc.yaml`, adding a powerwall entry:

```
log: debug
levels:
  cache: error
  powerwall: trace
```

bill340, 2024-08-20:
Ok, I put in the refresh token only for the battery meter and it still works.
But the "Netzladen" Tab is still missing.

bill340, 2024-08-20:
> > > [#15496 (comment)](https://github.com/evcc-io/evcc/issues/15496#issuecomment-2296890985)
> > 
> > 
> > Sorry, not familiar how to do this...
> 
> Adjust the logging section of `evcc.yaml`, adding a powerwall entry:
> 
> ```yaml
> log: debug
> levels:
>   cache: error
>   powerwall: trace
> ```

Thanks! Where can I find the log then?

bill340, 2024-08-20:
Ich habe gerade gesehen, auch die anderen Einstellungen auf der Batterie-Seite funktionieren nicht mehr.
Egal welchen %-Wert ich einstelle, es ändert nicht die Grafik und auch nicht das Lade-Verhalten.
Mit 0.129.0 hat das alles einwandfrei funktioniert. 
[image]

andig, 2024-08-20:
Können wie bitte beim Thema bleiben? Danke!

manuelgruber, 2024-08-20:
Für Token-Zugriff habe ich einen separaten Tesla-Account, der Zugriff auf meine Tesla-Produkte an Drittanbieter weitergibt.

Vor einigen Wochen habe ich dem Account den Powerwall-Zugriff entzogen. Und anscheinend meine EVcc-Instanz nicht mehr neugestartet. Ich vermute, dass das ist der Grund für die Fehlermeldung "Not found" ist. Also: Powerwall im Tesla-Account nicht gefunden.
@andig @GrimmiMeloni Ist das der Grund? 

Leider funktionieren die Einladungen zur Powerwall gerade irgendwie nicht... Werde es dann in den nächsten Tagen nochmal testen.

GrimmiMeloni, 2024-08-21:
> Vor einigen Wochen habe ich dem Account den Powerwall-Zugriff entzogen.

Wie soll es dann funktionieren?

closed 2024-09-02 by github-actions[bot]