evcc-io/evcc #15503: Tronity / vehicle recognition after plugin

sebnaf, 2024-08-19:
### Describe the bug

After plugging my Tesla back in, evcc does not recognize the tesla and uses the "gastfahrzeug".
Selecting my vehicle manually or restarting evcc pulls the right vehicle.

### Steps to reproduce

1. unplug tesla and drive
2. park and plug tesla in
3. "gastfahrzeug" stays active until selected different or evcc restart
...

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local
  port: 7070

interval: 20s # control cycle interval. Interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

plant: 2d22<snip>

# sponsor token enables optional features (request at https://sponsor.evcc.io)
# sponsortoken:
sponsortoken: eyJh<snip>

telemetry: true

# log settings
log: error
levels:
  site: info
  lp-1: info
  lp-2: info
  cache: error
  db: warn
  vehicle: info
  tesla: info
  elektronenpumpe: info

# modbus proxy for allowing external programs to reuse the evcc modbus connection
# each entry will start a proxy instance at the given port speaking Modbus TCP and
# relaying to the given modbus downstream device (either TCP or RTU, RS485 or TCP)
modbusproxy:
  #  - port: 5200
  #    uri: solar-edge:502
  #    # rtu: true
  #    # readonly: true # use `deny` to raise modbus errors

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for documentation see https://docs.evcc.io/docs/devices/meters
meters:
  - name: grid
    type: movingaverage
    decay: 0.2
    meter:
      type: custom
      power:
        source: sunspec
        #model: sunspec
        uri: 10.10.20.7:502
        id: 1
[... 255 lines cut ...]
[tronity] TRACE 2024/08/19 18:05:06 GET https://api.tronity.tech/tronity/vehicles/668aedd2d9d3250012584ce4/last_record
[tronity] TRACE 2024/08/19 18:05:06 {"odometer":33500.2,"range":193,"level":43.45,"charging":"Disconnected","plugged":false,"chargerPower":0,"chargeRemainingTime":null,"latitude":48.124068,"longitude":7.844362,"status":null,"timestamp":1724083341000,"lastUpdate":1724083341000}
[tronity] TRACE 2024/08/19 18:06:06 GET https://api.tronity.tech/tronity/vehicles/668aedd2d9d3250012584ce4/last_record
[tronity] TRACE 2024/08/19 18:06:06 {"odometer":33500.2,"range":193,"level":43.45,"charging":"Disconnected","plugged":false,"chargerPower":0,"chargeRemainingTime":null,"latitude":48.124068,"longitude":7.844362,"status":null,"timestamp":1724083341000,"lastUpdate":1724083341000}
[tronity] TRACE 2024/08/19 18:07:06 GET https://api.tronity.tech/tronity/vehicles/668aedd2d9d3250012584ce4/last_record
[tronity] TRACE 2024/08/19 18:07:06 {"odometer":33500.2,"range":193,"level":43.45,"charging":"Disconnected","plugged":false,"chargerPower":0,"chargeRemainingTime":null,"latitude":48.124068,"longitude":7.844362,"status":null,"timestamp":1724083341000,"lastUpdate":1724083341000}
[tronity] TRACE 2024/08/19 18:08:06 GET https://api.tronity.tech/tronity/vehicles/668aedd2d9d3250012584ce4/last_record
[tronity] TRACE 2024/08/19 18:08:06 {"odometer":33500.2,"range":193,"level":43.45,"charging":"Disconnected","plugged":false,"chargerPower":0,"chargeRemainingTime":null,"latitude":48.124068,"longitude":7.844362,"status":null,"timestamp":1724083341000,"lastUpdate":1724083341000}
[tronity] TRACE 2024/08/19 18:09:06 GET https://api.tronity.tech/tronity/vehicles/668aedd2d9d3250012584ce4/last_record
[tronity] TRACE 2024/08/19 18:09:06 {"odometer":33500.2,"range":193,"level":43.45,"charging":"Disconnected","plugged":false,"chargerPower":0,"chargeRemainingTime":null,"latitude":48.124068,"longitude":7.844362,"status":null,"timestamp":1724083341000,"lastUpdate":1724083341000}
[tronity] TRACE 2024/08/19 18:09:46 GET https://api.tronity.tech/tronity/vehicles/668aedd2d9d3250012584ce4/last_record
[tronity] TRACE 2024/08/19 18:09:46 {"odometer":33500.2,"range":193,"level":43.45,"charging":"Disconnected","plugged":false,"chargerPower":0,"chargeRemainingTime":null,"latitude":48.124068,"longitude":7.844362,"status":null,"timestamp":1724083341000,"lastUpdate":1724083341000}
[tronity] TRACE 2024/08/19 18:11:06 GET https://api.tronity.tech/tronity/vehicles/668aedd2d9d3250012584ce4/last_record
[tronity] TRACE 2024/08/19 18:11:06 {"odometer":33500.2,"range":193,"level":43.45,"charging":"Disconnected","plugged":false,"chargerPower":0,"chargeRemainingTime":null,"latitude":48.124068,"longitude":7.844362,"status":null,"timestamp":1724083341000,"lastUpdate":1724083341000}
[tronity] TRACE 2024/08/19 18:50:06 GET https://api.tronity.tech/tronity/vehicles/668aedd2d9d3250012584ce4/last_record
[tronity] TRACE 2024/08/19 18:50:06 {"odometer":33500.2,"range":192.2,"level":43.279,"charging":"Disconnected","plugged":false,"chargerPower":0,"chargeRemainingTime":null,"latitude":48.124068,"longitude":7.844362,"status":null,"timestamp":1724085884000,"lastUpdate":1724085884000}
```

### What type of operating system are you running?

Docker container

### Version

0.130.0

sebnaf, 2024-08-19:
evcc-20240819-185919-trace.log sent via email to support@evcc.io

andig, 2024-08-19:
> evcc does not recognize the tesla and uses the "gastfahrzeug".

Machen denn die Tronity Updates Sinn?

> Selecting my vehicle manually or restarting evcc pulls the right vehicle.

Das suggeriert, als würde da anderenfalls etwas nicht "gepulled"- was ist damit gemeint?

sebnaf, 2024-08-20:
> > evcc does not recognize the tesla and uses the "gastfahrzeug".
> 
> Machen denn die Tronity Updates Sinn?

Ja, in Tronity wird der Status selber korrekt angezeigt.

> 
> > Selecting my vehicle manually or restarting evcc pulls the right vehicle.
> 
> Das suggeriert, als würde da anderenfalls etwas nicht "gepulled"- was ist damit gemeint?

Wenn ich den Tesla ans Ladekabel anschließe, wird er in Tronity selber als "eingesteckt" angezeigt, sowie auch in der Tesla-App. evcc erkennt diesen Zustand nicht und bleibt auf "Gastfahrzeug" stehen.
Wechsle ich manuell auf meinen Tesla, wird alles sofort richtig erkannt.
Starte ich in obigem Zustand evcc neu, wird nach dem Neustart sofort alles richtig erkannt.

andig, 2024-08-20:
Und was zeigt das Log als Tronity Status?

sebnaf, 2024-08-20:
Hast Du einen Pointer, wonach ich im Log suchen kann?
Ist es: [main  ] DEBUG 2024/08/19 18:03:06 vehicle status: A (EM-IL 55E) ?

sebnaf, 2024-08-20:
Der Charger erkennt das Anschließen:

> [lp-1  ] DEBUG 2024/08/19 18:00:26 charger status: A
> [lp-1  ] DEBUG 2024/08/19 18:00:46 charger status: A
> [lp-1  ] DEBUG 2024/08/19 18:01:06 charger status: A
> [lp-1  ] DEBUG 2024/08/19 18:01:26 charger status: A
> [lp-1  ] DEBUG 2024/08/19 18:01:46 charger status: B
> [lp-1  ] DEBUG 2024/08/19 18:02:06 charger status: B

Charger erkennt B und fragt Tronity, die antworten, in evcc bleibt es aber "Gastfahrzeug"

```
[tronity] TRACE 2024/08/19 18:01:46 POST https://api.tronity.tech/authentication
[influx] TRACE 2024/08/19 18:01:46 write chargeCurrent=map[value:0] (map[loadpoint:Elektronenpumpe])
[influx] TRACE 2024/08/19 18:01:46 write sessionEnergy=map[value:2136.000000000422] (map[loadpoint:Elektronenpumpe])
[influx] TRACE 2024/08/19 18:01:46 write sessionSolarPercentage=map[value:70.25184337171713] (map[loadpoint:Elektronenpumpe])
[influx] TRACE 2024/08/19 18:01:46 write sessionPricePerKWh=map[value:0.1330140859809751] (map[loadpoint:Elektronenpumpe])
[influx] TRACE 2024/08/19 18:01:46 write sessionPrice=map[value:0.28411808765541896] (map[loadpoint:Elektronenpumpe])
[influx] TRACE 2024/08/19 18:01:46 write chargeTotalImport=map[value:7155.987] (map[loadpoint:Elektronenpumpe])
[pushover] DEBUG 2024/08/19 18:01:46 sending to ***
[influx] TRACE 2024/08/19 18:01:46 write effectivePriority=map[value:0] (map[loadpoint:Elektronenpumpe])
[influx] TRACE 2024/08/19 18:01:46 write effectivePlanSoc=map[value:0] (map[loadpoint:Elektronenpumpe])
[influx] TRACE 2024/08/19 18:01:46 write effectiveMinCurrent=map[value:5] (map[loadpoint:Elektronenpumpe])
[influx] TRACE 2024/08/19 18:01:46 write effectiveMaxCurrent=map[value:16] (map[loadpoint:Elektronenpumpe])
[influx] TRACE 2024/08/19 18:01:46 write effectiveLimitSoc=map[value:100] (map[loadpoint:Elektronenpumpe])
[influx] TRACE 2024/08/19 18:01:46 write sessionEnergy=map[value:0] (map[loadpoint:Elektronenpumpe])
[influx] TRACE 2024/08/19 18:01:46 write sessionSolarPercentage=map[value:0] (map[loadpoint:Elektronenpumpe])
[influx] TRACE 2024/08/19 18:01:46 write chargedEnergy=map[value:0] (map[loadpoint:Elektronenpumpe])
[influx] TRACE 2024/08/19 18:01:46 write chargeCurrent=map[value:0] (map[loadpoint:Elektronenpumpe])
[tronity] TRACE 2024/08/19 18:01:47 GET https://api.tronity.tech/tronity/vehicles/668aed<snip>4ce4/last_record
[tronity] TRACE 2024/08/19 18:01:47 **{"odometer":33500.2,"range":193,"level":43.45,"charging":"Disconnected","plugged":false,"chargerPower":0,"chargeRemainingTime":null,"latitude":48.124068,"longitude":7.844355,"status":null,"timestamp":1724083291000,"lastUpdate":1724083291000}**
[main  ] DEBUG 2024/08/19 18:01:47 vehicle status: A (EM-IL 55E)
[lp-1  ] DEBUG 2024/08/19 18:01:47 pv charge current: 1.1A = 0A + 1.1A (-253W @ 1p)
[lp-1  ] DEBUG 2024/08/19 18:01:47 pv enable timer reset
[lp-1  ] DEBUG 2024/08/19 18:01:47 pv timer inactive
[influx] TRACE 2024/08/19 18:01:47 write greenShareHome=map[value:1] (map[])
[influx] TRACE 2024/08/19 18:01:47 write greenShareLoadpoints=map[value:1] (map[])
[influx] TRACE 2024/08/19 18:01:47 write tariffPriceHome=map[value:0.065] (map[])
[influx] TRACE 2024/08/19 18:01:47 write tariffPriceLoadpoints=map[value:0.065] (map[])
```

andig, 2024-08-20:
Soweit ich sehe schickt Tronity da disconnected?

sebnaf, 2024-08-20:
Danke hier für den Pointer, `"lastUpdate":1724085884000` erzählt es: die Tesla-API war (mal wieder…) mehrfach offline.
Ich glaube ich bau mir mal nen Statusboard für die…

Ich werde mal schauen, ob evcc den Wechsel zu ´connected´ dann richtig erkennt.

closed 2024-08-20 by sebnaf