evcc-io/evcc #15504: 0.130 OCPP Fronius Wattpilot

mecstroeken, 2024-08-19:
### Describe the bug

EVCC startet nicht mehr - im Protokoll steht:

starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2024/08/19 18:48:11 evcc 0.130.0
[main  ] INFO 2024/08/19 18:48:11 using config file: /config/evcc.yaml
[db    ] INFO 2024/08/19 18:48:11 using sqlite database: /data/evcc.db
[mqtt  ] INFO 2024/08/19 18:48:12 connecting evcc-1655328720 at tcp://192.168.178.100:1883
[mqtt  ] DEBUG 2024/08/19 18:48:12 tcp://192.168.178.100:1883 connected
[main  ] INFO 2024/08/19 18:48:12 listening at :7070
[91046150-1] DEBUG 2024/08/19 18:48:14 waiting for chargepoint: 5m0s

### Steps to reproduce

1. Konfiguration in der Wattpilot-App für OCPP - Status ist "verbunden"
2. in HA läuft die OCPP-Integration sauber und zeigt alle Werte der Wallbox an
3.
...

### Configuration details

```yaml
Chargers:
    - name: my_charger
    type: template
    template: ocpp-fronius-wattpilot
    stationid: 91046150 # Station ID des Ladepunktes. Nur erforderlich wenn mehrere OCPP-Ladestationen eingerichtet sind um diese korrekt zuzuweisen. Eine einzelne OCPP-Ladestation kann auch automatisch zugeordnet werden. Hinweis: In Ausnahmefällen kann es erforderlich sein, diese ID manuell an die OCPP-URL der Ladestation in der Form ws://<evcc-adresse>:8887/<stationid> anzuhängen. Die meisten Ladestationen fügen die ID intern automatisch hinzu. (optional)
    # connector: 1 # Anschlussnummer bei Ladestationen mit mehreren Anschlüssen. Die Zählung beginnt bei 1 für den ersten Anschluss. (optional)
    # remotestart: # Diese Option nur aktivieren wenn keinerlei Möglichkeit besteht Transaktionen seitens des Ladepunktes zu initiieren! Das ist nur der Fall wenn z. B. kein RFID-Lesegerät vorhanden ist und Ladevorgänge grundsätzlich einzeln per App freigeschaltet werden müssten. Normalerweise sollte der Ladepunkt am Gerät immer so konfiguriert werden, dass entweder eine RFID-Karte zur Freischaltung verwendet wird oder der Ladepunkt auf "Autostart", "Freies Laden" o.ä. eingestellt ist. Zunächst die Dokumentation und die Konfigurationsmöglichkeiten des Ladepunktes prüfen, ggf. beim Hersteller nachfragen! (optional)
    # idtag: evcc # Diese Option ist nur in Ausnahmefällen erforderlich wenn der Ladepunkt für die Annahme externer Transaktionen einen spezifischen Token erfordert. (optional)
    connecttimeout: 5m # optional
    timeout: 30s # optional
    meterinterval: 10s # optional
    # metervalues: Energy.Active.Import.Register,Power.Active.Import,SoC,Current.Offered,Power.Offered,Current.Import,Voltage # optional
```

### Log details

```text
siehe oben
Der Chargepoint wird wohl nicht erkannt
Wenn ich zurück auf die ursprüngliche Konfiguration gehe, dann funktioniert es:

  # - name: my_charger
    # type: template
    # template: fronius-wattpilot
    # host: 192.168.178.94 # IP-Adresse oder Hostname
    # password: 'xxx'  # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen) (optional)
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.130

andig, 2024-08-19:
Es braucht bitte ein trace log der ocpp Kommunikation

premultiply, 2024-08-19:
Und reduziere die Konfig mal bitte nur auf

```yaml
chargers:
  - name: my_charger
    type: template
    template: ocpp-fronius-wattpilot
```

mecstroeken, 2024-08-19:
Ok. Kann ich aber erst morgen Abend probieren. Wie funktioniert die Log-Ausgabe, habe evcc unter HA als AddOn laufen?Soll ich die OPCC-Integration in HA deaktivieren, um diese als Fehlerquelle auszuschließen?Beste Grüße

premultiply, 2024-08-20:
Einfach ganz normal auf deinem System ausführen.
Windows, macOS, Linux, ...

mecstroeken, 2024-08-20:
[evcc-20240820-203824-trace.log](https://github.com/user-attachments/files/16680391/evcc-20240820-203824-trace.log)

premultiply, 2024-08-22:
> es funktioniert mit der "reduzierten" yaml-Konfig

Prima, dann mache ich hier mal zu.

closed 2024-08-22 by premultiply