evcc-io/evcc #15505: battery grid charge not visible in Desktop mode but visible on mobile phone (Home Assistant)

Presl1983, 2024-08-19:
### Describe the bug

battery grid charge not visible in Desktop mode but visible on mobile phone (Home Assistant)

### Steps to reproduce

1. Open on pc not visible
[image]
3. Open on mobile phone grid charge is visible

### Configuration details

```yaml
- name: my_battery
    type: template
    template: kostal-plenticore
    usage: battery

    modbus: tcpip
    id: 71
    host: 192.168.178.88 # Hostname
    port: 1502 # Port
    endianness: little # optional
    capacity: 7.7 # Akkukapazität in kWh (optional)
    minsoc: 25 # Ladung mit maximaler Geschwindigkeit bis zu dem angegeben Ladestand unabhängig PV-Erzeugung, wenn der Lademodus nicht auf 'Aus' steht (optional)
    maxsoc: # optional
    watchdog: 60s # optional
```

### Log details

```text
DEBUG
Alle Bereiche
[main ] INFO 2024/08/19 18:58:08 evcc 0.130.0
[main ] INFO 2024/08/19 18:58:08 using config file: /config/evcc.yaml
[db ] INFO 2024/08/19 18:58:08 using sqlite database: /data/evcc.db
[mqtt ] INFO 2024/08/19 18:58:08 connecting evcc-758460162 at tcp://192.168.178.128:1883
[mqtt ] DEBUG 2024/08/19 18:58:08 tcp://192.168.178.128:1883 connected
[main ] INFO 2024/08/19 18:58:08 listening at :7070
[tesla ] INFO 2024/08/19 18:58:10 !! tesla proxy for Lex Loreny: https://tesla.evcc.io/
[site ] WARN 2024/08/19 18:58:10 battery configured but residualPower is missing or <= 0 (add residualPower: 100 to site), see https://docs.evcc.io/en/docs/reference/configuration/site#residualpower
[site ] INFO 2024/08/19 18:58:12 site config:
[site ] INFO 2024/08/19 18:58:12 meters: grid ✓ pv ✓ battery ✓
[site ] INFO 2024/08/19 18:58:12 grid: power ✓ energy ✗ currents ✗
[site ] INFO 2024/08/19 18:58:12 pv 1: power ✓ energy ✓ currents ✗
[site ] INFO 2024/08/19 18:58:12 battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✓
[site ] INFO 2024/08/19 18:58:12 vehicles:
[site ] INFO 2024/08/19 18:58:12 vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✓
[lp-1 ] INFO 2024/08/19 18:58:12 loadpoint 1:
[lp-1 ] INFO 2024/08/19 18:58:12 mode: off
[lp-1 ] INFO 2024/08/19 18:58:12 charger: power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1 ] INFO 2024/08/19 18:58:12 meters: charge ✓
[lp-1 ] INFO 2024/08/19 18:58:12 charge: power ✓ energy ✗ currents ✗
[site ] DEBUG 2024/08/19 18:58:12 set buffer soc: 100
[site ] DEBUG 2024/08/19 18:58:12 set buffer start soc: 100
[site ] DEBUG 2024/08/19 18:58:12 set priority soc: 100
[site ] DEBUG 2024/08/19 18:58:12 set battery discharge control: true
[lp-1 ] DEBUG 2024/08/19 18:58:12 phase timer inactive
[lp-1 ] DEBUG 2024/08/19 18:58:12 pv timer inactive
[site ] DEBUG 2024/08/19 18:58:12 ----
[lp-1 ] DEBUG 2024/08/19 18:58:12 charge power: -8W
[site ] DEBUG 2024/08/19 18:58:12 pv power: 142W
[site ] DEBUG 2024/08/19 18:58:13 battery soc: 14%
[site ] DEBUG 2024/08/19 18:58:13 battery power: 410W
[site ] DEBUG 2024/08/19 18:58:13 grid meter: 19W
[site ] DEBUG 2024/08/19 18:58:13 site power: 429W
[lp-1 ] DEBUG 2024/08/19 18:58:13 charger status: B
[lp-1 ] INFO 2024/08/19 18:58:13 car connected
[lp-1 ] DEBUG 2024/08/19 18:58:13 vehicle api refresh
[... 549 lines cut ...]
[lp-1 ] DEBUG 2024/08/19 19:27:12 charge power: 472W
[site ] DEBUG 2024/08/19 19:27:12 pv power: 28W
[site ] DEBUG 2024/08/19 19:27:13 battery soc: 11%
[site ] DEBUG 2024/08/19 19:27:13 battery power: 11W
[site ] DEBUG 2024/08/19 19:27:13 grid meter: 472W
[site ] DEBUG 2024/08/19 19:27:13 site power: 483W
[lp-1 ] DEBUG 2024/08/19 19:27:13 charger status: B
[lp-1 ] DEBUG 2024/08/19 19:27:13 pv charge current: 0A = 0A + -2.1A (483W @ 1p)
[site ] DEBUG 2024/08/19 19:27:17 set grid charge limit: 0.0
[site ] DEBUG 2024/08/19 19:27:22 set grid charge limit: -0.1
[site ] DEBUG 2024/08/19 19:27:42 ----
[lp-1 ] DEBUG 2024/08/19 19:27:42 charge power: 465W
[site ] DEBUG 2024/08/19 19:27:42 set battery mode: normal
[site ] DEBUG 2024/08/19 19:27:42 pv power: 27W
[site ] DEBUG 2024/08/19 19:27:43 battery soc: 11%
[site ] DEBUG 2024/08/19 19:27:43 battery power: 11W
[site ] DEBUG 2024/08/19 19:27:43 grid meter: 472W
[site ] DEBUG 2024/08/19 19:27:43 site power: 483W
[lp-1 ] DEBUG 2024/08/19 19:27:43 charger status: B
[lp-1 ] DEBUG 2024/08/19 19:27:43 pv charge current: 0A = 0A + -2.1A (483W @ 1p)
```

### What type of operating system are you running?

HomeAssistant Add-on

andig, 2024-08-19:
> but visible on mobile phone (Home Assistant)

Wo sieht man das?

Presl1983, 2024-08-19:
[image]

Presl1983, 2024-08-19:
[image]

wies scheint hat ein restart von HA jetzt das Problem behoben - verstehe ich aber nicht wirklich muss ich gestehen...

closed 2024-08-19 by Presl1983

naltatis, 2024-08-19:
Experimental hast to be enabled, you must have a dynamic grid price and a compatible battery system.