evcc-io/evcc #15506: Aktive Batteriesteuerung für LG ESS home 8/10 hinzufügen

Matthias13MUC, 2024-08-19:
**Is your feature request related to a problem? Please describe.**
Die Entladesperre (aktive Batteriesteuerung) ist in EVCC für das System **LG ESS home 8/10** nicht "freigeschalten" und wäre ein hilfreiches Feature, um den Batteriespeicher nicht vollständig zu entleeren, insbesondere, wenn mann zwei unterschiedliche Tarife (Hausstrom, WP-Strom als Kaskade im gleichen Stromkreis hat. Im Winter reicht der PV-Strom und Speicher für eine WP nicht aus. Vielen Dank.

premultiply, 2024-08-19:
Dann einfach die die notwendige Doku hier einfügen wie das bei diesem System funktioniert.
Idealerweise entsprechend vorgetestet.

Bis dahin mache ich hier mal zu.

andig, 2024-08-19:
Anders gesagt: es ist nicht klar, wie das bei diesem Speicher gehen soll. Also gerne den Support mit entsprechenden Fragen nerven!

closed 2024-08-19 by andig