evcc-io/evcc #15507: evcc 0.130.0 not starting anymore since today 19.08.2024

Sensay2k, 2024-08-19:
### Describe the bug

Using Home Assistant 
Core 2024.8.2
Supervisor 2024.08.0
Operating System 13.0
Frontend
20240809.0 ⸱ legacy

evcc not starting anymore, did not change any hardware or configuration. Then I updated Home Asssistant and FEMS from Fenecon to most actual versions, but error persists. FEMS separately working properly.

Error message evcc:

add-on läuft nicht. Möchtest du es jetzt starten?

After confirmation, starting is not possible. Thanks in advance for any help!

### Steps to reproduce

1. Try to start evcc, always same error
[image]
[image]

2.
3.
...

### Configuration details

```yaml
Cannot copy & paste from file editor (iOS on MAC), adding PNG ccreenshots to show
```

### Log details

```text
github.com/spf13/cobra@v1.8.1/command.go:1117 +0x344
github.com/spf13/cobra.(*Command).Execute(...)
	github.com/spf13/cobra@v1.8.1/command.go:1041
github.com/evcc-io/evcc/cmd.Execute()
	github.com/evcc-io/evcc/cmd/root.go:107 +0x24
main.main()
	github.com/evcc-io/evcc/main.go:44 +0x38
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2024/08/19 19:48:56 evcc 0.130.0
[main  ] INFO 2024/08/19 19:48:56 using config file: /config/evcc.yaml
[db    ] INFO 2024/08/19 19:48:56 using sqlite database: /data/evcc.db
[main  ] INFO 2024/08/19 19:48:57 listening at :7070
[main  ] ERROR 2024/08/19 19:48:57 creating vehicle Mini failed: cannot create vehicle type 'template': cannot create vehicle type 'mini': unexpected status: 429 (Too Many Requests)
[main  ] ERROR 2024/08/19 19:48:58 creating vehicle ev5 failed: cannot create vehicle type 'template': cannot create vehicle type 'etron': 429 Too Many Requests
[site  ] WARN 2024/08/19 19:48:58 battery configured but residualPower is missing or <= 0 (add residualPower: 100 to site), see https://docs.evcc.io/en/docs/reference/configuration/site#residualpower
[site  ] WARN 2024/08/19 19:48:58 vehicle 'e-tron' does not support automatic detection
[site  ] WARN 2024/08/19 19:48:58 vehicle 'MINI Cooper SE' does not support automatic detection
[site  ] INFO 2024/08/19 19:48:58 site config:
[site  ] INFO 2024/08/19 19:48:58   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2024/08/19 19:48:58     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2024/08/19 19:48:58     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2024/08/19 19:48:58     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2024/08/19 19:48:58   vehicles:
[site  ] INFO 2024/08/19 19:48:58     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[site  ] INFO 2024/08/19 19:48:58     vehicle 2: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2024/08/19 19:48:58 loadpoint 1:
[lp-1  ] INFO 2024/08/19 19:48:58   mode:        pv
[lp-1  ] INFO 2024/08/19 19:48:58   charger:     power ✗ energy ✗ currents ✗ phases ✗ wakeup ✗
[lp-1  ] INFO 2024/08/19 19:48:58   meters:      charge ✗
[site  ] DEBUG 2024/08/19 19:48:58 set buffer soc: 80
[site  ] DEBUG 2024/08/19 19:48:58 set buffer start soc: 100
[site  ] DEBUG 2024/08/19 19:48:58 set priority soc: 80
[lp-1  ] DEBUG 2024/08/19 19:48:58 phase timer inactive
[lp-1  ] DEBUG 2024/08/19 19:48:58 pv timer inactive
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x20 pc=0x21ee570]

goroutine 1 [running]:
github.com/evcc-io/evcc/core.(*Loadpoint).Prepare(0x400150bc08, 0x0?, 0x57b8860?, 0x4001ab81b0?)
	github.com/evcc-io/evcc/core/loadpoint.go:651 +0xb90
github.com/evcc-io/evcc/core.(*Site).Prepare(0x400112a800, 0x4000562d20, 0x40005bcbd0)
	github.com/evcc-io/evcc/core/site.go:968 +0x15c
github.com/evcc-io/evcc/cmd.runRoot(0x57569a0, {0x2a670ca?, 0x7?, 0x2a5a098?})
	github.com/evcc-io/evcc/cmd/root.go:283 +0x191c
github.com/spf13/cobra.(*Command).execute(0x57569a0, {0x400012c1c0, 0x2, 0x2})
	github.com/spf13/cobra@v1.8.1/command.go:989 +0x81c
github.com/spf13/cobra.(*Command).ExecuteC(0x57569a0)
	github.com/spf13/cobra@v1.8.1/command.go:1117 +0x344
github.com/spf13/cobra.(*Command).Execute(...)
	github.com/spf13/cobra@v1.8.1/command.go:1041
github.com/evcc-io/evcc/cmd.Execute()
	github.com/evcc-io/evcc/cmd/root.go:107 +0x24
main.main()
	github.com/evcc-io/evcc/main.go:44 +0x38
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2024/08/19 19:57:00 evcc 0.130.0
[main  ] INFO 2024/08/19 19:57:00 using config file: /config/evcc.yaml
[db    ] INFO 2024/08/19 19:57:00 using sqlite database: /data/evcc.db
[main  ] INFO 2024/08/19 19:57:01 listening at :7070
[main  ] ERROR 2024/08/19 19:57:01 creating vehicle Mini failed: cannot create vehicle type 'template': cannot create vehicle type 'mini': unexpected status: 429 (Too Many Requests)
[site  ] WARN 2024/08/19 19:57:03 battery configured but residualPower is missing or <= 0 (add residualPower: 100 to site), see https://docs.evcc.io/en/docs/reference/configuration/site#residualpower
[site  ] WARN 2024/08/19 19:57:03 vehicle 'MINI Cooper SE' does not support automatic detection
[site  ] INFO 2024/08/19 19:57:03 site config:
[site  ] INFO 2024/08/19 19:57:03   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2024/08/19 19:57:03     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2024/08/19 19:57:03     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2024/08/19 19:57:03     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2024/08/19 19:57:03   vehicles:
[site  ] INFO 2024/08/19 19:57:03     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[site  ] INFO 2024/08/19 19:57:03     vehicle 2: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2024/08/19 19:57:03 loadpoint 1:
[lp-1  ] INFO 2024/08/19 19:57:03   mode:        pv
[lp-1  ] INFO 2024/08/19 19:57:03   charger:     power ✗ energy ✗ currents ✗ phases ✗ wakeup ✗
[lp-1  ] INFO 2024/08/19 19:57:03   meters:      charge ✗
[site  ] DEBUG 2024/08/19 19:57:03 set buffer soc: 80
[site  ] DEBUG 2024/08/19 19:57:03 set buffer start soc: 100
[site  ] DEBUG 2024/08/19 19:57:03 set priority soc: 80
[lp-1  ] DEBUG 2024/08/19 19:57:03 phase timer inactive
[lp-1  ] DEBUG 2024/08/19 19:57:03 pv timer inactive
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x20 pc=0x21ee570]

goroutine 1 [running]:
github.com/evcc-io/evcc/core.(*Loadpoint).Prepare(0x40012f4408, 0x0?, 0x57b8860?, 0x40017db1a0?)
	github.com/evcc-io/evcc/core/loadpoint.go:651 +0xb90
github.com/evcc-io/evcc/core.(*Site).Prepare(0x4000538c00, 0x4000654230, 0x40005c76c0)
	github.com/evcc-io/evcc/core/site.go:968 +0x15c
github.com/evcc-io/evcc/cmd.runRoot(0x57569a0, {0x2a670ca?, 0x7?, 0x2a5a098?})
	github.com/evcc-io/evcc/cmd/root.go:283 +0x191c
github.com/spf13/cobra.(*Command).execute(0x57569a0, {0x400012c1c0, 0x2, 0x2})
	github.com/spf13/cobra@v1.8.1/command.go:989 +0x81c
github.com/spf13/cobra.(*Command).ExecuteC(0x57569a0)
	github.com/spf13/cobra@v1.8.1/command.go:1117 +0x344
github.com/spf13/cobra.(*Command).Execute(...)
	github.com/spf13/cobra@v1.8.1/command.go:1041
github.com/evcc-io/evcc/cmd.Execute()
	github.com/evcc-io/evcc/cmd/root.go:107 +0x24
main.main()
	github.com/evcc-io/evcc/main.go:44 +0x38
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.130.0

closed 2024-08-19 by andig

Sensay2k, 2024-08-20:
Hi, ich konnte den Querverweis zu dem Ursprungsproblem nicht sehen, aber ähnliche Probleme hier bei Github gefunden. Habe nun bemerkt, dass es ein Update gibt. Dieses eingespielt, aber das Proplem besteht weiterhin, hier die leicht geänderte panic-Message:

panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x20 pc=0x21ee2b0]

Sensay2k, 2024-08-20:
Wäre super wenn ihr den Thread doch nochmal öffnet :)

dakky, 2024-08-20:
wäre schon nett, wenn hier ne referenz auf das duplicate gepostet worden wäre. und ja: hab dasselbe problem