evcc-io/evcc #15508: Netzladen der Hausbatterie beim Kostal Plenticore ohne Funktion

StevieC121176, 2024-08-19:
### Describe the bug

Ich habe zum testen mal die CO2 Grenze für das Netzladen der Hausbatterie auf <460g eingestellt.
Es wird im UI auch saubere Netzenergie angezeigt. Aber anstatt die Hausbatterie zu laden wird nur die Entladung der Batterie gesperrt und die Differenz zwischen noch verfügbarer PV Energie und Hausverbrauch aus dem Netz beglichen.
externe Batteriesteuerung über Modus sind im Kostal Plenticore eingestellt.

[evcc-20240819-201648-trace.log](https://github.com/user-attachments/files/16664056/evcc-20240819-201648-trace.log)

### Steps to reproduce

1. CO2 Grenze hoch einstellen (in meinem Fall war es 460g)

[image]
[image]

### Configuration details

```yaml
über EVCC UI konfiguriert
```

### Log details

```text
logfile sh. oben
```

### What type of operating system are you running?

Linux

### Version

0.130

Presl1983, 2024-08-19:
gleiches Problem habe ich auch!

deadrabbit87, 2024-08-19:
Kann auch nicht funktionieren, weil keines der Register von evcc beschrieben wird, die dazu notwendig wären.

Damit es geht, könnte man auf Register `1028` `-100` schreiben. 

Dann beginnt der WR mit seiner technisch max.  möglichen Leistung die Batterie zu laden. 

@premultiply Gibt es best-practice Beispiele wo das Netzladen umgesetzt ist und auch funktioniert?

https://cdn-production.kostal.com/-/media/document-library-folder---kse/2023/11/16/13/29/ba_kostal-interface-description-modbus-tcp_sunspec_hybrid.pdf?rev=-1&hash=6BEE952BED4A8A48ECE11BC37B369F06

andig, 2024-08-19:
@deadrabbit87 aktuell verwenden wir 1042=100(%). Dazu kommt dann der Watchdog.

deadrabbit87, 2024-08-19:
`1042` setzt aber nur den minsoc auf 100% und sperrt somit nur die Entladung beim Zielladen/ Schnellladen.

Denkbar wäre `1028` beim Zielladen mit 0 zu beschreiben und beim Batterienetzladen mit `-100`.

Watchdog braucht es, weil bei ausbleiben von Werten der WR auf interne Regelung zurückfällt.

`1042` wäre dann obsolet.

Chris45671, 2024-08-19:
Bei mir logischer Weise ebenfalls so. 
Wenn es eine spätere Lösung geben wird, bin ich gerne bereit zum Testen. 
Habe zwei von den betroffenen WR im Verbund am laufen und nur an einem hängt eine Batterie.

Presl1983, 2024-08-19:
ich habe das ganze im HA so konfiguriert:

hub: Kostal-BYD
address: 1034
unit: 71
value:
  - 0
  - 50528
falls das hilft...

deadrabbit87, 2024-08-19:
> ich habe das ganze im HA so konfiguriert und muss direkt das script alle 45 sekunden ausführen...
> 
> hub: Kostal-BYD
> address: 1034
> unit: 71
> value:
>   - 0
>   - 50528
> falls das hilft...

Macht das gleiche wie `1028` nur in Absolutwerten was aber unterschiedlich je nach Batterie ist.

deadrabbit87, 2024-08-23:
Mir ist bei meinem Vorschlag noch ein Fehler aufgefallen: Für `mode: normal` weiß ich nicht, wie man das erreichen könnte. 

Außer so lange nichts auf `1028` zu schreiben, bis der WR wieder von selbst auf die interne Regelung zurückfällt. Das gleiche Problem betrifft `1034` auch. 

@Presl1983 Wie erreichst du, dass der WR wieder selbst regelt?

Presl1983, 2024-08-24:
> Mir ist bei meinem Vorschlag noch ein Fehler aufgefallen: Für `mode: normal` weiß ich nicht, wie man das erreichen könnte.
> 
> Sobald ja etwas auf `1028` geschrieben wird, ist die interne Regelung deaktiviert und ich weiß nicht, wie man die wieder einschalten kann.
> 
> Außer so lange nichts auf `1028` zu schreiben, bis der WR wieder von selbst auf die interne Regelung zurückfällt. Das gleiche Problem betrifft `1034` auch.
> 
> @Presl1983 Wie erreichst du, dass der WR wieder selbst regelt?

Da sich der WR automatisch wieder retour stellt sobald er kein commands mehr bekommt muss man da nichts machen.

Ich habe gestern mal eine Kalkulation aufgestellt mit den stündlichen Tarifen.... Recht viel spart man sich da aktuell gar nicht mehr mit Akku pv und E-Auto...

iseeberg79, 2024-08-25:
Hallo,
mich hat das Thema auch sehr interessiert und die Funktion Netzladen seit langem verfolgt. Da ich auch einen Kostal-WR mit BYD Batterie betreibe, habe ich mit den Anmerkungen und den neuen Funktionen versucht, die Funktion auch für den Kostal-WR zu ermöglichen.

In den Templates für Kostal, SMA und Fronius bin ich fündig geworden und habe das aktuell in einem Custom-Type für die Batterie am Kostal-Plenticore WR umgesetzt. Aktuell habe ich auch ein Ladelimit (maxSoc) für die Netzladenfunktion mit eingestellt. Vielleicht ist das für das Kostal-Template und die aktuelle Diskussion eine Anregung? Ich teile hier mal die Konfiguration im Anhang. 

[DOKU_EVCC_KOSTAL_WR_GRIDLOAD.pdf](https://github.com/user-attachments/files/16741003/DOKU_EVCC_KOSTAL_WR_GRIDLOAD.pdf)

iseeberg79, 2024-08-25:
Finde ich auch nicht schön, es scheint aber auch in anderen Lösungen keine andere Option zu geben? Wäre praktisch, wenn nicht nur ein Timer ablaufen würde, sondern auch aktiv die interne Steuerung via Modbus wieder aktivierbar wäre.

deadrabbit87, 2024-08-25:
Ich kenne leider auch keine andere Lösung. 

Eine Frage zu deiner Umsetzung hätte ich noch. Du nutzt zum sperren der Batterie nach wie vor `1042` um den minsoc auf 100% zu setzen. 

Das wird aber nicht mehr zurück gesetzt. Mich wundert, dass das so funktioniert. Nach hold müsste doch ´1042` wieder zurückgesetzt werden?

iseeberg79, 2024-08-25:
Ja, das hab ich auch gedacht und zuerst so implementiert und dann stückweise auf das Notwendige reduziert. Die Entladung beginnt nach dem Ende des Watchdogs/Timers ohne zurücksetzen des Registers. Vermutlich weil die interne Regelung auf einen anderen internen Wert des Wechselrichters (minSoC auf WR: /de-de/#/service/battery ) zurückfällt?
Etwas anderes ist mir bei einem neuen Test jedoch aufgefallen: der case 1 darf nicht leer sein? EVCC reagiert sonst im Log mit einem Fehler und die Entladungssperre im UI bleibt lange hängen.
Ich schreibe da zur Lösung jetzt den minSoC hinein, mit dem Nebeneffekt das dann die Entladung direkter wieder fortgesetzt wird und das UI ohne Logeintrag aktuell ist:

(..)
```
        - case: 1 # normal
          set:
            source: const
            value: 5 # % (dummy operation, reset minSoC
            set:
              source: modbus
              id: 71
              uri: localhost:1502
              register:
                address: 1042 # limit soc (min)
                type: writemultiple
                encoding: float32s
```
(..)

deadrabbit87, 2024-08-25:
Und warum da nicht auch gleich Register `1028` mit 0 beschreiben?

Dann sollte er die Entladung ja auch stoppen.

iseeberg79, 2024-08-25:
In case 2? 
Ja, das geht auch. Setzt man den 1028 auf 0 ist die Batterieentladung auch gesperrt. Dann braucht es, wie oben schon diskutiert, das andere Register 1042 nicht. Gestern wollte ich noch nah am Orginal bleiben und primär die Netzladung hinzufügen...

Ich habe Register 1028 ausprobiert und Schnellladung mit Sperre, sowie Netzladung funktionieren - inkl. die Rückkehr in den Normalbetrieb.

Für den Normalbetrieb eignet sich dann auch in case 1 das Schreiben von 0 in 1028, so schwingt die Batterieladung bei Rückkehr aus der Netzladung (case 3) aus, und die Zeit für den Watchdog/Timeout ist gut genutzt?

Man würde dann nur noch Register 1028 beschreiben müssen, ggf. ergänzt durch eine konfigurierbare Maximalladung für den Netzladebetrieb (1044).

```
        - case: 1 # normal
          set:
            source: const
            value: 0 # % (reset grid loading)
            set:
              source: modbus
              id: 71
              uri: localhost:1502
              register:
                address: 1028 # Battery charge current (DC) setpoint, relative
                type: writemultiple
                encoding: float32s
        - case: 2 # hold/now
          set:
            source: const
            value: 0 # % (lock battery)
            set:
              source: modbus
              id: 71
              uri: localhost:1502
              register:
                address: 1028 # Battery charge current (DC) setpoint, relative
                type: writemultiple
                encoding: float32s
```

deadrabbit87, 2024-08-26:
Ja, meinte ich. 

Warum machst du keinen PR draus?

stna1981, 2024-08-26:
@StevieC121176 Hast du einen Plenticore G1 oder G2? Und auch in Verbindung mit dem KSEM?

StevieC121176, 2024-08-26:
> @StevieC121176 Hast du einen Plenticore G1 oder G2? Und auch in Verbindung mit dem KSEM?
> 
> Ich frage daher, weil bei meinem Plenticore G1 das setzen des minSOC bzw. Sperren des Speichers gegen Entladen nicht funktioniert. Mich würde daher zum einen interessieren, ob das nur bei mir so ist und andere mit gleichem Setup das Problem nicht haben und zum anderen, welche Settings im KSEM für die externe Modbus-Steuerung verwendet wurden.

Ich habe 2 Plenticore G1 im Verbund, eine BYD Batterie und den KSEM. Sperren der Batterie funktioniert , nur halt das Laden aus dem Netz noch nicht.

stna1981, 2024-08-26:
Wie sehen denn bei dir die Einstellungen im KSEM für die Modbus-Steuerung aus? Ich verstehe nicht, warum die minSOC-Steuerung bei mir plötzlich aufgehört hat, zu funktionieren. Du hast auch die aktuelle FW von Ende Februar auf den beiden Plenticores?

StevieC121176, 2024-08-26:
Sind die standard Einstellungen. 
Firmware ist überall die aktuellste drauf.

[image]

stna1981, 2024-08-26:
Und bei Modbus TCP hast du Slave aktiviert und als IP die des KSEM und als Port die 502 eingetragen?

StevieC121176, 2024-08-26:
bei mir ist nur Slave aktiviert, keine ip Adresse oder Port eingetragen.

stna1981, 2024-08-26:
Kann es vielleicht sein, dass es gar nicht am Plenicore, sondern am KSEM liegt? Ich habe nämlich schon das KSEM G2 bei mir installiert, ist es bei dir noch die G1?

StevieC121176, 2024-08-26:
ja, ich habe die G1 Version

stna1981, 2024-08-26:
Dann habe ich keine Ahnung, was es sein könnte. Wir haben beide einen Plenticore G1 mit aktueller FW, ein KSEM mit aktueller FW (müsste 2.4.2 sein oder?) und auch unsere Modbus Settings sind gleich. Bei dir klappt das Sperren, bei mir nicht. Das ist echt mies. Muss vielleicht sonst irgendwo im Plenticore oder im KSEM was aktiviert sein, damit das Setzen des minSOC von extern klappt?

Vielleicht hilft ja auch die Umstellung von Register 1042 auf 1028. 1028 war ja eh nur ein Workaround. Fingers crossed.

Chris45671, 2024-08-26:
Bei meinen Plenticore Plus wurde die externe Batteriesteuerung durch den Installateur eingestellt. Vielleicht ist dies die Bedingung damit es gehen kann?

stna1981, 2024-08-26:
Kann gut sein, das müsste der dann mal machen, wenn das nur über seinen Zugang geht. Was hat das denn alles für Auswirkungen, wenn man das umstellt, muss ich dann die Steuerung der Ladung komplett extern regeln oder macht er das dann immer noch selbständig?

Chris45671, 2024-08-26:
Da bin ich überfragt. Bei mir steuert extern eine GridX Box die grundlegenden Sachen was den Akku betreffen. Dies wurde uns von Eon aufgezwungen. 
Wie dies nur mit EVCC kann ich daher nicht sagen.

Swantu, 2024-08-26:
Hi. Möchte hier auch meinen Wunsch auf eine funktionierende Netzladung für den Kostal Plenticore anbringen - wie ich sehe bin ich damit auch nicht allein.
### Meine config
- evcc v0130.5 in docker
- meters in der yaml konfiguriert
### Das Problem
- der klick auf "Hausbatterie" führt zu keine Ergebnis (kommt kein popup)
- in den Logs steht`[site ] WARN 2024/08/26 21:37:00 battery configured but residualPower is missing or <= 0 (add residualPower: 100 to site), see https://docs.evcc.io/en/docs/reference/configuration/site#residualpower`
ABER: wenn ich residualpower: 100 zu "meters" hinzufüge, starter er nichtmehr `cannot create meter 'pv1': cannot create meter type 'template': invalid key: residualpower
`

Würde mich freuen, wenn es demnächste funktioniert. 
Bisher bin ich sehr happy mit der Entwicklung von EVCC 👍

iseeberg79, 2024-08-26:
@Chris45671 Soweit mir bekannt ist für das Schreiben der Register 1042 oder die anderen Register des externen Managements per Modbus das externe Management auch zu aktivieren (Installateurszugang). 
Man kann via HTTP auf dem WR auch per minSoC regeln, aber eben nicht per Modbus.
Sie Änderung auf 1028 funktioniert bei mir wg. des externen Managements und macht 1042 insoweit unnötig. Die Voraussetzung ist aber die Aktivierung des externen Management via Modbus.
Werden Werte geschrieben erwartet der Wechselrichter die Kontrolle durch die externe Steuerung. Das gilt solange, bis ein Timeout des WR abgelaufen ist, weil keine neuen Modbus Nachrichten für die Steuerung eingehen. Danach greift dann wieder die interne Steueung bis zum nächsten Schreiben von Modbus Registern des externen Managements.

stna1981, 2024-08-26:
Muss man den Wert für den Watchdog/Timeout in der evcc.yaml setzen und wenn ja, auf welchen Wert?

deadrabbit87, 2024-08-26:
https://github.com/evcc-io/evcc/pull/15715

Chris45671, 2024-08-26:
> Hi. Möchte hier auch meinen Wunsch auf eine funktionierende Netzladung für den Kostal Plenticore anbringen - wie ich sehe seit ihr schon fleißig dabei.
> ### Meine config
> 
>     * evcc v0130.5 in docker
> 
>     * meters in der yaml konfiguriert
> 
> 
> ### Das Problem
> 
>     * der klick auf "Hausbatterie" führt zu keine Ergebnis (kommt kein popup)
> 
>     * in den Logs steht`[site ] WARN 2024/08/26 21:37:00 battery configured but residualPower is missing or <= 0 (add residualPower: 100 to site), see https://docs.evcc.io/en/docs/reference/configuration/site#residualpower`
>       ABER: wenn ich residualpower: 100 zu "meters" hinzufüge, starter er nichtmehr `cannot create meter 'pv1': cannot create meter type 'template': invalid key: residualpower `
> 
> 
> Würde mich freuen, wenn es demnächste funktioniert. Bisher bin ich sehr happy mit der Entwicklung von EVCC 👍

Der Parameter muss unter Site eingrifflegt werden.

andig, 2024-09-01:
Ich hab den Überblick verloren. Gibt es hier noch ein konkretes Todo oder Fehlerbehebung? Welches?

deadrabbit87, 2024-09-01:
Es gibt hier einen PR:
https://github.com/evcc-io/evcc/pull/15709

Das einzig unschöne daran ist, das der `normal` oder `case 1` Zustand durch abwarten des timeouts im WR erreicht wird.

Presl1983, 2024-09-08:
jetzt muss ich kurz fragen nach dem evcc nicht mehr gestartet hat.

muss man min und max rausnehmen?

rolli09, 2024-09-13:
probiert mal die 0.130.9 ;)
bei mir läuft es :)

andig, 2024-09-14:
In Ermangelung eines Todo geschlossen.

closed 2024-09-14 by andig

stna1981, 2024-10-03:
> beachtet für die Funktion Netzladen den Wechsel auf ein anderes Template "kostal-plenticore-gen2" ab dem nächsten nightly, sowie ab 0.130.14 (PR #16414); aktuell mit Bedingung: nicht HW Version 0100 (~G1).

Und was ist, wenn man einen Plenticore G1 verwendet, kann man die Funktion dann nicht mehr nutzen?