evcc-io/evcc #15514: Huawei Batterie wird aus dem Netz geladen

jori0001, 2024-08-20:
### Describe the bug

Heute Morgen, als die Sonne noch nicht groß scheint, ist mir aufgefallen, dass seit der v0.130.0 meine Huawei Batterie aus dem Netz geladen wird, was mit der v0.131.0 auch noch so ist. Das soll natürlich so nicht sein, bei den aktuellen Preisen.
Denke, dass das noch ein Bug der neuen Funktion ist:
 
[image]

[image]

[evcc-20240820-070824-trace.log](https://github.com/user-attachments/files/16669277/evcc-20240820-070824-trace.log)

### Steps to reproduce

Einfach EVCC am Laufen haben, keine Sonne und nicht volle Batterie und schon wird aus dem Netz geladen.

### Configuration details

```yaml
hänge ich an, falls benötigt.
```

### Log details

```text
Trace File ist angehängt
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

v0.131.0

jori0001, 2024-08-20:
[evcc-20240820-070824-trace.log](https://github.com/user-attachments/files/16669429/evcc-20240820-070824-trace.log)

mrhund, 2024-08-20:
Haben auch grad mieses Wetter, Batterie heute morgen bei 6%. Das Verhalten kann ich aber nicht bestätigen - Batterie lädt nur mit PV Überschuss:

[image]

Habe 0.130.0 installiert.

jori0001, 2024-08-20:
Sehr seltsam. Bei mir wurde jetzt bis auf 100% aus dem Netz geladen. :(

AlexJoda, 2024-08-20:
Bei mir lädt die  Huawei Luna Batterie nur wenn die Preisgrenze oberhalb des aktuellen Preises ist , allerdings  nur mit 2,5 KW und nicht mit 5 KW wie sonst wenn man die Luna direkt lädt.

jori0001, 2024-08-20:
@AlexJoda 
Bei 2,5 kWh Laden hast du vermutlich nur ein 5 kWh Batterie-Modul.
Bleibt die Frage, warum lädt es bei mir mit 5 kWh aus dem Netz. Die Einstellungen sind im Screenshot zu sehen. Bin gespannt, ob jemand aus dem Team ne Idee hat.

AlexJoda, 2024-08-20:
@jori0001
Nein, hab ich nicht. Ich habe 3 Module mit insgesamt 15 kWh. Wenn ich die über die FusionSalar App lade (Zeitsteuerung) wird auch mit 5 kW geladen. Scheint ein Parameter von EVCC zu sein, kann den aber nirgendwo finden....

premultiply, 2024-08-20:
https://github.com/evcc-io/evcc/blob/4ed8a47b8970a41b116b079342ac83eb01a47bff/templates/definition/meter/huawei-sun2000-dongle-powersensor.yaml#L231

andig, 2024-08-20:
Sieht aus als verhielte sich das erwartungsgemäß. Oder gibt es jetzt noch ein Problem? Sonst könnten wir hier zu machen.

hstreitenberger, 2024-08-20:
Nein, noch nicht gelöst, die Ladeleistung ist u.U. zu niedrig!

Huawei-Speicher können je nach Anzahl Speicher-Module mit unterschiedlicher Ladeleistung laden:

1 Modul = 5kWh -> 2500W
2 Module =10kWh -> 5000W
3 Module = 15kWh -> 5000W

die neue Serie kann noch höher…

es wäre gut, wenn mit der maximalen Ladeleistung geladen werden könnte.
Ein einstellbarer Parameter wäre ggf. hilfreich.

Ich denke, das ist bei anderen Herstellern ggf. ähnlich?

jori0001, 2024-08-20:
Hallo?
Hier geht es darum, dass meine Batterie einfach zu nem horenten Preis aus dem Netz lädt. Obwohl ich das nicht will. 
Es geht nicht um 2,5 kWh oder 5 kWh. Sondern dass sie überhaupt aus dem Netz lädt.

premultiply, 2024-08-20:
Sorry, danke für den Hinweis! Durch das OT hab ich auch das eigentliche Problem aus den Augen verloren.

premultiply, 2024-08-20:
Bist du sicher dass diese Zwangsladung von evcc ausgelöst wurde?
Denn eigentlich müsste sie in dem Fall dann max. mit 2500W geladen werden.
Hier wird aber offensichtlich die volle Leistung von 5000W geladen.

jori0001, 2024-08-20:
Nein. Ich habe eine 10 kWh Luna. Die wird Max. mit 5 kWh geladen. Vor der v0.130.0 war das nicht der Fall. Heute Update gemacht und zack Full Speed aus dem Netz geladen. Berichte morgen früh, wie es sich verhält.

andig, 2024-08-20:
Du kannst das mit

    evcc meter —trace log —help

testen. Dann mal ein Log der Modi aus alter und neuer Version.

AlexJoda, 2024-08-21:
> Sorry, danke für den Hinweis! Durch das OT hab ich auch das eigentliche Problem aus den Augen verloren.

Sorry für den OT. Wollte kein eigenes Thema deshalb aufmachen wenn wir hier schon bei Huawei sind....

AlexJoda, 2024-08-21:
> Nein. Ich habe eine 10 kWh Luna. Die wird Max. mit 5 kWh geladen. Vor der v0.130.0 war das nicht der Fall. Heute Update gemacht und zack Full Speed aus dem Netz geladen. Berichte morgen früh, wie es sich verhält.

Das ist allerdings ein interessanter Punkt! Wenn EVCC meine 15 kWh Luna steuert lädt die mit 2,5 kW (werde ich jetzt ändern mit dem Parameter von @premultiply). Insofern scheint bei dir tatsächlich ein anderer Mechanismus zu greifen...

AlexJoda, 2024-08-21:
> huawei-sun2000-dongle-powersensor.yaml

Finde die huawei-sun2000-dongle-powersensor.yaml Datei nirgendwo auf meinem Ubuntu Server...

mucki12, 2024-08-22:
> Nein. Ich habe eine 10 kWh Luna. Die wird Max. mit 5 kWh geladen. Vor der v0.130.0 war das nicht der Fall. Heute Update gemacht und zack Full Speed aus dem Netz geladen. Berichte morgen früh, wie es sich verhält.

Da evcc ja mit dem Standardcode „nur“ mit 2,5 kWh laden würde, muss das Netzladen ja durch etwas anderes angestoßen werden. Nur „ins Blaue geschossen“: Eventuell macht der Huawei LUNA gerade eine seiner Zwangskalibrierungen?

closed 2024-08-23 by jori0001

andig, 2024-08-23:
Sehr merkwürdig- geändert haben wir da nix...

jori0001, 2024-08-23:
besser so, als anders. aber verstehen tue ich es auch nicht. :)

mucki12, 2024-08-23:
> besser so, als anders. aber verstehen tue ich es auch nicht. :)

Bleibe bei meiner Vermutung: Zwangskalibrierung des LUNA. Wäre ein typisches Verhalten...

AlexJoda, 2024-09-06:
> https://github.com/evcc-io/evcc/blob/4ed8a47b8970a41b116b079342ac83eb01a47bff/templates/definition/meter/huawei-sun2000-dongle-powersensor.yaml#L231

Wie kann man diese huawei-sun2000-dongle-powersensor.yaml Datei editieren? Hab die noch nicht mal auf meinem Ubuntu Server gefunden...