evcc-io/evcc #15518: evbox elvi - charger out of sync

michiproep, 2024-08-20:
### Describe the bug

After update to 0.130.1 I constantly receive "charger out of sync"

### Steps to reproduce

update from 0.129 to 0.130.1

### Configuration details

```yaml
- name: evbox
    type: template
    template: elvi
    stationid: EVB-xxx
    idtag: 'xxx' 
    connector: 1
    connecttimeout: 5m
    timeout: 2m
    meter: true  
    meterinterval: 4s
```

### Log details

```text
[lp-1  ] DEBUG 2024/08/20 08:45:02 charge total import: 5227.817kWh
[lp-1  ] INFO 2024/08/20 08:45:03 loadpoint 1:
[lp-1  ] INFO 2024/08/20 08:45:03   mode:        pv
[lp-1  ] INFO 2024/08/20 08:45:03   charger:     power ✓ energy ✓ currents ✗ phases ✗ wakeup ✗
[lp-1  ] INFO 2024/08/20 08:45:03   meters:      charge ✓
[lp-1  ] INFO 2024/08/20 08:45:03     charge:    power ✓ energy ✓ currents ✗
[lp-1  ] DEBUG 2024/08/20 08:45:03 phase timer inactive
[lp-1  ] DEBUG 2024/08/20 08:45:03 pv timer inactive
[lp-1  ] INFO 2024/08/20 08:45:03 vehicle updated: unknown -> Kia Niro
[lp-1  ] DEBUG 2024/08/20 08:45:04 max charge current: 6A
[lp-1  ] DEBUG 2024/08/20 08:45:04 charge power: 0W
[lp-1  ] DEBUG 2024/08/20 08:45:04 vehicle odometer: 4494km
[lp-1  ] DEBUG 2024/08/20 08:45:04 charge total import: 5227.817kWh
[lp-1  ] DEBUG 2024/08/20 08:45:04 charger status: A
[lp-1  ] INFO 2024/08/20 08:45:04 car disconnected
[lp-1  ] DEBUG 2024/08/20 08:45:04 set session soc limit: 0
[lp-1  ] DEBUG 2024/08/20 08:45:04 set session energy limit: 0
[lp-1  ] DEBUG 2024/08/20 08:45:04 charger disable
[lp-1  ] DEBUG 2024/08/20 08:45:04 wake-up timer: stop
[lp-1  ] DEBUG 2024/08/20 08:45:19 charge power: 0W
[lp-1  ] DEBUG 2024/08/20 08:45:34 charge power: 0W
[lp-1  ] DEBUG 2024/08/20 08:45:49 charge power: 0W
[lp-1  ] DEBUG 2024/08/20 08:46:04 charge power: 0W
[lp-1  ] DEBUG 2024/08/20 08:46:19 charge power: 0W
[lp-1  ] DEBUG 2024/08/20 08:46:19 vehicle odometer: 4494km
[lp-1  ] DEBUG 2024/08/20 08:46:19 charge total import: 5227.817kWh
[lp-1  ] DEBUG 2024/08/20 08:46:19 charger status: A
[lp-1  ] WARN 2024/08/20 08:46:19 charger out of sync: expected disabled, got enabled
[lp-1  ] DEBUG 2024/08/20 08:46:19 charger enable
[lp-1  ] DEBUG 2024/08/20 08:46:19 charger disable
[lp-1  ] DEBUG 2024/08/20 08:46:19 wake-up timer: stop
[lp-1  ] DEBUG 2024/08/20 08:46:34 charge power: 0W
[lp-1  ] DEBUG 2024/08/20 08:46:49 charge power: 0W
[lp-1  ] DEBUG 2024/08/20 08:47:04 charge power: 0W
[lp-1  ] DEBUG 2024/08/20 08:47:19 charge power: 0W
[lp-1  ] DEBUG 2024/08/20 08:47:34 charge power: 0W
[lp-1  ] DEBUG 2024/08/20 08:47:34 charge total import: 5227.817kWh
[lp-1  ] DEBUG 2024/08/20 08:47:34 charger status: A
[lp-1  ] WARN 2024/08/20 08:47:34 charger out of sync: expected disabled, got enabled
[lp-1  ] DEBUG 2024/08/20 08:47:34 charger enable
[lp-1  ] DEBUG 2024/08/20 08:47:34 charger disable
[lp-1  ] DEBUG 2024/08/20 08:47:34 wake-up timer: stop
[lp-1  ] DEBUG 2024/08/20 08:47:49 charge power: 0W
[lp-1  ] DEBUG 2024/08/20 08:47:59 charge power: 0W
[lp-1  ] DEBUG 2024/08/20 08:48:04 charge power: 0W
[lp-1  ] DEBUG 2024/08/20 08:48:19 charge power: 0W
[lp-1  ] DEBUG 2024/08/20 08:48:34 charge power: 0W
[lp-1  ] DEBUG 2024/08/20 08:48:49 charge power: 0W
[lp-1  ] DEBUG 2024/08/20 08:48:49 charge total import: 5227.817kWh
[lp-1  ] DEBUG 2024/08/20 08:48:49 charger status: A
[lp-1  ] WARN 2024/08/20 08:48:49 charger out of sync: expected disabled, got enabled
[lp-1  ] DEBUG 2024/08/20 08:48:49 charger enable
[lp-1  ] DEBUG 2024/08/20 08:48:49 charger disable
[lp-1  ] DEBUG 2024/08/20 08:48:49 wake-up timer: stop
[lp-1  ] DEBUG 2024/08/20 08:49:04 charge power: 0W
[lp-1  ] DEBUG 2024/08/20 08:49:19 charge power: 0W
[lp-1  ] DEBUG 2024/08/20 08:49:34 charge power: 0W
[lp-1  ] DEBUG 2024/08/20 08:49:49 charge power: 0W
[lp-1  ] DEBUG 2024/08/20 08:50:04 charge power: 0W
[lp-1  ] DEBUG 2024/08/20 08:50:04 charge total import: 5227.817kWh
[lp-1  ] DEBUG 2024/08/20 08:50:04 charger status: A
[lp-1  ] WARN 2024/08/20 08:50:04 charger out of sync: expected disabled, got enabled
[lp-1  ] DEBUG 2024/08/20 08:50:04 charger enable
[lp-1  ] DEBUG 2024/08/20 08:50:04 charger disable
[lp-1  ] DEBUG 2024/08/20 08:50:04 wake-up timer: stop
```

### What type of operating system are you running?

Docker container

### Version

0.130.1

michiproep, 2024-08-20:
After plug in the car:
[EVB-xx-1] WARN 2024/08/20 09:02:34 waiting for local authentication
[EVB-xx-1] WARN 2024/08/20 09:03:49 waiting for local authentication

mfuchs1984, 2024-08-20:
Concerning "waiting for local authentication", see

https://docs.evcc.io/blog#rfid-und-autorisierung

premultiply, 2024-08-20:
And you can reduce the config part to

```yaml
- name: evbox
    type: template
    template: elvi
```

michiproep, 2024-08-20:
Thank you. The remotestart:true seemed to fix the issue.
@premultiply : So, stationid and idtag is obsolete now?

closed 2024-08-20 by michiproep

premultiply, 2024-08-20:
The solution is setting the charger itself to "free charge" or what ever it is named. Check the charger setup.
Avoid `remotestart` where ever possible!

premultiply, 2024-08-20:
Please attach the output of `evcc charger --log trace --diagnose`. Something may still be wrong.

michiproep, 2024-08-20:
(Ich switche jetzt mal zu deutsch)
Ich hab die Wallbox jetzt auf Autostart gesetzt und die Konfig auf den Zweizeiler reduziert.
Aktuell sieht alles gut aus.

Aber eins ist mir noch nicht ganz klar:
Vorher war es möglich - falls meine Server (evcc) mal nicht liefen - mit der RFID den Ladevorgang manuell zu starten.
Jetzt steht die Wallbox zwar auf Autostart, startet aber (richtigerweise) nicht, sondern wird anscheinend nach dem Anstecken von evcc erstmal blockiert. Heißt das jetzt im Umkehrschluss, dass falls mein Netzwerk/Server tot sind, ich keine Möglichkeit mehr habe den Ladevorgang manuell zu starten, weil die Box erstmal grundsätzlich auf eine Freischaltung von evcc wartet?

premultiply, 2024-08-20:
Das kommt auf die Box und Firmware an. Ggf. geht sie nach Neustart wieder auf die Default-Werte und vergisst das Profil. Oder man kann Fallback-Einstellungen machen.
Aber das sind alles Implementierungsdetails des Herstellers.

Grundsätzlich funktioniert OCPP jetzt bei evcc so wie mit allen anderen Varianten üblich.

Das eigenmächtige Laden soll und darf ja gerade nicht sein.
Man denke an Lastmanagement und andere Konstellationen.

michiproep, 2024-08-20:
Verstanden. Danke!
Die alte Verhaltensweise mit remotestart und idtag in der Konfig wäre aber auch möglich, oder?

premultiply, 2024-08-20:
Die alte Verfahrensweise gibt es nicht mehr.

`remotestart` bewirkt nur, dass evcc statt wie vorgesehen die Box eine Transaktion startet. Das macht dann allerdings eine lokale Bedienung und Nutzung von RFID an der Box unmöglich.

Tidabblju, 2024-08-20:
Habe das selbe Problem. Ich habe mir gerade die Dokumentation auf evcc.io angesehen. Die EVBOX elvi erfordert nun einen Sponsortoken!? Kommt vielleicht daher das Problem?

michiproep, 2024-08-23:
Was auch komisch ist, dass ich jetzt ständig "charger out of sync: expected disabled, got enabled" bekomme

premultiply, 2024-08-23:
Da bräuchte es Log auf Trace-Level der OCPP-Kommunikation.

michiproep, 2024-08-23:
> Da bräuchte es Log auf Trace-Level der OCPP-Kommunikation.

Siehe https://github.com/evcc-io/evcc/discussions/15606#discussioncomment-10426142