evcc-io/evcc #15526: EBus (Elli) charing with 1 phase is reporting incorrect speed

augard, 2024-08-20:
### Describe the bug

While charing with 1 phase, it's reporting incorrect charging speed. It looks like when I was charging Kia EV9 with 3 phase charging power was correct. But not with PHEV car. 

[image]

### Steps to reproduce

1. Elli Pro charger
2. 1 Phase car
3. Charging

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: e54ca51a41c43d04f487b552158402752fc587cdf15340a89167bb662087424f

meters:
- name: grid_1
  type: template
  template: solax
  usage: grid
  modbus: tcpip
  id: 1
  host: 192.168.0.200
  port: 502

- name: pv_1
  type: template
  template: solax
  usage: pv
  modbus: tcpip
  id: 1
  host: 192.168.0.200
  port: 502

- name: battery_1
  type: template
  template: solax
  usage: battery
  modbus: tcpip
  id: 1
  host: 192.168.0.200
  port: 502

vehicles:
- name: vehicle_1
  type: template
  template: seat
  title: Cupra Formentor
  user: 
  password: 
  vin:
  capacity: 12.8
- name: vehicle_2
  type: template
  template: kia
  title: Kia EV9
  user: 
[... 463 lines cut ...]
[site  ] DEBUG 2024/08/20 14:10:06 battery soc: 99%
[site  ] DEBUG 2024/08/20 14:10:06 battery power: 0W
[site  ] DEBUG 2024/08/20 14:10:06 grid meter: -356W
[site  ] DEBUG 2024/08/20 14:10:06 site power: -6W
[lp-1  ] DEBUG 2024/08/20 14:10:06 !! session: chargeRater.chargedEnergy=5.9 - chargedAtStartup=0.0
[lp-1  ] DEBUG 2024/08/20 14:10:06 charger status: C
[lp-1  ] DEBUG 2024/08/20 14:10:06 soc estimated: 25.46% (vehicle: 20.00%)
[lp-1  ] DEBUG 2024/08/20 14:10:06 vehicle soc: 25%
[lp-1  ] DEBUG 2024/08/20 14:10:06 vehicle range: 5km
[eebus ] DEBUG 2024/08/20 14:10:06 !! OpEV.LoadControlLimits set: {a false true 11.7246}
[lp-1  ] DEBUG 2024/08/20 14:10:06 pv charge current: 11.7A = 11.7A + 0.0087A (-6W @ 3p)
[eebus ] DEBUG 2024/08/20 14:10:06 Send: d:_i:47859_Elli-Wallbox-2022A0RQG6 write 4012 LoadControlLimitListData
[lp-1  ] DEBUG 2024/08/20 14:10:06 max charge current: 11.7A
[eebus ] DEBUG 2024/08/20 14:10:06 Recv: d:_i:47859_Elli-Wallbox-2022A0RQG6:LoadControl to LoadControl notify 31858 LoadControlLimitListData
[eebus ] DEBUG 2024/08/20 14:10:06 ski 4a933ce6a732da2031c3a9f1c2ead588eb623e82 event cem-opev-DataUpdateLimit
[eebus ] DEBUG 2024/08/20 14:10:06 Recv: d:_i:47859_Elli-Wallbox-2022A0RQG6:LoadControl to LoadControl result 31859 4012 ResultData 0
```

### What type of operating system are you running?

Linux

### Version

0.130.1

StefanSchoof, 2024-08-20:
This is because of https://github.com/evcc-io/evcc/discussions/15367

The Firmware of Box sends sometimes wrong measurements and starting with 0.130 evcc will ignore all measurements from the box in the Elli template

augard, 2024-08-20:
Ok, that fixed it, thank you.

closed 2024-08-20 by augard