evcc-io/evcc #15528: [130.1] EVC04 charge total import: not available

Chris45671, 2024-08-20:
### Describe the bug

Hallo zusammen,
nach dem Update auf Version 130.1 bekommen ich die Fehlermeldung das "Charge Total" nicht mehr verfügbar ist.
Mit der Version 130.0 konnten die Werte noch ausgelesen werden.
Dies sieht man ganz gut im Logfile. Vor 15 Uhr mit der alten Version wurden die Werte noch ausgelesen. Danach bekomme ich immer die Fehlermeldung.

Vielleicht kann mir jemand helfen.

Fehlermeldung:
[lp-1 ] ERROR 2024/08/20 15:03:24 charge total import: not available

Wallbox: EVC04 (EON vBox) v3.118.0

Logdatei:
[evcc.log](https://github.com/user-attachments/files/16676478/evcc.log)

Schöne Grüße
Christian

### Steps to reproduce

1. Update auf Version 130.1
2. Start EVCC
3. Im Log erscheint die Meldung "[lp-1 ] ERROR 2024/08/20 15:03:24 charge total import: not available"
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: *****
  port: 7070
log: debug
levels:
  cache: error
plant: 26d4e5648b00d43476b26bc139aed784a2d1ad4b55fef45cf15016286a8578ed
sponsortoken: *****
interval: 10s # control cycle interval
tariffs:
  currency: EUR # (default EUR)
  grid:
    type: fixed
    price: 0.36 # EUR/kWh
  feedin:
    type: fixed
    price: 0.065 # EUR/kWh
  co2:
    type: grünstromindex
    zip: 85652
vehicles:
  - name: Genesis
    title: Genesis G80
    type: template
    template: offline
    icon: car # Icon in der Benutzeroberfläche (optional)
    capacity: 87 # Akkukapazität in kWh (optional)
    phases: 3 # Die maximale Anzahl der Phasen welche genutzt werden können (optional)
    mode: PV # Möglich sind Off, Now, MinPV und PV, oder leer wenn keiner definiert werden soll (optional)
  - name: Scotty
    type: template
    template: enyaq
    title: Scotty # Wird in der Benutzeroberfläche angezeigt (optional)
    user: *****
    password: *****
    capacity: 77 # Akkukapazität in kWh (optional)
    timeout: 10s # optional 
    maxCurrent: 14 # Definiert die maximale Stromstärke pro angeschlossener Phase mit welcher das Fahrzeug geladen werden soll (optional)
    cache: 5m
loadpoints: 
  - title: Garage
    charger: vBox
    mode: pv
    enable:
      threshold: -4000
      delay: 2m
    disable:
      threshold: 450
      delay: 2m
chargers:
  - name: vBox # EON Ladebox
    type: template
    template: ocpp
    stationid: EU_EDR_CB12001332
    
meters:
- type: template
  template: kostal-plenticore 
  id: 71  
  host: *****
  port: 1502  
  usage: pv  
  modbus: tcpip  
  name: pv1
- type: template
  template: kostal-plenticore 
  id: 71  
  host: *****
  port: 1502  
  usage: pv  
  modbus: tcpip  
  name: pv2
- type: template
  template: kostal-plenticore 
  id: 71  
  host: *****
  port: 1502  
  usage: battery  
  modbus: tcpip  
  name: battery2
- name: grid3
  type: template
  template: kostal-ksem
  usage: grid
  
  modbus: tcpip
  id: 71
  host: *****
  port: 502 # Port 
site:
  title: Zuhause
  meters:
    grid: grid3
    pv:
    - pv1
    - pv2
    battery:
    - battery2
  residualPower: 150
```

### Log details

```text
Datei im Anhang.
```

### What type of operating system are you running?

Linux

### Version

130.1

premultiply, 2024-08-20:
Das Log enthält keinerlei relevante Information.

`evcc charger --log trace --diagnose`

Chris45671, 2024-08-20:
Hallo @premultiply ,
anbei das Trace File für den Charger.
[Trace_Charger.txt](https://github.com/user-attachments/files/16678716/Trace_Charger.txt)

Hoffe dies hilft weiter.

Danke schon mal für die Unterstützung.

premultiply, 2024-08-20:
Der letzte Teil der Ausgabe fehlt offensichtlich.
Bitte nochmal vollständig.

premultiply, 2024-08-20:
Fixed by https://github.com/evcc-io/evcc/commit/d9dd50155f43d71821459d993022a1e8aed505a1

closed 2024-08-20 by premultiply

Chris45671, 2024-08-20:
> Fixed by [d9dd501](https://github.com/evcc-io/evcc/commit/d9dd50155f43d71821459d993022a1e8aed505a1)

Vielen Dank für die schnelle Lösung. Wird dieser Fix dann in der Nightly enthalten sein?

premultiply, 2024-08-20:
Ja, ist sogar jetzt schon im neusten Release drin.

Chris45671, 2024-08-20:
> Ja, ist sogar jetzt schon im neusten Release drin.

Vielen Dank für die schnelle Umsetzung

Chris45671, 2024-08-21:
@premultiply Leider ist bei mir nach der von Version 130.2 das Problem nicht behoben sondern durch weitere Meldungen verstärkt worden.

Ich hoffe ich habe dieses mal nichts falsch gemacht und es ist das komplette Tracelog.
[20240821_Trace_Log_Charger.log](https://github.com/user-attachments/files/16686234/20240821_Trace_Log_Charger.log)

Aktuelle Fehlermeldungen:
Aug 21 06:37:16 raspberrypi evcc[9686]: [site  ] WARN 2024/08/21 06:37:16 planner: no matching rate
Aug 21 06:51:18 raspberrypi evcc[9686]: [lp-1  ] ERROR 2024/08/21 06:51:18 charge voltages: not available
Aug 21 06:51:18 raspberrypi evcc[9686]: [lp-1  ] ERROR 2024/08/21 06:51:18 charge total import: not available

Vielleicht kannst Du es Dir bitte erneut anschauen.

Vielen Dank

premultiply, 2024-08-21:
Die Box schickt keine Meteringdaten.
Auch nicht vom Watchdog angetriggert.
Scheint aber auch kein Fahrzeug angeschlossen zu sein.

Probier bitte nochmal einen Reboot der Box nachdem evcc gestartet wurde und initialisiert wurde und probiere auch mal mit angeschlossenen Fahrzeug.

Es gibt alle Absurditäten in Wallbox-Firmware...

Chris45671, 2024-08-21:
Danke, werde ich diese Woche noch machen. Mich hat es halt nur gewundert, da dies vorher ohne Probleme funktioniert hat.
Vielleicht ist es ja wirklich so, das zwischendurch ein Fahrzeug angehängt sein muss damit die Box die Werte ausgibt.
Ich werde berichten.

`Aug 19 21:16:25 raspberrypi evcc[2836]: [site  ] DEBUG 2024/08/19 21:16:25 ----
Aug 19 21:16:25 raspberrypi evcc[2836]: [lp-1  ] DEBUG 2024/08/19 21:16:25 charge power: 0W
Aug 19 21:16:27 raspberrypi evcc[2836]: [site  ] DEBUG 2024/08/19 21:16:27 pv power: 0W
Aug 19 21:16:27 raspberrypi evcc[2836]: [site  ] DEBUG 2024/08/19 21:16:27 battery soc: 68%
Aug 19 21:16:27 raspberrypi evcc[2836]: [site  ] DEBUG 2024/08/19 21:16:27 battery power: 668W
Aug 19 21:16:27 raspberrypi evcc[2836]: [site  ] DEBUG 2024/08/19 21:16:27 grid meter: 20W
Aug 19 21:16:27 raspberrypi evcc[2836]: [site  ] DEBUG 2024/08/19 21:16:27 grid powers: [-80 -80 190]W
Aug 19 21:16:27 raspberrypi evcc[2836]: [site  ] DEBUG 2024/08/19 21:16:27 grid currents: [-0.98 -1.14 1.89]A
Aug 19 21:16:27 raspberrypi evcc[2836]: [site  ] DEBUG 2024/08/19 21:16:27 site power: 788W
Aug 19 21:16:27 raspberrypi evcc[2836]: [lp-1  ] DEBUG 2024/08/19 21:16:27 charge total import: 1541.287kWh
Aug 19 21:16:27 raspberrypi evcc[2836]: [lp-1  ] DEBUG 2024/08/19 21:16:27 charger status: A
Aug 19 21:16:55 raspberrypi evcc[2836]: [site  ] DEBUG 2024/08/19 21:16:55 ----`

Chris45671, 2024-08-22:
Hallo @premultiply ,
ich habe jetzt alles probiert inkl. Neustart Raspy, Neustart Wallbox und sogar das zurücksetzen auf Version 129, aber ich bekomme meinen Skoda Enyaq nicht mehr zu laden. Das Auto bricht mit roter LED ab.

Vielleicht hast Du einen Tipp woran es liegen könnte. 

Logdateien aus der Wallbox:
[ocpp-22-08-2024.zip](https://github.com/user-attachments/files/16707709/ocpp-22-08-2024.zip)
[smartcharge.zip](https://github.com/user-attachments/files/16707710/smartcharge.zip)

Tracelog aus EVCC:
[evcc-20240822-091928-trace.log](https://github.com/user-attachments/files/16707705/evcc-20240822-091928-trace.log)
[evcc-20240822-092714-trace.log](https://github.com/user-attachments/files/16707706/evcc-20240822-092714-trace.log)

Vielen Dank, wenn DU noch weitere Informationen benötigst, so lasse es mich wissen.

Schöne Grüße
Christian

Chris45671, 2024-08-22:
Habe die Nightly installiert, hoffe es hat soweit funktioniert.

Hier das benötigte Trace Log.
[20240822_1100__Trace_Log_Charger.log](https://github.com/user-attachments/files/16708047/20240822_1100__Trace_Log_Charger.log)

premultiply, 2024-08-22:
> [EU_EDR_CB12001332-1] WARN 2024/08/22 11:00:07 waiting for local authentication

Du denkst dran?
https://docs.evcc.io/blog#rfid-und-autorisierung
https://docs.evcc.io/docs/devices/chargers#ocpp-16j-kompatible-wallbox-mit-smart-charging-profil

Chris45671, 2024-08-22:
In den OCPP Einstellungen der Wallbox habe ich daher extra "FreeModeActive" auf "True" stehen. DIes dachte ich ist der Punkt wie im Blog geschrieben.
Muss ich dann trotzdem noch "remotestart: true" beim Charger angeben?

Den Punkt "OCPP 1.6J kompatible Wallbox mit Smart Charging Profil" hatte ich berücksichtigt, soweit es bei der Vestel möglich ist. Es hatte am Sontag mit der 129 noch funktioniert. Dann gab es eben mit der 130 die Fehler, wo wir hier geschrieben haben. diee auch mit 130.1 und 130.2 nicht gelöst waren. Dein Vorschlag war erstmal mit einem Fahrzeug versuchen. Dies hatte ich daher heute gemacht und bekomme keine Möglichkeit. Habe auch schon die RFID Karte drüber gezogen, aber auch dies hat nicht geholfen.

Die Wallbox zeigt den farbigen Ring (Rosa) für das Warten auf die Freigabe des Überschuss. So wie es bisher auch immer war.

Daher bin ich gerade komplett Ratlos was ich hier falsch mache.

Hier noch mal meine yaml Datei:

```
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  site: debug
  lp-1: trace
  opcc: trace
  cache: error
  db: error

# unique installation id
plant: xxxxxxx

sponsortoken: XXXXXX

interval: 30s # control cycle interval

tariffs:
  currency: EUR # (default EUR)
  grid:
    type: fixed
    price: 0.36 # EUR/kWh
  feedin:
    type: fixed
    price: 0.065 # EUR/kWh
  co2:
    type: grünstromindex
    zip: 85652

vehicles:
  - name: Genesis
    title: Genesis G80
    type: template
    template: offline
    icon: car # Icon in der Benutzeroberfläche (optional)
    capacity: 87 # Akkukapazität in kWh (optional)
    phases: 3 # Die maximale Anzahl der Phasen welche genutzt werden können (optional)
    mode: PV # Möglich sind Off, Now, MinPV und PV, oder leer wenn keiner definiert werden soll (optional)
  - name: Scotty
    type: template
    template: enyaq
    title: Scotty # Wird in der Benutzeroberfläche angezeigt (optional)
    user: XXXXXXX # Benutzerkonto (bspw. E-Mail Adresse, User Id, etc.)
    password: XXXXX # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen)
    capacity: 77 # Akkukapazität in kWh (optional)
    timeout: 10s # optional 
    cache: 5m

loadpoints: 
  - title: Garage
    charger: vBox
    mode: pv
    enable:
      threshold: 0
      delay: 1m
    disable:
      threshold: 450
      delay: 2m

chargers:
  - name: vBox # EON Ladebox
    type: template
    template: ocpp
    stationid: EU_EDR_CBXXXXXXXXX
    
meters:
- type: template
  template: kostal-plenticore 
  id: 71  
  host: 192.168.178.XX  
  port: 1502  
  usage: pv  
  modbus: tcpip  
  name: pv1
- type: template
  template: kostal-plenticore 
  id: 71  
  host: 192.168.178.XX  
  port: 1502  
  usage: pv  
  modbus: tcpip  
  name: pv2
- type: template
  template: kostal-plenticore 
  id: 71  
  host: 192.168.178.XX  
  port: 1502  
  usage: battery  
  modbus: tcpip  
  name: battery2
- name: grid3
  type: template
  template: kostal-ksem
  usage: grid 
  # Modbus TCP
  modbus: tcpip
  id: 71
  host: 192.168.178.XX # Hostname
  port: 502 # Port 

site:
  title: Zuhause
  meters:
    grid: grid3
    pv:
    - pv1
    - pv2
    battery:
    - battery2
  residualPower: 150

```

premultiply, 2024-08-22:
> In den OCPP Einstellungen der Wallbox habe ich daher extra "FreeModeActive" auf "True" stehen.

Mmmmh. Sollte eigentlich genau so passen...

Kannst ja trotzdem mal probieren was mit `remotestart: true` passiert (mit FreeModeActive an und aus).

Chris45671, 2024-08-22:
Ich habe es jetzt entsprechend in die yaml eingebaut, aber es geht nicht.
[EU_EDR_CB12001332-1] WARN 2024/08/22 12:14:39 failed to start remote transaction: Rejected 

Auto blinkt weiss und geht dann in rot über, während die Wallbox weiterhin rosa anzeigt.

TRACE aus der Oberfläche anbei:
[evcc-20240822-121733-trace.log](https://github.com/user-attachments/files/16709065/evcc-20240822-121733-trace.log)

premultiply, 2024-08-22:
Kannst du bitte nochmal mit dem aktuellen Nightly Build probieren?

Bitte auch beachten, dass man die Box nach einer Konfigurationsänderung (FreeModeActive an) möglicherweise komplett neustarten muss (und evcc anschließend auch) um die Änderungen auch tatsächlich zu übernehmen. Bei manchen Modellen ist es zusätzlich wichtig, dass währenddessen kein Fahrzeug und/oder Kabel angesteckt ist.

Chris45671, 2024-08-22:
Ich werde es morgen gleich testen. 

Mein angedachter Ablauf:
1. Kabel abstecken/kontrollieren
2. Einstellungen FreeMode kontrollieren und einstellen
3. Box mit der Sicherung vom Netz nehmen und danach wieder Sicherung rein
4. EVCC stoppen 
5. EVCC nightly Update durchführen
6. evcc charger --log trace --diagnose durchführen
7. EVCC Dienst starten
8. Auto mit der Wallbox verbinden
9. Trace ab Start danach aus der Servicefläche ziehen

Falls ich etwas vergessen habe, gerne noch Bescheid geben. 

Bisher danke für deine Hilfe. Hoffe ich kann das Auto dann wieder laden.

premultiply, 2024-08-22:
Klingt erstmal nach einem guten Plan :)

Ggf. nach dem Update prüfen ob evcc nicht automatisch schon wieder als Dienst gestartet wurde.

Chris45671, 2024-08-23:
@premultiply Leider hat es nicht komplet zum Erfolg geführt.
Ich habe heute morgen um 7:00 Uhr alles fertig gemacht wie gestern Abend geschrieben.
Her ist auch kein Fehler mehr aufgetaucht.
Alle Dateien zu dieser Uhrzeit:
[ocpp-23-08-2024.log](https://github.com/user-attachments/files/16725998/ocpp-23-08-2024.log)
[smartcharge.log](https://github.com/user-attachments/files/16726001/smartcharge.log)
[20240823_0710_Trace_Charger.log](https://github.com/user-attachments/files/16726002/20240823_0710_Trace_Charger.log)
[evcc-20240823-073157-trace.log](https://github.com/user-attachments/files/16726003/evcc-20240823-073157-trace.log)

Gegen 9:00 Uhr gab es einen Stromausfall beim Raspi. Danach erschien wieder die gewohnte Fehlermeldung.
[Trace Log nach dem Anstecken.txt](https://github.com/user-attachments/files/16726007/Trace.Log.nach.dem.Anstecken.txt)

Ich habe jetzt gegen 10:15 Uhr noch mal alles von neuem gemacht. Inkl. SIcherung der Wallbox ziehen.
Folgende Dateien habe ich jetzt dafür:

[20240823_1030_Trace_Charger.log](https://github.com/user-attachments/files/16726011/20240823_1030_Trace_Charger.log)
[evcc-20240823-103920-trace.log](https://github.com/user-attachments/files/16726052/evcc-20240823-103920-trace.log)

HIer noch mal das Log nach dem mehrmaligen Versuchen das AUto zu laden:
[evcc-20240823-104942-trace.log](https://github.com/user-attachments/files/16726179/evcc-20240823-104942-trace.log)

Laut Log weiterhin die folgenden Fehler:
[lp-1 ] ERROR 2024/08/23 10:49:19 charge power: not available
[lp-1 ] ERROR 2024/08/23 10:49:19 charge currents: not available
[site ] WARN 2024/08/23 10:49:19 planner: no matching rate 

Was ist weiterhin das Problem?
- Das Auto (Skoda Enyaq) lässt sich nicht laden. Es wird von EVCC erkannt das ein Auto angesteckt wird, es wird auch der Skoda erkannt und das Auto zeigt am Ladeport die rote LED nach ca. 5-10 Sekunden an.

Einstellungen in der OCPP Schnittstelle bei Vestel, vielleicht ist hier nach der Änderung auf die EVCC Version 130 doch noch etwas einzustellen, was ich übersehen habe.

[image]

Chris45671, 2024-08-23:
Hier noch die beiden Log Dateien direkt aus der Vestel EVC04 um die Zeit wo das Auto auch angesteckt wurde.
Vielleicht kann dies auch weiterhelfen.

[ocpp-23-08-2024_1100.log](https://github.com/user-attachments/files/16726324/ocpp-23-08-2024_1100.log)
[smartcharge_1100.log](https://github.com/user-attachments/files/16726325/smartcharge_1100.log)

Chris45671, 2024-08-24:
Hier noch das Trace Log nach dem neuen Nightly Update.

[20240824_1215_Trace_Charger.log](https://github.com/user-attachments/files/16736257/20240824_1215_Trace_Charger.log)

Chris45671, 2024-08-24:
@premultiply Nach dem Log habe ich jetzt mal etwas anderes ausprobiert. Die Wallbox war von mir per Default auf PV gestellt.
Hier scheint es seit Version 130.x jetzt ein Problem zwischen Wallbox und dem Auto zu geben.
Ich habe die Wallbox vor dem Anstecken auf MinPV eingestellt und das Auto wurde wie bisher ohne Probleme erkannt, aber auch das Auto wollte laden und hat keine rote LED gebracht.
Anbei das Trace aus der Weboberfläche, da ich nicht weiss während dem Ladevorgang wie ich an ein anderes Log komme.
[evcc-20240824-134527-trace.log](https://github.com/user-attachments/files/16736478/evcc-20240824-134527-trace.log)

Könntest Du Dir das Thema bitte noch mal anschauen und mir helfen?
Was mache ich vielleicht falsch und wo müsste ich etwas ändern?

Vielen Dank

Chris45671, 2024-08-26:
@premultiply Entschuldige wenn ich nerve. Ich wollte nur kurz nachfragen ob die gelieferten Informationen reichen oder ob noch etwas benötigt wird an Informationen?