evcc-io/evcc #15533: "cannot create charger" after update 0.130.1 ABB Terra AC over OCPP

mik5z, 2024-08-20:
### Describe the bug

After I updated from 0.129.0 to 0.130.1 evcc will no longer start. It gives the following error and keeps restarting: [main  ] FATAL 2024/08/20 17:43:45 cannot create charger 'my_charger': cannot create charger type 'ocpp': ocpp message (): GenericError - client disconnected, no response received from client

Rolling back to 0.129.0 fixes the problem. 

### Steps to reproduce

1. have evcc running without issues on 0.129.0
2. update from 0.129.0 to 0.130.1
3. wait for evcc to start
...

### Configuration details

```yaml
Home Assistant OS 13.0 with core 2024.8.2
Wallbox ABB TAC-W11-G5-R-0 with firmware 1.8.21 (latest) 

chargers:
  - name: my_charger
    type: ocpp
    timeout: 10m
    stationid: TACW11[my station id]
    metervalues: Energy.Active.Import.Register 
    getconfiguration: false
```

### Log details

```text
[site  ] DEBUG 2024/08/20 17:40:21 ----
[lp-1  ] DEBUG 2024/08/20 17:40:21 charge power: 0W
[site  ] DEBUG 2024/08/20 17:40:21 pv power: 1745W
[site  ] DEBUG 2024/08/20 17:40:21 battery soc: 100%
[site  ] DEBUG 2024/08/20 17:40:21 battery power: 0W
[site  ] DEBUG 2024/08/20 17:40:21 grid meter: -863W
[site  ] DEBUG 2024/08/20 17:40:21 grid powers: [-752 -171 60]W
[site  ] DEBUG 2024/08/20 17:40:21 grid currents: [-3.45 -1.39 1.83]A
[site  ] DEBUG 2024/08/20 17:40:21 site power: -763W
[lp-1  ] DEBUG 2024/08/20 17:40:21 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[lp-1  ] DEBUG 2024/08/20 17:40:21 charger status: A
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2024/08/20 17:40:57 evcc 0.130.1
[main  ] INFO 2024/08/20 17:40:57 using config file: /config/evcc.yaml
[main  ] INFO 2024/08/20 17:40:59 listening at :7070
[main  ] FATAL 2024/08/20 17:43:45 cannot create charger 'my_charger': cannot create charger type 'ocpp': ocpp message (): GenericError - client disconnected, no response received from client
[main  ] FATAL 2024/08/20 17:43:45 will attempt restart in: 15m0s
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2024/08/20 17:44:13 evcc 0.130.1
[main  ] INFO 2024/08/20 17:44:13 using config file: /config/evcc.yaml
[main  ] INFO 2024/08/20 17:44:14 listening at :7070
[ocpp  ] ERROR 2024/08/20 17:44:32 ocpp message (5929564): NotSupported - Unsupported feature SecurityEventNotification ([2, "5929564", "SecurityEventNotification", {"type": "StartupOfTheDevice", "timestamp": "2024-08-20T15:43:25.000Z"}])
[ocpp  ] ERROR 2024/08/20 17:44:32 ocpp message (8454120): NotSupported - Unsupported feature SecurityEventNotification ([2, "8454120", "SecurityEventNotification", {"type": "SettingSystemTime", "techInfo": "ocppBoot", "timestamp": "2024-08-20T15:44:32.000Z"}])
[main  ] FATAL 2024/08/20 17:45:00 cannot create charger 'my_charger': cannot create charger type 'ocpp': ocpp message (): GenericError - client disconnected, no response received from client
[main  ] FATAL 2024/08/20 17:45:00 will attempt restart in: 15m0s
[ocpp  ] ERROR 2024/08/20 17:45:19 ocpp message (5929564): NotSupported - Unsupported feature SecurityEventNotification ([2, "5929564", "SecurityEventNotification", {"type": "StartupOfTheDevice", "timestamp": "2024-08-20T15:44:42.000Z"}])
[ocpp  ] ERROR 2024/08/20 17:45:19 ocpp message (8454120): NotSupported - Unsupported feature SecurityEventNotification ([2, "8454120", "SecurityEventNotification", {"type": "SettingSystemTime", "techInfo": "ocppBoot", "timestamp": "2024-08-20T15:45:19.000Z"}])
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2024/08/20 17:47:29 evcc 0.130.1
[main  ] INFO 2024/08/20 17:47:29 using config file: /config/evcc.yaml
[main  ] INFO 2024/08/20 17:47:29 listening at :7070
[main  ] FATAL 2024/08/20 17:50:17 cannot create charger 'my_charger': cannot create charger type 'ocpp': ocpp message (): GenericError - client disconnected, no response received from client
[main  ] FATAL 2024/08/20 17:50:17 will attempt restart in: 15m0s
```

### What type of operating system are you running?

Linux

### Version

0.130.1

mik5z, 2024-08-20:
Hi @premultiply,

thanks for the quick answer! Changing the charger configuration does give me the information that it sees the charge point connecting:
[ocpp  ] INFO 2024/08/20 18:22:59 charge point connected, registering: TACW1142021G1553

but it's still producing the same error afterwards: 

[site  ] DEBUG 2024/08/20 18:20:23 ----
[lp-1  ] DEBUG 2024/08/20 18:20:23 charge power: 0W
[site  ] DEBUG 2024/08/20 18:20:24 pv power: 2839W
[site  ] DEBUG 2024/08/20 18:20:24 battery soc: 99%
[site  ] DEBUG 2024/08/20 18:20:24 battery power: 2091W
[site  ] DEBUG 2024/08/20 18:20:24 grid meter: 85W
[site  ] DEBUG 2024/08/20 18:20:24 grid powers: [-1731 2830 -1014]W
[site  ] DEBUG 2024/08/20 18:20:24 grid currents: [-7.33 12 -4.58]A
[site  ] DEBUG 2024/08/20 18:20:24 site power: 2276W
[lp-1  ] DEBUG 2024/08/20 18:20:24 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[lp-1  ] DEBUG 2024/08/20 18:20:25 charger status: A
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2024/08/20 18:20:42 evcc 0.130.1
[main  ] INFO 2024/08/20 18:20:42 using config file: /config/evcc.yaml
[main  ] INFO 2024/08/20 18:20:43 listening at :7070
[ocpp  ] INFO 2024/08/20 18:22:59 charge point connected, registering: TACW1142021G1553
[main  ] FATAL 2024/08/20 18:23:29 cannot create charger 'my_charger': cannot create charger type 'ocpp': ocpp message (): GenericError - client disconnected, no response received from client
[main  ] FATAL 2024/08/20 18:23:29 will attempt restart in: 15m0s
[ocpp  ] ERROR 2024/08/20 18:23:47 ocpp message (5929564): NotSupported - Unsupported feature SecurityEventNotification ([2, "5929564", "SecurityEventNotification", {"type": "StartupOfTheDevice", "timestamp": "2024-08-20T16:23:10.000Z"}])
[ocpp  ] ERROR 2024/08/20 18:23:47 ocpp message (8454120): NotSupported - Unsupported feature SecurityEventNotification ([2, "8454120", "SecurityEventNotification", {"type": "SettingSystemTime", "techInfo": "ocppBoot", "timestamp": "2024-08-20T16:23:47.000Z"}])
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2024/08/20 18:24:00 evcc 0.130.1
[main  ] INFO 2024/08/20 18:24:00 using config file: /config/evcc.yaml
[main  ] INFO 2024/08/20 18:24:00 listening at :7070

premultiply, 2024-08-20:
What is the output of `evcc charger --log trace --diagnose`?

mik5z, 2024-08-20:
How do I reach the evcc console? I run Home Assistant OS as a VM, but don't know how to reach the evcc-specific console

premultiply, 2024-08-20:
Just run the binary on your native system.
Windows, macOS, Linux, ...

mik5z, 2024-08-21:
The "problem" is, that I'm running it as an add-on within Home Assistant, and cannot access the container's console from there. I'm now trying to recreate the container within Synology's Container Manager with an identical set-up and evcc.yaml, but can't access the console from within Home Assistant.

premultiply, 2024-08-21:
Just run the binary on a native system.
Windows, macOS, Linux, ...

mik5z, 2024-08-21:
This is the output of the command: 
```
[main  ] INFO 2024/08/21 14:28:38 evcc 0.130.2                                   
[main  ] INFO 2024/08/21 14:28:38 using config file: /etc/evcc.yaml              
[TACW1142021G1553-1] DEBUG 2024/08/21 14:28:39 waiting for chargepoint: 5m0s     
[main  ] FATAL 2024/08/21 14:33:39 cannot create charger 'my_charger': cannot create charger type 'ocpp': timeout          
```

During boot the following log is created:
```
[main  ] FATAL 2024/08/21 14:28:54 cannot create charger 'my_charger': cannot cre
ate charger type 'ocpp': ocpp message (): GenericError - client disconnected, no 
response received from client                                                    
[main  ] FATAL 2024/08/21 14:28:54 will attempt restart in: 15m0s                
[ocpp  ] ERROR 2024/08/21 14:29:15 ocpp message (5929564): NotSupported - Unsuppo
rted feature SecurityEventNotification ([2, "5929564", "SecurityEventNotification
", {"type": "StartupOfTheDevice", "timestamp": "2024-08-21T12:28:38.000Z"}])     
[ocpp  ] ERROR 2024/08/21 14:29:15 ocpp message (8454120): NotSupported - Unsuppo
rted feature SecurityEventNotification ([2, "8454120", "SecurityEventNotification
", {"type": "SettingSystemTime", "techInfo": "ocppBoot", "timestamp": "2024-08-21
T12:29:15.000Z"}])
```

premultiply, 2024-08-21:
It does not help any further.

It needs a full log with ocpp in trace level including everything else at least on debug level.

mik5z, 2024-08-21:
I have now put all logs levels to trace except db, and now I get the following messages: 

```
[cache ] TRACE 2024/08/21 14:54:12 fatal: <nil>                                  
[cache ] TRACE 2024/08/21 14:54:12 interval: 30s                                 
[cache ] TRACE 2024/08/21 14:54:12 network: {http 192.168.1.10 7070}             
[cache ] TRACE 2024/08/21 14:54:12 mqtt: { evcc   false}                         
[cache ] TRACE 2024/08/21 14:54:12 influx: {   }                                 
[cache ] TRACE 2024/08/21 14:54:12 sponsor: {trial 2024-08-23 09:00:00 +0000 UTC 
true}                                                                            
[cache ] TRACE 2024/08/21 14:54:12 auth: {map[]}                                 
[main  ] FATAL 2024/08/21 14:54:12 cannot create charger 'my_charger': cannot cre
ate charger type 'ocpp': ocpp message (): GenericError - client disconnected, no 
response received from client                                                    
[cache ] TRACE 2024/08/21 14:54:12 version: 0.130.2                              
[main  ] FATAL 2024/08/21 14:54:12 will attempt restart in: 15m0s                
[cache ] TRACE 2024/08/21 14:54:12 fatal: {0xc00196f7b8}                         
[ocpp  ] ERROR 2024/08/21 14:54:29 ocpp message (5929564): NotSupported - Unsuppo
rted feature SecurityEventNotification ([2, "5929564", "SecurityEventNotification
", {"type": "StartupOfTheDevice", "timestamp": "2024-08-21T12:53:53.000Z"}])     
[ocpp  ] ERROR 2024/08/21 14:54:29 ocpp message (8454120): NotSupported - Unsuppo
rted feature SecurityEventNotification ([2, "8454120", "SecurityEventNotification
", {"type": "SettingSystemTime", "techInfo": "ocppBoot", "timestamp": "2024-08-21
T12:54:29.000Z"}])
```

premultiply, 2024-08-21:
Please provide a full log from start.

mik5z, 2024-08-22:
So I've tried this several times, but evcc v0.130.2 is not giving me any more output than the logs posted yesterday:
```
[main  ] INFO 2024/08/22 19:01:27 listening at :7070                             
[cache ] TRACE 2024/08/22 19:02:41 fatal: <nil>                                  
[cache ] TRACE 2024/08/22 19:02:41 interval: 30s                                 
[cache ] TRACE 2024/08/22 19:02:41 network: {http 192.168.1.10 7070}             
[cache ] TRACE 2024/08/22 19:02:41 mqtt: { evcc   false}                         
[cache ] TRACE 2024/08/22 19:02:41 influx: {   }                                 
[cache ] TRACE 2024/08/22 19:02:41 sponsor: {trial 2024-08-23 09:00:00 +0000 UTC 
true}                                                                            
[cache ] TRACE 2024/08/22 19:02:41 auth: {map[]}                                 
[main  ] FATAL 2024/08/22 19:02:41 cannot create charger 'my_charger': cannot cre
ate charger type 'ocpp': ocpp message (): GenericError - client disconnected, no 
response received from client                                                    
[main  ] FATAL 2024/08/22 19:02:41 will attempt restart in: 15m0s                
[cache ] TRACE 2024/08/22 19:02:41 version: 0.130.2                              
[cache ] TRACE 2024/08/22 19:02:41 fatal: {0xc000c39620}                         
[ocpp  ] ERROR 2024/08/22 19:03:03 ocpp message (5929564): NotSupported - Unsuppo
rted feature SecurityEventNotification ([2, "5929564", "SecurityEventNotification
", {"type": "StartupOfTheDevice", "timestamp": "2024-08-22T17:02:27.000Z"}])     
[ocpp  ] ERROR 2024/08/22 19:03:03 ocpp message (8454120): NotSupported - Unsuppo
rted feature SecurityEventNotification ([2, "8454120", "SecurityEventNotification
", {"type": "SettingSystemTime", "techInfo": "ocppBoot", "timestamp": "2024-08-22
T17:03:03.000Z"}])          
```

When I run the command "evcc charger --log trace --diagnose" all I get is:
```
[main  ] INFO 2024/08/22 19:01:32 evcc 0.130.2                                   
[main  ] INFO 2024/08/22 19:01:32 using config file: /etc/evcc.yaml              
[TACW1142021G1553-1] DEBUG 2024/08/22 19:01:37 waiting for chargepoint: 5m0s     
[main  ] FATAL 2024/08/22 19:06:37 cannot create charger 'my_charger': cannot cre
ate charger type 'ocpp': timeout
```

Here is the full log for evcc version 0.129.0 (which works for me) 
[evcc-20240822-191311-trace.log](https://github.com/user-attachments/files/16715279/evcc-20240822-191311-trace.log)

premultiply, 2024-08-22:
Your log is still missing the relevant parts.
Please provide a full **TRACE LEVEL** log for OCPP from start.

mik5z, 2024-08-22:
How can I specify for it to specifically put OCPP logs in trace? 

log: info
levels:
  site: trace
  lp-1: trace
  lp-2: trace
  cache: trace
  db: error
Would adding the following line do anything? 
  ocpp: trace

premultiply, 2024-08-22:
```yaml
log: debug

levels:
  cache: error
  db: error
  ocpp: trace
```

premultiply, 2024-08-22:
> By the way, the [evcc-20240822-191311-trace.log](https://github.com/user-attachments/files/16715279/evcc-20240822-191311-trace.log) file was missing relevant parts? Looking through it, I do see [ocpp ] TRACE events in there

`[main  ] INFO 2024/08/22 19:09:01 evcc 0.129.0` is outdated and not relevant here.

mik5z, 2024-08-22:
> > By the way, the [evcc-20240822-191311-trace.log](https://github.com/user-attachments/files/16715279/evcc-20240822-191311-trace.log) file was missing relevant parts? Looking through it, I do see [ocpp ] TRACE events in there
> 
> `[main ] INFO 2024/08/22 19:09:01 evcc 0.129.0` is outdated version and not relevant here.

I am aware of that, but as I mentioned when uploading the log, version 0.130.2 is completely stuck and did not provide any log other than the small "chunk" I've already posted. 

I will update evcc.yaml with the log configuration you posted and try again in 0.130.2.

premultiply, 2024-08-22:
> version 0.130.2 is completely stuck and did not provide any log other than the small "chunk" I've already posted.

This was a result of the log configuration and has hidden the relevant ocpp communication in the background.

mik5z, 2024-08-22:
I'm getting the following log entries after changing the config:

```
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2024/08/22 20:12:16 evcc 0.130.2
[main  ] INFO 2024/08/22 20:12:16 using config file: /config/evcc.yaml
[main  ] INFO 2024/08/22 20:12:19 listening at :7070
[TACW1142021G1553-1] DEBUG 2024/08/22 20:12:20 waiting for chargepoint: 5m0s
[ocpp  ] DEBUG 2024/08/22 20:14:27 charge point connected: TACW1142021G1553
[ocpp  ] TRACE 2024/08/22 20:14:28 send TACW1142021G1553: [2,"2924647174","GetConfiguration",{}]
[ocpp  ] TRACE 2024/08/22 20:14:28 recv TACW1142021G1553: [2, "3019962", "StatusNotification", {"connectorId": 0, "errorCode": "NoError", "info": "null", "status": "Available", "vendorErrorCode": "0x0000"}]
[ocpp  ] TRACE 2024/08/22 20:14:28 send TACW1142021G1553: [3,"3019962",{}]
[ocpp  ] TRACE 2024/08/22 20:14:28 recv TACW1142021G1553: [2, "7843122", "StatusNotification", {"connectorId": 1, "errorCode": "NoError", "info": "null", "status": "Available", "vendorErrorCode": "0x0000"}]
[ocpp  ] TRACE 2024/08/22 20:14:28 send TACW1142021G1553: [3,"7843122",{}]
[ocpp  ] DEBUG 2024/08/22 20:14:54 charge point disconnected: TACW1142021G1553
[main  ] FATAL 2024/08/22 20:14:54 cannot create charger 'my_charger': cannot create charger type 'ocpp': ocpp message (): GenericError - client disconnected, no response received from client
[main  ] FATAL 2024/08/22 20:14:54 will attempt restart in: 15m0s
[ocpp  ] DEBUG 2024/08/22 20:15:14 charge point connected: TACW1142021G1553
[ocpp  ] TRACE 2024/08/22 20:15:15 recv TACW1142021G1553: [2, "7053156", "BootNotification", {"chargePointModel": "CDT_TACW11::NET_ETH", "chargePointVendor": "ABB", "chargeBoxSerialNumber": "TACW1142021G1553", "chargePointSerialNumber": "TACW1142021G1553", "firmwareVersion": "TAC3Z9119006710273::V1.8.21", "meterType": "V1"}]
[ocpp  ] TRACE 2024/08/22 20:15:15 send TACW1142021G1553: [3,"7053156",{"currentTime":"2024-08-22T18:15:15Z","interval":60,"status":"Accepted"}]
[ocpp  ] TRACE 2024/08/22 20:15:15 recv TACW1142021G1553: [2, "1893258", "StatusNotification", {"connectorId": 0, "errorCode": "NoError", "info": "null", "status": "Available", "vendorErrorCode": "0x0000"}]
[ocpp  ] TRACE 2024/08/22 20:15:15 send TACW1142021G1553: [3,"1893258",{}]
[ocpp  ] TRACE 2024/08/22 20:15:15 recv TACW1142021G1553: [2, "7784418", "StatusNotification", {"connectorId": 1, "errorCode": "NoError", "info": "null", "status": "Available", "vendorErrorCode": "0x0000"}]
[ocpp  ] TRACE 2024/08/22 20:15:15 send TACW1142021G1553: [3,"7784418",{}]
[ocpp  ] TRACE 2024/08/22 20:15:15 recv TACW1142021G1553: [2, "5929564", "SecurityEventNotification", {"type": "StartupOfTheDevice", "timestamp": "2024-08-22T18:14:38.000Z"}]
[ocpp  ] ERROR 2024/08/22 20:15:15 ocpp message (5929564): NotSupported - Unsupported feature SecurityEventNotification ([2, "5929564", "SecurityEventNotification", {"type": "StartupOfTheDevice", "timestamp": "2024-08-22T18:14:38.000Z"}])
[ocpp  ] TRACE 2024/08/22 20:15:15 recv TACW1142021G1553: [2, "8454120", "SecurityEventNotification", {"type": "SettingSystemTime", "techInfo": "ocppBoot", "timestamp": "2024-08-22T18:15:15.000Z"}]
[ocpp  ] ERROR 2024/08/22 20:15:15 ocpp message (8454120): NotSupported - Unsupported feature SecurityEventNotification ([2, "8454120", "SecurityEventNotification", {"type": "SettingSystemTime", "techInfo": "ocppBoot", "timestamp": "2024-08-22T18:15:15.000Z"}])
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2024/08/22 20:15:33 evcc 0.130.2
[main  ] INFO 2024/08/22 20:15:33 using config file: /config/evcc.yaml
[main  ] INFO 2024/08/22 20:15:33 listening at :7070
[TACW1142021G1553-1] DEBUG 2024/08/22 20:15:35 waiting for chargepoint: 5m0s
```

premultiply, 2024-08-22:
Ok, this is very strange.
Seems the firmware just crashes on the simple `GetConfiguration` request and reboots.

mik5z, 2024-08-22:
You mean the charger itself crashes because of that request?

premultiply, 2024-08-22:
Yes exactly. It firmware crashes and reboots the charger.

And it just "worked" in 0.129.0 because of `getconfiguration: false` quirks mode.

This definitely needs to be fixed by ABB.

mik5z, 2024-08-22:
I see, so it was not working before. As I had that set because I was having issues and that seemed to resolve the problem I was having at the time.

It does not seem to help any more, though.

premultiply, 2024-08-22:
Yeah, the old quirks mode without `GetConfiguration` was removed.
We need the provided informations to adapt to the chargers properties.
Otherwise, many things would have to be configured manually, which are normally not known.

And it is not usual that a very very basic OCPP command like `GetConfiguration` has such big issues in the firmware.

mik5z, 2024-08-22:
I understand. And I agree, I was not expecting that to be an issue. Especially for ABB, who isn't a small player in the EV chanrging business

mik5z, 2024-08-22:
> Yes exactly. It firmware crashes and reboots the charger.
> 
> This only "worked" in version in 0.129.0 because of the `getconfiguration: false` quirks mode.
> 
> This definitely needs to be fixed by ABB.

I will try and reach out to ABB tomorrow and see if they are aware of this issue

premultiply, 2024-08-24:
Please have a look here:

https://github.com/evcc-io/evcc/issues/15493#issuecomment-2308143746

This guy mentioned that he had similiar problems until he resetted the box to factory defaults.

mik5z, 2024-08-24:
Hello,

I'll give that a try. I'll let you know if it fixes the problem

Mit freundlichen Grüßen/ Kind regards,

Mike Maassen

Please have a look here:

#15493 (comment)<https://github.com/evcc-io/evcc/issues/15493#issuecomment-2308143746>

mik5z, 2024-08-24:
So after the factory reset I am indeed able to start evcc, but I'm running into new probems: it's now rejecting the 'set charging profile'

Here is the full log after the update: 
[evcc-20240824-102021-debug.log](https://github.com/user-attachments/files/16735954/evcc-20240824-102021-debug.log)

mik5z, 2024-08-24:
Here's another log that includes a reboot of the charger: 
[evcc-20240824-103806-debug.log](https://github.com/user-attachments/files/16735982/evcc-20240824-103806-debug.log)

mfuchs1984, 2024-08-24:
Im Log steht "waiting for local authentication", das bedeutet, die Wallbox wartet auf Freigabe durch einen RFID tag oder ähnliches. Kannst du sie auf free charging konfigurieren? Siehe https://docs.evcc.io/blog#rfid-und-autorisierung
https://docs.evcc.io/docs/devices/chargers#ocpp-16j-kompatible-wallbox-mit-smart-charging-profil

mik5z, 2024-08-24:
Also hier meine Konfiguration der wallbox:
[image]

Free vending ist an. Auch nach eine zweite factory reset mit direkt free vending an bekomme ich die gleiche Fehler 

Hier die neueste Log:
[evcc-20240824-152809-debug.log](https://github.com/user-attachments/files/16736797/evcc-20240824-152809-debug.log)

mik5z, 2024-08-24:
Auch hier wenn ich zurück gehe auf 0.129.0 habe ich keine Probleme ohne etwas zu ändern 

[evcc-20240824-202611-debug.log](https://github.com/user-attachments/files/16737619/evcc-20240824-202611-debug.log)

andig, 2024-08-24:
> So after the factory reset I am indeed able to start evcc, but I'm running into new probems: it's now rejecting the 'set charging profile'

@mik5z that is https://github.com/evcc-io/evcc/issues/15493, please lets continue there.

closed 2024-08-24 by andig

matti1993, 2024-08-25:
bin wieder auf 0.129.0 zurück, kann mir jemand sagen ob das Problem mit v130.3 jetzt behoben ist?