evcc-io/evcc #15535: waiting for local authentication/SetChargingProfile rejected (Pulsar Plus)

dscho, 2024-08-20:
### Describe the bug

After upgrading from v0.128.4 to v0.130.1, I get the warning "waiting for local authentication" (most likely the root cause is identical to the one of https://github.com/evcc-io/evcc/issues/15529, and I guess that the `ocpp-pulsarplus` template needs to be updated to set `remotestart: true` by default because I cannot find any of those indicated options like "Autostart" in Pulsar's App).

More serious than that, though, is the fact that `SetChargingProfile` gets rejected. I see that someone helpful tried to point out in https://github.com/evcc-io/evcc/issues/15493 that this issue happens not only with ABB but also with Pulsar Plus, but was met with replies that I can only imagine are borne out of the frustration that this new OCPP thing isn't working as was hoped. Nevertheless, I strongly suspect that the ABB issue has the exact same root cause as the Pulsar Plus issue.

### Steps to reproduce

1. Connect car to a Pulsar Plus charger
2. observe the log (and the errors in the UI)

### Configuration details

```yaml
log: debug
levels:
  cache: error
  ocpp: trace

meters:
- type: template
  template: senec-home
  usage: grid
  schema: https
  host: 192.168.178.84
  name: grid1
- type: template
  template: senec-home
  usage: pv
  schema: https
  host: 192.168.178.84
  name: bsh
- type: template
  template: senec-home
  usage: battery
  schema: https
  host: 192.168.178.84
  name: senec

chargers:
- type: template
  template: pulsarplus
  name: pulsar

vehicles:
- type: template
  name: id3
  title: id.3
  template: id
  [... login data ...]

loadpoints:
- title: Garage
  charger: pulsar
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16

site:
  title: [... title ...]
  meters:
    grid: grid1
    pv:
    - bsh
    battery:
    - senec
  residualPower: 300
  maxGridSupplyWhileBatteryCharging: 50
```
[... 131 lines cut ...]
Aug 20 12:01:12 raspberrypi evcc[637]: [lp-1  ] DEBUG 2024/08/20 12:01:12 vehicle status: B
Aug 20 12:01:12 raspberrypi evcc[637]: [lp-1  ] DEBUG 2024/08/20 12:01:12 vehicle soc: 79%
Aug 20 12:01:12 raspberrypi evcc[637]: [lp-1  ] DEBUG 2024/08/20 12:01:12 vehicle soc limit: 80%
Aug 20 12:01:12 raspberrypi evcc[637]: [lp-1  ] DEBUG 2024/08/20 12:01:12 vehicle range: 327km
Aug 20 12:01:12 raspberrypi evcc[637]: [lp-1  ] DEBUG 2024/08/20 12:01:12 pv charge current: 5.61A = 0A + 5.61A (-3868W @ 3p)
Aug 20 12:01:12 raspberrypi evcc[637]: [lp-1  ] DEBUG 2024/08/20 12:01:12 pv disable timer reset
Aug 20 12:01:12 raspberrypi evcc[637]: [lp-1  ] DEBUG 2024/08/20 12:01:12 pv timer inactive
Aug 20 12:01:12 raspberrypi evcc[637]: [ocpp  ] TRACE 2024/08/20 12:01:12 send evcc: [2,"2640631095","SetChargingProfile",{"connectorId":1,"csChargingProfiles":{"chargingProfileId":1,"stackLevel":1,"chargingProfilePurpose":"TxDefaultProfile","chargingProfileKind":"Absolute","chargingSchedule":{"startSchedule":"2024-08-20T11:01:12Z","chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":6}]}}}]
Aug 20 12:01:12 raspberrypi evcc[637]: [ocpp  ] TRACE 2024/08/20 12:01:12 recv evcc: [3,"2640631095",{"status": "Rejected"}]
Aug 20 12:01:12 raspberrypi evcc[637]: [lp-1  ] ERROR 2024/08/20 12:01:12 max charge current 6A: set charging profile: Rejected
Aug 20 12:01:35 raspberrypi evcc[637]: [ocpp  ] TRACE 2024/08/20 12:01:35 recv evcc: [2,"1317886357","Heartbeat",{}]
Aug 20 12:01:35 raspberrypi evcc[637]: [ocpp  ] TRACE 2024/08/20 12:01:35 send evcc: [3,"1317886357",{"currentTime":"2024-08-20T11:01:35Z"}]
Aug 20 12:01:35 raspberrypi evcc[637]: [site  ] DEBUG 2024/08/20 12:01:35 ----
[...]
```

</details>

### What type of operating system are you running?

Linux

### Version

evcc version 0.130.1

dscho, 2024-08-20:
> [premultiply](https://github.com/premultiply)
changed the title waiting for local authentication/SetChargingProfile rejected (Pulsar Plus) Wallbox Pulsar Plus: No Autostart in charger config?

@premultiply  umm, no. This ticket is about _two_ issues. The more critical one is not that Autostart one. It's that `SetChargingProfile` gets rejected. Please be careful when you edit tickets not to drop important stuff like that. I put a lot of care into writing this ticket.

dscho, 2024-08-20:
Oh, and I just noticed that clicking the "Off" button has no effect, it continues charging with 11kW even though production is currently at 1.0kW and the car should not charge at all.

premultiply, 2024-08-20:
First of all setup your charger to "autostart", "free vending" or however it is called.
Use your App or Web-UI config of your charger to configure it correctly.

Or just put an RFID card to the reader.

https://github.com/evcc-io/evcc/discussions/15471
https://docs.evcc.io/en/blog#ocpp-stability-features--sponsoring
https://docs.evcc.io/en/docs/devices/chargers#ocpp-16j-compatible-charger-with-smart-charging-profile

Please do not mix up everything or I will just ignore this issue.

premultiply, 2024-08-20:
> I get the warning "waiting for local authentication" and I guess that the ocpp-pulsarplus template needs to be updated to set remotestart: true by default because I cannot find any of those indicated options like "Autostart" in Pulsar's App).

Just do it and validate if all of your issues are gone?

typxxi, 2024-08-20:
Same issue here, lots of new errors never seen before

typxxi, 2024-08-20:
For all those struggling with Pulsar Plus which can not find an autostart or whatsoever in the app nor in the my.wallbox.com/chargers webui there is a solution for evcc.yaml found in the [latest blog post from august 17th](https://docs.evcc.io/en/blog#rfid-and-authorization) in the rfid and authorisation topic which I have linked here to find a rather quickly.

remotestart: true

premultiply, 2024-08-20:
https://docs.evcc.io/en/docs/devices/chargers#ocpp-16j-compatible-charger-with-smart-charging-profile