evcc-io/evcc #15540: Shelly smart plugs with attached vehicle are always heaters

pohlinkzei, 2024-08-20:
### Describe the bug

My setup: Shelly 3EM as grid meter, Shelly Plus Plug S as switch socket for the OEM AC charger brick and a Shelly 1PM for a Tesla UMC2 (16A). Since the Solar power is not online yet, I just play around with evcc for easy "charge-to-80%". It is working very well, but the Shelly Plug is always showing temperature and not SoC. After adding the Shelly 1PM as second load point, it also shows temperature, but neither is integrated_device set nor heater in the config file.
[image]

### Steps to reproduce

Its that way from the very beginning using just a shelly Plus Plug S. The config settings 'heater' and 'integrateddevice' seem to be ignored.

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: 67b3f3821860ff6393c85845baa87dc93d22896c6c71253c55619dc02d60a82a

interval: 30s # control cycle interval

meters:
- type: template
  template: shelly-3em
  usage: grid
  host: 192.168.178.30
  name: grid1

chargers:
- type: template
  template: shelly
  host: 192.168.178.72
  channel: 0
  standbypower: 15
  integrateddevice: false
  heating: false
  icon: car
  name: wallbox3
- type: template
  template: shelly
  host: 192.168.178.71
  channel: 0
  standbypower: 15
  integrateddevice: false
  heating: false
  icon: car
  name: wallbox4

vehicles:
- type: template
  template: nissan
  title: Nissan
  icon: car
  user: *****
  password: *****
  vin: *****
  capacity: 40
  phases: 1
  cache: 15m
  mode: now
  minCurrent: 6
  maxCurrent: 20
  name: ev2
- type: template
  template: nissan
  title: Nissan
  icon: car
  user: *****
  password: *****
  vin: *****
  capacity: 40
  phases: 1
  cache: 15m
  mode: now
  minCurrent: 6
  maxCurrent: 20
  name: ev1

loadpoints:
- title: Garage
  charger: wallbox3
  vehicle: ev2
  mode: now
- title: Hof
  charger: wallbox4
  vehicle: ev1
  mode: now

site:
  title: Zuhause
  meters:
    grid: grid1

influx:
  url: http://localhost:8086
  database: evcc
  user:  evcc
  password: evcc
```

### Log details

```text
I dont know what logs to collect. Couldnt find anything about Heater or temperature in the Logs.
```

### What type of operating system are you running?

Linux

### Version

v0.129.0

andig, 2024-08-20:
Was sagt denn /api/state?

pohlinkzei, 2024-08-20:
{"result":{"auth":{},"availableVersion":"0.130.1","battery":[],"batteryDischargeControl":false,"batteryMode":"unknown","bufferSoc":0,"bufferStartSoc":0,"currency":"EUR","fatal":null,"greenShareHome":0,"greenShareLoadpoints":0,"gridConfigured":true,"gridCurrents":[0.53,19.42,0.89],"gridEnergy":213.032,"gridPower":4779.69,"gridPowers":[73.59,4550.63,155.47],"homePower":2461.69,"influx":{"url":"http://localhost:8086","database":"evcc","org":"","user":"evcc"},"interval":30,"loadpoints":[{"chargeCurrent":10,"chargeDuration":4440,"chargePower":2318,"chargeRemainingDuration":17521,"chargeRemainingEnergy":11281.517,"chargeTotalImport":46.797,"chargedEnergy":2890.82,"chargerFeatureHeating":true,"chargerFeatureIntegratedDevice":true,"chargerIcon":"car","chargerPhases1p3p":false,"chargerPhysicalPhases":1,"charging":true,"connected":true,"connectedDuration":0,"disableThreshold":0,"effectiveLimitSoc":80,"effectiveMaxCurrent":10,"effectiveMinCurrent":6,"effectivePlanSoc":0,"effectivePlanTime":null,"effectivePriority":0,"enableThreshold":0,"enabled":true,"limitEnergy":20,"limitSoc":80,"maxCurrent":10,"minCurrent":6,"mode":"now","phaseAction":"inactive","phaseRemaining":0,"phasesActive":1,"phasesConfigured":3,"phasesEnabled":3,"planEnergy":0,"planOverrun":0,"planProjectedEnd":null,"planProjectedStart":null,"planTime":null,"priority":0,"pvAction":"inactive","pvRemaining":0,"sessionCo2PerKWh":null,"sessionEnergy":2890.82,"sessionPrice":null,"sessionPricePerKWh":null,"sessionSolarPercentage":0,"smartCostActive":false,"smartCostNextStart":null,"title":"Garage","vehicleClimaterActive":null,"vehicleDetectionActive":false,"vehicleLimitSoc":0,"vehicleName":"ev2","vehicleOdometer":0,"vehicleRange":143,"vehicleSoc":54.617,"vehicleWelcomeActive":false},{"chargeCurrent":16,"chargeDuration":0,"chargePower":0,"chargeRemainingEnergy":11555.556,"chargeTotalImport":6.182,"chargedEnergy":0,"chargerFeatureHeating":true,"chargerFeatureIntegratedDevice":true,"chargerIcon":"car","chargerPhases1p3p":false,"chargerPhysicalPhases":1,"charging":false,"connected":true,"connectedDuration":0,"disableThreshold":0,"effectiveLimitSoc":80,"effectiveMaxCurrent":16,"effectiveMinCurrent":6,"effectivePlanSoc":0,"effectivePlanTime":null,"effectivePriority":0,"enableThreshold":0,"enabled":true,"limitEnergy":0,"limitSoc":80,"maxCurrent":16,"minCurrent":6,"mode":"now","phaseAction":"inactive","phaseRemaining":0,"phasesActive":1,"phasesConfigured":3,"phasesEnabled":3,"planEnergy":0,"planOverrun":0,"planProjectedEnd":null,"planProjectedStart":null,"planTime":null,"priority":0,"pvAction":"inactive","pvRemaining":0,"sessionCo2PerKWh":null,"sessionEnergy":0,"sessionPrice":null,"sessionPricePerKWh":null,"sessionSolarPercentage":0,"smartCostActive":false,"smartCostNextStart":null,"title":"Hof","vehicleClimaterActive":null,"vehicleDetectionActive":false,"vehicleLimitSoc":0,"vehicleName":"ev1","vehicleOdometer":0,"vehicleRange":143,"vehicleSoc":54,"vehicleWelcomeActive":false}],"maxGridSupplyWhileBatteryCharging":0,"mqtt":{"broker":"","topic":"evcc","user":"","clientID":"","insecure":false},"network":{"schema":"http","host":"evcc.local","port":7070},"prioritySoc":0,"pv":[],"pvPower":0,"residualPower":0,"siteTitle":"Zuhause","smartCostType":null,"sponsor":{"name":"","expiresAt":"0001-01-01T00:00:00Z"},"statistics":{"30d":{"avgCo2":0,"avgPrice":0,"chargedKWh":131.146693,"solarPercentage":0},"365d":{"avgCo2":0,"avgPrice":0,"chargedKWh":131.146693,"solarPercentage":0},"total":{"avgCo2":0,"avgPrice":0,"chargedKWh":131.146693,"solarPercentage":0}},"vehicles":{"ev1":{"title":"Nissan","icon":"car","capacity":40},"ev2":{"title":"Nissan","icon":"car","capacity":40}},"version":"0.129.0"}}

andig, 2024-08-21:
Ich sehe (bzw. man könnte es sehen wenn es formatiert wäre):

    "chargerFeatureHeating": true,
    "chargerFeatureIntegratedDevice": true,

naltatis, 2024-08-21:
Meine Vermutung ist, dass wir hier ein Parsing/Interpretationsproblem haben. Lösch mal die Zeilen.

```
  integrateddevice: false
  heating: false
```

naltatis, 2024-08-21:
Ja, ist wie vermutet. Beim Rendern der Templates gehen uns die [Typ-Informationen](https://github.com/evcc-io/evcc/blob/master/util/templates/defaults.yaml#L336-L337) verloren und wir prüfen an dieser Stelle nur auf Existenz des Parameters, aber nicht auf seinen wirklichen Wert:

https://github.com/evcc-io/evcc/blob/master/util/templates/includes/switchsocket.tpl#L4

`integrateddevice` ist hier vom Typ `string` und hat den Wert `"false"`.

@andig Magst du da mal reinschauen? Ich hab’ auf die Schnelle nicht die richtige Stelle gefunden, um das zu fixen.

andig, 2024-08-21:
Der Fix is (somewhat) klar. Der Code ist allerdings 9 Monate alt. @pohlinkzei ging das in einer alten Version mal "richtig"?

pohlinkzei, 2024-08-21:
Nein, ich bin neuuser.

VolkerK62, 2024-08-21:
Ich vermute schwer, das ist bisher nur nicht aufgefallen, weil niemand `false` benutzt hat.
Wenn man es nicht braucht, trägt man es gar nicht erst ein.

andig, 2024-08-21:
Closed in https://github.com/evcc-io/evcc/commit/aca405b37324783da144658d1bc85a6b012d327f

closed 2024-08-21 by andig

pohlinkzei, 2024-08-21:
Die config habe ich größtenteils mit dem evcc -config (oder so) erstellt, die die beiden false Werte eigefügt hat.

VolkerK62, 2024-08-21:
stimmt, kommt im advanced mode. Sorry, hab ich vorher noch nie ausprobiert