evcc-io/evcc #15542: seit 0.130.x werden nach Abzug des Ladekabels falsche Werte beim ID.4 angezeigt

CarG70, 2024-08-20:
### Describe the bug

wie bereits in der Diskussion unter #15522 mit @StefanSchoof geschrieben.
1. neuerdings wir beim Fahrzeug im UI dauerhaft die Reichweite angezeigt.
[image]

3. während des Ladevorgangs ist alles wie gewünscht.
4. nach dem Beenden des Ladevorgangs werden beim Ladestand die aktuell zugeladenen Prozente und die aktuelle Gesamtreichweite angezeigt. Anzeige rechts unter Ladelimit Ziel-SoC und eine falsch berechnete Zielreichweite.
[image]
5. Log zeigt einen falschen SoC von 0% obwohl die evcc vehicle die richtigen Werte liefert.
[image]
[image]
6. Evcc Neustart entfernt die falsche Anzeige, jedoch bleibt die aktuelle Reichweite dauerhaft angezeigt.

### Steps to reproduce

1. Auto laden
2. Laden beenden
3. Anzeige prüfen
...

### Configuration details

```yaml
wird bei Bedarf nachgeliefert, sollte aber nicht relevant sein.
```

### Log details

```text
kirzer Ausschnitt siehe oben
```

### What type of operating system are you running?

Linux

### Version

0.130.1

andig, 2024-08-20:
Unfortunately your report is missing a detailed log file. Without log, we don‘t have the information for diagnosing the issue described. See https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen.

CarG70, 2024-08-21:
anbei 3 Logfiles, die die Problematik zeigen sollten:
**Log 01: Fahrzeug nicht verbunden.**
ab 14:44:34 werden regelmäßig vehicle sic, vehicle soc limit und vehicle range abgefragt. Vehicle Soc wir immer mit 0% ausgegeben. Im UI wird unter dauerhaft der Ladestand angezeigt (auch wenn das Fahrzeug nicht lädt.
**Log 02: Fahrzeug wird mit Wallbox verbunden und lädt.**
ab 15:02:33 passt dann beim Laden der vehicle soc mit 75%. Das Laden erfolgt wie erwartet/gewünscht.
**Log 03: Laden beendet.**
15:31:32 ist der vehicle soc bei 77%
15:32:32 ist der vehicle soc bei 3% und wird dauerhaft als Ladestand im UI angezeigt.

bis 0.130.1 wurden zusätzlich noch die 3% mit Reichweite 388km kombiniert und daraus 10.346,66 km für 80% angezeigt.
Mich verwundert grundsätzlich, dass die Reichweite nun dauerhaft (und nicht nur beim Laden) angezeigt wird, da ja damit ständig die API abgefragt wird.
[Log_01_ohne Auto evcc-20240821-144545-debug.log](https://github.com/user-attachments/files/16692633/Log_01_ohne.Auto.evcc-20240821-144545-debug.log)
[Log_02_Laden evcc-20240821-150411-debug.log](https://github.com/user-attachments/files/16692634/Log_02_Laden.evcc-20240821-150411-debug.log)
[Log_03_Laden beendet evcc-20240821-153244-debug.log](https://github.com/user-attachments/files/16692635/Log_03_Laden.beendet.evcc-20240821-153244-debug.log)

andig, 2024-08-21:
Ich sehe nur ein Issue:

```
[lp-1  ] DEBUG 2024/08/21 15:32:05 vehicle soc: 77%
...
[eebus ] DEBUG 2024/08/21 15:32:05 !! OpEV.LoadControlLimits set: {a false true 6}
[lp-1  ] DEBUG 2024/08/21 15:32:05 pv charge current: min 6A > 3.58A (-2473W @ 3p, battery: true)
[eebus ] DEBUG 2024/08/21 15:32:06 Send: d:_i:47859_Elli-Wallbox-2045A11TNB notify 1003 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/08/21 15:32:07 Recv: d:_i:47859_Elli-Wallbox-2045A11TNB:NodeManagement to NodeManagement notify 1451 NodeManagementUseCaseData
[eebus ] DEBUG 2024/08/21 15:32:07 Recv: d:_i:47859_Elli-Wallbox-2045A11TNB:NodeManagement to NodeManagement notify 1452 NodeManagementDetailedDiscoveryData
[eebus ] DEBUG 2024/08/21 15:32:07 ski 829ce7814380c70379edd6a82dd6cf28b1d31ad1 event cem-evcc-EvDisconnected
[eebus ] DEBUG 2024/08/21 15:32:07 ski 829ce7814380c70379edd6a82dd6cf28b1d31ad1 event cem-evcem-UseCaseSupportUpdate
[eebus ] DEBUG 2024/08/21 15:32:07 ski 829ce7814380c70379edd6a82dd6cf28b1d31ad1 event cem-opev-UseCaseSupportUpdate
[eebus ] DEBUG 2024/08/21 15:32:07 ski 829ce7814380c70379edd6a82dd6cf28b1d31ad1 event cem-evcc-UseCaseSupportUpdate
[eebus ] DEBUG 2024/08/21 15:32:07 ski 829ce7814380c70379edd6a82dd6cf28b1d31ad1 event cem-oscev-UseCaseSupportUpdate
[site  ] DEBUG 2024/08/21 15:32:32 ----
...
[lp-1  ] DEBUG 2024/08/21 15:32:32 charger status: A
[lp-1  ] INFO 2024/08/21 15:32:32 car disconnected
[lp-1  ] DEBUG 2024/08/21 15:32:32 set session soc limit: 0
[lp-1  ] DEBUG 2024/08/21 15:32:32 set session energy limit: 0
[lp-1  ] DEBUG 2024/08/21 15:32:32 soc estimated: 3.09% (vehicle: 0.00%)
[lp-1  ] DEBUG 2024/08/21 15:32:32 vehicle soc: 3%
```

Der Estimator scheint aufgrund Poll-Mode "always" weiter zu laufen, obwohl das Fahrzeug abgeklemmt ist.

@premultiply müssen wir hier noch was ergänzen?

CarG70, 2024-08-21:
ich habe aber keinen Poll-Mode "always" eingerichtet.
Zur Vollständigkeit hier noch meine yaml.
[config.txt](https://github.com/user-attachments/files/16695384/config.txt)

andig, 2024-08-21:
Ah, stimmt. Hier ist die Elli der Auslöser:

    lp.chargerHasFeature(api.IntegratedDevice) || lp.vehicleSocPollAllowed()

In beiden Fällen wird dann der Estimator gefüttert- was bei `StatusA` aber nicht soviel Sinn macht:

    lp.socEstimator.Soc(lp.getChargedEnergy())

CarG70, 2024-08-21:
bis einschl. 0.129.0 lief meine Konfiguration relativ problemlos. Trotz Elli. Wenn das Auto nicht angeschlossen ist, muss ich auf der Übersicht unter Ladestand nicht unbedingt Reichweite/SoC sehen. War das von irgendjemanden gewünscht?

andig, 2024-08-21:
Ahh, das war der Hinweis. Ich habs beim Umbau kaputt gemacht:

```go
// Soc implements the api.Battery interface
func (c *EEBus) Soc() (float64, error) {
	evEntity, ok := c.isEvConnected()
	if !ok {
		return 0, nil
	}

	if !c.uc.EvSoc.IsScenarioAvailableAtEntity(evEntity, 1) {
		return 0, api.ErrNotAvailable
	}

	soc, err := c.uc.EvSoc.StateOfCharge(evEntity)
	if err != nil {
		return 0, api.ErrNotAvailable
	}

	return soc, nil
}
```

Die erste Abfrage hier kommt ohne Fehler zurück, sollte sie aber.

/cc @DerAndereAndi

closed 2024-08-21 by andig