evcc-io/evcc #15546: Easee Wallbox stoppt gelegentlich den Ladevorgang wegen eines zu hohen Ladestromes

iseeberg79, 2024-08-20:
### Describe the bug

Gelegentlich wird der Ladevorgang abgebrochen, weil die Easee Wallbox in einen Fehlerzustand (invalid opmode=5) gerät. Diese lässt sich nur durch entfernen des Ladesteckers aufheben. In der Regel kann der Ladevorgang nach zurücksetzen des Ladevorgangs erfolgreich beendet werden.
Der Fehler trat vor einiger Zeit bereits gelegentlich auf, konnte dann aber mit der Einstellung "zweiphasiges Laden" am Fahrzeug bislang erfolgreich vermieden werden. Seit kurzer Zeit tritt der Fehler jedoch wieder auf.
Das Fahrzeug (e-Golf) unterstützt nur zweiphasiges Laden mit 7.2kW, während die Wallbox mit 11kW konfiguriert und installiert ist. In der Konfiguration ist das Fahrzeug mit zwei Phasen und einer Ladeleistung von mindestens 5A, die Wallbox mit automatischer Phasenumschaltung konfiguriert.

### Steps to reproduce

1. Verbinden des Fahrzeugs am Abend zur Ladung im PV-Modus am nächsten Tag
2. der Ladevorgang startet, nachdem der Speicher einen Mindestladezustand erreicht hat und genug Überschuss vorhanden ist
3. der Ladevorgang wird gelegentlich - vermeindlich nach Reduzierung und Erhöhung der Anzahl der Phasen, mit dem benannten Fehler abgebrochen
4. erst das Ladekabel von der Wallbox zu trennen, behebt den Fehlerzustand und eine Fortsetzung des Ladevorgangs ist meist ohne weitere Unterbrechung bis zum Erreichen des maximalen Ladezustandes möglich 

### Configuration details

```yaml
(..)
log: info 
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  easee: trace

interval: 30s
(..)
vehicles:
- type: template
  template: id
  title: e-Golf
  user: xxxxx@xxx.yy
  password: xxxxxxxxxxxxxx
  vin: WVWZZZXXXXXXXXX  
  capacity: 35.8  
  name: e-Golf
  mincurrent: 5
  phases: 2
(..)
chargers:
  - name: Wallbox
    type: template
    template: easee
    user: xxxxxx@xxx.yy
    password: xxxxxxx
    charger: EH35WRYQ
(..)
loadpoints:
  - title: Garage
    charger: Wallbox
    mode: pv
    soc:
      poll:
        mode: always
        interval: 60m
      estimate: true
    phases: 0
    enable: 
      delay: 1m
      threshold: 0
    disable:
      delay: 3m
      threshold: 800
(..)
```

### Log details

```text
Aug 16 15:38:16 unifi evcc[1712070]: [site  ] DEBUG 2024/08/16 15:38:16 battery soc: 92%
Aug 16 15:38:16 unifi evcc[1712070]: [site  ] DEBUG 2024/08/16 15:38:16 battery power: -394W
[... 76 lines cut ...]
Aug 16 15:38:59 unifi evcc[1712070]: [easee ] TRACE 2024/08/16 15:38:59 ProductUpdate EH35WRYQ: (2024-08-16 13:38:56 +0000 UTC) OUTPUT_CURRENT 0
Aug 16 15:38:59 unifi evcc[1712070]: [easee ] TRACE 2024/08/16 15:38:59 ProductUpdate EH35WRYQ: (2024-08-16 13:38:57 +0000 UTC) CIRCUIT_TOTAL_ALLOCATED_PHASE_CONDUCTOR_CURRENT_L1 0
Aug 16 15:39:00 unifi evcc[1712070]: [easee ] TRACE 2024/08/16 15:39:00 ProductUpdate EH35WRYQ: (2024-08-16 13:38:59 +0000 UTC) OUT_VOLT_PIN1_3 0.9240000247955322
Aug 16 15:39:10 unifi evcc[1712070]: [easee ] TRACE 2024/08/16 15:39:10 ProductUpdate EH35WRYQ: (2024-08-16 13:39:07 +0000 UTC) CIRCUIT_TOTAL_PHASE_CONDUCTOR_CURRENT_L3 0.009999999776482582
Aug 16 15:39:15 unifi evcc[1712070]: [site  ] DEBUG 2024/08/16 15:39:15 ----
Aug 16 15:39:15 unifi evcc[1712070]: [lp-1  ] ERROR 2024/08/16 15:39:15 charge power: invalid opmode: 5
Aug 16 15:39:15 unifi evcc[1712070]: [lp-1  ] DEBUG 2024/08/16 15:39:15 charge currents: [0.446 0.012 0.02]A
Aug 16 15:39:16 unifi evcc[1712070]: [site  ] DEBUG 2024/08/16 15:39:16 pv power: 3798W
Aug 16 15:39:16 unifi evcc[1712070]: [site  ] DEBUG 2024/08/16 15:39:16 battery soc: 92%
Aug 16 15:39:16 unifi evcc[1712070]: [site  ] DEBUG 2024/08/16 15:39:16 battery power: -3218W
Aug 16 15:39:16 unifi evcc[1712070]: [site  ] DEBUG 2024/08/16 15:39:16 grid meter: 6W
Aug 16 15:39:16 unifi evcc[1712070]: [site  ] DEBUG 2024/08/16 15:39:16 site power: -3412W
Aug 16 15:39:16 unifi evcc[1712070]: [lp-1  ] DEBUG 2024/08/16 15:39:16 !! session: chargeRater.chargedEnergy=6.9 - chargedAtStartup=0.0
Aug 16 15:39:16 unifi evcc[1712070]: [lp-1  ] DEBUG 2024/08/16 15:39:16 charge total import: 3821.256kWh
Aug 16 15:39:16 unifi evcc[1712070]: [lp-1  ] ERROR 2024/08/16 15:39:16 charger status: invalid opmode: 5
Aug 16 15:39:45 unifi evcc[1712070]: [site  ] DEBUG 2024/08/16 15:39:45 ----
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.129.0 (9ad7399b) and before

andig, 2024-08-20:
@GrimmiMeloni darf ich dich hier wieder um Hilfe bitten?

GrimmiMeloni, 2024-08-21:
@iseeberg79 das muß sich der Easee Support anschauen. Der Charger meldet hier einen Fehler bzgl. Überspannung am Fahrzeug ...

```
Aug 16 15:38:54 unifi evcc[1712070]: [easee ] TRACE 2024/08/16 15:38:54 ProductUpdate EH35WRYQ: (2024-08-16 13:38:52 +0000 UTC) ERROR_STRING [ERROR] Charger: Overcurrent EV=14514 > 6000 mA
Aug 16 15:38:54 unifi evcc[1712070]: [easee ] TRACE 2024/08/16 15:38:54 ProductUpdate EH35WRYQ: (2024-08-16 13:38:52 +0000 UTC) ERROR_CODE 2048
```

... und schaltet danach ab, was Ihn in den OpMode 5 (Error) führt. Daher ist der in evcc gemeldete Fehler erstmal kein evcc Problem. Sieht für mich nach einem Timing Problem im Charger aus. Dieser bestätigt einerseits 6A current und sieht kurz danach noch 14A am Auto. Das darf so ja nicht sein, denn es würde ja bedeuten der Charger hat gar nicht runter geregelt.

closed 2024-08-21 by andig

iseeberg79, 2024-08-21:
Hallo, erstmal ein dankeschön für's wirklich schnelle ansehen! Ich werden Kontakt zum Easee Support aufnehmen. Ehrlicherweise hatte ich die merkwürdige Umschaltung auf 3-phasiges Laden im Verdacht. Die Umschaltung hätte eigentlich nicht passieren dürfen, da das Fahrzeug die Konfiguration nicht akzeptiert. Immer wenn der Fehler auftritt, ist in EVCC der Ladestatus opmode=5 mit einer Konfiguration für 3-phasiges Laden. Der Fehler bzgl. des Overcurrent tritt ja auf, nachdem die Wallbox durch EVCC statt in 2-phasiges Laden in 3-phasiges gesetzt wird?
Ich habe noch mehr Zeilen aus dem Protokoll (ca. 10 Minuten), dort erkennt man den Wechsel der Verwendung einer Phase zu 2, zurück 1-phasig und dann ins 3-phasige Laden mit dem anschliessenden Fehler. Ich hatte bewusst das Prokoll auf die zentrale Stelle des Umschaltens konzentriert. 
Aber wenn die Aussage lautet, die Umschaltung auf 3-phasiges Laden ist nicht ursächlich für den Fehler, gebe ich das an den Easee Support weiter.

GrimmiMeloni, 2024-08-21:
> Der Fehler bzgl. des Overcurrent tritt ja auf, nachdem die Wallbox durch EVCC statt in 2-phasiges Laden in 3-phasiges gesetzt wird?

Es geht nicht um die Anzahl der Phasen. So wie ich es sehe, ist das Problem das die Wallbox eine Reduzierung auf 6A bestätigt

```
Aug 16 15:38:16 unifi evcc[1712070]: [easee ] TRACE 2024/08/16 15:38:16 POST https://api.easee.com/api/chargers/EH35WRYQ/settings
Aug 16 15:38:17 unifi evcc[1712070]: [easee ] TRACE 2024/08/16 15:38:17 {"dynamicChargerCurrent":6}
Aug 16 15:38:17 unifi evcc[1712070]: --
Aug 16 15:38:17 unifi evcc[1712070]: [{"device":"EH35WRYQ","commandId":48,"ticks":638594122971046500}]
Aug 16 15:38:18 unifi evcc[1712070]: [easee ] TRACE 2024/08/16 15:38:18 ProductUpdate EH35WRYQ: (2024-08-16 13:38:16 +0000 UTC) DYNAMIC_CHARGER_CURRENT 6
Aug 16 15:38:18 unifi evcc[1712070]: [easee ] TRACE 2024/08/16 15:38:18 CommandResponse EH35WRYQ: {SerialNumber:EH35WRYQ ID:48 Timestamp:2024-08-16 13:38:17.10465 +0000 UTC DeliveredAt:2024-08-16 13:38:16
.554 +0000 UTC WasAccepted:true ResultCode:0 Comment: Ticks:638594122971046500}
```

... und ferner diese auch im Anschluß entsprechende Soll-Werte meldet:

```
Aug 16 15:38:19 unifi evcc[1712070]: [easee ] TRACE 2024/08/16 15:38:19 ProductUpdate EH35WRYQ: (2024-08-16 13:38:17 +0000 UTC) OUTPUT_CURRENT 6
Aug 16 15:38:19 unifi evcc[1712070]: [easee ] TRACE 2024/08/16 15:38:19 ProductUpdate EH35WRYQ: (2024-08-16 13:38:17 +0000 UTC) CIRCUIT_TOTAL_ALLOCATED_PHASE_CONDUCTOR_CURRENT_L1 6
```

.... und dann aber selbst 30 Sekunden später die Physik immer noch 1-phasig und vorallem mit 14.1 A lädt.

```
Aug 16 15:38:45 unifi evcc[1712070]: [lp-1  ] DEBUG 2024/08/16 15:38:45 charge currents: [14.1 0.012 0.02]A
...
Aug 16 15:38:54 unifi evcc[1712070]: [easee ] TRACE 2024/08/16 15:38:54 ProductUpdate EH35WRYQ: (2024-08-16 13:38:52 +0000 UTC) ERROR_STRING [ERROR] Charger: Overcurrent EV=14514 > 6000 mA
Aug 16 15:38:54 unifi evcc[1712070]: [easee ] TRACE 2024/08/16 15:38:54 ProductUpdate EH35WRYQ: (2024-08-16 13:38:52 +0000 UTC) ERROR_CODE 2048
```

Was anscheinend auch korrekt ist, denn, zumindest der letzte Messwert zur Spannung ist 2 Minuten alt (was nichts heißt, passiert hier auch schon mal) - und somit immer noch 14A. 

```
Aug 16 15:36:33 unifi evcc[1712070]: [easee ] TRACE 2024/08/16 15:36:33 ProductUpdate EH35WRYQ: (2024-08-16 13:36:32 +0000 UTC) IN_CURRENT_T3 14.060999870300293
```

> Oder ist mein Verständnis falsch und es wurde in "Aug 16 15:38:16" nicht umgeschaltet?

Laut Wallbox wurde umgeschaltet. Um genau zu sein, wurde auf dem Circuit auf den Phasen L2 und L3 der MaxCurrent auf 16A gesetzt

```
Aug 16 15:38:18 unifi evcc[1712070]: [easee ] TRACE 2024/08/16 15:38:18 ProductUpdate EH35WRYQ: (2024-08-16 13:38:16 +0000 UTC) DYNAMIC_CIRCUIT_CURRENT_P1 16
Aug 16 15:38:18 unifi evcc[1712070]: [easee ] TRACE 2024/08/16 15:38:18 ProductUpdate EH35WRYQ: (2024-08-16 13:38:16 +0000 UTC) DYNAMIC_CIRCUIT_CURRENT_P3 16
Aug 16 15:38:18 unifi evcc[1712070]: [easee ] TRACE 2024/08/16 15:38:18 ProductUpdate EH35WRYQ: (2024-08-16 13:38:16 +0000 UTC) DYNAMIC_CIRCUIT_CURRENT_P2 16
```

Fazit: Die Phasenschaltung ist vielleicht nur Beiwerk. Für mich sieht es eher danach aus, daß die Box nicht mehr der Anweisung zum dynamischen Spannungswechsel folgt, und dann irgendwann ein Watchdog(?) innerhalb der Box zuschlägt. Die Fehlermeldung sagt ja auch genau das aus. Selbst die Box ist der Meinung es sollten nur 6A maximum sein, die Box glaubt aber gleichzeitig am eigenen Meter noch 14A zu messen, und schaltet ab.

iseeberg79, 2024-08-21:
Danke für die ausführliche Erklärung. Das hilft mir im Umgang und der Argumentation mit dem Easee Support Das Ticket ist erstellt. Ich frage mich jedoch auch noch, warum überhaupt auf mehr als 1-Phase umgeschaltet wurde? Wir haben hier, zumindest verstehe ich das aus den Logs so, einen Grenzfall der Umschaltung? Eigentlich ist kaum genug Leistung für das triggern der Umschaltung auf mehr als eine Phase. Mehrfach reicht der Wert nicht aus, einmal ist er dann höher als die Schwelle. Dann wieder niedriger, es wird aber trotzdem umgeschaltet und nicht, wie ich erwartet hätte, der Timer zurückgesetzt und bei einphasigem Laden geblieben. Direkt nach der Umschaltung wird dann wieder ein niedriger Leistungswert erwartet, der auch einphasig (14A < 16A) noch erreicht werden könnte.

GrimmiMeloni, 2024-08-21:
Aus den Logs geht hervor, daß über den Zeitraum von 1 Minute (das ist der PV Timer) über dem Schwellwert von 6a@3p Leistung verfügbar ist. Fogllich erfolgt die Umschaltung.