evcc-io/evcc #15547: XPENG Support

rgries-debug, 2024-08-20:
**Is your feature request related to a problem? Please describe.**
I want to ask whether you are on implementing the vehicles of the brand XPENG. I‘m interested in buying an ev of this brand and using the great application evcc.

andig, 2024-08-20:
Gibts eine Doku zur Schnittstelle?

rgries-debug, 2024-08-20:
Eine gute Frage. Noch bin ich ja nur interessiert. Ich weiß bisher nur, das es eine APP gibt. Ob die eine API haben, die dokumentiert und für die Kunden offen ist, ist mir bis dato nicht bekannt.Mit freundlichen Gruß,Ralf Grieser

andig, 2024-08-21:
Gerne melden wenn es Informationen gibt. Uns ist bisher nichts bekannt.

closed 2024-08-21 by andig

NCO3, 2024-12-25:
I would be interested as well as I am about to purchase an XPeng G6
According to the XPeng support, there is no API in place yet.
I’m still trying to find out if there is something in the works or planned for the future.

Silencer80, 2025-03-15:
Hallo zusammen,

Ich schließe mich dem Wunsch nach einer Integration von XPeng einmal 
freundlichst an.

Aufgrund von "Meinungsverschiedenheiten mit dem aktuellen Chef von Tesla"
gebe ich im Dezember zum Leasing-Ende mein Model 3 LR auf und suche aktuell 
nach Alternativen. Aufgrund der Konzeption der Fahrzeuge tendiere ich aktuell 
zu XPeng und favorisiere den G6 oder den P7+ (falls der im Dezember in der EU
verfügbar sein sollte.).

Ein freundlicher Händler ermöglicht mir vom 28.03-30.03 eine ausführliche 
Probefahrt des G6 Performance, danach weiß ich ob das Fahrzeug meine 
Erwartungen erfüllen kann.

Aktuell habe ich mein Model 3 via BLE-Proxy integriert und lade 
über meine Elli Gen. 1 Wallbox. 

Sollte bei XPeng wirklich jegliche Fahrzeugintegration verloren gehen,
ist das schon fast das stärkste Argument gegen XPeng.

Ich habe mich ebenfalls mal an den Support gewendet und nach einer API
gefragt. Selbige muss es ja geben, da Enode sie benutzt.

Folgende News lässt mich aber glauben, das XPeng eine exklusive 
Vereinbarung haben könnte und das eine weiter Öffnung nicht
vorgesehen sein könnte:

https://enode.com/blog/Partnerships/xpeng-innovating-enode-partnership

Vielen Dank!
Boris

NCO3, 2025-03-15:
Gratulation zu der Entscheidung keinen Tesla mehr zu fahren.
Ich hoffe auch sehr, dass XPeng sich da bewegt.
Dann hätten Sie gute Chancen Tesla in Europa abzulösen.

Silencer80, 2025-03-15:
> Gratulation zu der Entscheidung keinen Tesla mehr zu fahren. Ich hoffe auch sehr, dass XPeng sich da bewegt. Dann hätten Sie gute Chancen Tesla in Europa abzulösen.

Vielleicht hilft es ja, wenn möglichst viele nach einer API fragen? Das Anfrageformular ist auf der Webseite und die Anfragen werden bestimmt in München gelesen.

andig, 2025-03-15:
Ein Herstellerapi zu unterstützen wäre deutlich charmanter als einen weiteren Bezahlservice hinzu zu fügen.

NCO3, 2025-03-16:
@Silencer80 
Das habe ich schon nach meiner ersten Probefahrt gemacht 😊

Silencer80, 2025-03-17:
@NCO3 Ich habe noch eine zweite Anfrage über die APP gestellt. Bisher alles unbeantwortet. Ich bitte aber den Händler am 29.03 um Unterstützung. Wenn das alles nix bring, dann rufe ich auch einfach mal in München bei der XPeng Motors (Germany) GmbH an. 

@andig Das sehe ich genauso. Umso wichtiger ist es, den Hersteller zu überzeugen eine API rauszurücken. Vielleicht bin ich nach meiner Probefahrt auch desillusioniert, aber aktuell gehe ich davon aus, das ein G6 Performance oder ein P7+ ein guter Ersatz für meinen Tesla sein könnte. MAGNA schraubt XPeng wohl ab Juni teilweise in Österreich zusammen, was das Ganze finanziell noch attraktiver machen könnte, da Zölle entfallen.

Silencer80, 2025-03-19:
Ich habe Antwort vom XPeng Support erhalten: 
Aktuell keine API, mein Wunsch wird aber an die Produktentwicklung weitergeleitet. 
Ich habe nochmals nachgehakt und auf die API (die Enode verwendet) verwiesen. Vielleicht findet sich ja williger Unterstützer.

christophwen, 2025-03-23:
Hallo zusammen,

hier eine kurze Information von mir zu dem Thema.
Nach Rücksprache mit @andig habe ich am 05.03. versucht, Kontakt mit XPeng aufzunehmen. 
Dazu habe ich die folgende E-Mail an diese drei Adressen gesendet:

- pr@xiaopeng.com
- ir@xiaopeng.com
- xpeng_de@xiaopeng.com

> Dear Xpeng Team,
> 
> We at the evcc project develop an open-source solution for intelligent PV surplus charging management for electric vehicles. Many of our users are interested in integrating Xpeng vehicles into their energy management systems to optimize charging in combination with their photovoltaic installations.
> 
> I am personally considering purchasing an Xpeng P7 and would like to explore the possibility of integrating it directly into evcc. A public discussion has mentioned "enode" as an API, but we are specifically looking for a direct interface from Xpeng to enable an efficient and sustainable integration.
> 
> Is there an official way to access vehicle data, either through an API or another interface? Additionally, would it be possible to directly access the same interface that enode is using? If so, what are the access requirements, and how can we, as an open-source project, implement such an integration?
> 
> If this inquiry is directed to the wrong department or person, I would appreciate it if you could kindly point me to the right contact.
> 
> I would appreciate any further information and the opportunity to discuss potential collaboration.
> 
> Thank you for your support!

Leider ist noch keine Antwort zurückgekommen und ich denke, dass sich da auch nicht mehr viel tun wird. 
Andere oder bessere Anlaufstellen habe ich ungünstigerweise nicht gefunden.
Wenn ihr noch eine Anlaufstelle wisst, würde ich auch dort hinschreiben. Danke!

Silencer80, 2025-03-23:
@christophwen 

Finde ich ganz toll! Ich hatte Kontakt zu customerservice@heyxpeng.com. 

Leider sind die Antworten als geheim eingestuft: 

"This email and any files transmitted with it are confidential and intended solely for the use of the individual or entity to whom they are addressed."

Sinngemäß habe ich die Antworten auf meine Tickets aber oben wiedergegeben. 

Nehmt doch ebenfalls auch diese E-Mail Adresse. Dauert ein paar Tage, aber es wird beantwortet. 

Danke für euren Einsatz!
-Boris

christophwen, 2025-03-23:
Danke für den Tipp!
E-Mail ist raus.

andig, 2025-03-23:
Da bleibt erstmal nur, diese Dinger nicht zu kaufen!

Silencer80, 2025-03-23:
Ich möchte noch nicht aufgeben...

NCO3, 2025-03-24:
@Silencer80 
Ich hab auch noch einmal eine E-Mail rausgefeuert.
:-)

themuck, 2025-05-13:
Gibt es an der Baustelle Neuigkeiten?

manu-5759, 2025-05-17:
?? würde mich auch interessieren

christophwen, 2025-05-17:
Hallo zusammen,
ich habe leider nie eine Antwort auf die Anfragen an XPENG bekommen. Mehr kann ich dazu auch nicht sagen.

ChristophWeigert, 2025-05-18:
Braucht man bei nur einem E-Auto (XPeng G6) überhaupt die Daten?
Ich möchte eigentlich nur PV Laden, das sollte zu 99% für uns ausreichen.
EVCC in Verbindung mit Warp3 Pro ist aufgesetzt. Das generische Fahrzeug steht jedoch immer bei "Ladebereit. Warte auf Fahrzeug …". Eine Idee was das sein kann? 

Ich hätte zumindest gedacht, wenn ich auf Schnell wechsel, dass dann das Auto lädt (inkl. Strom vom Netz). 

`# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

# unique installation id
plant: DUMMY
sponsortoken: DUMMY

interval: 30s # control cycle interval

meters:
- type: template
  template: fronius-gen24 
  usage: grid  
  host: 192.168.1.92  
  port: 502  
  name: grid
- type: template
  template: fronius-gen24 # Fronius Symo Gen 24 Hybrid 8KW + 10.24KW BYD Akku 
  usage: pv  
  host: 192.168.1.92  
  port: 502  
  name: pv-sued
  maxacpower: 8010
- type: template
  template: fronius-gen24 
  usage: battery  
  host: 192.168.1.92  
  port: 502  
  name: battery-byd
  maxchargerate: 100
  capacity: 10.24
- name: pv-ost-west # Fronius Symo Advanced 10kw
  type: template
  template: fronius-solarapi-v1
  usage: pv
  host: 192.168.1.176
  user: customer

chargers:
- type: template
  template: tinkerforge-warp3 
  host: 192.168.1.208 # ha ip address due to mqtt server host 
  port: 1883  
  topic: warp3/2edw  
  timeout: 30s  
  name: wallbox-warp3-nord
  user: correctuser
  password: correctpassword

vehicles:
- type: template
  template: offline 
  title: XPeng G6  
  capacity: 84  
  name: xpeng-g6

loadpoints:
- title: Garage
  charger: wallbox-warp3-nord
  vehicle: xpeng-g6
  mode: pv

site:
  title: Mein Zuhause
  meters:
    grid: grid
    pv:
    - pv-sued
    - pv-ost-west
    battery:
    - battery-byd
`

themuck, 2025-05-18:
Wir haben 3x Warp2 in EVCC und es ist schon nett wenn man den Ladestand angezeigt bekommt und welches Fahrzeug gerade wo angeklemmt ist. Wir können dann eben auch anders Priorisieren wenn ein Fahrzeug halt schon 75% hat dann kann ich es Ausschalten. Sind 3 separate Haushalte ;).

ChristophWeigert, 2025-05-18:
Ja da ergibt das natürlich Sinn.
Magst du nachschauen, wie du EVCC/Warp eingestellt hast?

Ich habe gerade:

- Warp3 resettet
- Warp3 in MQTT (Home Assistant Host) eingebunden (Verbindung passt)
- Warp3 Energiemanagement > Wallboxen: Fremd­steuer­ung auf Fremdgesteuert gestellt 
- EVCC konfig (siehe oben) auch MQTT klappt

Wenn ich nun auf "Schnell" stelle, würde ich erwarten, dass vom Netz/Batterie geladen wird.

themuck, 2025-05-18:
Nein,

`Warp3 Energiemanagement > Wallboxen: Fremd­steuer­ung auf Fremdgesteuert gestellt`

ist falsch. Unter MQTT gibt es oben noch einen Reiter ;) 

`Hierdurch kann die Wallbox über den kon­figurier­ten MQTT-Broker kon­trol­liert werden. [MQTT-API-Dokumen­tation](https://www.warp-charger.com/api.html)`

Das muss aktiviert sein...

ChristophWeigert, 2025-05-18:
Danke! In der zwischenzeit hatte ich das auch abgeschalten. Die Doku dazu ist bei Tinkerforge/Warp etwas verwirrend. Erst nachdem ich das abgeschaltten hatte, war auch in evcc die automatische Phasenumschaltung sichtbar. 

Das Laden mit PV Überschuss klappt nun auch schon 👍 
Vielen Dank!

datafoxdd, 2025-05-18:
Guten Abend, 

ab demnächst steht hier nebenan ein XPeng G6 und ich wurde gebeten, mir die Einstellungen anzuschauen (bisher Basis-Wissen mit 2 Autos, Smart und Enyaq, die dort wechselseitig laden).

Wenn jetzt der G6 dazu kommt, wäre das o.g. Setting ausreichend?

 vehicles:
- type: template
  template: offline 
  title: XPeng G6  
  capacity: 84  
  name: xpeng-g6

Wäre alternativ der Hinweis aus dem [Lifehack Wiki] nutzbar(https://github.com/evcc-io/evcc/wiki/aaa-Lifehacks#start-soc-bei-einem-offline-fahrzeug-eintragen)?

Vielen Dank und Grüße!

ChristophWeigert, 2025-07-01:
So wie es oben von @Silencer80 beschrieben ist, hab ich es bei mir im Einsatz.
Hier mal mein vehicle Abschnitt:

- a.b.c durch ein Home Assistant long lived Token ersetzen
- 192.168.1.208 durch die HA IP ersetzten
- Ggf. die Sensoren anpassen, sollte aber so direkt funktionieren. 
- Bei Capacity bin ich für den G6 bei dem offiziellen Wert von 87.5 beblieben
  - EVLinkHA meldet hier 91.4 (was vermutlich wirklich die komplette Kapazität ist)

```yaml
vehicles:
- type: custom
  title: XPeng G6  
  capacity: 87.5  
  name: xpeng-g6
  icon: car
  soc: # battery soc (%)
    source: http
    uri: http://192.168.1.208:8123/api/states/sensor.evlinkha_battery_level
    method: GET
    headers:
    - authorization: Bearer a.b.c
    - content-type: application/json
    jq: .state | tonumber
  range: # optional electric range (km)
    source: http
    uri: http://192.168.1.208:8123/api/states/sensor.evlinkha_range
    method: GET
    headers:
    - authorization: Bearer a.b.c
    - content-type: application/json
    jq: .state | tonumber
  limitsoc: # optional electric range (km)
    source: http
    uri: http://192.168.1.208:8123/api/states/sensor.evlinkha_charge_limit
    method: GET
    headers:
    - authorization: Bearer a.b.c
    - content-type: application/json
    jq: .state | tonumber
```

andig, 2025-07-01:
Perfekt! Gerne PR um das in ein Template zu verpacken.

ChristophWeigert, 2025-07-01:
Das geht aber an die HA-API und nicht die von EVLinkHA. Trotzdem PR erstellen? Dann versuche ich mich die Tage mal daran.

ChristophWeigert, 2025-07-01:
Ja laut Webseite tested & verified für: 
- XPENG
- NIO

"Other supported brands"
- Audi
- BMW
- Cupra
- DS
- Hyundai
- Jaguar
- Kia
- Land Rover
- Lexus
- Mercedes-Benz
- Mini
- Nissan
- Polestar
- Porsche
- Renault
- Rivian
- SEAT
- Skoda
- Tesla
- Toyota
- Volkswagen
- Volvo