evcc-io/evcc #15548: Renault Zoe Phase 1: climater: unexpected status: 404 (Not Found)

tosog, 2024-08-20:
### Describe the bug

I'm getting this error on the gui and on the Log:

[lp-1 ] ERROR 2024/08/20 22:35:57 climater: unexpected status: 404 (Not Found)

I'm having two cars, a Skoda Enyaq and a Renault Zoe P1, the error message comes from the Zoe.
Already tried to use the latest nightly (d9dd50155f43d71821459d993022a1e8aed505a1)

credentials can be shared if needed.

### Steps to reproduce

1. ./evcc vehicle zoe
2. output:
`[main  ] INFO 2024/08/20 22:40:22 using config file: /opt/evcc/evcc.yaml                                                                                                                                                                                                                                                                                                                                                                                                                                                Soc:           33%                                                                                                                                                                                                                                          Capacity:      41.0kWh                                                                                                                                                                                                                                      Charge status: B                                                                                                                                                                                                                                            Range:         75km                                                                                                                                                                                                                                         Odometer:      53032km                                                                                                                                                                                                                                      Finish time:   not available                                                                                                                                                                                                                                Climater:      unexpected status: 404 (Not Found)                                                                                                                                                                                                           Position:      unexpected status: 404 (Not Found) `

### Configuration details

```yaml
- name: zoe                                                                                                                                                                                                                                                   type: template                                                                                                                                                                                                                                              template: renault                                                                                                                                                                                                                                           title: Zoe                                                                                                                                                                                                                                                  user: *****                                                                                                                                                                                                                                                 password: *****                                                                                                                                                                                                                                             capacity: 41                                                                                                                                                                                                                                                  #    vin: *****
```

### Log details

```text
[site ] DEBUG 2024/08/20 22:45:27 ----
[lp-1 ] DEBUG 2024/08/20 22:45:27 charge power: 0W
[lp-1 ] DEBUG 2024/08/20 22:45:27 charge currents: [0 0 0]A
[lp-2 ] DEBUG 2024/08/20 22:45:27 charge power: 0W
[lp-2 ] DEBUG 2024/08/20 22:45:27 charge currents: [0 0 0]A
[site ] DEBUG 2024/08/20 22:45:27 pv power: 0W
[site ] DEBUG 2024/08/20 22:45:27 battery soc: 64%
[site ] DEBUG 2024/08/20 22:45:27 battery power: 849W
[site ] DEBUG 2024/08/20 22:45:27 grid meter: 12W
[site ] DEBUG 2024/08/20 22:45:27 site power: 861W
[lp-2 ] DEBUG 2024/08/20 22:45:27 charge total import: 2207.391kWh
[lp-2 ] DEBUG 2024/08/20 22:45:27 charger status: A
[site ] DEBUG 2024/08/20 22:45:57 ----
[lp-1 ] DEBUG 2024/08/20 22:45:57 charge power: 0W
[lp-1 ] DEBUG 2024/08/20 22:45:57 charge currents: [0 0 0]A
[lp-2 ] DEBUG 2024/08/20 22:45:57 charge power: 0W
[lp-2 ] DEBUG 2024/08/20 22:45:57 charge currents: [0 0 0]A
[site ] DEBUG 2024/08/20 22:45:57 pv power: 0W
[site ] DEBUG 2024/08/20 22:45:57 battery soc: 64%
[site ] DEBUG 2024/08/20 22:45:57 battery power: 860W
[site ] DEBUG 2024/08/20 22:45:57 grid meter: 69W
[site ] DEBUG 2024/08/20 22:45:57 site power: 929W
[lp-1 ] DEBUG 2024/08/20 22:45:57 charge total import: 2223.258kWh
[lp-1 ] DEBUG 2024/08/20 22:45:57 charger status: B
[lp-1 ] DEBUG 2024/08/20 22:45:57 next soc poll remaining time: 20m0s
[lp-1 ] DEBUG 2024/08/20 22:45:57 pv charge current: 0A = 0A + -4.04A (929W @ 1p)
[lp-1 ] ERROR 2024/08/20 22:45:57 climater: unexpected status: 404 (Not Found)
```

### What type of operating system are you running?

Linux

### Version

d9dd50155f43d71821459d993022a1e8aed505a1

andig, 2024-08-20:
Sounds like Renault changed the api. Does anybody know how?

premultiply, 2024-08-20:
Me too - for a long time. 😁 

I don't know of any changes. Maybe Renault dropped it?

Maybe we should just drop the climater API from renault?

premultiply, 2024-08-20:
Ich habe immer gedacht nur in meinem Renault-App-Cloud-Abo-Chaos-Gedöhns wäre das nur so. Habe da aber nie näher nach geschaut weil es mir bislang einfach zu unwichtig war.

tosog, 2024-08-20:
mich "stört" nur das rote Warndreieck rechts oben :)
[image]

closed 2024-08-22 by andig

hpbock, 2024-08-25:
[image]

It looks like this in my Renault App.

WordsOfMe, 2024-08-25:
Mit meinem Zoe gab es mit evcc v0.130.2 ebenfalls keine Probleme (Position unkenntlich gemacht):
```
/app # evcc vehicle -c /config/evcc.yaml 
[main  ] INFO 2024/08/25 22:03:36 evcc 0.130.2
[main  ] INFO 2024/08/25 22:03:36 using config file: /config/evcc.yaml
Soc:            94%
Capacity:       52.0kWh
Charge status:  B
Range:          249km
Odometer:       42259km
Finish time:    2024-08-25 21:59:00 +0200 CEST
Climate active: false
Position:       ***
OnIdentified:   Mode:pv
```
Bitte die Climate-Abfrage wieder hinzufügen, kann nicht am API liegen. Eher daran, dass die "vernetzten Services" nicht gegen Geld bei Renault verlängert wurden.

tosog, 2024-08-25:
@WordsOfMe die habe ich eigentlich aktiviert. Das einzige, was ich kaufen kann, sind die TomTom Services. Brauche ich diese dafür? 

Womöglich geht das aber auch nur ab Phase 2 Zoes und meine Phase 1 ist zu alt?

[image]
[image]

WordsOfMe, 2024-08-25:
Ja, meine ist eine Phase 2 ZOE, liegt vielleicht daran. Liefert das Renault-Plugin von Home Assistant mehr Daten für deinen Account?

pellenbeck, 2024-08-28:
Ich möchte an der Stelle darauf hinweisen, dass es laut dem Renault Kundenservice Deutschland letzte Woche einen längeren (mehrtätigen) Ausfall beim Service Provider Orange in Frankreich gab und diverse Services bei Renault nicht verfügbar waren.