evcc-io/evcc #15552: OCPP EN+ AC7000-AE-45 unstable

jeffborg, 2024-08-20:
### Describe the bug

Since upgrade from 0.129.0 to 0.130.2 occp has been unstable, I removed all custom ocpp config to allow the new ocpp code to do it's thing. 

new issues on 0.130.2
1. charging will not occur without turning the wallbox on/off see log - car says no power so wallbox is disabled
2. wallbox will disconnect and not reconnect again - I think ocpp is too chatty now previous config was very stable
3. getting errors about voltages and currents not being available.

previous config for 0.129 below
```
chargers:
  - name: garage
    type: template
    template: ocpp
    timeout: 2m
    getconfiguration: false
    metervalues: Energy.Active.Import.Register,Power.Active.Import,Current.Import
    bootnotification: false
    meterinterval: 10s
```

The only way to get it working is to power cycle the wallbox and restart evcc at the same time.

### Steps to reproduce

1. restart evcc
2. wall box reconnects
3. no charging occurs

### Configuration details

```yaml
chargers:
  - name: garage
    type: template
    template: ocpp
    timeout: 2m
    remotestart: true
```

### Log details

```text
[ocpp-1] DEBUG 2024/08/21 08:17:32 waiting for chargepoint: 5m0s
[ocpp ] INFO 2024/08/21 08:17:38 charge point connected, registering: SN10052307203612
[ocpp ] TRACE 2024/08/21 08:17:38 recv SN10052307203612: [2,"159","DataTransfer",{"vendorId":"EN+","messageId":"gatewayInfo","data":"{\"SN\":\"SN10052307203612\",\"fwVer\":\"V43.0.552\",\"gateCode\":107,\"measureType\":1,\"disconnectReason\":0,\"maxCurr\":320,\"phase\":1,\"sdCardStatus\":0}"}]
[ocpp ] TRACE 2024/08/21 08:17:38 send SN10052307203612: [3,"159",{"status":"Accepted"}]
[ocpp ] TRACE 2024/08/21 08:17:39 send SN10052307203612: [2,"2635029249","GetConfiguration",{}]
[ocpp ] TRACE 2024/08/21 08:17:40 recv SN10052307203612: [3,"2635029249",{"configurationKey":[{"key":"vendorId","value":"EN+","readonly":false},{"key":"chargePointSN","value":"","readonly":false},{"key":"AuthorizationKey","value":"ABCDEFGHIJ123456","readonly":false},{"key":"ChargingParametersEnabled","value":"true","readonly":false},{"key":"AllowOfflineTxForUnknownId","value":"false","readonly":false},{"key":"AuthorizationCacheEnabled","value":"false","readonly":false},{"key":"AuthorizeRemoteTxRequests","value":"false","readonly":false},{"key":"BlinkRepeat","value":"0","readonly":false},{"key":"ClockAlignedDataInterval","value":"0","readonly":false},{"key":"ConnectionTimeOut","value":"30","readonly":false},{"key":"GetConfigurationMaxKeys","value":"50","readonly":true},{"key":"HeartbeatInterval","value":"60","readonly":false},{"key":"LocalAuthorizeOffline","value":"false","readonly":false},{"key":"LocalPreAuthorize","value":"false","readonly":false},{"key":"MaxEnergyOnInvalidId","value":"3000","readonly":false},{"key":"MeterValuesAlignedData","value":"Energy.Active.Import.Register","readonly":false},{"key":"MeterValuesSampledData","value":"Power.Active.Import,Energy.Active.Import.Register,Current.Import,Voltage,Current.Offered,Power.Offe","readonly":false},{"key":"MeterValueSampleInterval","value":"10","readonly":false},{"key":"MinimumStatusDuration","value":"0","readonly":false},{"key":"ResetRetries","value":"0","readonly":false},{"key":"ConnectorPhaseRotation","value":"","readonly":false},{"key":"StopTransactionOnEVSideDisconnect","value":"false","readonly":false},{"key":"StopTransactionOnInvalidId","value":"false","readonly":false},{"key":"StopTxnAlignedData","value":"","readonly":false},{"key":"StopTxnSampledData","value":"Energy.Active.Import.Register","readonly":false},{"key":"TransactionMessageAttempts","value":"0","readonly":false},{"key":"TransactionMessageRetryInterval","value":"15","readonly":false},{"key":"UnlockConnectorOnEVSideDisconnect","value":"false","readonly":false},{"key":"WebSocketPingInterval","value":"0","readonly":false},{"key":"NumberOfConnectors","value":"1","readonly":true},{"key":"MeterValuesAlignedDataMaxLength","value":"120","readonly":true},{"key":"MeterValuesSampledDataMaxLength","value":"120","readonly":true},{"key":"ConnectorPhaseRotationMaxLength","value":"120","readonly":true},{"key":"StopTxnAlignedDataMaxLength","value":"120","readonly":true},{"key":"StopTxnSampledDataMaxLength","value":"120","readonly":true},{"key":"SupportedFeatureProfiles","value":"Core,Reservation,FirmwareManagement,LocalAuthListManagement,RemoteTrigger,SmartCharging","readonly":true},{"key":"SupportedFeatureProfilesMaxLength","value":"120","readonly":true},{"key":"ReserveConnectorZeroSupported","value":"false","readonly":true},{"key":"LocalAuthListEnabled","value":"true","readonly":false},{"key":"LocalAuthListMaxLength","value":"10","readonly":true},{"key":"SendLocalListMaxLength","value":"10","readonly":true},{"key":"ChargeProfileMaxStackLevel","value":"5","readonly":true},{"key":"ChargingScheduleAllowedChargingRateUnit","value":"Current","readonly":true},{"key":"ChargingScheduleMaxPeriods","value":"5","readonly":true},{"key":"ConnectorSwitch3to1PhaseSupported","value":"false","readonly":true},{"key":"MaxChargingProfilesInstalled","value":"3","readonly":true}],"unknownKey":[]}]
[ocpp ] TRACE 2024/08/21 08:17:40 send SN10052307203612: [2,"2093885981","TriggerMessage",{"requestedMessage":"BootNotification"}]
[ocpp ] TRACE 2024/08/21 08:17:41 recv SN10052307203612: [3,"2093885981",{"status":"Accepted"}]
[ocpp ] TRACE 2024/08/21 08:17:41 recv SN10052307203612: [2,"160","BootNotification",{"chargePointModel":"23","chargePointSerialNumber":"SN10052307203612","chargePointVendor":"EN+","firmwareVersion":"1.3.53"}]
[ocpp ] TRACE 2024/08/21 08:17:41 send SN10052307203612: [3,"160",{"currentTime":"2024-08-20T22:17:41Z","interval":60,"status":"Accepted"}]
[ocpp ] TRACE 2024/08/21 08:17:41 send SN10052307203612: [2,"226426864","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Power.Active.Import"}]
[ocpp ] TRACE 2024/08/21 08:17:42 recv SN10052307203612: [3,"226426864",{"status":"Accepted"}]
[ocpp ] TRACE 2024/08/21 08:17:42 send SN10052307203612: [2,"2616666845","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Energy.Active.Import.Register"}]
[ocpp ] TRACE 2024/08/21 08:17:43 recv SN10052307203612: [3,"2616666845",{"status":"Accepted"}]
[ocpp ] TRACE 2024/08/21 08:17:43 send SN10052307203612: [2,"3115775950","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Current.Import"}]
[ocpp ] TRACE 2024/08/21 08:17:44 recv SN10052307203612: [3,"3115775950",{"status":"Accepted"}]
[ocpp ] TRACE 2024/08/21 08:17:44 send SN10052307203612: [2,"1736058610","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Voltage"}]
[ocpp ] TRACE 2024/08/21 08:17:44 recv SN10052307203612: [3,"1736058610",{"status":"Accepted"}]
[ocpp ] TRACE 2024/08/21 08:17:44 send SN10052307203612: [2,"4088406183","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Current.Offered"}]
[ocpp ] TRACE 2024/08/21 08:17:45 recv SN10052307203612: [3,"4088406183",{"status":"Accepted"}]
[ocpp ] TRACE 2024/08/21 08:17:45 send SN10052307203612: [2,"2742721241","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Power.Offered"}]
[ocpp ] TRACE 2024/08/21 08:17:46 recv SN10052307203612: [3,"2742721241",{"status":"Accepted"}]
[ocpp ] TRACE 2024/08/21 08:17:46 send SN10052307203612: [2,"4023502275","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"SoC"}]
[ocpp ] TRACE 2024/08/21 08:17:47 recv SN10052307203612: [3,"4023502275",{"status":"Accepted"}]
[ocpp ] TRACE 2024/08/21 08:17:47 send SN10052307203612: [2,"2903194887","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Power.Active.Import,Energy.Active.Import.Register,Current.Import,Voltage,Current.Offered,Power.Offered,SoC"}]
[ocpp ] TRACE 2024/08/21 08:17:48 recv SN10052307203612: [3,"2903194887",{"status":"Accepted"}]
[... 106 lines cut ...]
[ocpp ] TRACE 2024/08/21 08:21:47 send SN10052307203612: [2,"3722534888","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp ] TRACE 2024/08/21 08:21:48 recv SN10052307203612: [3,"3722534888",{"status":"Accepted"}]
[ocpp ] TRACE 2024/08/21 08:21:48 recv SN10052307203612: [2,"187","MeterValues",{"connectorId":1,"meterValue":[{"timestamp":"2024-08-20T22:21:47Z","sampledValue":[{"measurand":"Current.Offered","unit":"A","value":"32.0","context":"Trigger"},{"measurand":"Current.Import","unit":"A","phase":"L1","value":"0.0"},{"measurand":"Voltage","unit":"V","phase":"L1","value":"248.5","context":"Trigger"},{"measurand":"Energy.Active.Import.Register","unit":"Wh","value":"1.0","contex [... cut ...] xt":"Trigger"},{"measurand":"Current.Import","unit":"A","phase":"L1","value":"0.0"},{"measurand":"Voltage","unit":"V","phase":"L1","value":"248.5","context":"Trigger"},{"measurand":"Energy.Active.Import.Register","unit":"Wh","value":"1.0","context":"Trigger"},{"measurand":"Power.Active.Import","unit":"W","value":"0.0","context":"Trigger"}]}]}]
[ocpp ] TRACE 2024/08/21 08:22:24 send SN10052307203612: [3,"190",{}]
```

### What type of operating system are you running?

Linux

### Version

0.130.2

premultiply, 2024-08-21:
As any other information than ocpp is missing in the log can not see what is the operational state.
None of the "new issues" is visible in the provided log.

The only thing visible is
```
[ocpp ] TRACE 2024/08/21 08:18:50 recv SN10052307203612: [2,"169","StatusNotification",{"connectorId":1,"status":"Finishing","errorCode":"NoError","timestamp":"2024-08-20T22:17:46Z","vendorId":"EN+","info":"{\"reason\":\"plugInGun\",\"cpv\":0,\"rv\":0}"}]
```
meaning that the transaction is closed (evcc will never do this) and you have to plug out the vehicle. No further operation is possible until the vehicle is disconnected.

If you want any help please provide usefull logfiles showing the issues (including context) you describe.

jeffborg, 2024-08-21:
So before with the config in the initial part on 0.129.0 was working really well stable didn't need to power cycle the evse. now with 0.130.2 I have to to get things going

The car was plugged in and that was myself unplugging the car, but the car was ready to accept charge but the evse wasn't in an enabled state until after the evse was turned on off and evcc restarted.

Between the start 8:17:38 and 8:18:50 the car should have been charging and it wasn't evcc was just saying waiting for vehicle. Even after that point the car was plugged back in and nothing was happening.

premultiply, 2024-08-21:
Unfortunately your report is missing a detailed log file. Without log, we don‘t have the information for diagnosing the issue described. See https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen.

andig, 2024-08-21:
@jeffborg we'll need a logfile, OCPP: trace and the rest on debug.

jeffborg, 2024-08-22:
@andig logs with levels set to debug for all and trace for ocpp now.

prior to 0.130.0 this was never an issue https://github.com/evcc-io/evcc/issues/15552#issuecomment-2299986099 evcc would restart the transaction on it's own even if this isn't strictly part of the ocpp spec.

yes physically unplugging the car and plugging it back in again gets things going but that makes **no sense for home charging**.

```
[main  ] INFO 2024/08/23 08:43:58 evcc 0.130.2
[main  ] INFO 2024/08/23 08:43:58 using config file: /config/evcc.yaml
[db    ] INFO 2024/08/23 08:43:58 using sqlite database: /config/.evcc/evcc.db
[mqtt  ] INFO 2024/08/23 08:44:00 connecting evcc-1527597213 at tcp://mqtt-mosquitto:1883
[mqtt  ] DEBUG 2024/08/23 08:44:00 tcp://mqtt-mosquitto:1883 connected
[main  ] INFO 2024/08/23 08:44:00 listening at :7070
[ocpp-1] DEBUG 2024/08/23 08:44:00 waiting for chargepoint: 5m0s
[ocpp  ] INFO 2024/08/23 08:44:04 charge point connected, registering: SN10052307203612
[ocpp  ] TRACE 2024/08/23 08:44:04 recv SN10052307203612: [2,"3328","DataTransfer",{"vendorId":"EN+","messageId":"gatewayInfo","data":"{\"SN\":\"SN10052307203612\",\"fwVer\":\"V43.0.552\",\"gateCode\":107,\"measureType\":1,\"disconnectReason\":0,\"maxCurr\":320,\"phase\":1,\"sdCardStatus\":0}"}]
[ocpp  ] TRACE 2024/08/23 08:44:04 send SN10052307203612: [3,"3328",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/08/23 08:44:05 send SN10052307203612: [2,"1893634325","GetConfiguration",{}]
[ocpp  ] TRACE 2024/08/23 08:44:06 recv SN10052307203612: [3,"1893634325",{"configurationKey":[{"key":"vendorId","value":"EN+","readonly":false},{"key":"chargePointSN","value":"","readonly":false},{"key":"AuthorizationKey","value":"ABCDEFGHIJ123456","readonly":false},{"key":"ChargingParametersEnabled","value":"true","readonly":false},{"key":"AllowOfflineTxForUnknownId","value":"false","readonly":false},{"key":"AuthorizationCacheEnabled","value":"false","readonly":false},{"key":"AuthorizeRemoteTxRequests","value":"false","readonly":false},{"key":"BlinkRepeat","value":"0","readonly":false},{"key":"ClockAlignedDataInterval","value":"0","readonly":false},{"key":"ConnectionTimeOut","value":"30","readonly":false},{"key":"GetConfigurationMaxKeys","value":"50","readonly":true},{"key":"HeartbeatInterval","value":"60","readonly":false},{"key":"LocalAuthorizeOffline","value":"false","readonly":false},{"key":"LocalPreAuthorize","value":"false","readonly":false},{"key":"MaxEnergyOnInvalidId","value":"3000","readonly":false},{"key":"MeterValuesAlignedData","value":"Energy.Active.Import.Register","readonly":false},{"key":"MeterValuesSampledData","value":"Power.Active.Import,Energy.Active.Import.Register,Current.Import,Voltage,Current.Offered,Power.Offe","readonly":false},{"key":"MeterValueSampleInterval","value":"10","readonly":false},{"key":"MinimumStatusDuration","value":"0","readonly":false},{"key":"ResetRetries","value":"0","readonly":false},{"key":"ConnectorPhaseRotation","value":"","readonly":false},{"key":"StopTransactionOnEVSideDisconnect","value":"false","readonly":false},{"key":"StopTransactionOnInvalidId","value":"false","readonly":false},{"key":"StopTxnAlignedData","value":"","readonly":false},{"key":"StopTxnSampledData","value":"Energy.Active.Import.Register","readonly":false},{"key":"TransactionMessageAttempts","value":"0","readonly":false},{"key":"TransactionMessageRetryInterval","value":"15","readonly":false},{"key":"UnlockConnectorOnEVSideDisconnect","value":"false","readonly":false},{"key":"WebSocketPingInterval","value":"0","readonly":false},{"key":"NumberOfConnectors","value":"1","readonly":true},{"key":"MeterValuesAlignedDataMaxLength","value":"120","readonly":true},{"key":"MeterValuesSampledDataMaxLength","value":"120","readonly":true},{"key":"ConnectorPhaseRotationMaxLength","value":"120","readonly":true},{"key":"StopTxnAlignedDataMaxLength","value":"120","readonly":true},{"key":"StopTxnSampledDataMaxLength","value":"120","readonly":true},{"key":"SupportedFeatureProfiles","value":"Core,Reservation,FirmwareManagement,LocalAuthListManagement,RemoteTrigger,SmartCharging","readonly":true},{"key":"SupportedFeatureProfilesMaxLength","value":"120","readonly":true},{"key":"ReserveConnectorZeroSupported","value":"false","readonly":true},{"key":"LocalAuthListEnabled","value":"true","readonly":false},{"key":"LocalAuthListMaxLength","value":"10","readonly":true},{"key":"SendLocalListMaxLength","value":"10","readonly":true},{"key":"ChargeProfileMaxStackLevel","value":"5","readonly":true},{"key":"ChargingScheduleAllowedChargingRateUnit","value":"Current","readonly":true},{"key":"ChargingScheduleMaxPeriods","value":"5","readonly":true},{"key":"ConnectorSwitch3to1PhaseSupported","value":"false","readonly":true},{"key":"MaxChargingProfilesInstalled","value":"3","readonly":true}],"unknownKey":[]}]
[ocpp  ] TRACE 2024/08/23 08:44:06 send SN10052307203612: [2,"1160000174","TriggerMessage",{"requestedMessage":"BootNotification"}]
[ocpp  ] TRACE 2024/08/23 08:44:07 recv SN10052307203612: [3,"1160000174",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/08/23 08:44:07 recv SN10052307203612: [2,"3329","BootNotification",{"chargePointModel":"23","chargePointSerialNumber":"SN10052307203612","chargePointVendor":"EN+","firmwareVersion":"1.3.53"}]
[ocpp  ] TRACE 2024/08/23 08:44:07 send SN10052307203612: [3,"3329",{"currentTime":"2024-08-22T22:44:07Z","interval":60,"status":"Accepted"}]
[... cut ...]
[ocpp  ] TRACE 2024/08/23 08:54:46 send SN10052307203612: [3,"3405",{}]
[site  ] DEBUG 2024/08/23 08:54:51 ----
[lp-1  ] DEBUG 2024/08/23 08:54:51 charge power: 0W
[lp-1  ] ERROR 2024/08/23 08:54:52 charge currents: not available
[circuit-main] DEBUG 2024/08/23 08:54:52 power: 785.93W
[circuit-main] DEBUG 2024/08/23 08:54:52 current: 0A
[site  ] DEBUG 2024/08/23 08:54:52 pv power: 1871W
[site  ] DEBUG 2024/08/23 08:54:52 grid meter: 786W
[site  ] DEBUG 2024/08/23 08:54:52 site power: 786W
[lp-1  ] ERROR 2024/08/23 08:54:52 charge voltages: not available
```

premultiply, 2024-08-23:
On start-up of the log your charger is in "finishing" state.
This state is only entered if the transaction is closed by the charger. This may be caused by the unavailability of the ocpp server (evcc).
You have to disconnect your car to get out of this state. It is not possible to start a new transaction in this state.
You should set your charger to "free charge mode" and remove the `remotestart: true` from the config to fix this issue.

The meter messages are because it is a one phase charger.
We are currently not prepared to parse the specific measurands from one phase only.
This needs to be implemented.

jeffborg, 2024-08-23:
@premultiply I can't use free charge mode as this will disable the ocpp connection.

jeffborg, 2024-08-23:
Also the evse just disconnected around 10:22:42 , and I need to power cycle it. Before on 0.129.2 with the config set to

```
    getconfiguration: false
    metervalues: Energy.Active.Import.Register,Power.Active.Import,Current.Import
    bootnotification: false
``` 

There was far less chatter on the ocpp websocket and only the meter values and ping messages and the evse stayed online much much much longer.

log of no more ocpp messages - not much to see here they just stop at 10:22:42am

```
[ocpp  ] TRACE 2024/08/23 10:22:27 send SN10052307203612: [2,"3953051613","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp  ] TRACE 2024/08/23 10:22:28 recv SN10052307203612: [3,"3953051613",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/08/23 10:22:28 recv SN10052307203612: [2,"4468","MeterValues",{"connectorId":1,"meterValue":[{"timestamp":"2024-08-23T00:22:27Z","sampledValue":[{"measurand":"Current.Offered","unit":"A","value":"32.0","context":"Trigger"},{"measurand":"Current.Import","unit":"A","phase":"L1","value":"0.0"},{"measurand":"Voltage","unit":"V","phase":"L1","value":"247.3","context":"Trigger"},{"measurand":"Energy.Active.Import.Register","unit":"Wh","value":"1939.0","context":"Trigger"},{"measurand":"Power.Active.Import","unit":"W","value":"0.0","context":"Trigger"}]}],"transactionId":3}]
[ocpp  ] TRACE 2024/08/23 10:22:28 send SN10052307203612: [3,"4468",{}]
[ocpp  ] TRACE 2024/08/23 10:22:29 recv SN10052307203612: [2,"4469","DataTransfer",{"vendorId":"EN+","messageId":"cpChargingParameterRpt","data":"{\"duty\":0,\"cpVolt\":931,\"transactionId\":3,\"totalPowerFactor\":256,\"aPhasesPowerFactor\":0,\"bPhasesPowerFactor\":6,\"cPhasesPowerFactor\":0,\"ccVolt\":256,\"offset0\":2020,\"offset1\":2028,\"leakcurr\":0,\"AMBTemp\":312,\"lockStatus\":1,\"ctCurrent\":9}"}]
[ocpp  ] TRACE 2024/08/23 10:22:29 send SN10052307203612: [3,"4469",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/08/23 10:22:31 recv SN10052307203612: [2,"4470","Heartbeat",{}]
[ocpp  ] TRACE 2024/08/23 10:22:31 send SN10052307203612: [3,"4470",{"currentTime":"2024-08-23T00:22:31Z"}]
[ocpp  ] TRACE 2024/08/23 10:22:39 send SN10052307203612: [2,"1160893016","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp  ] TRACE 2024/08/23 10:22:40 recv SN10052307203612: [3,"1160893016",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/08/23 10:22:41 send SN10052307203612: [2,"1547743844","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp  ] TRACE 2024/08/23 10:22:41 recv SN10052307203612: [2,"4471","MeterValues",{"connectorId":1,"meterValue":[{"timestamp":"2024-08-23T00:22:39Z","sampledValue":[{"measurand":"Current.Offered","unit":"A","value":"32.0","context":"Trigger"},{"measurand":"Current.Import","unit":"A","phase":"L1","value":"0.0"},{"measurand":"Voltage","unit":"V","phase":"L1","value":"247.5","context":"Trigger"},{"measurand":"Energy.Active.Import.Register","unit":"Wh","value":"1939.0","context":"Trigger"},{"measurand":"Power.Active.Import","unit":"W","value":"0.0","context":"Trigger"}]}],"transactionId":3}]
[ocpp  ] TRACE 2024/08/23 10:22:41 send SN10052307203612: [3,"4471",{}]
[ocpp  ] TRACE 2024/08/23 10:22:42 recv SN10052307203612: [3,"1547743844",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/08/23 10:22:42 recv SN10052307203612: [2,"4472","MeterValues",{"connectorId":1,"meterValue":[{"timestamp":"2024-08-23T00:22:41Z","sampledValue":[{"measurand":"Current.Offered","unit":"A","value":"32.0","context":"Trigger"},{"measurand":"Current.Import","unit":"A","phase":"L1","value":"0.0"},{"measurand":"Voltage","unit":"V","phase":"L1","value":"247.5","context":"Trigger"},{"measurand":"Energy.Active.Import.Register","unit":"Wh","value":"1939.0","context":"Trigger"},{"measurand":"Power.Active.Import","unit":"W","value":"0.0","context":"Trigger"}]}],"transactionId":3}]
[ocpp  ] TRACE 2024/08/23 10:22:42 send SN10052307203612: [3,"4472",{}]
[ocpp  ] TRACE 2024/08/23 10:22:42 recv SN10052307203612: [2,"4473","DataTransfer",{"vendorId":"EN+","messageId":"cpChargingParameterRpt","data":"{\"duty\":0,\"cpVolt\":931,\"transactionId\":3,\"totalPowerFactor\":256,\"aPhasesPowerFactor\":0,\"bPhasesPowerFactor\":6,\"cPhasesPowerFactor\":0,\"ccVolt\":256,\"offset0\":2020,\"offset1\":2028,\"leakcurr\":0,\"AMBTemp\":312,\"lockStatus\":1,\"ctCurrent\":9}"}]
[ocpp  ] TRACE 2024/08/23 10:22:42 send SN10052307203612: [3,"4473",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/08/23 10:22:43 recv SN10052307203612: [2,"4474","DataTransfer",{"vendorId":"EN+","messageId":"cpChargingParameterRpt","data":"{\"duty\":0,\"cpVolt\":931,\"transactionId\":3,\"totalPowerFactor\":256,\"aPhasesPowerFactor\":0,\"bPhasesPowerFactor\":6,\"cPhasesPowerFactor\":0,\"ccVolt\":256,\"offset0\":2020,\"offset1\":2028,\"leakcurr\":0,\"AMBTemp\":312,\"lockStatus\":1,\"ctCurrent\":9}"}]
[ocpp  ] TRACE 2024/08/23 10:22:43 send SN10052307203612: [3,"4474",{"status":"Accepted"}]
[site  ] DEBUG 2024/08/23 10:22:51 ----
[lp-1  ] DEBUG 2024/08/23 10:22:51 charge power: 0W
[lp-1  ] ERROR 2024/08/23 10:22:52 charge currents: not available
[circuit-main] DEBUG 2024/08/23 10:22:52 power: -1443.9W
[circuit-main] DEBUG 2024/08/23 10:22:52 current: 0A
[site  ] DEBUG 2024/08/23 10:22:52 pv power: 3951W
[site  ] DEBUG 2024/08/23 10:22:52 grid meter: -1444W
[site  ] DEBUG 2024/08/23 10:22:52 site power: -1444W
[lp-1  ] ERROR 2024/08/23 10:22:52 charge voltages: not available
[... cut ...]
[circuit-main] DEBUG 2024/08/23 10:25:22 power: -554.95W
[circuit-main] DEBUG 2024/08/23 10:25:22 current: 0A
[site  ] DEBUG 2024/08/23 10:25:22 pv power: 3639W
[site  ] DEBUG 2024/08/23 10:25:22 grid meter: -555W
[site  ] DEBUG 2024/08/23 10:25:22 site power: -555W
[lp-1  ] ERROR 2024/08/23 10:25:22 charge voltages: timeout
[lp-1  ] DEBUG 2024/08/23 10:25:22 !! session: chargeRater.chargedEnergy=1.1 - chargedAtStartup=0.0
[lp-1  ] ERROR 2024/08/23 10:25:22 charge total import: timeout
[lp-1  ] ERROR 2024/08/23 10:25:22 charger status: timeout
```

jeffborg, 2024-08-23:
@premultiply I just power cycled the evse but didn't restart evcc it still ended up in the finishing state

It seems to stop the evcc transaction with a softreset at 10:41:53, goes into Preparing then jumps to finishing, I haven't touched the car it was plugged in the entire time.

I'll physically unplug the car and plug back in and post the log in another comment (I suspect it will jump out of finishing state)

```
[circuit-main] DEBUG 2024/08/23 10:40:52 current: 0A
[site  ] DEBUG 2024/08/23 10:40:52 pv power: 1751W
[site  ] DEBUG 2024/08/23 10:40:52 grid meter: 1025W
[site  ] DEBUG 2024/08/23 10:40:52 site power: 795W
[lp-1  ] ERROR 2024/08/23 10:40:52 charge voltages: timeout
[lp-1  ] DEBUG 2024/08/23 10:40:52 !! session: chargeRater.chargedEnergy=1.1 - chargedAtStartup=0.0
[lp-1  ] ERROR 2024/08/23 10:40:52 charge total import: timeout
[lp-1  ] ERROR 2024/08/23 10:40:52 charger status: timeout
[site  ] DEBUG 2024/08/23 10:41:21 ----
[lp-1  ] ERROR 2024/08/23 10:41:22 charge power: timeout
[lp-1  ] ERROR 2024/08/23 10:41:23 charge currents: timeout
[circuit-main] DEBUG 2024/08/23 10:41:23 power: 567.81W
[circuit-main] DEBUG 2024/08/23 10:41:23 current: 0A
[site  ] DEBUG 2024/08/23 10:41:23 pv power: 2262W
[site  ] DEBUG 2024/08/23 10:41:23 grid meter: 568W
[site  ] DEBUG 2024/08/23 10:41:23 site power: 338W
[lp-1  ] ERROR 2024/08/23 10:41:23 charge voltages: timeout
[lp-1  ] DEBUG 2024/08/23 10:41:23 !! session: chargeRater.chargedEnergy=1.1 - chargedAtStartup=0.0
[lp-1  ] ERROR 2024/08/23 10:41:23 charge total import: timeout
[lp-1  ] ERROR 2024/08/23 10:41:23 charger status: timeout
[ocpp  ] DEBUG 2024/08/23 10:41:47 charge point connected: SN10052307203612
[ocpp  ] TRACE 2024/08/23 10:41:47 send SN10052307203612: [2,"544576103","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp  ] TRACE 2024/08/23 10:41:48 recv SN10052307203612: [3,"544576103",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/08/23 10:41:49 send SN10052307203612: [2,"1278990172","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp  ] TRACE 2024/08/23 10:41:50 recv SN10052307203612: [3,"1278990172",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/08/23 10:41:51 send SN10052307203612: [2,"3228471321","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[site  ] DEBUG 2024/08/23 10:41:51 ----
[ocpp  ] TRACE 2024/08/23 10:41:52 recv SN10052307203612: [3,"3228471321",{"status":"Accepted"}]
[lp-1  ] ERROR 2024/08/23 10:41:52 charge power: timeout
[ocpp  ] TRACE 2024/08/23 10:41:52 recv SN10052307203612: [2,"5","BootNotification",{"chargePointModel":"23","chargePointSerialNumber":"SN10052307203612","chargePointVendor":"EN+","firmwareVersion":"1.3.53"}]
[ocpp  ] TRACE 2024/08/23 10:41:52 send SN10052307203612: [3,"5",{"currentTime":"2024-08-23T00:41:52Z","interval":60,"status":"Accepted"}]
[lp-1  ] ERROR 2024/08/23 10:41:52 charge currents: timeout
[circuit-main] DEBUG 2024/08/23 10:41:52 power: 957.69W
[circuit-main] DEBUG 2024/08/23 10:41:52 current: 0A
[site  ] DEBUG 2024/08/23 10:41:52 pv power: 2422W
[site  ] DEBUG 2024/08/23 10:41:52 grid meter: 958W
[site  ] DEBUG 2024/08/23 10:41:52 site power: 728W
[lp-1  ] ERROR 2024/08/23 10:41:52 charge voltages: timeout
[lp-1  ] DEBUG 2024/08/23 10:41:52 !! session: chargeRater.chargedEnergy=1.1 - chargedAtStartup=0.0
[lp-1  ] ERROR 2024/08/23 10:41:52 charge total import: timeout
[lp-1  ] DEBUG 2024/08/23 10:41:52 charger status: B
[lp-1  ] DEBUG 2024/08/23 10:41:52 pv charge current: 0A = 0A + -3.16A (728W @ 1p)
[ocpp  ] TRACE 2024/08/23 10:41:52 recv SN10052307203612: [2,"6","DataTransfer",{"vendorId":"EN+","messageId":"gatewayInfo","data":"{\"SN\":\"SN10052307203612\",\"fwVer\":\"V43.0.552\",\"gateCode\":107,\"measureType\":1,\"disconnectReason\":4,\"maxCurr\":320,\"phase\":1,\"sdCardStatus\":0}"}]
[ocpp  ] TRACE 2024/08/23 10:41:52 send SN10052307203612: [3,"6",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/08/23 10:41:53 recv SN10052307203612: [2,"offline-1","StopTransaction",{"meterStop":1939,"transactionId":3,"transactionData":[{"sampledValue":[{"measurand":"Energy.Active.Import.Register","context":"Transaction.End","value":"1.939","unit":"kWh"}],"timestamp":"2024-08-23T00:39:55Z"}],"timestamp":"2024-08-23T00:39:55Z","idTag":"evcc","reason":"SoftReset"}]
[ocpp  ] TRACE 2024/08/23 10:41:53 send SN10052307203612: [3,"offline-1",{"idTagInfo":{"status":"Accepted"}}]
[ocpp  ] TRACE 2024/08/23 10:41:53 recv SN10052307203612: [2,"7","StatusNotification",{"connectorId":0,"status":"Preparing","errorCode":"NoError","timestamp":"2024-08-23T00:41:50Z","vendorId":"EN+","info":"{\"reason\":\"plugInGun\",\"cpv\":0,\"rv\":0}"}]
[ocpp  ] TRACE 2024/08/23 10:41:53 send SN10052307203612: [3,"7",{}]
[ocpp  ] TRACE 2024/08/23 10:41:53 recv SN10052307203612: [2,"8","StatusNotification",{"connectorId":1,"status":"Preparing","errorCode":"NoError","timestamp":"2024-08-23T00:41:50Z","vendorId":"EN+","info":"{\"reason\":\"plugInGun\",\"cpv\":0,\"rv\":0}"}]
[ocpp  ] TRACE 2024/08/23 10:41:53 send SN10052307203612: [3,"8",{}]
[ocpp  ] TRACE 2024/08/23 10:41:53 recv SN10052307203612: [2,"9","Heartbeat",{}]
[ocpp  ] TRACE 2024/08/23 10:41:53 send SN10052307203612: [3,"9",{"currentTime":"2024-08-23T00:41:53Z"}]
[ocpp  ] TRACE 2024/08/23 10:41:55 recv SN10052307203612: [2,"offline-4","MeterValues",{"connectorId":1,"meterValue":[{"timestamp":"1970-01-01T00:00:19Z","sampledValue":[{"measurand":"Current.Offered","unit":"A","value":"32.0","context":"Trigger"},{"measuran [... cut ...]
[ocpp  ] TRACE 2024/08/23 10:47:42 recv SN10052307203612: [2,"45","MeterValues",{"connectorId":1,"meterValue":[{"timestamp":"2024-08-23T00:47:41Z","sampledValue":[{"measurand":"Current.Offered","unit":"A","value":"32.0","context":"Trigger"},{"measurand":"Current.Import","unit":"A","phase":"L1","value":"0.0"},{"measurand":"Voltage","unit":"V","phase":"L1","value":"245.7","context":"Trigger"},{"measurand":"Energy.Active.Import.Register","unit":"Wh","value":"0.0","context":"Trigger"},{"measurand":"Power.Active.Import","unit":"W","value":"0.0","context":"Trigger"}]}]}]
[ocpp  ] TRACE 2024/08/23 10:47:42 send SN10052307203612: [3,"45",{}]
```

jeffborg, 2024-08-23:
@premultiply as suspected after physically unplugging and plugging back in here is the logs covering that time period.

evse seems to go from

- Available at 10:52:38
- preparing at 10:52:53
- evcc sends remote start transaction at 10:53:22
- finishing at 10:53:23
- suspend evse at 10:53:27
- charging at 10:53:32

```
[lp-1  ] DEBUG 2024/08/23 10:51:52 !! session: chargeRater.chargedEnergy=1.1 - chargedAtStartup=0.0
[lp-1  ] DEBUG 2024/08/23 10:51:52 charge total import: 0.000kWh
[lp-1  ] DEBUG 2024/08/23 10:51:52 charger status: B
[lp-1  ] DEBUG 2024/08/23 10:51:52 pv charge current: min 6A > 5.19A (-1193W @ 1p, battery: false)
[ocpp  ] TRACE 2024/08/23 10:51:52 recv SN10052307203612: [2,"70","Heartbeat",{}]
[ocpp  ] TRACE 2024/08/23 10:51:52 send SN10052307203612: [3,"70",{"currentTime":"2024-08-23T00:51:52Z"}]
[ocpp  ] TRACE 2024/08/23 10:51:53 send SN10052307203612: [2,"1745643414","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp  ] TRACE 2024/08/23 10:51:54 recv SN10052307203612: [3,"1745643414",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/08/23 10:51:54 recv SN10052307203612: [2,"71","MeterValues",{"connectorId":1,"meterValue":[{"timestamp":"2024-08-23T00:51:53Z","sampledValue":[{"measurand":"Current.Offered","unit":"A","value":"32.0","context":"Trigger"},{"measurand":"Current.Import","unit":"A","phase":"L1","value":"0.0"},{"measurand":"Voltage","unit":"V","phase":"L1","value":"245.7","context":"Trigger"},{"measurand":"Energy.Active.Import.Register","unit":"Wh","value":"0.0","context":"Trigger"},{"measurand":"Power.Active.Import","unit":"W","value":"0.0","context":"Trigger"}]}]}]
[ocpp  ] TRACE 2024/08/23 10:51:54 send SN10052307203612: [3,"71",{}]
[ocpp  ] TRACE 2024/08/23 10:52:05 send SN10052307203612: [2,"3254324430","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp  ] TRACE 2024/08/23 10:52:06 recv SN10052307203612: [3,"3254324430",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/08/23 10:52:06 recv SN10052307203612: [2,"72","MeterValues",{"connectorId":1,"meterValue":[{"timestamp":"2024-08-23T00:52:05Z","sampledValue":[{"measurand":"Current.Offered","unit":"A","value":"32.0","context":"Trigger"},{"measurand":"Current.Import","unit":"A","phase":"L1","value":"0.0"},{"measurand":"Voltage","unit":"V","phase":"L1","value":"245.7","context":"Trigger"},{"measurand":"Energy.Active.Import.Register","unit":"Wh","value":"0.0","context":"Trigger"},{"measurand":"Power.Active.Import","unit":"W","value":"0.0","context":"Trigger"}]}]}]
[ocpp  ] TRACE 2024/08/23 10:52:06 send SN10052307203612: [3,"72",{}]
[ocpp  ] TRACE 2024/08/23 10:52:17 send SN10052307203612: [2,"4223284156","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp  ] TRACE 2024/08/23 10:52:18 recv SN10052307203612: [3,"4223284156",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/08/23 10:52:18 recv SN10052307203612: [2,"73","MeterValues",{"connectorId":1,"meterValue":[{"timestamp":"2024-08-23T00:52:17Z","sampledValue":[{"measurand":"Current.Offered","unit":"A","value":"32.0","context":"Trigger"},{"measurand":"Current.Import","unit":"A","phase":"L1","value":"0.0"},{"measurand":"Voltage","unit":"V","phase":"L1","value":"245.7","context":"Trigger"},{"measurand":"Energy.Active.Import.Register","unit":"Wh","value":"0.0","context":"Trigger"},{"measurand":"Power.Active.Import","unit":"W","value":"0.0","context":"Trigger"}]}]}]
[ocpp  ] TRACE 2024/08/23 10:52:18 send SN10052307203612: [3,"73",{}]
[site  ] DEBUG 2024/08/23 10:52:21 ----
[lp-1  ] DEBUG 2024/08/23 10:52:21 charge power: 0W
[lp-1  ] ERROR 2024/08/23 10:52:22 charge currents: not available
[circuit-main] DEBUG 2024/08/23 10:52:22 power: -965.57W
[circuit-main] DEBUG 2024/08/23 10:52:22 current: 0A
[site  ] DEBUG 2024/08/23 10:52:22 pv power: 4300W
[site  ] DEBUG 2024/08/23 10:52:22 grid meter: -966W
[site  ] DEBUG 2024/08/23 10:52:22 site power: -1196W
[lp-1  ] ERROR 2024/08/23 10:52:22 charge voltages: not available
[lp-1  ] DEBUG 2024/08/23 10:52:22 !! session: chargeRater.chargedEnergy=1.1 - chargedAtStartup=0.0
[lp-1  ] DEBUG 2024/08/23 10:52:22 charge total import: 0.000kWh
[lp-1  ] DEBUG 2024/08/23 10:52:22 charger status: B
[lp-1  ] DEBUG 2024/08/23 10:52:22 pv charge current: min 6A > 5.2A (-1196W @ 1p, battery: false)
[ocpp  ] TRACE 2024/08/23 10:52:29 send SN10052307203612: [2,"2409293760","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp  ] TRACE 2024/08/23 10:52:30 recv SN10052307203612: [3,"2409293760",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/08/23 10:52:30 recv SN10052307203612: [2,"74","MeterValues",{"connectorId":1,"meterValue":[{"timestamp":"2024-08-23T00:52:29Z","sampledValue":[{"measurand":"Current.Offered","unit":"A","value":"32.0","context":"Trigger"},{"measurand":"Current.Import","unit":"A","phase":"L1","value":"0.0"},{"measurand":"Voltage","unit":"V","phase":"L1","value":"245.7","context":"Trigger"},{"measurand":"Energy.Active.Import.Register","unit":"Wh","value":"0.0","context":"Trigger"},{"measurand":"Power.Active.Import","unit":"W","value":"0.0","context":"Trigger"}]}]}]
[ocpp  ] TRACE 2024/08/23 10:52:30 send SN10052307203612: [3,"74",{}]
[ocpp  ] TRACE 2024/08/23 10:52:38 recv SN10052307203612: [2,"75","DataTransfer",{"vendorId":"EN+","messageId":"cpChargingParameterRpt","data":"{\"duty\":0,\"cpVolt\":1253,\"transactionId\":0,\"totalPowerFactor\":253,\"aPhasesPowerFactor\":0,\"bPhasesPowerFactor\":6,\"cPhasesPowerFactor\":1,\"ccVolt\":253,\"offset0\":2015,\"offset1\":201 [... cut ...]
[ocpp  ] TRACE 2024/08/23 10:55:17 recv SN10052307203612: [2,"115","DataTransfer",{"vendorId":"EN+","messageId":"cpChargingParameterRpt","data":"{\"duty\":116,\"cpVolt\":615,\"transactionId\":4,\"totalPowerFactor\":250,\"aPhasesPowerFactor\":0,\"bPhasesPowerFactor\":6,\"cPhasesPowerFactor\":1,\"ccVolt\":250,\"offset0\":2015,\"offset1\":2019,\"leakcurr\":40,\"AMBTemp\":286,\"lockStatus\":1,\"ctCurrent\":12}"}]
[ocpp  ] TRACE 2024/08/23 10:55:17 send SN10052307203612: [3,"115",{"status":"Accepted"}]
[site  ] DEBUG 2024/08/23 10:55:21 ----
```

andig, 2024-08-24:
What's missing here? Looks like the box is 1p and we don't handle that yet for currents?

jeffborg, 2024-08-24:
> What's missing here? Looks like the box is 1p and we don't handle that yet for currents?

@andig yes my box is 1p. It does appear the en+ firmware supports 3p boxes also

closed 2024-08-24 by andig

jeffborg, 2024-08-29:
@andig issue is prior to 0.130.0 evse would stay online for days at a time

0.129.0 config
```
  - name: solax
    type: template
    template: ocpp
    timeout: 2m
    getconfiguration: false
    metervalues: Energy.Active.Import.Register,Power.Active.Import,Current.Import
    bootnotification: false
    meterinterval: 10s
```
Now with the above config and 0.129.0 the only messages sent on a regular basis were the meter values, I had the same issue with the evse disconnecting with I had getconfiguration enabled which is why it's disabled, it was even more stable if you don't request the metervalues!

Now in 0.130.0 the regular messages also include this DataTransfer message below

config for 0.130+
```
chargers:
  - name: garage
    type: template
    template: ocpp
    timeout: 2m
    remotestart: true
```

regular messages
```
[ocpp  ] TRACE 2024/08/29 20:37:34 send SN10052307203612: [2,"2871585613","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp  ] TRACE 2024/08/29 20:37:35 recv SN10052307203612: [3,"2871585613",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/08/29 20:37:35 recv SN10052307203612: [2,"882","MeterValues",{"connectorId":1,"meterValue":[{"timestamp":"2024-08-29T10:37:34Z","sampledValue":[{"measurand":"Current.Offered","unit>
[ocpp  ] TRACE 2024/08/29 20:37:35 send SN10052307203612: [3,"882",{}]
[ocpp  ] TRACE 2024/08/29 20:37:36 recv SN10052307203612: [2,"883","DataTransfer",{"vendorId":"EN+","messageId":"cpChargingParameterRpt","data":"{\"duty\":0,\"cpVolt\":931,\"transactionId\":8,\"totalP>
[ocpp  ] TRACE 2024/08/29 20:37:36 send SN10052307203612: [3,"883",{"status":"Accepted"}]
```

The extra messages make the evse more unstable in keeping the network connection going and it will disconnect within less than 24 hours, before it would stay working for weeks. I can only put it down to the volume of traffic on the websocket and I'm guessing there is some sort of memory leak in the evse.

andig, 2024-08-30:
> 0.129.0 config

You can still have `metervalues` in current version. Did you try that?

>  issue is prior to 0.130.0 evse would stay online for days at a time

That sounds both horrible and like a charger issue more than an evcc issue. Did you open a ticket with vendor support?

> The extra messages make the evse more unstable

It's unclear where those are coming from. evcc doesn't do anything to trigger specific messages. If these are cause of instability that would be another question to vendor.

Sorry for not having better news- but there is no issue with evcc that we can see here.

jeffborg, 2024-08-31:
@andig without a flag now to consider finished state the same as preparing 0.130.0 is almost unusable. I have spent far too much time in trying to get a charge happening where this was no where near the issue this is in 0.129.0

andig, 2024-09-01:
@jeffborg sorry for the bad experience. However, I'm entirely unclear how we could help since there is no obvious problem with evcc. I've asked a couple of questions above, especially regarding `metervalues` but you didn't answer.

> without a flag now to consider finished state the same as preparing

You're making assumptions. We don't need a flag since it is already implemented that way and hasn't changed:

	case
		core.ChargePointStatusPreparing,     // "Preparing"
		core.ChargePointStatusSuspendedEVSE, // "SuspendedEVSE"
		core.ChargePointStatusSuspendedEV,   // "SuspendedEV"
		core.ChargePointStatusFinishing:     // "Finishing"
		return api.StatusB, nil

It should be possible from logfile to see if and that evcc will again enable the charger in this state.

That said I'm happy to help but right now there's nothing to do for me. Sorry.

jeffborg, 2024-09-01:
@andig seems somehow mine got stuck in a state with 0.129.0 where it would only ever charge at 32amps and never lower.

I have forked evcc and first change I did to avoid having to unplug the car, Just assume finishing is the same as preparing. It's not where you pointed out above.

https://github.com/jeffborg/evcc/commit/b94391e8293bb95d5bebce837ae98a9480bd934f

I think it's still going to be too chatty but I can deal with going to the garage and turning on/off the evse, The entire unplug the car thing was crazy and is totally unnecessary

premultiply, 2024-09-02:
Starting a new transaction in the "terminating" state would never allow the cable to be unplugged again.

@andig needsTransaction() should be moved from the Status() API to the StatusNotification callback on the connector so that it can be called immediately before the charger's internal connection timeout expires.

jeffborg, 2024-09-02:
@premultiply that should work by starting the transaction without waiting after Getting the first status message, note if you make a change to nightly I can't test due to https://github.com/evcc-io/evcc/issues/15852 but that looks like it's fixed now 👍

andig, 2024-09-02:
> @andig needsTransaction() should be moved from the Status() API to the StatusNotification callback on the connector so that it can be called immediately before the charger's internal connection timeout expires.

Could you explain a little more? What ist "charger's internal connection timeout" (what connection?)? Which status would be notified in the callback to react on? Apparently not `core.ChargePointStatusFinishing`?

andig, 2024-09-02:
Ich verstehe nicht, was der Link mit `finishing` zu tun hat. Da muss Du bitte die Implementierung anschauen.

jeffborg, 2024-09-02:
Got a firmware update for the the unit overnight, I'm hoping 🤞 the network stability issues are resolved. I power cycled it and it still does the Preparing -> Finishing thing but it took 15 seconds as before it took 6 so the proposed change https://github.com/evcc-io/evcc/issues/15552#issuecomment-2323573075 is still required.