evcc-io/evcc #15556: easee: Strange switching behaviour incl. 1/3-phase issue

MaTr74, 2024-08-21:
### Describe the bug

I reference to #14996 and #14387

After #14387 was solved within new release v.0.130.0 I upgraded my docker container to this version. After that a guest has plugged in his car to one of our two easee chargers (left-charger/lp-1). The charging mode was set to min+pv and min current of 6A. In the chart you can see a strange behaviour. I don't know if the car or the charger is causing this behaviour. The guest was telling me, that he got a bunch of error messages on his phone from the car.
[image]

Sometime he was annoyed from that and switched to the second charger (right-charger/lp2). The configuration was the same as of the first charger, min+pv and 6 A minimum. From 23:00 to 23:59 and beginning from 01:00 the mode was set to fast from my OpenHAB rule through evcc. At theses points the charger behavour was like I would expect, but during the min+pv period I would expect to get 6 A per phase.
[image]

I would put the first part of the first session log and the switching part to fast mode of the second session log into this thread. Please let me know what, if you like see the other parts.

It would be very nice if you could have a look on that.

Update: The log is not fitting in this report. I try to add it later.

### Steps to reproduce

1. Set mode to pv+min
2. Plugin EV

### Configuration details

```yaml
site:
  - title: Hotel
    meters:
      grid: grid-meter
      pv:
      - sungrow1
      - sungrow2

loadpoints:
  - title: Wallbox-links
    charger: easee-master
    mode: minpv
    enable: 
      threshold: 0
      delay: 1m
    disable:
      threshold: 6000
      delay: 3m
      #phases: 3
    priority: 1
      #    mincurrent: 6
      #    maxcurrent: 32
  - title: Wallbox-rechts
    charger: easee-slave
    mode: minpv
    enable: 
      threshold: 0
      delay: 1m
    disable:
      threshold: 6000
      delay: 3m
        #phases: 3
    priority: 1
      #    mincurrent: 6
      #    maxcurrent: 32
  - title: Klimatruhe
    charger: klimatruhe
    mode: off
    enable:
      threshold: 0
      delay: 1m
    disable:
      threshold: 100
      delay: 3m
    guardduration: 5m
      #phases: 1
[... 296 lines cut ...]
evcc    | [easee ] TRACE 2024/08/19 18:30:49 ProductUpdate [charger-left]: (2024-08-19 16:24:14 +0000 UTC) IN_VOLT_T2_T3 242.3209991455078
evcc    | [easee ] TRACE 2024/08/19 18:30:49 ProductUpdate [charger-left]: (2024-08-19 16:24:14 +0000 UTC) IN_VOLT_T2_T5 230.98500061035156
evcc    | [easee ] TRACE 2024/08/19 18:30:49 ProductUpdate [charger-left]: (2022-10-30 20:57:24 +0000 UTC) IN_VOLT_T3_T5 205.5050048828125
evcc    | [easee ] TRACE 2024/08/19 18:30:49 ProductUpdate [charger-left]: (2022-10-30 20:57:24 +0000 UTC) IN_VOLT_T1_T5 106.49500274658203
evcc    | [easee ] TRACE 2024/08/19 18:30:49 ProductUpdate [charger-left]: (2024-07-21 13:40:49 +0000 UTC) ObservationID(233) 0
evcc    | [easee ] TRACE 2024/08/19 18:30:49 ProductUpdate [charger-left]: (2022-10-30 20:57:24 +0000 UTC) IN_VOLT_T3_T4 415.16400146484375
evcc    | [easee ] TRACE 2024/08/19 18:30:49 ProductUpdate [charger-left]: (2022-10-30 20:57:24 +0000 UTC) IN_VOLT_T4_T5 316.0039978027344
evcc    | [easee ] TRACE 2024/08/19 18:30:49 ProductUpdate [charger-left]: (2024-08-19 01:20:35.268 +0000 UTC) CLOUD_DISCONNECT_REASON DUPLICATE_CLIENTID
evcc    | [easee ] TRACE 2024/08/19 18:30:49 ProductUpdate [charger-left]: (2024-07-21 13:40:49 +0000 UTC) ObservationID(234) 0
evcc    | [easee ] TRACE 2024/08/19 18:30:49 ProductUpdate [charger-left]: (2024-07-21 13:40:49 +0000 UTC) ObservationID(219) 0
evcc    | [easee ] TRACE 2024/08/19 18:30:49 ProductUpdate [charger-left]: (2024-08-19 01:20:35.332 +0000 UTC) CONNECTED_TO_CLOUD true
evcc    | [easee ] TRACE 2024/08/19 18:30:50 ProductUpdate [charger-left]: (2024-08-19 16:30:48 +0000 UTC) CHARGER_OP_MODE 6
evcc    | [easee ] TRACE 2024/08/19 18:30:51 ProductUpdate [charger-left]: (2024-08-19 16:30:49 +0000 UTC) OUT_VOLT_PIN1_3 0.9290000200271606
evcc    | [easee ] TRACE 2024/08/19 18:31:02 ProductUpdate [charger-left]: (2024-08-19 16:30:59 +0000 UTC) LEGACY_EV_STATUS 2
evcc    | [lp-1  ] DEBUG 2024/08/19 18:31:06 charge power: 0W
evcc    | [lp-1  ] DEBUG 2024/08/19 18:31:06 charge currents: [0.007 0.008 0.008]A
```

### What type of operating system are you running?

Linux

### Version

v0.130.0

MaTr74, 2024-08-21:
[evcc-lp1.log](https://github.com/user-attachments/files/16686298/evcc-lp1.log)
[evcc-lp2.log](https://github.com/user-attachments/files/16686299/evcc-lp2.log)

andig, 2024-08-21:
So what is the actual problem? Current too high for minpv?

MaTr74, 2024-08-21:
I would say the chaotic switching bevaviour during the min+pv mode.

andig, 2024-08-21:
Seems I couldn‘t find it in the report. What chaotic behaviour?

MaTr74, 2024-08-21:
I mean the switching behaviour from 0 to 2 A and back interrupted by short periods of 16 A. I would expect to see charging at constant 6 A. For me it looks chaotic and is different from charging bevor with evcc version v0.128.4 which i used before. Also, the customer got a lot of error messages from his car during this period, what intent to believe there could be any problem between the car, the charger and/or evcc. Maybe you can find any hints within the logs?

pasqualito79, 2024-08-23:
same here, after version 130 the automatic phase switching doesn't work anymore.

andig, 2024-08-23:
/cc @GrimmiMeloni

GrimmiMeloni, 2024-08-24:
@pasqualito79 
> same here, after version 130 the automatic phase switching doesn't work anymore

Nope - that's not the same. This is not about phase switching not working, but erratic(?) power consumption. Please open a separate issue of the problem persists.

GrimmiMeloni, 2024-08-24:
_@MaTr74 thanks for the logs. For the future - please leave the entire log intact. I appreciate you wanting to help by pre-filtering, but it is usually also helpful for us to see the site category, to distinguish intervals. So best if you just leave everything in._

I checked the first log. From what I can tell charge starts with 32A@3p just as expected for an Easee after plugging in. Also very normal, evcc detects the "false start", adjust its state accordingly and then scales the charge down to 6A.
Then it gets interesting. 4 Seconds after the step down to 6A, evcc loses the SignalR connection to the charger. Once the connection to the charger comes back, it fhas already entered a (I presume) fallback mode, running suddenly at 5A@1p. 

Also noteworthy is that when the charger reconnects the last known state for the circuit max amps is 0 on all three phases. Seems like that is the last state prior to the disconnect?

Your second scenario looks exactly the same. The log starts a bit late, but the disconnect is captured and afterwards also here the chargers is at5A@1p. Here, I can also see a second disconnect roughly 3 mins after the first. Maybe the chargers are crashing?

As far as I can tell from here, evcc does not do anything that would cause this. Seems to be a hardware problem.

andig, 2024-08-24:
> Seems to be a hardware problem.

Closing.

MaTr74, 2024-08-25:
Hi @GrimmiMeloni, thanks for checking the logs! I will let the logs intact for future issues. I would consider that the car has caused the crash, because i think it's highly unlikely that both chargers are got hardware problems at the same time after they reacted not so in the past.

MaTr74, 2024-09-10:
In the meantime I observed the charging behaviour. As the next car was charging a few weeks later I could observe the same strange behaviour:
[image]
I will attach the appropriate log file. 

Then I stopped evcc to observe the charger's behaviour without evcc. I limited the charging power to 6 A by the easee app to get a similar charging pre condition. Now, the behaviour is like expected:
[image]
The short disruption at 12:30 was caused by myself by removing the 6A limit and pausing the charging session temporarly.

After this observations I would assume that the strange behaviour is at least influenced by evcc and it's not necessarily a hardware problem. What can I do to investigate this problem?
[evcc-2024-09-03.log](https://github.com/user-attachments/files/16946855/evcc-2024-09-03.log)

GrimmiMeloni, 2024-09-11:
I have given your new log a look. So we might be on to something.
From what you shared, your installation is limited to 25A per phase.

```
evcc    | [easee ] TRACE 2024/09/03 14:41:32 ProductUpdate [charger-left]: (2024-07-21 13:41:20 +0000 UTC) CIRCUIT_MAX_CURRENT_P2 25
evcc    | [easee ] TRACE 2024/09/03 14:41:32 ProductUpdate [charger-left]: (2024-07-21 13:41:20 +0000 UTC) CIRCUIT_MAX_CURRENT_P1 25
evcc    | [easee ] TRACE 2024/09/03 14:41:32 ProductUpdate [charger-left]: (2024-07-21 13:41:20 +0000 UTC) CIRCUIT_MAX_CURRENT_P3 25
```

I also see the loadbalancing between your chargers adhering to this

```
evcc    | [easee ] TRACE 2024/09/03 14:40:49 ProductUpdate [charger-left]: (2024-09-03 12:40:46 +0000 UTC) CIRCUIT_TOTAL_ALLOCATED_PHASE_CONDUCTOR_CURRENT_L1 25
evcc    | [easee ] TRACE 2024/09/03 14:40:49 ProductUpdate [charger-left]: (2024-09-03 12:40:46 +0000 UTC) CIRCUIT_TOTAL_ALLOCATED_PHASE_CONDUCTOR_CURRENT_L2 25
evcc    | [easee ] TRACE 2024/09/03 14:40:49 ProductUpdate [charger-left]: (2024-09-03 12:40:46 +0000 UTC) CIRCUIT_TOTAL_ALLOCATED_PHASE_CONDUCTOR_CURRENT_L3 25
```

The output current of your charger also never seems to go above this value, e.g.

```
evcc    | [easee ] TRACE 2024/09/03 14:40:49 ProductUpdate [charger-left]: (2024-09-03 12:40:46 +0000 UTC) OUTPUT_CURRENT 25
```

It would be great if you could confirm this being the case, by checking the Easee App (or the site portal).

At the same time though, evcc is asking dynamic current to be set to 32A. Currently this is hardcoded, which is not good in a case like yours.

So, assuming you confirm the 25A limit on installation level, we can state that evcc is asking for Dynamic Charge Current to go above the circuit limit. Now I would **_expect_** such a request to either be ignored, or automatically limited down to the circuit limit. That's somewhat backed by the App's interface, which also presents an elaborate list why the Amps are limited, and that shows that "physics" (circuit, cables, ...) come first. But given your experience, maybe that is not the case, and this actually causes the charger to crash, or at least abort the charge.

It's a theory at this point, but at least we should adhere to certain limits. Setting 6A instead of 32A would be the save bet - a more optimized version might pick the maximum of 6A, circuit limit, loadpoint limit. I have to think this through for a moment, while I wait for you to confirm the 25A limit on the circuit.

MaTr74, 2024-09-11:
Hi @GrimmiMeloni,

thanks for your answer and looking on the logs! Yes that's true, there is a 25A max restriction on circuit level:
[image]

I'v configured the loadpoint's max current to 25A also now. So I'll wait for the next charging session.

closed 2024-09-16 by andig

MaTr74, 2024-09-21:
The last charging session looked pretty good:
[image]

GrimmiMeloni, 2024-09-21:
Thanks for confirming.

PR #16116 (2024-09-16): Easee: obey current limits based on charger configuration
fix #15556

Note: This change serves the purpose of preventing evcc to request higher currents than allowed according to the site/circuit configuration on the Easee side. One would expect (hope?) that the charger simply ignores (or limits) requests above the configured amps allowed according to the installation. Based on the protocols shared in the linked case however, it seems that requests above the physical limits, cause erratic charger behavior (possibly resets). 

This change does not replace the limits configured in the loadpoint. [... cut]