evcc-io/evcc #15559: [OCPP] ABL eMH3

etecprojekt, 2024-08-21:
### Describe the bug

Hallo zusammen,

nach dem Update von Version 0.129.0 auf 0.130.2 funktioniert die Ansteuerung der ABL eMH3 (3W2263) nicht mehr.
Nachfolgende Fehlermeldungen aus dem Log:

`[ocpp  ] ERROR 2024/08/21 07:04:20 ocpp message (2383823729): OccurrenceConstraintViolation - Field CallResult.Payload.ChargingSchedule.ChargingRateUnit required but not found for feature GetCompositeSchedule ([3, "2383823729" ,{"status":"Accepted","connectorId":1,"scheduleStart":"2024-08-21T05:04:19.385+00:00","chargingSchedule":{"duration":60,"startSchedule":"2024-08-21T05:04:19.385+00:00","chargingSchedulePeriod":[{"startPeriod":0,"limit":-1.0}]}}])
`
Die Ladung wird laut Log wohl freigegeben aber es erfolgt keine Ladung.
Haben jetzt erstmal wieder auf Version 0.129.0 zurück.

### Steps to reproduce

PV-Überschussladen, Min+PV ohne Funktion.

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error
  ocpp: trace

# https://docs.evcc.io/docs/reference/configuration/log/

# unique installation id
plant: 

interval: 30s # control cycle interval

sponsortoken: 

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: sma-data-manager 
  id: 2  
  host: 192.168.10.194  
  port: 502  
  usage: grid  
  modbus: tcpip  
  name: grid1

- type: template
  template: sma-data-manager 
  id: 2  
  host: 192.168.10.194  
  port: 502  
  usage: pv  
  modbus: tcpip  
  name: pv1

#- type: template
#  name: wallboxen
#  template: sma-energy-meter
#  usage: charge
#  host: 192.168.10.125

chargers:
- type: template
[... 677 lines cut ...]
[site  ] DEBUG 2024/08/21 08:56:10 grid meter: -1879W
[modbus] TRACE 2024/08/21 08:56:10 modbus: send 06 9b 00 00 00 06 02 03 7b 2f 00 02
[modbus] TRACE 2024/08/21 08:56:10 modbus: recv 06 9b 00 00 00 07 02 03 04 00 00 14 3d
[modbus] TRACE 2024/08/21 08:56:10 modbus: send 06 9b 00 00 00 06 02 03 7b 31 00 02
[modbus] TRACE 2024/08/21 08:56:10 modbus: recv 06 9b 00 00 00 07 02 03 04 ff ff e7 f6
[modbus] TRACE 2024/08/21 08:56:10 modbus: send 06 9b 00 00 00 06 02 03 7b 33 00 02
[modbus] TRACE 2024/08/21 08:56:10 modbus: recv 06 9b 00 00 00 07 02 03 04 00 00 23 46
[site  ] DEBUG 2024/08/21 08:56:10 grid currents: [-5.18 6.15 -9.03]A
[site  ] DEBUG 2024/08/21 08:56:10 giving loadpoint priority for additional: 3473W
[site  ] DEBUG 2024/08/21 08:56:10 site power: -5352W
[lp-4  ] ERROR 2024/08/21 08:56:10 charge voltages: not available
[lp-4  ] ERROR 2024/08/21 08:56:10 charge total import: not available
[lp-4  ] DEBUG 2024/08/21 08:56:10 charger status: A
[ocpp  ] TRACE 2024/08/21 08:56:10 send ABL_3W226305065: [2,"140010820","GetCompositeSchedule",{"connectorId":2,"duration":60}]
[ocpp  ] TRACE 2024/08/21 08:56:10 recv ABL_3W226305065: [3, "140010820" ,{"status":"Accepted","connectorId":2,"scheduleStart":"2024-08-21T06:56:09.340+00:00","chargingSchedule":{"duration":60,"startSchedule":"2024-08-21T06:56:09.340+00:00","chargingSchedulePeriod":[{"startPeriod":0,"limit":16.0}]}}]
[ocpp  ] ERROR 2024/08/21 08:56:10 ocpp message (140010820): OccurrenceConstraintViolation - Field CallResult.Payload.ChargingSchedule.ChargingRateUnit required but not found for feature GetCompositeSchedule ([3, "140010820" ,{"status":"Accepted","connectorId":2,"scheduleStart":"2024-08-21T06:56:09.340+00:00","chargingSchedule":{"duration":60,"startSchedule":"2024-08-21T06:56:09.340+00:00","chargingSchedulePeriod":[{"startPeriod":0,"limit":16.0}]}}])
```

### What type of operating system are you running?

Linux

### Version

0.130.2

etecprojekt, 2024-08-21:
Anbei noch die volle Logdatei

[[circuit-main] DEBUG 20240821 07034.txt](https://github.com/user-attachments/files/16687643/circuit-main.DEBUG.20240821.07034.txt)

premultiply, 2024-08-21:
> [ABL_3W226305065-1] WARN 2024/08/21 08:52:10 waiting for local authentication

Die Box wartet auf die lokale Freigabe via RFID oder Free Charge-Mode.

https://github.com/evcc-io/evcc/discussions/15471

premultiply, 2024-08-21:
> [ocpp  ] ERROR 2024/08/21 08:56:10 ocpp message (140010820): OccurrenceConstraintViolation - Field CallResult.Payload.ChargingSchedule.ChargingRateUnit required but not found for feature [...]

Das ist ein Fehler in der OCPP-Implementierung der Box. Verletzung des Standards. Bitte damit an den Hersteller wenden.

etecprojekt, 2024-08-21:
Hallo,

vielen Dank für die Info. Da wird seitens ABL wohl nichts mehr kommen, da die eMH3 aus dem Programm genommen wurde. Also heißt das, dauerhaft die Version 0.129.0 nutzen?

Vielen Dank.

premultiply, 2024-08-21:
Erstmal Problem 1 bei dir lösen, der Error ist zweitrangig und nur ein Hinweis.

etecprojekt, 2024-08-21:
Die ID für freies Laden wird doch in der Konfiguration (idtag: 10101010101010) mit übergeben,
wieso sollte die Box dann auf eine Authentifizierung waren? 
Es hat so bislang immer funktioniert, und es wurde nichts geändert oder neugestartet. 
Ich werde nochmal auf die neuste Version gehen und es nochmal beobachten.

premultiply, 2024-08-21:
RFID-Karte vorhalten oder
Wallbox in deren OCPP-Setup auf "Free charge", "Autostart", "Free vending" o.ä. konfigurieren.
Ref https://github.com/evcc-io/evcc/pull/14888 und https://docs.evcc.io/blog/2024/08/17/highlights-14a-enwg-ocpp-loadmanagement-elli#rfid-und-autorisierung

premultiply, 2024-08-22:
Bitte die Konfig der beiden OCPP-Ladepunkte mal auf

```yaml
- type: template
  template: ocpp
  stationid: ABL_3W226305065
  connector: 1
  name: wb_re_1
- type: template
  template: ocpp
  stationid: ABL_3W226305065
  connector: 2
  name: wb_re_2
```
reduzieren und den FreeCharge Mode aktivieren.

Schau auch mal bitte ob du in der Konfiguration der Box irgendwo das Meter Sample Interval Box auf 10s reduzieren kannst.

Dann mal bitte den Output von `evcc charger --log trace --diagnose`

premultiply, 2024-08-22:
Also, ich habe jetzt dein Log mal durchgesehen.

> [ocpp  ] DEBUG 2024/08/22 12:30:40 charge point connected: ABL_3W226305065

Ab hier funktioniert es wie es soll.
Auf Grund der Logeinträge gehe ich davon aus, dass das der Zeitpunkt war nachdem du den FreeChargeMode aktiviert hast und die Box neugestartet hat/wurde.

Bitte auch noch beachten, da du insgesamt 4 Ladepunkte hast, dass pro Intervall immer nur ein LP bearbeitet wird und es daher zu längeren Verzögerungen zwischen Aktion in der UI und tatsächlicher Änderung am Charger kommen kann.
Daher ist es vielleicht hilfreich das evcc interval in der Konfig ggf. etwas zu reduzieren.
Probier mal 10s. Das sollte das System deutlich responsiver machen.

Das einzige was etwas irritierend ist, ist dass die Meteringdaten von evcc in dem Log nicht erkannt wurden und daher vermutlich keine sinnvolle Anzeige in der UI kam. Das kann aber mit der Neuverbindung durch die Änderung auf FreeChargeMode zusammenhängen.

Das würde ich gerne mit einem aktuellen Log nochmal genauer ansehen.

Hier also bitte nochmal mit dem aktuellen Nightly Build testen. Dazu bitte FreeChargeMode anschalten, dann die Box vollständig neustarten und dann erst evcc (neu)starten.

etecprojekt, 2024-08-23:
Fahrzeug wurde angesteckt, erst mit 6A geladen, dann mit 16A und dann gestoppt, obwohl kein Überschuss verfügbar war, idTag ist in der Config nicht mehr drin.

Trace:
[evcc-20240823-065208-trace.log](https://github.com/user-attachments/files/16723631/evcc-20240823-065208-trace.log)
W
Wallbox:
[image]
[image]

etecprojekt, 2024-08-23:
Eine Manuelle Ansteuerung über Min+PV funktioniert, Zählerwerte werden gesendet, allerdings wird die Ladeleistung in EVCC nicht erkannt.

[evcc-20240823-080348-trace.log](https://github.com/user-attachments/files/16724342/evcc-20240823-080348-trace.log)

premultiply, 2024-08-23:
Die Anzeige der Leistung des Ladepunkts hat nicht mit der Überschussregelung zu tun. Diese basiert ausschließlich auf den Daten des Netzanschlusses (grid).

premultiply, 2024-08-23:
Abgesehen von den diversen Fehlermeldungen auf Grund von Fehlern in der Firmware funktioniert es nach dem erstem Blick in das Log aber nun eigentlich wie es soll.

Bei der Ausgabe der nicht kritischen Fehler werden wir sicherlich noch nachjustieren.

Die Sache mit den Meteringdaten stellt mich aber gerade noch vor ein Rätsel.

/cc @andig Ich stehe auf dem Schlauch. Weshalb werden hier die Ladeleistungen und Energie mit NotAvailable abgelehnt? Hast du eine Idee?

etecprojekt, 2024-08-23:
Ich glaube du hast recht, er merkt also das die Box Strom zieht, zeigt es glaube ich nur nicht an.
Würde es nicht merken, würde er ja immer wieder an und ausschalten, da er nicht weiß, wieviel Anteil Ladeleistung in Grid steckt.
Anbei ein Log mit einem Fahrzeug dran, welches 1p läd und geregelt wird.

[evcc-20240823-105932-trace.log](https://github.com/user-attachments/files/16726307/evcc-20240823-105932-trace.log)

mfuchs1984, 2024-08-23:
@premultiply die MeterValues messages ist auf den ersten Blick Das ist anders als bswp. bei meinem Bender Charger, es handelt sich um ein Array mit 2 Objekten, das erste enthält die erfolgreich geparsten currents, das zweite den Rest und nochmals die Currents. Evtl. ist das die Ursache?

```
[
    2,
    "39118670",
    "MeterValues",
    {
        "connectorId": 1,
        "transactionId": 1,
        "meterValue": [
            {
                "timestamp": "2024-08-23T06:02:45.311+00:00",
                "sampledValue": [
                    {
                        "value": "6.0",
                        "context": "Sample.Periodic",
                        "format": "Raw",
                        "measurand": "Current.Import",
                        "phase": "L1",
                        "location": "Body",
                        "unit": "A"
                    },
                    {
                        "value": "6.0",
                        "context": "Sample.Periodic",
                        "format": "Raw",
                        "measurand": "Current.Import",
                        "phase": "L2",
                        "location": "Body",
                        "unit": "A"
                    },
                    {
                        "value": "5.9",
                        "context": "Sample.Periodic",
                        "format": "Raw",
                        "measurand": "Current.Import",
                        "phase": "L3",
                        "location": "Body",
                        "unit": "A"
                    }
                ]
            },
            {
                "timestamp": "2024-08-23T06:02:45.337+00:00",
                "sampledValue": [
                    {
                        "value": "10280.792",
                        "context": "Sample.Periodic",
                        "format": "Raw",
                        "measurand": "Energy.Active.Import.Register",
                        "location": "Outlet",
                        "unit": "kWh"
                    },
                    {
                        "value": "4185.3",
                        "context": "Sample.Periodic",
                        "format": "Raw",
                        "measurand": "Power.Active.Import",
                        "location": "Outlet",
                        "unit": "W"
                    },
                    {
                        "value": "5.957",
                        "context": "Sample.Periodic",
                        "format": "Raw",
                        "measurand": "Current.Import",
                        "phase": "L1",
                        "location": "Outlet",
                        "unit": "A"
                    },
                    {
                        "value": "6.027",
                        "context": "Sample.Periodic",
                        "format": "Raw",
                        "measurand": "Current.Import",
                        "phase": "L2",
                        "location": "Outlet",
                        "unit": "A"
                    },
                    {
                        "value": "5.983",
                        "context": "Sample.Periodic",
                        "format": "Raw",
                        "measurand": "Current.Import",
                        "phase": "L3",
                        "location": "Outlet",
                        "unit": "A"
                    },
                    {
                        "value": "236.3",
                        "context": "Sample.Periodic",
                        "format": "Raw",
                        "measurand": "Voltage",
                        "phase": "L1-N",
                        "location": "Outlet",
                        "unit": "V"
                    },
                    {
                        "value": "234.8",
                        "context": "Sample.Periodic",
                        "format": "Raw",
                        "measurand": "Voltage",
                        "phase": "L2-N",
                        "location": "Outlet",
                        "unit": "V"
                    },
                    {
                        "value": "235.8",
                        "context": "Sample.Periodic",
                        "format": "Raw",
                        "measurand": "Voltage",
                        "phase": "L3-N",
                        "location": "Outlet",
                        "unit": "V"
                    },
                    {
                        "value": "6.0",
                        "context": "Sample.Periodic",
                        "format": "Raw",
                        "measurand": "Current.Offered",
                        "location": "Outlet",
                        "unit": "A"
                    },
                    {
                        "value": "0.0",
                        "context": "Sample.Periodic",
                        "format": "Raw",
                        "measurand": "Power.Offered",
                        "location": "Outlet",
                        "unit": "kW"
                    }
                ]
            }
        ]
    }
]
```

premultiply, 2024-08-23:
Autsch, ja!! Super!
Hier gibt es zwei Objekte: Einmal von der Messposition Body (für uns eigentlich eher irrelevant) und einmal Outlet.

Da muss ich erstmal überlegen wie wir damit sinnvoll umgehen.

andig, 2024-08-23:
> Da aber conn.clock.Now() bei Verarbeitung des ersten Objekts schon später als der Timestamp des zweiten Objekts ist, wird der Inhalt ignoriert.

Dann sollten wir das einfach nach Timestamps sortieren (anstatt mit location rumzufrickeln)? Vorschlag kommt.

premultiply, 2024-08-23:
Die Location ist hier schon auch relevant, da das hier eine Box mit zwei LPs ist und der Ladestrom beider LPs auf den Body-Werten auftaucht, wir uns hier aber nur für die Outlets (Connector) interessieren. Die Body-Werte verwirren hier nur - kann aber bei anderen Boxen wieder anders sein.

mfuchs1984, 2024-08-23:
Wäre es nicht am robustesten, mit dem neuesten timestamp anzufangen und dann bis zum ältesten Objekt weiterzumachen und measurands, die noch nicht gefunden wurden, daraus zu ergänzen?

andig, 2024-08-23:
> Dabei Outlet vor Body priorisieren?

Sind die irgendwo definiert?

> mit dem neuesten timestamp anzufangen und dann bis zum ältesten Objekt weiterzumachen und measurands

Umgekehrt und dann wird einfach überschrieben.

premultiply, 2024-08-23:
Das würde hier vielleicht funktionieren aber in anderen Fällen (andere Boxen) ist die Reihenfolge ggf. eine andere?
Ist ja leider nicht näher definiert. Und wie wir wissen macht da jeder Hersteller irgendwas anderes (wie man hier ja auch sieht).
Theoretisch können die auch alle mit unterschiedlichen Locations unter einem Timestamp stehen.

Aktuell sehe ich noch keine elegante Möglichkeit das möglichst universell zu lösen.
Doch irgendwie eine Priorität nach der Location machen?

andig, 2024-08-23:
Ich sehe hier gar kein Problem. Oder wollen wir Spannungsabfall über das Kabel messen???

premultiply, 2024-08-23:
Nein, aber wir wollen auch nicht zufällige Sprünge in den Currents sehen?

andig, 2024-08-23:
...dann könnte man in anderer Logik nach Zeit und Location sortieren. Dafür müssten die verschachtelten Listen erst gemerged werden?

premultiply, 2024-08-23:
Das geht aus meiner Sicht in die richtige Richtung.

mfuchs1984, 2024-08-23:
Mit dem PR bleibt das Grundroblem doch bestehen, oder? Da conn.meterUpdated auf conn.clock.Now(), wird alles nach dem ersten Objekt in der nun sortierten Liste nicht verarbeitet. Steht zufällig das benötigte im ersten Objekt der sortierte Liste, ist alles gut, wenn nicht, fehlt was. Oder übersehe ich was?
```
for _, meterValue := range sortByAge(request.MeterValue) {
      // ignore old meter value requests
      if meterValue.Timestamp.Time.After(conn.meterUpdated) {
	      for _, sample := range meterValue.SampledValue {
		      sample.Value = strings.TrimSpace(sample.Value)
		      conn.measurements[getSampleKey(sample)] = sample
		      conn.meterUpdated = conn.clock.Now()
	      }
      }
}
```

andig, 2024-08-23:
> Das geht aus meiner Sicht in die richtige Richtung.

Dann brauchts bitte eine Precedence eincshl. "leer". Welche Werte "stechen"?

premultiply, 2024-08-23:
Das ist gar nicht so einfach, da es für diverse Messwerte unterschiedlich ist.

Voltage: Inlet, Body, Outlet, ""
Currents/Power/Energy: "", Outlet, Body, Inlet
Current.Offered/Power.Offered/SoC: egal?

kschwar, 2024-08-23:
Wie kann ich den pr #15623 testen?

andig, 2024-08-24:
>  da es für diverse Messwerte unterschiedlich ist.

Ist das so? Muss uns das wirklich interessieren? Ich sehe nichtmal in den Beispielmesswerten hier ein echtes Problem da die disjunkt sein?

That said: der aktuelle PR sollte schon reichen. Kein Grund für Overengineering.

premultiply, 2024-08-24:
Ja, ist schon so.
Irgendwo wird es uns sehr bald wieder auf die Füße fallen. ;)
Aber gut, fürs Erste hier erstmal dann hoffentlich gefixt.

Ich lass hier aber mal auf, da das Problem mit den Error-Ausgaben auf die OCPP-Schnittstelle noch eines Fixes bedarf.
Komme ich aber erst heute Abend dazu.

etecprojekt, 2024-08-24:
Wann erscheint #15623 im Docker nightly Build?
Würde es dann einmal testen und logs bereitstellen.

mfuchs1984, 2024-08-24:
> Mit dem PR bleibt das Grundroblem doch bestehen, oder? Da conn.meterUpdated auf conn.clock.Now() gesetzt wird, was definitiv später als die Timestamps ist, wird alles nach dem ersten Objekt in der nun sortierten Liste nicht verarbeitet. Steht zufällig das Benötigte im ersten Objekt der sortierte Liste, ist alles gut, wenn nicht, fehlt was. Oder übersehe ich was?
> 
> ```
> for _, meterValue := range sortByAge(request.MeterValue) {
>       // ignore old meter value requests
>       if meterValue.Timestamp.Time.After(conn.meterUpdated) {
> 	      for _, sample := range meterValue.SampledValue {
> 		      sample.Value = strings.TrimSpace(sample.Value)
> 		      conn.measurements[getSampleKey(sample)] = sample
> 		      conn.meterUpdated = conn.clock.Now()
> 	      }
>       }
> }
> ```
> 
> Man müsste meines Erachtens nur einen der Timestamp im request für die Prüfung meterValue.Timestamp.Time.After(conn.meterUpdated) heranziehen, danach alle Objekte verarbeiten und anschließend erst conn.meterUpdated = conn.clock.Now() aufrufen.

@andig @premultiply 
Sorry falls ich es nicht ganz verstehe, aber ich bin nach wie vor der Meinung, dass der PR nicht hilft. In dem Fall hier sind sie Objekte ja schon nach timestamp sortiert, aber das zweite wird nicht verarbeitet aufgrund des Vergleichs mit conn.meterupdated, welches nach Verarbeitung des ersten Objekt auf conn.clock.Now() steht, was wohl der aktuellen Systemzeit entspricht und somit neuer als der timestamp des zweiten Objekts ist. Will man das verhindern, müsste man auf den timestamp des zuletzt verarbeiteten Objekt setzen oder wie oben vorgeschlagen nur einen timestamp vergleichen.

andig, 2024-08-24:
Good catch. Siehe PR- der Vergleich ist umzudrehen.

> In dem Fall hier sind sie Objekte ja schon nach timestamp sortiert

Check- aber darauf kann man sich vmtl. auch nicht verlassen, daher der andere PR...

etecprojekt, 2024-08-24:
> Good catch. Siehe PR- der Vergleich ist umzudrehen.
> 
> > In dem Fall hier sind sie Objekte ja schon nach timestamp sortiert
> 
> Check- aber darauf kann man sich vmtl. auch nicht verlassen, daher der andere PR...

Wann erscheint das ganze im Docker nightly Build?
Würde es dann einmal testen und logs bereitstellen.

etecprojekt, 2024-08-25:
Mit der aktuellen nightly, kann der Ladepunkt nicht mehr erstellt werden.
Nachfolgend der Log: [evcc-20240825-103747-trace.log](https://github.com/user-attachments/files/16739518/evcc-20240825-103747-trace.log)

[image]

andig, 2024-08-25:
@premultiply ich verstehe nicht was hier los ist. Die Kommunikation funktioniert, aber es gibt einen Timeout und alle Meßwerte werden hier einzeln konfiguriert?

```
[ocpp  ] TRACE 2024/08/25 10:31:09 send ABL_3W226305065: [2,"2656132676","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Power.Active.Import"}]
[ocpp  ] TRACE 2024/08/25 10:31:09 send ABL_3W226305065: [2,"2489658164","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Energy.Active.Import.Register"}]
[ocpp  ] TRACE 2024/08/25 10:31:09 send ABL_3W226305065: [2,"1788724089","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Current.Import"}]
[ocpp  ] TRACE 2024/08/25 10:31:09 send ABL_3W226305065: [2,"553884136","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Voltage"}]
[ocpp  ] TRACE 2024/08/25 10:31:09 send ABL_3W226305065: [2,"3257290741","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Current.Offered"}]
[ocpp  ] TRACE 2024/08/25 10:31:09 send ABL_3W226305065: [2,"3995821291","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Power.Offered"}]
[ocpp  ] TRACE 2024/08/25 10:31:09 send ABL_3W226305065: [2,"2798628732","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"SoC"}]
[ocpp  ] TRACE 2024/08/25 10:31:09 send ABL_3W226305065: [2,"1399715684","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Power.Active.Import,Energy.Active.Import.Register,Current.Import,Voltage,Current.Offered,Power.Offered,SoC"}]
[ocpp  ] TRACE 2024/08/25 10:31:09 send ABL_3W226305065: [2,"145743017","ChangeConfiguration",{"key":"MeterValueSampleInterval","value":"10"}]
```

mfuchs1984, 2024-08-25:
Auffällig ist an der Stelle des timeou, dass der erste der beiden timestamps in den MeterValues stark veraltet ist, evtl. ein Seiteneffekt des PR #15643?

```
[ocpp  ] TRACE 2024/08/25 10:35:05 recv ABL_3W226305065: [2, "20839185", "MeterValues" ,{"connectorId":1,"transactionId":1,"meterValue":[{"timestamp":"2024-08-25T08:34:44.584+00:00","sampledValue":[{"value":"15.9","context":"Sample.Periodic","format":"Raw","measurand":"Current.Import","phase":"L1","location":"Body","unit":"A"},{"value":"15.9","context":"Sample.Periodic","format":"Raw","measurand":"Current.Import","phase":"L2","location":"Body","unit":"A"},{"value":"15.8","context":"Sample.Periodic","format":"Raw","measurand":"Current.Import","phase":"L3","location":"Body","unit":"A"}]},{"timestamp":"2024-08-25T08:35:04.062+00:00","sampledValue":[{"value":"10286.317","context":"Sample.Periodic","format":"Raw","measurand":"Energy.Active.Import.Register","location":"Outlet","unit":"kWh"},{"value":"11274.0","context":"Sample.Periodic","format":"Raw","measurand":"Power.Active.Import","location":"Outlet","unit":"W"},{"value":"15.896","context":"Sample.Periodic","format":"Raw","measurand":"Current.Import","phase":"L1","location":"Outlet","unit":"A"},{"value":"15.934","context":"Sample.Periodic","format":"Raw","measurand":"Current.Import","phase":"L2","location":"Outlet","unit":"A"},{"value":"15.983","context":"Sample.Periodic","format":"Raw","measurand":"Current.Import","phase":"L3","location":"Outlet","unit":"A"},{"value":"237.0","context":"Sample.Periodic","format":"Raw","measurand":"Voltage","phase":"L1-N","location":"Outlet","unit":"V"},{"value":"236.4","context":"Sample.Periodic","format":"Raw","measurand":"Voltage","phase":"L2-N","location":"Outlet","unit":"V"},{"value":"236.5","context":"Sample.Periodic","format":"Raw","measurand":"Voltage","phase":"L3-N","location":"Outlet","unit":"V"},{"value":"16.0","context":"Sample.Periodic","format":"Raw","measurand":"Current.Offered","location":"Outlet","unit":"A"},{"value":"0.0","context":"Sample.Periodic","format":"Raw","measurand":"Power.Offered","location":"Outlet","unit":"kW"}]}]}]
[ocpp  ] TRACE 2024/08/25 10:35:05 send ABL_3W226305065: [3,"20839185",{}]
[main  ] FATAL 2024/08/25 10:35:09 cannot create charger 'wb_re_1': cannot create charger type 'template': cannot create charger type 'ocpp': timeout
```

premultiply, 2024-08-25:
> alle Meßwerte werden hier einzeln konfiguriert?

Das ist die normale Autokonfiguration die die einzelnen Measurands auf Gültigkeit durchprobiert und anschließend alle gültigen konfuguriert.

andig, 2024-08-25:
Ahhh, danke :). Hast Du eine Idee, woher jetzt der Timeout kommt? Ich habe https://github.com/evcc-io/evcc/pull/15644 im Verdacht, sehe aber nicht warum?

andig, 2024-08-25:
@mfuchs1984 kannst Du selbst kompilieren? Falls ja probier mal bitte

    git revert e622e67 && make

mfuchs1984, 2024-08-25:
Ich kann kompilieren, habe aber nicht die betroffene Wallbox, sondern eine Bender

andig, 2024-08-25:
@etecprojekt siehe https://github.com/evcc-io/evcc/issues/15559#issuecomment-2308750931. Kannst Du compilieren oder alternativ Deine Box mal auf meinen Rechner als OCPP Server umbiegen? In dem Fall bitte bei info@evcc.io melden- vielen Dank!

kschwar, 2024-08-25:
Habe dir eine E-Mail geschrieben, danke 😉

etecprojekt, 2024-08-25:
Startet wieder und verbindet. Werde es ggf. heute Nachmittag nochmal mit einem Ladevorgang testen.
Nachfolgen ein Logs ohne verbundene Fahrzeuge: [evcc-20240825-132443-trace.log](https://github.com/user-attachments/files/16740074/evcc-20240825-132443-trace.log)

etecprojekt, 2024-08-25:
Nachfolgend ein Log von einem kurzen Ladevorgang:
[evcc-20240825-151251-trace.log](https://github.com/user-attachments/files/16740370/evcc-20240825-151251-trace.log)

etecprojekt, 2024-08-26:
Wenn ein Fahrzeug eingesteckt ist, kein Überschuss verfügbar ist, laufen die charge abfragen alle auf timeout und werden in der UI als Fehler angezeigt.

[image]
[evcc-20240826-072041-trace.log](https://github.com/user-attachments/files/16744344/evcc-20240826-072041-trace.log)

mfuchs1984, 2024-08-26:
Sieht aus, als würde die Wallbox in dem Fall aufhören, die MeterValues zu senden.

etecprojekt, 2024-08-26:
Genau, da keine Ladung aktiv ist.

mfuchs1984, 2024-08-26:
Eine Transaction sollte aber aktiv sein. Und dann gilt

>Interval between sampling of metering (or other) data, intended to be transmitted by "MeterValues" PDUs. For charging session data (ConnectorId>0), samples are acquired and transmitted periodically at this interval from the start of the charging transaction.

etecprojekt, 2024-08-26:
Nachfolgend ein Trace nach dem Update auf aktuellen Docker Nightly ohne Fahrzeug, da werden die Werte gesendet:
[evcc-20240826-093211-trace.log](https://github.com/user-attachments/files/16745546/evcc-20240826-093211-trace.log)

mfuchs1984, 2024-08-26:
In diesem Fall fragt evcc aktiv nach den Werten, vermutlich weil keine Transaktion läuft, während man im anderen trace mit angestecktem Fahrzeug die requests nicht sieht, vermutlich weil erwartet wird, dass die Wallbox sich an die Spec halt und die Werte dann selbst sendet. Ich denke der Experte ist hier @premultiply.

etecprojekt, 2024-08-26:
Aktuellen Nightly installiert... es dauert 300 Sekunden (timeout), bis vom start von evcc, bis zum erreichen der Weboberfläche.
Er fragt immer wieder von connectorId die MeterValues ab, welches auch zurückgesendet wird und geht dann in:
[ABL_3W226305065-2] DEBUG 2024/08/26 15:19:51 BootNotification timeout
Daraufhin läuft evcc. Die Parameter bootnotification und getconfiguration habe ich schon aus der config rausgenommen.

[evcc_logs (2).txt](https://github.com/user-attachments/files/16749133/evcc_logs.2.txt)

andig, 2024-08-26:
@premultiply das scheint das beschriebene Problem mit mehreren Connectoren zu sein: es wird 2x die Availability geändert, gibt zunächst aber nur 1x BootNotification zurück. Später- nach TriggerMessage kommt sie dann. Warum es am Ende dennoch in Timeout läuft ist mir auf die Schnelle nicht klar.

Ich mache nochmal auf, auch wenn es jetzt ein anderes Fehlerbild ist.

etecprojekt, 2024-08-27:
4 Fahrzeuge dran, ein Fahrzeug ist nicht voll. 
Der Port steht auf "waiting for local authentication", läd trotzdem. 

[image]
[image]

andig, 2024-08-27:
@etecprojekt das ist ein eigenständiges Problem das wir uns anschauen sollten wenn das Setup erstmal vernünftig mit Deiner Box funktioniert. Eins nach dem Anderen!

andig, 2024-08-29:
Angehängter PR soll jetzt auch das Verhalten mit mehreren Connectoren fixen und Timingissues beim Start auflösen.

etecprojekt, 2024-08-30:
@andig kannst du den PR in die Docker Nightly mit reinnehmen? 
Glaube der ist aktuell nicht mit drin oder? 
Dann könnte ich das einmal testen.

andig, 2024-09-01:
Der PR kommt- wie jeder andere auch- ins nightly wenn er gemerged wird. Damit da nix kaputt geht wäre ein Test vorab schön.

closed 2024-09-01 by andig

etecprojekt, 2024-09-02:
Mit aktueller nightly, kein starten möglich:
[image]

[evcc_logs (3).txt](https://github.com/user-attachments/files/16832377/evcc_logs.3.txt)
[evcc-20240902-065301-trace.log](https://github.com/user-attachments/files/16832378/evcc-20240902-065301-trace.log)

[7 more replies]